thesofproject/sof #9268: [BUG] DSP panic with Zephyr on Intel MTL, regression 27th June

kv2019i, 2024-06-27:
**Describe the bug**
A DSP panic started showing up in CI runs on 27th June. No individual PR merged recently shows this in test results, so suspcion is a combination of PRs merged. Current suspect:
 - #9241 
 - #9232 

Both passed tested independently, but in together there seem to be DSP panics.

**To Reproduce**
https://sof-ci.01.org/sofpr/PR9267/build6051/devicetest/index.html

**Reproduction Rate**
Very high (most PR CI runs on 27th June)

**Expected behavior**
No DSP panics

**Impact**
Blocking CI 

**Environment**
https://sof-ci.01.org/sofpr/PR9267/build6051/devicetest/index.html

kv2019i, 2024-06-27:
Also seen in:
- https://sof-ci.01.org/sofpr/PR7782/build9265/devicetest/index.html
- https://sof-ci.01.org/sofpr/PR9260/build6001/devicetest/index.html
- https://sof-ci.01.org/sofpr/PR9258/build5967/devicetest/index.html

lyakh, 2024-06-27:
`git bisect` reliably brought to an SOF commit, that *obviously* cannot be the real reason - in a sense that that commit only adds 2 new functions without calling them and without touching a single line of runnable code, so it's purely modifying the build layout.
The crash happens at https://github.com/zephyrproject-rtos/zephyr/blob/b82b5b0734c30490368e21627423f10036487343/soc/intel/intel_adsp/ace/power_down.S#L64
That location is trying to lock in cache a variable on caller's stack https://github.com/zephyrproject-rtos/zephyr/blob/b82b5b0734c30490368e21627423f10036487343/soc/intel/intel_adsp/ace/power.c#L342 which all looks valid. However, it causes an exception. As a WA moving that variable to `.bss` by making it `static` seems to eliminate the problem.

lyakh, 2024-06-27:
@ceolin any ideas?

lyakh, 2024-06-27:
#9270 https://github.com/zephyrproject-rtos/zephyr/pull/75108

teburd, 2024-06-27:
What does the assembly dump look like for pm_state_set and power_down?

andyross, 2024-06-27:
> Also, I note there are two "MOVI" pseudo instructions after you start locking cache lines

Oooh, and there are lots more in asm_memory_management.h in contexts where you've already started shutting down memory!  I'm going to place my chips on this as the culprit.  I give it 60%+ odds.

Someone needs to comb through these files and make sure there's no "MOVI" usage (which again to be clear: is **only a CPU instruction for small values**, for big ones the compiler gets fancy and emits a .literals record for the linker to find and place).  I mean, maybe I'll lose the bet and these will turn out to all be valid/non-loading variants.  But still I think style would demand this be cleaned up.

lyakh, 2024-06-28:
> Oooh, and there are lots more in asm_memory_management.h in contexts where you've already started shutting down memory! I'm going to place my chips on this as the culprit. I give it 60%+ odds. [...]

@andyross ouch, yes, that sounds like a problem. Thanks for finding it! Now somebody just needs to fix it...

lyakh, 2024-06-28:
> Also, I note there are two "MOVI" pseudo instructions after you start locking cache lines which are going to end up pointing into an arbitrary literals location. You need an explicit L32R to be sure it lands in valid memory, not a compiler-generated MOVI.

@andyross I looked again at these. And TBH I don't see a problem with those specific ones. Here are the lines we're talking about:
https://github.com/zephyrproject-rtos/zephyr/blob/cfbe2adabc511663776642616cdc75510db882d3/soc/intel/intel_adsp/ace/power_down.S#L53-L61
Firstly, they just lock some cache lines, all the memory is still powered on. The problem *would* occur if we tried to access RAM after it's powered off, correct? So, those locations before powering down RAM should be safe? Same about these two lines https://github.com/zephyrproject-rtos/zephyr/blob/cfbe2adabc511663776642616cdc75510db882d3/soc/intel/intel_adsp/ace/power_down.S#L67-L68
> Oooh, and there are lots more in asm_memory_management.h in contexts where you've already started shutting down memory! I'm going to place my chips on this as the culprit. I give it 60%+ odds.

These ones - yes, agree, look dangerous...

lyakh, 2024-06-28:
> These ones - yes, agree, look dangerous...

@andyross @teburd that was a nice theory, but it doesn't seem to help: https://github.com/zephyrproject-rtos/zephyr/pull/75174 doesn't fix the problem. What exactly did we bet, Andy? ;-)

andyross, 2024-06-28:
Bah.  I was sure I had it.  Is the panic you're seeing on the DPFL instruction itself?  Reading the ISA ref, that's only supposed to happen if:

1. The MPU or MMU doesn't provide access to the virtual address.  Obviously not relevant on MTL which has neither.
2. The cache in question (dcache or icache) does not have two[1] free lines/ways available at the addressed index.

Possibility 2 seems not... entirely impossible?  This gets down to details about the hardware cache layout, which aren't completely clear from core-isa.h.  But my read is that MTL has a 48k dcache laid out in three ways with a 16k stride.  So if you have two cache lines pinned in the dcache at the same address modulo 16k, you can't add a third.  I see the code here adding two essentially unrestricted dcache addresses (the literals and the stack).  Is it possible there's another somewhere else, maybe leftover from the ROM?  If so, then bad luck with memory layout would (I think) be able to make this happen.

[1] Presumably "two" so that there's always one available to populates

andyross, 2024-06-28:
FWIW, I'm staring at that assembly, and... I think that stack word is a noop anyway? 

1. It lives in the second C function argument in A3, labeled "pu32_hpsram_mask" in the assembly.
2. The address it specifies is pinned via DPFL, seemingly correctly
3. It's then tested vs. NULL, and if so it skips the HPSRAM shutdown
4. That's the last thing that happens with it

Note:
* The pointer **can't be null**, because it's constructed in the caller as the address of a stack variable
* The pointer **is never used as a load address**, so why bother pinning it?

I think this is a bug in the assembly, the [BEQZ on line 82](https://github.com/zephyrproject-rtos/zephyr/blob/b82b5b0734c30490368e21627423f10036487343/soc/intel/intel_adsp/ace/power_down.S#L82) appears to have forgotten to load the value before testing it.

Finally: why bother anyway?  If it's just a boolean flag that means "skip the HPSRAM shutdown":

1. Pass it as a boolean/integer/whatever and not a pointer
2. Don't call it a "mask" (maybe the idea was it would specify a set of banks?)
3. Absolutely don't play games with cache pinning, just load it into a register in regular code before you start messing with RAM

andyross, 2024-06-28:
Anyway, given that: my recommendation is just to remove the second argument from the function, shut down HPSRAM unconditionallly, don't pin it, and see if that works around the issue?  That doesn't really explain the root cause here, but might evade the proximate one by deleting it.

andyross, 2024-06-28:
(Deleted a comment again to avoid confusion: thought I had it, but missed a spot where it's loading the mask values.)

FWIW, regarding the earlier point: it wouldn't be too hard in the failing configuration to dump the actual addresses and see if the low 14 bits of the mask and literals regions match (there are two lines in literals).

And if that does turn out to be the problem, you could resolve it by reserving space in that literals area and coping the mask words there.  That way they live sequentially in memory and can't collide on cache index.

marc-hb, 2024-06-29:
Daily tests are mostly red right now because of this, which means other, unrelated regressions WILL sneak unnoticed. 

tl;dr: CI is dead right now.

ceolin, 2024-07-01:
> (Deleted a comment again to avoid confusion: thought I had it, but missed a spot where it's loading the mask values.)
>
> FWIW, regarding the earlier point: it wouldn't be too hard in the failing [...]

If I have to bet I would put my money here. I was looking the documentation and stumbled in the same implementation notes that you commented. That would explain why moving that variable out of the stack solves the problem.

lyakh, 2024-07-01:
There's also an incorrectness in the Zephyr code at https://github.com/zephyrproject-rtos/zephyr/blob/2c34da96f0e3ba07764db3ac7def9b400bbd1729/soc/intel/intel_adsp/ace/power_down.S#L92-L104 - that code expects `pu32_hpsram_mask` to point to an array of `MAX_MEMORY_SEGMENTS` masks. Whereas when locking at https://github.com/zephyrproject-rtos/zephyr/blob/2c34da96f0e3ba07764db3ac7def9b400bbd1729/soc/intel/intel_adsp/ace/power_down.S#L63-L64 the code only locks one cache-line. And it isn't just that it knows, that the array will fit in one cache line, it isn't even guaranteed that it's cache-line aligned. So if the array even just had 2 elements, it could cross 2 cache lines. And the caller at https://github.com/zephyrproject-rtos/zephyr/blob/2c34da96f0e3ba07764db3ac7def9b400bbd1729/soc/intel/intel_adsp/ace/power.c#L342-L352 doesn't bother with an array at all, it just uses a single 32-bit value for a mask.

kv2019i, 2024-07-01:
One more idea. The exception seems to be right after first reference to windowed registers set up by call8. This is when the window overflow would be handled. The registers are stored to the stack frame, so these writes may go to the same cache line as we are trying to lock with dpfl (on caller's stack) -- or some other interaction between windowoverflow and dpfl. An ugly change following this hypothesis seems to be holding up -> https://github.com/zephyrproject-rtos/zephyr/pull/75285 -- let's see full results.

andyross, 2024-07-01:
@lyakh I agree that the code as committed seems a little fragile.  I got curious over the weekend and started on a reverse-engineered version that puts the enumeration and masking code in C and precomputes a single pointer table that can be pinned in a single region for a much smaller asm block.  I'm not set up to test it and I'm not really the one who should be writing this stuff (also I probably got at least a few things wrong!), but if you are heading down that path and want inspiration, it looked like this (note that I didn't get as far as transcribing the tiny IPC stub at the end of the process, though that's just ~3 fixed register writes I think):

https://gist.github.com/andyross/8c77ff1360084a56c70955c900a5d78b

kv2019i, 2024-07-02:
One observation not recorded here that it seems reproduction requires a debug build (that includes asserts). This explains why the failures are not seen in all CI jobs (some test with asserts some without).

andyross, 2024-07-02:
(Reposting here, put it in the wrong place originally)

Just a digression while we're here: **why are we even trying to do this at all?** Is there really no control over DSP SRAM power from the host CPU, or if not there the CSME or PMC or some other coprocessor? This kind of "shut down my own memory" trick is just outrageously complicated, surely there's a cleaner solution?

kv2019i, 2024-07-03:
@andyross wrote:
> Just a digression while we're here: **why are we even trying to do this at all?** Is there really no control over DSP SRAM power from the host CPU, or if not there the CSME or PMC or some other [...]

I'm not expert on the PM flows, so better I don't comment on the question itself. But just a note, this approach has been used in many generations of products and is used with XTOS based builds. Just in SOF, this has been used pretty much as-is since first version of SOF (see 2018 commit a292d45ba1b61664046209d2500978402a9b9e44 ). And based on git history, aside this current problem we see on Zephyr debug builds, there has not been many (if any) changes to this code, so even if complex, it seems to have been fairly low-maintainance to keep using this approach. That's all sidestepping the question whether this could be done differently -- that I cannot answer.

kv2019i, 2024-07-04:
Fix merged via #9278 , test run now clean with a fresh PR test https://sof-ci.01.org/sofpr/PR9174/build6213/devicetest/index.html

marc-hb, 2024-07-19:
>  so even if complex, it seems to have been fairly low-maintainance to keep using this approach.

Did you say low maintenance?

- https://github.com/thesofproject/sof/issues/9308

Sorry if I'm jumping to conclusions; I couldn't resist :-)

lyakh, 2024-09-05:
It's back https://sof-ci.01.org/sofpr/PR9430/build7664/devicetest/index.html

lyakh, 2024-09-05:
and a new attempt to fix it https://github.com/zephyrproject-rtos/zephyr/pull/78057

andyross, 2024-09-05:
Posted this in the PR by accident, copying here:

FWIW: I still bet that if you whiteboxed a rig to enumerate all the dcache indexes and check how many of them can be pinned, we'd discover that something in the boot ROM or elsewhere has left a line pinned accidentally, preventing more pinning at that index, and we're just hitting that by bad luck in the linker.

Would be non-trivial assembly to write, and tedious to debug as the only feedback is a panic, but not "hard" hopefully.

lyakh, 2024-09-06:
@andyross I think what I've tested that back when debugging this last time by doing either one or both of the following things:
1. trying to lock (and unlock) cache lines at all offsets to cover the entire associativity set - all succeeded
2. by trying to speculatively unlock the entire data cache by using `DIU`
```
diff --git a/soc/intel/intel_adsp/ace/power.c b/soc/intel/intel_adsp/ace/power.c
index 20efc8e6f97..61830897397 100644
--- a/soc/intel/intel_adsp/ace/power.c
+++ b/soc/intel/intel_adsp/ace/power.c
@@ -46,6 +46,9 @@ __imr void power_init(void)
 	cache_data_flush_range((__sparse_force void *)
 			sys_cache_cached_ptr_get(&adsp_pending_buffer),
 			sizeof(adsp_pending_buffer));
+	/* unlock entire data cache */
+	for (unsigned int i = 0; i < 16 * 1024; i += 64)
+		__asm__ __volatile__("diu %0, 0" : "r"(0xa0000000 + i));
 #endif /* CONFIG_SOC_INTEL_ACE15_MTPM */
 }
 ```
Neither had helped

andyross, 2024-09-07:
@lyakh my read is there are three ways, so you need to prove you can lock **two** lines at each index to saturate.  IIRC there are two separate dcache regions in the code in question, right?  So if those overlap in index and happen to collide with something already in the cache, then we'd be in a situation where we're trying to lock the same index in all three ways, which is illegal.

Maybe.  :)

lyakh, 2024-09-09:
@andyross sure, but as I said - I also tried unlocking all cache lines and it didn't help either

wszypelt, 2024-09-20:
@lyakh is everything working now? Can we close this issue?

marc-hb, 2024-09-20:
@wszypelt the last words are "... and it didn't help either"

?

lyakh, 2024-09-23:
> @wszypelt the last words are "... and it didn't help either"
> 
> ?

@marc-hb that was a reply to a specific question, not a statement about the state of this bug

lyakh, 2024-09-23:
> @lyakh is everything working now? Can we close this issue?

@wszypelt yes, so far https://github.com/zephyrproject-rtos/zephyr/pull/78283 has fixed it

closed 2024-09-23 by lyakh