thesofproject/sof #9271: [BUG] sof-audio-pci-intel-mtl: no matching blob

VitalyOstanin, 2024-07-01:
**Describe the bug**
Internal mic does not work on laptop Huawei matebook x pro 2024 U9

using sof-bin v2024.03
kernel 6.8.0-36-generic on Ubuntu 24.04 + oracular unstable
internal speakers work well using pipewire/wireplumber

dmesg (full output is attached as well):
```
sof-audio-pci-intel-mtl 0000:00:1f.3: DSP detected with PCI class/subclass/prog-if 0x040100
sof-audio-pci-intel-mtl 0000:00:1f.3: DSP detected with PCI class/subclass/prog-if info 0x040100
sof-audio-pci-intel-mtl 0000:00:1f.3: Digital mics found on Skylake+ platform, using SOF driver
sof-audio-pci-intel-mtl 0000:00:1f.3: DSP detected with PCI class/subclass/prog-if 0x040100
sof-audio-pci-intel-mtl 0000:00:1f.3: bound 0000:00:02.0 (ops i915_audio_component_bind_ops [i915])
sof-audio-pci-intel-mtl 0000:00:1f.3: use msi interrupt mode
sof-audio-pci-intel-mtl 0000:00:1f.3: hda codecs found, mask 5
sof-audio-pci-intel-mtl 0000:00:1f.3: using HDA machine driver skl_hda_dsp_generic now
sof-audio-pci-intel-mtl 0000:00:1f.3: DMICs detected in NHLT tables: 4
sof-audio-pci-intel-mtl 0000:00:1f.3: Firmware paths/files for ipc type 1:
sof-audio-pci-intel-mtl 0000:00:1f.3:  Firmware file:     intel/sof-ipc4/mtl/sof-mtl.ri
sof-audio-pci-intel-mtl 0000:00:1f.3:  Firmware lib path: intel/sof-ipc4-lib/mtl
sof-audio-pci-intel-mtl 0000:00:1f.3:  Topology file:     intel/sof-ace-tplg/sof-hda-generic-4ch.tplg
sof-audio-pci-intel-mtl 0000:00:1f.3: Loaded firmware library: ADSPFW, version: 2.9.0.1
sof-audio-pci-intel-mtl 0000:00:1f.3: Booted firmware version: 2.9.0.1
sof-audio-pci-intel-mtl 0000:00:1f.3: Topology: ABI 3:29:0 Kernel ABI 3:23:0
sof-audio-pci-intel-mtl 0000:00:1f.3: no matching blob for sample rate: 48000 sample width: 32 channels: 4
sof-audio-pci-intel-mtl 0000:00:1f.3: failed to prepare widget dai-copier.DMIC.dmic01.capture
sof-audio-pci-intel-mtl 0000:00:1f.3: Failed to prepare connected widgets
sof-audio-pci-intel-mtl 0000:00:1f.3: error: failed widget list set up for pcm 6 dir 1
sof-audio-pci-intel-mtl 0000:00:1f.3: ASoC: error at snd_soc_pcm_component_hw_params on 0000:00:1f.3: -22
```

**To Reproduce**
Install Ubuntu 24.04 - does not work.
Update ubuntu to latest noble-proposed/oracular unstable - does not work.

**Reproduction Rate**
Always

**Expected behavior**
Working internal mic.

**Impact**
showstopper
[dmesg.sof.gz](https://github.com/user-attachments/files/16050475/dmesg.sof.gz)

lgirdwood, 2024-07-01:
@bardliao @plbossart looks like NHLT is reporting 4 DMICs, but could it be missing a valid DMIC config blob ? (I'm assuming driver would select DMIC NHLT if it looked valid).

VitalyOstanin, 2024-07-01:
I'm sorry, this issue looks like a duplicate of
https://github.com/thesofproject/linux/issues/4973

VitalyOstanin, 2024-07-01:
Duplicate of
https://github.com/thesofproject/linux/issues/4973

bardliao, 2024-07-02:
@VitalyOstanin Can you try with `options snd_sof_intel_hda_common sof_use_tplg_nhlt=1`? You can add the option to `/etc/modprobe.d/alsa-base.conf`

VitalyOstanin, 2024-07-02:
@bardliao 
Yes, I've tried it as suggested in
https://github.com/thesofproject/linux/issues/4973#issuecomment-2096173970
and mic doesn't work.

I think the problem is fully investigated in
https://github.com/thesofproject/linux/issues/4973
and I have exactly the same laptop model.

All I need is to wait for new ubuntu kernel 6.10 + fixed sof-hda-generic-4ch.tplg or sof release 2.10. I can't build custom ubuntu kernel myself yet.

VitalyOstanin, 2024-07-02:
Finally recording from internal mic works.

kernel 6.10.0-9005-oem

sof-hda-generic-4ch.tplg from
https://github.com/thesofproject/linux/issues/4973#issuecomment-2099838538
```
arecord -fS32_LE -r 48000 -D hw:0,6 -c4 -d 5 test.wav
aplay test.wav
```
Don't forget to enable capture on mic channels in alsamixer: F6/Select sof-hda-dsp/F5/Press space on dmic0 Front/Rear.

reyhanboim, 2024-07-04:
I have the problem also on Huawei Matebook D14. in the settings menu there is only a dummy output

kv2019i, 2024-07-09:
Thanks @VitalyOstanin  . I'll close this FW bug instance as duplicate of LInux kernel driver bug https://github.com/thesofproject/linux/issues/4973

@reyhanboim If the fixes in 4973 don't help on your device, can you please file a new bug with Matebook D14 in the title (use https://github.com/thesofproject/linux/issues/4973 as example)?

closed 2024-07-09 by kv2019i