thesofproject/sof #9283: [BUG] No sound after fresh install of voidlinux

lambda2501, 2024-07-05:
**Describe the bug**
No sound at all after a fresh install of VoidLinux, and sof-firmware from xbps repositories.

VoidLinux version : `2024-03-14`
Kernel : `Linux 6.6.35_1 x86_64`
SoF firmware package : `sof-firmware-2023.12.1_1`
Result of `lspci` : `0000:00:1f.3 Multimedia audio controller: Intel Corporation Alder Lake Smart Sound Sound Technology Audio Controller (rev 01)`
Result of `aplay -l` : 
```
**** Liste des périphériques matériels PLAYBACK ****
carte 0 : sofsoundwire [sof-soundwire], périphérique 2 : Speaker (*) []
   Sous-périphérique : 1/1
   Sous-périphérique #0 : subdevice #0
carte 0 : sofsoundwire [sof-soundwire], périphérique 5 : HDMI 1 (*) []
   Sous-périphérique : 1/1
   Sous-périphérique #0 : subdevice #0
carte 0 : sofsoundwire [sof-soundwire], périphérique 6 : HDMI 2 (*) []
   Sous-périphérique : 1/1
   Sous-périphérique #0 : subdevice #0
carte 0 : sofsoundwire [sof-soundwire], périphérique 7 : HDMI 3 (*) []
   Sous-périphérique : 1/1
   Sous-périphérique #0 : subdevice #0
```

Result of `aplay -L` (nothing):
```
null
   Discard all samples (playback) or generate zero samples (capture)
```

dmesg output (with grep sof) :
```
 [    4.333904] sof-audio-pci-intel-tgl 0000:00:1f.3: DSP detected with PCI class/subclass/prog-if info 0x040100
 [    4.333944] sof-audio-pci-intel-tgl 0000:00:1f.3: SoundWire enabled on CannonLake+ platform, using SOF driver
 [    4.333974] sof-audio-pci-intel-tgl 0000:00:1f.3: enabling device (0000 -> 0002)
 [    4.334554] sof-audio-pci-intel-tgl 0000:00:1f.3: DSP detected with PCI class/subclass/prog-if 0x040100
 [    4.334727] sof-audio-pci-intel-tgl 0000:00:1f.3: bound 0000:00:02.0 (ops i915_audio_component_bind_ops [i915])
 [    4.341580] sof-audio-pci-intel-tgl 0000:00:1f.3: use msi interrupt mode
 [    4.359560] sof-audio-pci-intel-tgl 0000:00:1f.3: DMICs detected in NHLT tables: 0
 [    4.359568] sof-audio-pci-intel-tgl 0000:00:1f.3: hda codecs found, mask 4
 [    4.362027] sof-audio-pci-intel-tgl 0000:00:1f.3: Firmware info: version 2:2:0-57864
 [    4.362033] sof-audio-pci-intel-tgl 0000:00:1f.3: Firmware: ABI 3:22:1 Kernel ABI 3:23:0
 [    4.362046] sof-audio-pci-intel-tgl 0000:00:1f.3: unknown sof_ext_man header type 3 size 0x30
 [    4.463629] sof-audio-pci-intel-tgl 0000:00:1f.3: Firmware info: version 2:2:0-57864
 [    4.463641] sof-audio-pci-intel-tgl 0000:00:1f.3: Firmware: ABI 3:22:1 Kernel ABI 3:23:0
 [    4.489361] sof-audio-pci-intel-tgl 0000:00:1f.3: Topology: ABI 3:22:1 Kernel ABI 3:23:0
 [    4.489740] sof_sdw sof_sdw: ASoC: Parent card not yet available, widget card binding deferred
 [    4.513034] sof_sdw sof_sdw: hda_dsp_hdmi_build_controls: no PCM in topology for HDMI converter 3
 [    4.536795] input: sof-soundwire HDMI/DP,pcm=5 as /devices/pci0000:00/0000:00:1f.3/sof_sdw/sound/card0/input22
 [    4.536971] input: sof-soundwire HDMI/DP,pcm=6 as /devices/pci0000:00/0000:00:1f.3/sof_sdw/sound/card0/input23
 [    4.537123] input: sof-soundwire HDMI/DP,pcm=7 as /devices/pci0000:00/0000:00:1f.3/sof_sdw/sound/card0/input24
```

**To Reproduce**
Install a fresh voidlinux, update/upgrade distrib, then installing alsa/sof packages :
```
$ xbps-install -Su
$ xbps-install alsa-tools alsa-utils alsa-firmware sof-firmware
```

**Reproduction Rate**
In 2 same installation retries, 100%

**Expected behavior**
Get sound from speakers/headphones

**Impact**
Very annoying, obviously

**Others**
I can provide any other information if needed. Just ask.

lgirdwood, 2024-07-05:
@lambda2501 do you have a dmesg from your previous Linux install ? it looks like an ingredient is missing that the driver needs to bind all the endpoints. Adding @plbossart @bardliao and @ujfalusi .

lambda2501, 2024-07-05:
I have no previous install. It's a new Dell XPS 13 9315 (2022 I think). It is delivered with Windows that I wiped out.
I have another computer, an older XPS 13 with voidlinux too, no problem, but it is dying, that's why I bought a new one. The soundcard is not the same, so I can't relate.

If you need the whole dmesg, I can share it.

plbossart, 2024-07-05:
please add this file 
[sof-dyndbg.conf.txt](https://github.com/user-attachments/files/16112091/sof-dyndbg.conf.txt)
as /etc/modprobe.d/sof-dyndbg.conf, reboot and attach the dmesg log and the output of 'alsa-info'.