thesofproject/sof #9285: [BUG] HiFi3 sat_int32 Saturation Functions Incorrectly Handle INT64_MAX and INT64_MIN Edge Cases

ShriramShastry, 2024-07-06:
**Describe the bug**
A clear and concise description of what the bug is.
What have you tried to diagnose or workaround this issue?

The HiFi3 `sat_int32(int64_t x)` 

[https://github.com/thesofproject/sof/blob/15e57eec60bdf1c90f04bb4d8e03874a52ac044e/src/include/sof/audio/format_hifi3.h](url) 

saturation functions for 32-bit integers are not correctly handling saturation for INT64_MAX and INT64_MIN values. The expected behavior is for these edge cases to return INT32_MAX and INT32_MIN, respectively, matching the original Branching implementation. However, the HiFi3 optimized functions are returning incorrect results for these cases

Please also read https://thesofproject.github.io/latest/contribute/process/bug-tracking.html for further information on submitting bugs.

**To Reproduce**
Steps to reproduce the behavior: (e.g. list commands or actions used to reproduce the bug)

We have used HiFi3  saturation functions . 

```
static inline int32_t sat_int32(int64_t x)
{
	return (ae_int32)AE_ROUND32F48SSYM(AE_SLAI64(x, 16));
}
```

The functions fail to handle the edge cases of INT64_MAX and INT64_MIN correctly. 

**Reproduction Rate**
How often does the issue happen ? i.e. 1/10 (once in ten attempts), 1/1000 or all the time.
Always reproducible.

Does the reproduction rate vary with any other configuration or user action, if so please describe and show the new reproduction rate.

**Expected behavior**
A clear and concise description of what you expected to happen.

With input values of `-9223372036854775808` and `9223372036854775807` . It is expected to return `-2147483648` and `2147483647` but we are observing `0` and `-1` 

**Impact**
What impact does this issue have on your progress (e.g., annoyance, showstopper)

Incorrect return values from saturation functions can compromise the correctness, reliability, and performance of a system. It is crucial to ensure that saturation functions behave correctly for all input values, including edge cases, to maintain system integrity and prevent unexpected behavior.

Example
(1) Signal Processing: Incorrect saturation can cause clipping or distortion in the audio output, leading to a poor user experience
(2) Data Integrity:  With the saturation logic fails for edge cases like INT64_MAX and INT64_MIN, it could result in corrupted data being processed or stored
(3) Numerical Algorithms:  Incorrect saturation can lead to erroneous results, which might propagate through the system and cause larger errors or system failures.
(4) Performance: Since sat_int32() is a part performance critical loop, this will reduce overall performance of the system

**Environment**
1) Branch name and commit hash of the 2 repositories: sof (firmware/topology) and linux (kernel driver).
    * Kernel: {SHA}
    * SOF: {SHA}
2) Name of the topology file
    * Topology: {FILE}
3) Name of the platform(s) on which the bug is observed.
    * Platform: {PLATFORM}

**Screenshots or console output**
If applicable, add a screenshot (drag-and-drop an image), or console logs
(cut-and-paste text and put a code fence (\`\`\`) before and after, to help
explain the issue.

[image]

Please also include the relevant sections from the firmware log and kernel log in the report (and attach the full logs for complete reference). Kernel log is taken from *dmesg* and firmware log from *sof-logger*. See https://thesofproject.github.io/latest/developer_guides/debugability/logger/index.html

kv2019i, 2024-09-05:
@ShriramShastry Does #9344 address this completely or still work to do?

ShriramShastry, 2024-09-05:
> @ShriramShastry Does #9344 address this completely or still work to do?

Yes, it is address by #9344

closed 2024-09-05 by kv2019i