thesofproject/sof #9292: [BUG] Cadence codec is dead code in-tree, needs a stub

andyross, 2024-07-09:
The Cadence Codec component is dead code in the tree and not built in CI or via any testing with open source tooling.  This means it can't be exercised when APIs change and needs to play catch-up via out-of-tree testing.  See for example #9287, which had to be rushed in to implement an API change from #9261 that missed it because nothing built it.

Check out the following kconfigs and the code they gate for examples:

* CONFIG_DTS_CODEC_STUB
* CONFIG_GOOGLE_RTC_AUDIO_PROCESSING_MOCK
* CONFIG_COMP_IGO_NR_STUB
* CONFIG_WAVES_CODEC_STUB
* CONFIG_COMP_RTNR_STUB
* CONFIG_MAXIM_DSM_STUB

Then enable the component in `app/stub_build_all_ipc3/4.conf` file (and/or set it to default =y if CONFIG_COMP_STUBS=y) and CI will pick it up automatically and build it vs. all new changes.

dbaluta, 2024-07-12:
@andyross thanks for creating this. I will take care of it.

AnneOnciulescu, 2024-07-12:
@dbaluta  please assign me to this.

closed 2024-07-30 by dbaluta