thesofproject/sof #9295: [BUG] LNLM_RVP_RT712 and LNLM_RVP_RT722 fails to load topology

ssavati, 2024-07-10:
**Describe the bug**

LNLM_RVP_RT712 and LNLM_RVP_RT722 fails to load topology but it works for LNLM_SDW_AIOC

we seen similar failure on MTLP_RVP_SDW and it fixed by PR https://github.com/thesofproject/sof/pull/9284

```
Jul 10 13:20:44 ba-lnlm-rvp-sdw-02 kernel: sof_sdw sof_sdw: ASoC: physical link iDisp1 (id 5) not exist
Jul 10 13:20:44 ba-lnlm-rvp-sdw-02 kernel: sof_sdw sof_sdw: ASoC: topology: could not load header: -22
Jul 10 13:20:44 ba-lnlm-rvp-sdw-02 kernel: sof-audio-pci-intel-lnl 0000:00:1f.3: error: tplg component load failed -22
Jul 10 13:20:44 ba-lnlm-rvp-sdw-02 kernel: sof-audio-pci-intel-lnl 0000:00:1f.3: error: failed to load DSP topology -22
Jul 10 13:20:44 ba-lnlm-rvp-sdw-02 kernel: sof-audio-pci-intel-lnl 0000:00:1f.3: ASoC: error at snd_soc_component_probe on 0000:00:1f.3: -22
Jul 10 13:20:44 ba-lnlm-rvp-sdw-02 kernel: sof_sdw sof_sdw: ASoC: failed to instantiate card -22
Jul 10 13:20:44 ba-lnlm-rvp-sdw-02 kernel: sof_sdw sof_sdw: error -EINVAL: snd_soc_register_card failed -22
Jul 10 13:20:44 ba-lnlm-rvp-sdw-02 kernel: sof_sdw sof_sdw: probe with driver sof_sdw failed with error -22
```

**To Reproduce**
Deploy latest SOF and Kernel and boot devices

**Reproduction Rate**
100%

**Expected behavior**
Topology should load and audio should list without any error

**Impact**
Devices can not be used for validation

**Environment**
1) Branch name and commit hash of the 2 repositories: sof (firmware/topology) and linux (kernel driver).
 Linux Branch: 
[topic/sof-dev](https://github.com/thesofproject/linux/tree/topic/sof-dev)
Linux Commit: 
[8245a1411fef](https://github.com/thesofproject/linux/commit/8245a1411fef45a208f182c176660107b7f582aa)
   SOF Branch: 
[main](https://github.com/thesofproject/sof/tree/main)
SOF Commit: 
[2a9b40f8a69a](https://github.com/thesofproject/sof/commit/2a9b40f8a69a41713c1f51437099f55f7d549172)

2) Name of the topology file
    * Topology:  
    * LNLM_RVP_RT722 : sof-lnl-rt722-l0.tplg
    * LNLM_RVP_RT712 : sof-lnl-rt712-l2-rt1712-l3.tplg
   
3) Name of the platform(s) on which the bug is observed.
    * Platform: LNLM_RVP_RT712 and LNLM_RVP_RT722

ssavati, 2024-07-12:
Added P1 label as our two of CI devices are blocked for CI validation and missing SDW validation coverage.

plbossart, 2024-07-15:
@ssavati can you attach or point to the dmesg log? It's a simple fix but we need to know what the registered index is - the dailink default 5 doesn't seem to work. Thanks.

ssavati, 2024-07-15:
@plbossart here is dmesg logs  [dmesg_hdmi_fail.txt](https://github.com/user-attachments/files/16233280/dmesg_hdmi_fail.txt)

plbossart, 2024-07-15:
I don't think you added the relevant log @ssavati, it mentions rt711 and  rt1316, the issue is about rt712 and  rt722

ssavati, 2024-07-15:
> I don't think you added the relevant log @ssavati, it mentions rt711 and rt1316, the issue is about rt712 and rt722

sorry i have attached wrong log files. here attaching new logs files

[dmesg_722.txt](https://github.com/user-attachments/files/16236157/dmesg_722.txt)
[dmesg_712.txt](https://github.com/user-attachments/files/16236163/dmesg_712.txt)

bardliao, 2024-07-16:
@ssavati @plbossart After adding `options snd_soc_sdw_utils dyndbg=+pmf` to sof-dyndbg.conf, I will get below dmesg log
```
[    5.453986] snd_soc_sof_sdw:sof_card_dai_links_create: sof_sdw sof_sdw: sdw 4, ssp 0, dmic 2, hdmi 0, bt: 0
[    5.453989] snd_soc_sdw_utils:asoc_sdw_init_dai_link: sof_sdw sof_sdw: create dai link SDW0-Playback-SimpleJack, id 0
[    5.454027] snd_soc_sdw_utils:asoc_sdw_init_dai_link: sof_sdw sof_sdw: create dai link SDW0-Capture-SimpleJack, id 1
[    5.454030] snd_soc_sdw_utils:asoc_sdw_init_dai_link: sof_sdw sof_sdw: create dai link SDW0-Playback-SmartAmp, id 2
[    5.454032] snd_soc_sdw_utils:asoc_sdw_init_dai_link: sof_sdw sof_sdw: create dai link SDW0-Capture-SmartMic, id 4
[    5.454033] snd_soc_sdw_utils:asoc_sdw_init_dai_link: sof_sdw sof_sdw: create dai link dmic01, id 5
[    5.454035] snd_soc_sdw_utils:asoc_sdw_init_dai_link: sof_sdw sof_sdw: create dai link dmic16k, id 6
[    5.454037] snd_soc_sdw_utils:asoc_sdw_init_dai_link: sof_sdw sof_sdw: create dai link iDisp1, id 7
[    5.454038] snd_soc_sdw_utils:asoc_sdw_init_dai_link: sof_sdw sof_sdw: create dai link iDisp2, id 8
[    5.454040] snd_soc_sdw_utils:asoc_sdw_init_dai_link: sof_sdw sof_sdw: create dai link iDisp3, id 9
```
The problem is that there are both SDW DMIC and PCH DMIC links are created. For rt712 and rt722, we may want to disable PCH DMIC from BIOS.

plbossart, 2024-07-16:
Thanks @bardliao, please make sure this change is added to the CI default.

I think it's indeed a configuration issue at the BIOS level, it makes no sense to have both NHLT definitions for DMIC and SoundWire definitions for SoundWire mic codec.

@ssavati please update the boards and let us know the resuilt

ssavati, 2024-07-16:
> @ssavati please update the boards and let us know the resuilt

Sure I will check PR https://github.com/thesofproject/sof/pull/9302 with  DMIC disabled

plbossart, 2024-07-16:
I think you just need main with DMIC disabled, PR #9302 is probably incorrect.

fredoh9, 2024-07-16:
I quickly checked with LNLM_RVP_RT722.
In BIOS we need to disable both places

1. disable DMIC #1 & DMIC #2 in HD Audio config
2. disable DMIC Quad (or DMIC stereo) in DSP Feature

After this, I'm able to load sof-lnl-rt722-l0.tplg

ssavati, 2024-07-17:
@fredoh9 thanks for confirming. I too checked with the above BIOS settings and devices are listed.

@plbossart @kv2019i Is this default BIOS settings going forward for 712/722 ? or is it's workaround till the actual issue is fixed in topology/Kernel ? 
I mean can we close this issue or keep it open ?

plbossart, 2024-07-17:
These BIOS settings are going to be the default IMHO, so we can close the issue.

closed 2024-07-17 by plbossart