thesofproject/sof #9303: [BUG] For ADL-N Varient Soundwire topology load fails due BT can't connect DAI

ssavati, 2024-07-15:
**Describe the bug**
For ADL-N Varient Soundwire topology load fails due BT can't connect DAI

```
[    7.104085] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: error: can't connect DAI SSP2.OUT stream SSP2-BT
[    7.104090] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: failed to add widget type 27 name : SSP2.OUT stream SSP2-BT
[    7.104099] kernel: sof_sdw sof_sdw: ASoC: failed to load widget SSP2.OUT
[    7.104102] kernel: sof_sdw sof_sdw: ASoC: topology: could not load header: -22
[    7.104171] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: error: tplg component load failed -22
[    7.104182] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: error: failed to load DSP topology -22
[    7.104185] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ASoC: error at snd_soc_component_probe on 0000:00:1f.3: -22
[    7.104216] kernel: sof_sdw sof_sdw: ASoC: failed to instantiate card -22
[    7.110450] kernel: sof_sdw sof_sdw: error -EINVAL: snd_soc_register_card failed -22
[    7.110515] kernel: sof_sdw sof_sdw: probe with driver sof_sdw failed with error -22
```

**To Reproduce**
Boot system with firmware and toplogy built from SOF branch : cavs2.5-001-drop-stable

**Reproduction Rate**
100%

**Expected behavior**
Linux boots and sound card listed 

**Impact**
All test validtaion blocked

**Environment**
1) Branch name and commit hash of the 2 repositories: sof (firmware/topology) and linux (kernel driver).
    * Linux Branch:  [topic/sof-dev](https://github.com/thesofproject/linux/tree/topic/sof-dev)
      Linux Commit: [8245a1411fef](https://github.com/thesofproject/linux/commit/8245a1411fef45a208f182c176660107b7f582aa)
    * SOF Branch: [cavs2.5-001-drop-stable](https://github.com/thesofproject/sof/tree/cavs2.5-001-drop-stable)
      SOF Commit: [84d4de2d91dc](https://github.com/thesofproject/sof/commit/84d4de2d91dcda46d296a219b213f621f9c6314c)

2) Name of the topology file
    * Topology: sof-adl-rt711-4ch.tplg
3) Name of the platform(s) on which the bug is observed.
    * Platform: ADL-N Varient

ssavati, 2024-07-15:
@kv2019i  If we use the Topology file built from sof branch stable-v2.10. Issue not observed
Initially thought like the Remove BT_OFFLOAD patch (patch[ link](https://github.com/thesofproject/sof/commit/a5ec5f84143b2312e74d4daefc6d0a923c4d3c02)) missing in cavs2.5-001-drop-stable but when checked [CMakeLists.txt](https://github.com/thesofproject/sof/blob/cavs2.5-001-drop-stable/tools/topology/topology1/CMakeLists.txt). sof-adl-rt711-4ch does not have any BT_OFFLOAD. Not sure what else missing here.

```
"sof-tgl-rt711-rt1308\;sof-adl-rt711-4ch\;-DCHANNELS=4\;-DDMICPROC_FILTER1=eq_iir_coef_highpass_40hz_20db_48khz.m4\;-DDMIC16KPROC_FILTER1=eq_iir_coef_highpass_40hz_20db_16khz.m4\;-DPLATFORM=adl"
	"sof-tgl-rt711-rt1308\;sof-adl-rt711\;-DCHANNELS=0\;-DPLATFORM=adl"
```

plbossart, 2024-07-16:
@ssavati what are you trying to test? cavs2.5-001-drop-stable and  stable-v2.10 are completely different IPCs.

ssavati, 2024-07-16:
@plbossart  Yes it's different. It's created for the upcoming  platform ADL-N Variant with IPC3 for upcoming milestone.

plbossart, 2024-07-16:
are you sure you're using the correct branch? https://github.com/thesofproject/sof/commit/a5ec5f84143b2312e74d4daefc6d0a923c4d3c02 was committed on stable-2.2

kv2019i, 2024-07-17:
I think this is ok. stable-v2.2 is the usptream SOF branch (for all vendors) and happens to be the last stable branch where Intel has support for IPC3 platforms (including this one). This is the branch where we pick FW binaries and topologies for upstream sof-bin releases. So @plbossart 's patch https://github.com/thesofproject/sof/pull/9275 is a needed one.

cavs2.5-001-drop-stable is an Intel maintained and Intel specific branch for a set of hardware and this is only used for Chromebooks and this branch never goes to upstream sof-bin releases.

So I'd say we can fix this as a wont-fix. To test upstream topologies on these platforms, one should use stable-v2.2 branch, which is the upstream main for these platforms.

plbossart, 2024-07-18:
we are all in agreement, let's close this one.

closed 2024-07-18 by plbossart