thesofproject/sof #9307: [BUG] SoundWire topologies: headset and mic capture are not processed consistently between IPC3 and IPC4 solutions

plbossart, 2024-07-17:
**Describe the bug**

in IPC3 products, the headset capture does not have any EQIIR processing, while the mic capture does.

[image]

In IPC4 products, the headset capture does have an EQIIR processing while the mic capture does not.

[image]

This is inconsistent and leads to confusion on capabilities, as noted in https://github.com/thesofproject/linux/issues/3766

**To Reproduce**
use existing topologies

**Reproduction Rate**

100%

**Expected behavior**

Consistent behavior for all new IPC4 products. We are not going to fix IPC3 at this point.

**Impact**

user experience, specifically pop-clicks on capture start

@singalsu please chime in with recommendations.

We can 
a) add the EQIIR on mic capture and keep it active
b) remove EQIIR on all paths
c) all the EQIIR on mic capture but have the default as bypassed.

singalsu, 2024-07-19:
Option a) probably. A low frequency cutoff highpass (e.g. 40 Hz) is a good bet for all acoustical mic applications. Option c) also works, we will be able to use UCM to set up the EQ when need.

Option b) is not feasible. The IIRs have been added after reports about strong DC pulses in capture begin or with headset button presses.

plbossart, 2024-07-19:
ok for a), makes sense to deal with button-press and other shenanigans, and IPC4 topology will be better than IPC3 ones.

@ranj063 is this something we can do with the pipeline extension stuff or are there still some dependencies to be worked out with the topology compiler?

kv2019i, 2024-09-05:
@singalsu v2.12 stuff?

plbossart, 2024-09-09:
I added the EQIIR for all IPC4 capture path in 
https://github.com/thesofproject/sof/pull/9331

So I think this can be closed?

closed 2024-09-09 by kv2019i