thesofproject/sof #9339: [BUG] Intel Corporation Tiger Lake error: hda_dsp_core_reset_enter: timeout on HDA_DSP_REG_ADSPCS read

real-t0mg, 2024-07-29:
On ThinkPad T14s Gen 2i, kernel 6.6.43, Tiger Lake-LP Smart Sound Technology Audio Controller firmware loading fails with the following message: 

```
root@haxboxen:~# date 
Mon Jul 29 08:38:41 -03 2024
root@haxboxen:~# uname -a
Linux haxboxen.lan 6.6.43-ST1NKPAD #1 SMP PREEMPT Sun Jul 28 21:09:38 -03 2024 x86_64 11th Gen Intel(R) Core(TM) i7-1165G7 @ 2.80GHz GenuineIntel GNU/Linux
root@haxboxen:~# dmidecode | grep -i lenovo 
        Vendor: LENOVO
        Manufacturer: LENOVO
        SKU Number: LENOVO_MT_20WN_BU_Think_FM_ThinkPad T14s Gen 2i
        Manufacturer: LENOVO
        Manufacturer: LENOVO
root@haxboxen:~# dmesg | grep -E 'sof-|snd|error'
[    2.486683] snd_hda_intel 0000:00:1f.3: DSP detected with PCI class/subclass/prog-if info 0x040380
[    2.490542] snd_hda_intel 0000:00:1f.3: Digital mics found on Skylake+ platform, using SOF driver
[    2.558058] sof-audio-pci-intel-tgl 0000:00:1f.3: DSP detected with PCI class/subclass/prog-if info 0x040380
[    2.561240] sof-audio-pci-intel-tgl 0000:00:1f.3: Digital mics found on Skylake+ platform, using SOF driver
[    2.565309] sof-audio-pci-intel-tgl 0000:00:1f.3: DSP detected with PCI class/subclass/prog-if 0x040380
[    2.568591] sof-audio-pci-intel-tgl 0000:00:1f.3: bound 0000:00:02.0 (ops 0xffffffff86f1bfd0)
[    2.579079] sof-audio-pci-intel-tgl 0000:00:1f.3: use msi interrupt mode
[    2.605083] sof-audio-pci-intel-tgl 0000:00:1f.3: hda codecs found, mask 5
[    2.607598] sof-audio-pci-intel-tgl 0000:00:1f.3: using HDA machine driver skl_hda_dsp_generic now
[    2.610140] sof-audio-pci-intel-tgl 0000:00:1f.3: DMICs detected in NHLT tables: 2
[    5.672888] sof-audio-pci-intel-tgl 0000:00:1f.3: hda_cl_copy_fw: timeout with rom_status_reg (0x80000) read
[    5.673419] sof-audio-pci-intel-tgl 0000:00:1f.3: ------------[ DSP dump start ]------------
[    5.673420] sof-audio-pci-intel-tgl 0000:00:1f.3: Firmware download failed
[    5.673421] sof-audio-pci-intel-tgl 0000:00:1f.3: fw_state: SOF_FW_BOOT_IN_PROGRESS (3)
[    5.673437] sof-audio-pci-intel-tgl 0000:00:1f.3: 0x80000006: module: ROM, state: VERIFY_FEATURE_MASK, not running
[    5.673443] sof-audio-pci-intel-tgl 0000:00:1f.3: error code: 0x14 (unknown)
[    5.673455] sof-audio-pci-intel-tgl 0000:00:1f.3: extended rom status:  0x80000006 0x14 0x0 0x0 0x0 0x0 0x25101c2 0x0
[    5.673456] sof-audio-pci-intel-tgl 0000:00:1f.3: ------------[ DSP dump end ]------------
[    5.673496] sof-audio-pci-intel-tgl 0000:00:1f.3: Failed to start DSP
[    5.673498] sof-audio-pci-intel-tgl 0000:00:1f.3: error: failed to boot DSP firmware -110
[    5.723880] sof-audio-pci-intel-tgl 0000:00:1f.3: error: hda_dsp_core_reset_enter: timeout on HDA_DSP_REG_ADSPCS read
[    5.723884] sof-audio-pci-intel-tgl 0000:00:1f.3: error: dsp core reset failed: core_mask 1
[    5.724076] sof-audio-pci-intel-tgl 0000:00:1f.3: error: sof_probe_work failed err: -110
root@haxboxen:~# lspci -tv | grep -i sound
           +-1f.3  Intel Corporation Tiger Lake-LP Smart Sound Technology Audio Controller
root@haxboxen:~# 
```

plbossart, 2024-07-29:
@real-t0mg can you add this file 
[sof-dyndbg.conf.txt](https://github.com/user-attachments/files/16413307/sof-dyndbg.conf.txt)
as /etc/modprobe.d/sof-dyndbg.conf, reboot and attach the dmesg log? Thanks!

real-t0mg, 2024-07-29:
Hi @plbossart , will need to recompile kernel for that to work

```
root@haxboxen:~# dmesg | grep -E 'sof-|snd|error|dyndbg'
[    2.448945] dyndbg param is supported only in CONFIG_DYNAMIC_DEBUG builds
[    2.453032] dyndbg param is supported only in CONFIG_DYNAMIC_DEBUG builds
[    2.488211] snd_hda_intel 0000:00:1f.3: DSP detected with PCI class/subclass/prog-if info 0x040380
[    2.490870] snd_hda_intel 0000:00:1f.3: Digital mics found on Skylake+ platform, using SOF driver
[    2.498746] dyndbg param is supported only in CONFIG_DYNAMIC_DEBUG builds
[    2.543914] dyndbg param is supported only in CONFIG_DYNAMIC_DEBUG builds
[    2.554163] dyndbg param is supported only in CONFIG_DYNAMIC_DEBUG builds
[    2.558622] dyndbg param is supported only in CONFIG_DYNAMIC_DEBUG builds
[    2.564033] dyndbg param is supported only in CONFIG_DYNAMIC_DEBUG builds
[    2.566765] dyndbg param is supported only in CONFIG_DYNAMIC_DEBUG builds
[    2.569473] dyndbg param is supported only in CONFIG_DYNAMIC_DEBUG builds
[    2.572402] dyndbg param is supported only in CONFIG_DYNAMIC_DEBUG builds
[    2.577977] dyndbg param is supported only in CONFIG_DYNAMIC_DEBUG builds
[    2.580494] dyndbg param is supported only in CONFIG_DYNAMIC_DEBUG builds
[    2.583185] sof-audio-pci-intel-tgl 0000:00:1f.3: DSP detected with PCI class/subclass/prog-if info 0x040380
[    2.586844] sof-audio-pci-intel-tgl 0000:00:1f.3: Digital mics found on Skylake+ platform, using SOF driver
[    2.591289] sof-audio-pci-intel-tgl 0000:00:1f.3: DSP detected with PCI class/subclass/prog-if 0x040380
[    2.593660] sof-audio-pci-intel-tgl 0000:00:1f.3: bound 0000:00:02.0 (ops 0xffffffff83f1bfd0)
[    2.605068] sof-audio-pci-intel-tgl 0000:00:1f.3: use msi interrupt mode
[    2.623246] sof-audio-pci-intel-tgl 0000:00:1f.3: hda codecs found, mask 5
[    2.625383] sof-audio-pci-intel-tgl 0000:00:1f.3: using HDA machine driver skl_hda_dsp_generic now
[    2.628294] sof-audio-pci-intel-tgl 0000:00:1f.3: DMICs detected in NHLT tables: 2
[    5.692883] sof-audio-pci-intel-tgl 0000:00:1f.3: hda_cl_copy_fw: timeout with rom_status_reg (0x80000) read
[    5.693419] sof-audio-pci-intel-tgl 0000:00:1f.3: ------------[ DSP dump start ]------------
[    5.693427] sof-audio-pci-intel-tgl 0000:00:1f.3: Firmware download failed
[    5.693432] sof-audio-pci-intel-tgl 0000:00:1f.3: fw_state: SOF_FW_BOOT_IN_PROGRESS (3)
[    5.693442] sof-audio-pci-intel-tgl 0000:00:1f.3: 0x80000006: module: ROM, state: VERIFY_FEATURE_MASK, not running
[    5.693445] sof-audio-pci-intel-tgl 0000:00:1f.3: error code: 0x14 (unknown)
[    5.693456] sof-audio-pci-intel-tgl 0000:00:1f.3: extended rom status:  0x80000006 0x14 0x0 0x0 0x0 0x0 0x25101c2 0x0
[    5.693457] sof-audio-pci-intel-tgl 0000:00:1f.3: ------------[ DSP dump end ]------------
[    5.693485] sof-audio-pci-intel-tgl 0000:00:1f.3: Failed to start DSP
[    5.693486] sof-audio-pci-intel-tgl 0000:00:1f.3: error: failed to boot DSP firmware -110
[    5.743840] sof-audio-pci-intel-tgl 0000:00:1f.3: error: hda_dsp_core_reset_enter: timeout on HDA_DSP_REG_ADSPCS read
[    5.743841] sof-audio-pci-intel-tgl 0000:00:1f.3: error: dsp core reset failed: core_mask 1
[    5.745374] sof-audio-pci-intel-tgl 0000:00:1f.3: error: sof_probe_work failed err: -110
root@haxboxen:~# 
```

real-t0mg, 2024-07-29:
Attaching dmesg output as requested. Let me know if this works for debugging on your end:
[dmesg_log.txt](https://github.com/user-attachments/files/16414059/dmesg_log.txt)

```
tomg@haxboxen:~$  dmesg | grep -E 'sof-|snd|error|dyndbg'
[    2.485989] snd_hda_intel 0000:00:1f.3: DSP detected with PCI class/subclass/prog-if info 0x040380
[    2.490462] snd_intel_sdw_acpi:sdw_intel_scan_controller: acpi PRP00001:00: ACPI reports 4 SDW Link devices
[    2.492476] snd_intel_sdw_acpi:sdw_intel_scan_controller: acpi PRP00001:00: Link 0 not selected in firmware
[    2.492478] snd_intel_sdw_acpi:sdw_intel_scan_controller: acpi PRP00001:00: Link 1 not selected in firmware
[    2.492480] snd_intel_sdw_acpi:sdw_intel_scan_controller: acpi PRP00001:00: Link 2 not selected in firmware
[    2.492482] snd_intel_sdw_acpi:sdw_intel_scan_controller: acpi PRP00001:00: Link 3 not selected in firmware
[    2.492484] snd_hda_intel 0000:00:1f.3: Digital mics found on Skylake+ platform, using SOF driver
[    2.558147] sof-audio-pci-intel-tgl 0000:00:1f.3: DSP detected with PCI class/subclass/prog-if info 0x040380
[    2.564732] snd_intel_sdw_acpi:sdw_intel_scan_controller: acpi PRP00001:00: ACPI reports 4 SDW Link devices
[    2.564737] snd_intel_sdw_acpi:sdw_intel_scan_controller: acpi PRP00001:00: Link 0 not selected in firmware
[    2.564738] snd_intel_sdw_acpi:sdw_intel_scan_controller: acpi PRP00001:00: Link 1 not selected in firmware
[    2.564740] snd_intel_sdw_acpi:sdw_intel_scan_controller: acpi PRP00001:00: Link 2 not selected in firmware
[    2.564747] snd_intel_sdw_acpi:sdw_intel_scan_controller: acpi PRP00001:00: Link 3 not selected in firmware
[    2.564748] sof-audio-pci-intel-tgl 0000:00:1f.3: Digital mics found on Skylake+ platform, using SOF driver
[    2.567386] snd_sof_pci:sof_pci_probe: sof-audio-pci-intel-tgl 0000:00:1f.3: PCI DSP detected
[    2.567518] sof-audio-pci-intel-tgl 0000:00:1f.3: DSP detected with PCI class/subclass/prog-if 0x040380
[    2.570268] sof-audio-pci-intel-tgl 0000:00:1f.3: bound 0000:00:02.0 (ops 0xffffffff93f203b0)
[    2.583802] snd_sof_intel_hda_common:hda_dsp_ctrl_get_caps: sof-audio-pci-intel-tgl 0000:00:1f.3: checking for capabilities at offset 0xc00
[    2.583808] snd_sof_intel_hda_common:hda_dsp_ctrl_get_caps: sof-audio-pci-intel-tgl 0000:00:1f.3: found ML capability at 0xc00
[    2.583809] snd_sof_intel_hda_common:hda_dsp_ctrl_get_caps: sof-audio-pci-intel-tgl 0000:00:1f.3: checking for capabilities at offset 0x800
[    2.583814] snd_sof_intel_hda_common:hda_dsp_ctrl_get_caps: sof-audio-pci-intel-tgl 0000:00:1f.3: found DSP capability at 0x800
[    2.583815] snd_sof_intel_hda_common:hda_dsp_ctrl_get_caps: sof-audio-pci-intel-tgl 0000:00:1f.3: checking for capabilities at offset 0x500
[    2.583822] snd_sof_intel_hda_common:hda_dsp_ctrl_get_caps: sof-audio-pci-intel-tgl 0000:00:1f.3: found GTS capability at 0x500
[    2.583822] snd_sof_intel_hda_common:hda_dsp_ctrl_get_caps: sof-audio-pci-intel-tgl 0000:00:1f.3: checking for capabilities at offset 0x1f00
[    2.583826] snd_sof_intel_hda_common:hda_dsp_ctrl_get_caps: sof-audio-pci-intel-tgl 0000:00:1f.3: found DRSM capability at 0x1f00
[    2.583826] snd_sof_intel_hda_common:hda_dsp_ctrl_get_caps: sof-audio-pci-intel-tgl 0000:00:1f.3: checking for capabilities at offset 0x700
[    2.583833] snd_sof_intel_hda_common:hda_dsp_ctrl_get_caps: sof-audio-pci-intel-tgl 0000:00:1f.3: found SPIB capability at 0x700
[    2.583853] snd_sof_intel_hda_common:hda_dsp_stream_init: sof-audio-pci-intel-tgl 0000:00:1f.3: hda global caps = 0x1009701
[    2.583854] snd_sof_intel_hda_common:hda_dsp_stream_init: sof-audio-pci-intel-tgl 0000:00:1f.3: detected 9 playback and 7 capture streams
[    2.583980] sof-audio-pci-intel-tgl 0000:00:1f.3: use msi interrupt mode
[    2.587404] snd_sof_intel_hda_common:hda_dsp_probe: sof-audio-pci-intel-tgl 0000:00:1f.3: using IPC IRQ 183
[    2.587494] snd_sof:snd_sof_pci_update_bits_unlocked: sof-audio-pci-intel-tgl 0000:00:1f.3: Debug PCIR: 00000000 at  00000044
[    2.587496] snd_sof_intel_hda_common:hda_init_caps: sof-audio-pci-intel-tgl 0000:00:1f.3: PP capability, will probe DSP later.
[    2.590620] snd_sof:snd_sof_pci_update_bits_unlocked: sof-audio-pci-intel-tgl 0000:00:1f.3: Debug PCIR: 00000040 at  00000048
[    2.590627] snd_sof:snd_sof_pci_update_bits_unlocked: sof-audio-pci-intel-tgl 0000:00:1f.3: Debug PCIW: 00000000 at  00000048
[    2.595090] snd_sof_intel_hda:hda_codec_detect_mask: sof-audio-pci-intel-tgl 0000:00:1f.3: codec_mask = 0x5
[    2.595100] snd_sof:snd_sof_pci_update_bits_unlocked: sof-audio-pci-intel-tgl 0000:00:1f.3: Debug PCIR: 00000000 at  00000048
[    2.595102] snd_sof:snd_sof_pci_update_bits_unlocked: sof-audio-pci-intel-tgl 0000:00:1f.3: Debug PCIW: 00000040 at  00000048
[    2.596608] snd_sof_intel_hda_mlink:hda_bus_ml_init: sof-audio-pci-intel-tgl 0000:00:1f.3: HDAudio Multi-Link count: 2
[    2.596611] snd_sof_intel_hda_mlink:hdaml_lnk_enum: sof-audio-pci-intel-tgl 0000:00:1f.3: Link 0: HDAudio - lsdiid=3
[    2.596613] snd_sof_intel_hda_mlink:hdaml_lnk_enum: sof-audio-pci-intel-tgl 0000:00:1f.3: Link 1: HDAudio - lsdiid=4
[    2.596646] snd_intel_sdw_acpi:sdw_intel_scan_controller: acpi PRP00001:00: ACPI reports 4 SDW Link devices
[    2.596648] snd_intel_sdw_acpi:sdw_intel_scan_controller: acpi PRP00001:00: Link 0 not selected in firmware
[    2.596649] snd_intel_sdw_acpi:sdw_intel_scan_controller: acpi PRP00001:00: Link 1 not selected in firmware
[... cut ...]
[    5.724301] snd_sof_intel_hda_common:hda_dsp_core_reset_enter: sof-audio-pci-intel-tgl 0000:00:1f.3: FW Poll Status: reg[0x4]=0x1d003c timedout
[    5.724304] sof-audio-pci-intel-tgl 0000:00:1f.3: error: hda_dsp_core_reset_enter: timeout on HDA_DSP_REG_ADSPCS read
[    5.724305] sof-audio-pci-intel-tgl 0000:00:1f.3: error: dsp core reset failed: core_mask 1
[    5.724595] snd_sof:sof_set_fw_state: sof-audio-pci-intel-tgl 0000:00:1f.3: fw_state change: 4 -> 0
[    5.724598] sof-audio-pci-intel-tgl 0000:00:1f.3: error: sof_probe_work failed err: -110
tomg@haxboxen:~$ 

```

plbossart, 2024-07-29:
if you don't have dynamic debug enabled, chances are you are already in a rather specific configuration out of the beaten path... Please make sure you enable the ME engine and/or try with another distro with a USB  live image.

real-t0mg, 2024-07-29:
@plbossart  ME engine is enabled, and in current level per the latest lenovo available, it was even reinstalled as some old bug report recommended re-installation of the ME engine. Suggestions other than "options snd_intel_dspcfg dsp_driver=1" are welcome.

```
root@haxboxen:~# fwupdmgr reinstall 
Choose a device:
0.      Cancel
1.      2292ae5236790b47884e37cf162dcf23bfcd1c60 (Embedded Controller)
2.      349bb341230b1a86e5effe7dfe4337e1590227bd (Intel Management Engine)
3.      0d5d05911800242bb1f35287012cdcbd9b381148 (Prometheus)
4.      04e17fcf7d3de91da49a163ffe4907855c3648be (SKHynix HFS512GDE9X081N)
5.      a45df35ac0e948ee180fe216a5f703f32dda163f (System Firmware)
2
╔══════════════════════════════════════════════════════════════════════════════╗
║ Reinstall Intel Management Engine to 240.47.2473?                            ║
╠══════════════════════════════════════════════════════════════════════════════╣
║ Version 15.0.47.2473 (LVFS:240.47.2473)                                      ║
║                                                                              ║
║ • Fixed an issue where no video output might occurs when monitors            ║
║ connected to Thunderbolt Dock.                                               ║
║                                                                              ║
║ 20WNS85W00 must remain plugged into a power source for the duration of the   ║
║ update to avoid damage.                                                      ║
╚══════════════════════════════════════════════════════════════════════════════╝

Perform operation? [Y|n]: n
Request canceled
root@haxboxen:~# 

```

plbossart, 2024-07-29:
@mrhpearson FYI, not sure how to deal with this issue.

real-t0mg, 2024-07-29:
for the record, as this will hurt other ThinkPad users, I could only make it work with release version v2023.12.1, no other combination of sof-tgl.ri and sof-tplg/sof-hda-generic-2ch.tplg works for me. If you are in pain with a ThinkPad  T14s Gen 2i , use this release  (https://github.com/thesofproject/sof-bin/releases/tag/v2023.12.1), 

Next time, buy AMD. 

dmesg for the bug record
[dmesg_work_with_v2023.12.1.txt](https://github.com/user-attachments/files/16415368/dmesg_work_with_v2023.12.1.txt)

for the record:

```
tomg@haxboxen:~$ date 
Mon Jul 29 12:11:10 -03 2024
tomg@haxboxen:~$ uname -a
Linux haxboxen.lan 6.6.43-ST1NKPAD #2 SMP PREEMPT Mon Jul 29 10:25:09 -03 2024 x86_64 11th Gen Intel(R) Core(TM) i7-1165G7 @ 2.80GHz GenuineIntel GNU/Linux
tomg@haxboxen:~$ dmesg | grep -E 'sof-|snd|error|dyndbg' 
[    2.445353] snd_hda_intel 0000:00:1f.3: DSP detected with PCI class/subclass/prog-if info 0x040380
[    2.449457] snd_intel_sdw_acpi:sdw_intel_scan_controller: acpi PRP00001:00: ACPI reports 4 SDW Link devices
[    2.457100] snd_intel_sdw_acpi:sdw_intel_scan_controller: acpi PRP00001:00: Link 0 not selected in firmware
[    2.463556] snd_intel_sdw_acpi:sdw_intel_scan_controller: acpi PRP00001:00: Link 1 not selected in firmware
[    2.463558] snd_intel_sdw_acpi:sdw_intel_scan_controller: acpi PRP00001:00: Link 2 not selected in firmware
[    2.463560] snd_intel_sdw_acpi:sdw_intel_scan_controller: acpi PRP00001:00: Link 3 not selected in firmware
[    2.463562] snd_hda_intel 0000:00:1f.3: Digital mics found on Skylake+ platform, using SOF driver
[    2.517062] sof-audio-pci-intel-tgl 0000:00:1f.3: DSP detected with PCI class/subclass/prog-if info 0x040380
[    2.521326] snd_intel_sdw_acpi:sdw_intel_scan_controller: acpi PRP00001:00: ACPI reports 4 SDW Link devices
[    2.521328] snd_intel_sdw_acpi:sdw_intel_scan_controller: acpi PRP00001:00: Link 0 not selected in firmware
[    2.521330] snd_intel_sdw_acpi:sdw_intel_scan_controller: acpi PRP00001:00: Link 1 not selected in firmware
[    2.521331] snd_intel_sdw_acpi:sdw_intel_scan_controller: acpi PRP00001:00: Link 2 not selected in firmware
[    2.521332] snd_intel_sdw_acpi:sdw_intel_scan_controller: acpi PRP00001:00: Link 3 not selected in firmware
[    2.521333] sof-audio-pci-intel-tgl 0000:00:1f.3: Digital mics found on Skylake+ platform, using SOF driver
[    2.524628] snd_sof_pci:sof_pci_probe: sof-audio-pci-intel-tgl 0000:00:1f.3: PCI DSP detected
[    2.524739] sof-audio-pci-intel-tgl 0000:00:1f.3: DSP detected with PCI class/subclass/prog-if 0x040380
[    2.528548] sof-audio-pci-intel-tgl 0000:00:1f.3: bound 0000:00:02.0 (ops 0xffffffffaef203b0)
[    2.539878] snd_sof_intel_hda_common:hda_dsp_ctrl_get_caps: sof-audio-pci-intel-tgl 0000:00:1f.3: checking for capabilities at offset 0xc00
[    2.539884] snd_sof_intel_hda_common:hda_dsp_ctrl_get_caps: sof-audio-pci-intel-tgl 0000:00:1f.3: found ML capability at 0xc00
[    2.539885] snd_sof_intel_hda_common:hda_dsp_ctrl_get_caps: sof-audio-pci-intel-tgl 0000:00:1f.3: checking for capabilities at offset 0x800
[    2.539898] snd_sof_intel_hda_common:hda_dsp_ctrl_get_caps: sof-audio-pci-intel-tgl 0000:00:1f.3: found DSP capability at 0x800
[    2.539898] snd_sof_intel_hda_common:hda_dsp_ctrl_get_caps: sof-audio-pci-intel-tgl 0000:00:1f.3: checking for capabilities at offset 0x500
[    2.539901] snd_sof_intel_hda_common:hda_dsp_ctrl_get_caps: sof-audio-pci-intel-tgl 0000:00:1f.3: found GTS capability at 0x500
[    2.539902] snd_sof_intel_hda_common:hda_dsp_ctrl_get_caps: sof-audio-pci-intel-tgl 0000:00:1f.3: checking for capabilities at offset 0x1f00
[    2.539909] snd_sof_intel_hda_common:hda_dsp_ctrl_get_caps: sof-audio-pci-intel-tgl 0000:00:1f.3: found DRSM capability at 0x1f00
[    2.539909] snd_sof_intel_hda_common:hda_dsp_ctrl_get_caps: sof-audio-pci-intel-tgl 0000:00:1f.3: checking for capabilities at offset 0x700
[    2.539925] snd_sof_intel_hda_common:hda_dsp_ctrl_get_caps: sof-audio-pci-intel-tgl 0000:00:1f.3: found SPIB capability at 0x700
[    2.539948] snd_sof_intel_hda_common:hda_dsp_stream_init: sof-audio-pci-intel-tgl 0000:00:1f.3: hda global caps = 0x1009701
[    2.539949] snd_sof_intel_hda_common:hda_dsp_stream_init: sof-audio-pci-intel-tgl 0000:00:1f.3: detected 9 playback and 7 capture streams
[    2.540081] sof-audio-pci-intel-tgl 0000:00:1f.3: use msi interrupt mode
[    2.542438] snd_sof_intel_hda_common:hda_dsp_probe: sof-audio-pci-intel-tgl 0000:00:1f.3: using IPC IRQ 167
[    2.542505] snd_sof:snd_sof_pci_update_bits_unlocked: sof-audio-pci-intel-tgl 0000:00:1f.3: Debug PCIR: 00000000 at  00000044
[    2.542507] snd_sof_intel_hda_common:hda_init_caps: sof-audio-pci-intel-tgl 0000:00:1f.3: PP capability, will probe DSP later.
[    2.545357] snd_sof:snd_sof_pci_update_bits_unlocked: sof-audio-pci-intel-tgl 0000:00:1f.3: Debug PCIR: 00000040 at  00000048
[    2.545376] snd_sof:snd_sof_pci_update_bits_unlocked: sof-audio-pci-intel-tgl 0000:00:1f.3: Debug PCIW: 00000000 at  00000048
[    2.550592] snd_sof_intel_hda:hda_codec_detect_mask: sof-audio-pci-intel-tgl 0000:00:1f.3: codec_mask = 0x5
[    2.550600] snd_sof:snd_sof_pci_update_bits_unlocked: sof-audio-pci-intel-tgl 0000:00:1f.3: Debug PCIR: 00000000 at  00000048
[    2.550602] snd_sof:snd_sof_pci_update_bits_unlocked: sof-audio-pci-intel-tgl 0000:00:1f.3: Debug PCIW: 00000040 at  00000048
[    2.551858] snd_sof_intel_hda_mlink:hda_bus_ml_init: sof-audio-pci-intel-tgl 0000:00:1f.3: HDAudio Multi-Link count: 2
[... cut ...]
[   41.564178] snd_sof:snd_sof_pci_update_bits_unlocked: sof-audio-pci-intel-tgl 0000:00:1f.3: Debug PCIR: 00000010 at  00000044
[   41.564191] snd_sof_intel_hda:hda_codec_i915_display_power: sof-audio-pci-intel-tgl 0000:00:1f.3: Turning i915 HDAC power 0
[   41.564192] snd_sof_intel_hda_common:hda_dsp_state_log: sof-audio-pci-intel-tgl 0000:00:1f.3: Current DSP power state: D3
[   41.564193] snd_sof:sof_set_fw_state: sof-audio-pci-intel-tgl 0000:00:1f.3: fw_state change: 7 -> 0
tomg@haxboxen:~$ 

```

mrhpearson, 2024-07-29:
I created internal ticket  LO-3221 for tracking - but a few questions

@real-t0mg - Can you let me know what distro you're running please? Also it would be good to know the BIOS and EC versions (likely not important here - but having it handy in case the FW team ask is good)

Would you be able to try with Fedora 40 Live USB - that should have 2024.03 (I don't think anybody has 2024.06 packaged yet - I know I'm still fighting with it for the Debian packaging). 
It would be good to rule out this being a kernel/FW mismatch as you're on a slightly older kernel.

Unfortunately I don't have a T14s G2 intel system on hand - but once we have more details we'll see if we can reproduce in the lab.
Mark

real-t0mg, 2024-07-29:
@mrhpearson, sure: **Slackware 15.0** (http://www.slackware.com/). Thanks for tracking the issue.

BIOS:

```
BIOS Information
        Vendor: LENOVO
        Version: N35ET58W (1.58 )
        Release Date: 04/18/2024
        Address: 0xE0000
        Runtime Size: 128 kB
        ROM Size: 32 MB
        Characteristics:
                PCI is supported
                PNP is supported
                BIOS is upgradeable
                BIOS shadowing is allowed
                Boot from CD is supported
                Selectable boot is supported
                EDD is supported
                3.5"/720 kB floppy services are supported (int 13h)
                Print screen service is supported (int 5h)
                8042 keyboard services are supported (int 9h)
                Serial services are supported (int 14h)
                Printer services are supported (int 17h)
                CGA/mono video services are supported (int 10h)
                ACPI is supported
                USB legacy is supported
                BIOS boot specification is supported
                Targeted content distribution is supported
                UEFI is supported
        BIOS Revision: 1.58
        Firmware Revision: 1.44
```
        
 EC: 

```
Handle 0x0032, DMI type 140, 15 bytes
ThinkPad Embedded Controller Program
        Version ID: N35HT44W
        Release Date: 10/25/2023
```

Regarding the live cd boot with Fedora, I may later today and report back. Full time running a distro with systemd is not a viable alternative for me, but I can downlaod and compile my own kernels to test if that helps. I prefer running long-term support kernel alternatives if at all possible help to debug this. 

I'm reluctant to attach a full dmidecode output as it has the laptop's serial number that I registered to hava a valid warranty, but I can email it if you need it.

real-t0mg, 2024-07-29:
The **topology on 2024.03 does not work** (https://github.com/thesofproject/sof-bin/releases/download/v2024.03/sof-bin-2024.03.tar.gz) on my hardware:
[dmesg_on_2024.03.txt](https://github.com/user-attachments/files/16416166/dmesg_on_2024.03.txt)

interesting bit:

```
[    2.750977] snd_sof:sof_route_load: sof-audio-pci-intel-tgl 0000:00:1f.3: sink HDA4.OUT\xd7\x07 control none source BUF7.1
[    2.750978] sof-audio-pci-intel-tgl 0000:00:1f.3: error: sink HDA4.OUT\xd7\x07 not found
[    2.758687] sof-audio-pci-intel-tgl 0000:00:1f.3: error: tplg component load failed -22
[    2.763373] sof-audio-pci-intel-tgl 0000:00:1f.3: error: failed to load DSP topology -22
[    2.763375] sof-audio-pci-intel-tgl 0000:00:1f.3: ASoC: error at snd_soc_component_probe on 0000:00:1f.3: -22
[    2.763433] skl_hda_dsp_generic: probe of skl_hda_dsp_generic failed with error -22

```

topology fails:

```
tomg@haxboxen:~$ date
Mon Jul 29 13:24:20 -03 2024
tomg@haxboxen:~$ uname -a
Linux haxboxen.lan 6.6.43-ST1NKPAD-DEBUG #3 SMP PREEMPT Mon Jul 29 12:29:00 -03 2024 x86_64 11th Gen Intel(R) Core(TM) i7-1165G7 @ 2.80GHz GenuineIntel GNU/Linux
tomg@haxboxen:~$ 
tomg@haxboxen:~$  dmesg | grep -E 'sof-|snd|error|dyndbg'
[    2.481018] snd_hda_intel 0000:00:1f.3: DSP detected with PCI class/subclass/prog-if info 0x040380
[    2.483242] snd_intel_sdw_acpi:sdw_intel_scan_controller: acpi PRP00001:00: ACPI reports 4 SDW Link devices
[    2.483245] snd_intel_sdw_acpi:sdw_intel_scan_controller: acpi PRP00001:00: Link 0 not selected in firmware
[    2.483246] snd_intel_sdw_acpi:sdw_intel_scan_controller: acpi PRP00001:00: Link 1 not selected in firmware
[    2.483247] snd_intel_sdw_acpi:sdw_intel_scan_controller: acpi PRP00001:00: Link 2 not selected in firmware
[    2.483249] snd_intel_sdw_acpi:sdw_intel_scan_controller: acpi PRP00001:00: Link 3 not selected in firmware
[    2.483250] snd_hda_intel 0000:00:1f.3: Digital mics found on Skylake+ platform, using SOF driver
[    2.545782] sof-audio-pci-intel-tgl 0000:00:1f.3: DSP detected with PCI class/subclass/prog-if info 0x040380
[    2.551876] snd_intel_sdw_acpi:sdw_intel_scan_controller: acpi PRP00001:00: ACPI reports 4 SDW Link devices
[    2.551879] snd_intel_sdw_acpi:sdw_intel_scan_controller: acpi PRP00001:00: Link 0 not selected in firmware
[    2.551880] snd_intel_sdw_acpi:sdw_intel_scan_controller: acpi PRP00001:00: Link 1 not selected in firmware
[    2.551882] snd_intel_sdw_acpi:sdw_intel_scan_controller: acpi PRP00001:00: Link 2 not selected in firmware
[    2.551883] snd_intel_sdw_acpi:sdw_intel_scan_controller: acpi PRP00001:00: Link 3 not selected in firmware
[    2.551885] sof-audio-pci-intel-tgl 0000:00:1f.3: Digital mics found on Skylake+ platform, using SOF driver
[    2.556108] snd_sof_pci:sof_pci_probe: sof-audio-pci-intel-tgl 0000:00:1f.3: PCI DSP detected
[    2.567613] sof-audio-pci-intel-tgl 0000:00:1f.3: DSP detected with PCI class/subclass/prog-if 0x040380
[    2.571049] sof-audio-pci-intel-tgl 0000:00:1f.3: bound 0000:00:02.0 (ops 0xffffffff91f203b0)
[    2.579599] snd_sof_intel_hda_common:hda_dsp_ctrl_get_caps: sof-audio-pci-intel-tgl 0000:00:1f.3: checking for capabilities at offset 0xc00
[    2.579610] snd_sof_intel_hda_common:hda_dsp_ctrl_get_caps: sof-audio-pci-intel-tgl 0000:00:1f.3: found ML capability at 0xc00
[    2.579611] snd_sof_intel_hda_common:hda_dsp_ctrl_get_caps: sof-audio-pci-intel-tgl 0000:00:1f.3: checking for capabilities at offset 0x800
[    2.579616] snd_sof_intel_hda_common:hda_dsp_ctrl_get_caps: sof-audio-pci-intel-tgl 0000:00:1f.3: found DSP capability at 0x800
[    2.579617] snd_sof_intel_hda_common:hda_dsp_ctrl_get_caps: sof-audio-pci-intel-tgl 0000:00:1f.3: checking for capabilities at offset 0x500
[    2.579622] snd_sof_intel_hda_common:hda_dsp_ctrl_get_caps: sof-audio-pci-intel-tgl 0000:00:1f.3: found GTS capability at 0x500
[    2.579623] snd_sof_intel_hda_common:hda_dsp_ctrl_get_caps: sof-audio-pci-intel-tgl 0000:00:1f.3: checking for capabilities at offset 0x1f00
[    2.579630] snd_sof_intel_hda_common:hda_dsp_ctrl_get_caps: sof-audio-pci-intel-tgl 0000:00:1f.3: found DRSM capability at 0x1f00
[    2.579631] snd_sof_intel_hda_common:hda_dsp_ctrl_get_caps: sof-audio-pci-intel-tgl 0000:00:1f.3: checking for capabilities at offset 0x700
[    2.579634] snd_sof_intel_hda_common:hda_dsp_ctrl_get_caps: sof-audio-pci-intel-tgl 0000:00:1f.3: found SPIB capability at 0x700
[    2.579655] snd_sof_intel_hda_common:hda_dsp_stream_init: sof-audio-pci-intel-tgl 0000:00:1f.3: hda global caps = 0x1009701
[    2.579656] snd_sof_intel_hda_common:hda_dsp_stream_init: sof-audio-pci-intel-tgl 0000:00:1f.3: detected 9 playback and 7 capture streams
[    2.580691] sof-audio-pci-intel-tgl 0000:00:1f.3: use msi interrupt mode
[    2.584208] snd_sof_intel_hda_common:hda_dsp_probe: sof-audio-pci-intel-tgl 0000:00:1f.3: using IPC IRQ 167
[    2.584709] snd_sof:snd_sof_pci_update_bits_unlocked: sof-audio-pci-intel-tgl 0000:00:1f.3: Debug PCIR: 00000000 at  00000044
[    2.584712] snd_sof_intel_hda_common:hda_init_caps: sof-audio-pci-intel-tgl 0000:00:1f.3: PP capability, will probe DSP later.
[    2.587402] snd_sof:snd_sof_pci_update_bits_unlocked: sof-audio-pci-intel-tgl 0000:00:1f.3: Debug PCIR: 00000040 at  00000048
[    2.587406] snd_sof:snd_sof_pci_update_bits_unlocked: sof-audio-pci-intel-tgl 0000:00:1f.3: Debug PCIW: 00000000 at  00000048
[... cut ...]
[    2.750977] snd_sof:sof_route_load: sof-audio-pci-intel-tgl 0000:00:1f.3: sink HDA4.OUT\xd7\x07 control none source BUF7.1
[    2.750978] sof-audio-pci-intel-tgl 0000:00:1f.3: error: sink HDA4.OUT\xd7\x07 not found
[    2.758687] sof-audio-pci-intel-tgl 0000:00:1f.3: error: tplg component load failed -22
[    2.763373] sof-audio-pci-intel-tgl 0000:00:1f.3: error: failed to load DSP topology -22
[    2.763375] sof-audio-pci-intel-tgl 0000:00:1f.3: ASoC: error at snd_soc_component_probe on 0000:00:1f.3: -22
[    2.763433] skl_hda_dsp_generic: probe of skl_hda_dsp_generic failed with error -22
tomg@haxboxen:~$ 
```

plbossart, 2024-07-29:
ok so the good news is that there's at least one version that works, thanks for trying older versions @real-t0mg 

The last try shows that the firmware version 2024.03 boots as expected, if that is confirmed then only the last version has a problem.

I am not sure what changed between the firmware v2023.12.1/v2024.3 and v2024.06, in theory the keys are identical. The only thing that comes to my mind is compatibility between tools used for the signature by Intel and tools used by Lenovo for the authentication in the ME, or one of the 'Feature' flags that changed on the target. We've seen both before.

Wrt to the topology problem, IIRC this is a kernel bug that was corrected. The additional trailing characters in the widget names, e.g. "sink HDA4.OUT\xd7\x07" remind me of something, I just can't find the actual fix. @ujfalusi, @bardliao @ranj063 does this ring a bell?.

real-t0mg, 2024-07-29:
the bad news is that reverting to the version that worked (sof-bin-2023.12.1.tar.gz), no longer works and now it fails with a different error. Let me know what additional info I may provide, now no version works for me :-( . 

dmesg log: 
[dmesg_2023.12.1_not_working_now.txt](https://github.com/user-attachments/files/16417100/dmesg_2023.12.1_not_working_now.txt)

interesting bit: 

```
[    2.692259] snd_sof:sof_route_load: sof-audio-pci-intel-tgl 0000:00:1f.3: sink MIXER1.0\xc0\xa8)\x85:\xe5\xff\xff control none source BUF30.1
[    2.692260] sof-audio-pci-intel-tgl 0000:00:1f.3: error: sink MIXER1.0\xc0\xa8)\x85:\xe5\xff\xff not found
[    2.705721] sof-audio-pci-intel-tgl 0000:00:1f.3: error: tplg component load failed -22
[    2.710233] sof-audio-pci-intel-tgl 0000:00:1f.3: error: failed to load DSP topology -22
[    2.715651] sof-audio-pci-intel-tgl 0000:00:1f.3: ASoC: error at snd_soc_component_probe on 0000:00:1f.3: -22
[    2.723275] skl_hda_dsp_generic: probe of skl_hda_dsp_generic failed with error -22
```

fail:

```
tomg@haxboxen:~$ date
Mon Jul 29 14:56:32 -03 2024
tomg@haxboxen:~$ uname -a
Linux haxboxen.lan 6.6.43-ST1NKPAD-DEBUG #3 SMP PREEMPT Mon Jul 29 12:29:00 -03 2024 x86_64 11th Gen Intel(R) Core(TM) i7-1165G7 @ 2.80GHz GenuineIntel GNU/Linux
tomg@haxboxen:~$ 
tomg@haxboxen:~$  dmesg | grep -E 'sof-|snd|error|dyndbg'
[    2.430284] snd_hda_intel 0000:00:1f.3: DSP detected with PCI class/subclass/prog-if info 0x040380
[    2.432688] snd_intel_sdw_acpi:sdw_intel_scan_controller: acpi PRP00001:00: ACPI reports 4 SDW Link devices
[    2.432691] snd_intel_sdw_acpi:sdw_intel_scan_controller: acpi PRP00001:00: Link 0 not selected in firmware
[    2.432693] snd_intel_sdw_acpi:sdw_intel_scan_controller: acpi PRP00001:00: Link 1 not selected in firmware
[    2.432694] snd_intel_sdw_acpi:sdw_intel_scan_controller: acpi PRP00001:00: Link 2 not selected in firmware
[    2.432695] snd_intel_sdw_acpi:sdw_intel_scan_controller: acpi PRP00001:00: Link 3 not selected in firmware
[    2.432697] snd_hda_intel 0000:00:1f.3: Digital mics found on Skylake+ platform, using SOF driver
[    2.508163] sof-audio-pci-intel-tgl 0000:00:1f.3: DSP detected with PCI class/subclass/prog-if info 0x040380
[    2.511127] snd_intel_sdw_acpi:sdw_intel_scan_controller: acpi PRP00001:00: ACPI reports 4 SDW Link devices
[    2.511129] snd_intel_sdw_acpi:sdw_intel_scan_controller: acpi PRP00001:00: Link 0 not selected in firmware
[    2.511131] snd_intel_sdw_acpi:sdw_intel_scan_controller: acpi PRP00001:00: Link 1 not selected in firmware
[    2.511132] snd_intel_sdw_acpi:sdw_intel_scan_controller: acpi PRP00001:00: Link 2 not selected in firmware
[    2.511133] snd_intel_sdw_acpi:sdw_intel_scan_controller: acpi PRP00001:00: Link 3 not selected in firmware
[    2.511134] sof-audio-pci-intel-tgl 0000:00:1f.3: Digital mics found on Skylake+ platform, using SOF driver
[    2.514276] snd_sof_pci:sof_pci_probe: sof-audio-pci-intel-tgl 0000:00:1f.3: PCI DSP detected
[    2.514388] sof-audio-pci-intel-tgl 0000:00:1f.3: DSP detected with PCI class/subclass/prog-if 0x040380
[    2.518933] sof-audio-pci-intel-tgl 0000:00:1f.3: bound 0000:00:02.0 (ops 0xffffffff89f203b0)
[    2.527874] snd_sof_intel_hda_common:hda_dsp_ctrl_get_caps: sof-audio-pci-intel-tgl 0000:00:1f.3: checking for capabilities at offset 0xc00
[    2.527880] snd_sof_intel_hda_common:hda_dsp_ctrl_get_caps: sof-audio-pci-intel-tgl 0000:00:1f.3: found ML capability at 0xc00
[    2.527881] snd_sof_intel_hda_common:hda_dsp_ctrl_get_caps: sof-audio-pci-intel-tgl 0000:00:1f.3: checking for capabilities at offset 0x800
[    2.527886] snd_sof_intel_hda_common:hda_dsp_ctrl_get_caps: sof-audio-pci-intel-tgl 0000:00:1f.3: found DSP capability at 0x800
[    2.527887] snd_sof_intel_hda_common:hda_dsp_ctrl_get_caps: sof-audio-pci-intel-tgl 0000:00:1f.3: checking for capabilities at offset 0x500
[    2.527894] snd_sof_intel_hda_common:hda_dsp_ctrl_get_caps: sof-audio-pci-intel-tgl 0000:00:1f.3: found GTS capability at 0x500
[    2.527895] snd_sof_intel_hda_common:hda_dsp_ctrl_get_caps: sof-audio-pci-intel-tgl 0000:00:1f.3: checking for capabilities at offset 0x1f00
[    2.527898] snd_sof_intel_hda_common:hda_dsp_ctrl_get_caps: sof-audio-pci-intel-tgl 0000:00:1f.3: found DRSM capability at 0x1f00
[    2.527898] snd_sof_intel_hda_common:hda_dsp_ctrl_get_caps: sof-audio-pci-intel-tgl 0000:00:1f.3: checking for capabilities at offset 0x700
[    2.527905] snd_sof_intel_hda_common:hda_dsp_ctrl_get_caps: sof-audio-pci-intel-tgl 0000:00:1f.3: found SPIB capability at 0x700
[    2.527923] snd_sof_intel_hda_common:hda_dsp_stream_init: sof-audio-pci-intel-tgl 0000:00:1f.3: hda global caps = 0x1009701
[    2.527924] snd_sof_intel_hda_common:hda_dsp_stream_init: sof-audio-pci-intel-tgl 0000:00:1f.3: detected 9 playback and 7 capture streams
[    2.528043] sof-audio-pci-intel-tgl 0000:00:1f.3: use msi interrupt mode
[    2.532253] snd_sof_intel_hda_common:hda_dsp_probe: sof-audio-pci-intel-tgl 0000:00:1f.3: using IPC IRQ 183
[    2.532322] snd_sof:snd_sof_pci_update_bits_unlocked: sof-audio-pci-intel-tgl 0000:00:1f.3: Debug PCIR: 00000000 at  00000044
[    2.532324] snd_sof_intel_hda_common:hda_init_caps: sof-audio-pci-intel-tgl 0000:00:1f.3: PP capability, will probe DSP later.
[    2.536258] snd_sof:snd_sof_pci_update_bits_unlocked: sof-audio-pci-intel-tgl 0000:00:1f.3: Debug PCIR: 00000040 at  00000048
[... cut ...]
[    2.692260] sof-audio-pci-intel-tgl 0000:00:1f.3: error: sink MIXER1.0\xc0\xa8)\x85:\xe5\xff\xff not found
[    2.705721] sof-audio-pci-intel-tgl 0000:00:1f.3: error: tplg component load failed -22
[    2.710233] sof-audio-pci-intel-tgl 0000:00:1f.3: error: failed to load DSP topology -22
[    2.715651] sof-audio-pci-intel-tgl 0000:00:1f.3: ASoC: error at snd_soc_component_probe on 0000:00:1f.3: -22
[    2.723275] skl_hda_dsp_generic: probe of skl_hda_dsp_generic failed with error -22
tomg@haxboxen:~$ 
```

real-t0mg, 2024-07-29:
without changing the firmware nor the topology files, the error messages regarding the topology feel random (i.e. different HDA#.OUT accross reboots): 

dmesg log:
[dmesg_fail_on_topology.txt](https://github.com/user-attachments/files/16417901/dmesg_fail_on_topology.txt)

sample: 

```
[    2.728557] snd_sof:sof_route_load: sof-audio-pci-intel-tgl 0000:00:1f.3: sink BUF8.0 control none source PCM4P
[    2.728559] snd_sof:sof_route_load: sof-audio-pci-intel-tgl 0000:00:1f.3: sink PGA8.0 control none source BUF8.0
[    2.728561] snd_sof:sof_route_load: sof-audio-pci-intel-tgl 0000:00:1f.3: sink BUF8.1 control none source PGA8.0
[    2.728564] snd_sof:sof_route_load: sof-audio-pci-intel-tgl 0000:00:1f.3: sink HDA5.OUT\x80\x0f\x08\xc0P\x15\xcfqG\x80\x9c\x94\xa0\x81\x01\xd9ǀ#\xa0\x01\x86\x80\xe0\xa2\xc1\x0f\xf2\x81\xe0 \x86%\xf4\xe1\xb9#\xf2\xcfp\x88\xc0x \xb5x \x80 control none source BUF8.1
[    2.728565] sof-audio-pci-intel-tgl 0000:00:1f.3: error: sink HDA5.OUT\x80\x0f\x08\xc0P\x15\xcfqG\x80\x9c\x94\xa0\x81\x01\xd9ǀ#\xa0\x01\x86\x80\xe0\xa2\xc1\x0f\xf2\x81\xe0 \x86%\xf4\xe1\xb9#\xf2\xcfp\x88\xc0x \xb5x \x80 not found
[    2.740829] sof-audio-pci-intel-tgl 0000:00:1f.3: error: tplg component load failed -22
[    2.748411] sof-audio-pci-intel-tgl 0000:00:1f.3: error: failed to load DSP topology -22
[    2.748412] sof-audio-pci-intel-tgl 0000:00:1f.3: ASoC: error at snd_soc_component_probe on 0000:00:1f.3: -22
[    2.775662] skl_hda_dsp_generic: probe of skl_hda_dsp_generic failed with error -22
```

plbossart, 2024-07-29:
Now I remember, it was a fairly recent thread in https://lore.kernel.org/linux-sound/171812236450.201359.3019210915105428447.b4-ty@kernel.org/T/#m8c4bd5abf453960fde6f826c4b7f84881da63e9d

Maybe you have a bad patch that was added to your kernel, best to try with 6.10 if you can to rule out a backport issue.

real-t0mg, 2024-07-29:
@plbossart you were right about kernel 6.10. Running out of the box with **v2024.06** (https://github.com/thesofproject/sof-bin/releases/tag/v2024.06). 

Running as expected on **non-debug  6.10.2 kernel**. With the added benefit I can run the Xe drivers too... 

dmesg output for completeness:
[dmesg_works_6.10.2_v2024.06.txt](https://github.com/user-attachments/files/16418928/dmesg_works_6.10.2_v2024.06.txt)

[image]

code: 

```
tomg@haxboxen:~$ date
Mon Jul 29 17:41:53 -03 2024
tomg@haxboxen:~$ 
tomg@haxboxen:~$ uname -a
Linux haxboxen.lan 6.10.2-ST1NKP4D #1 SMP PREEMPT Mon Jul 29 17:16:23 -03 2024 x86_64 11th Gen Intel(R) Core(TM) i7-1165G7 @ 2.80GHz GenuineIntel GNU/Linux
tomg@haxboxen:~$ 
tomg@haxboxen:~$ dmesg | grep -E 'sof-|snd|error|dyndbg'
[    3.177013] dyndbg param is supported only in CONFIG_DYNAMIC_DEBUG builds
[    3.184413] dyndbg param is supported only in CONFIG_DYNAMIC_DEBUG builds
[    3.187981] snd_hda_intel 0000:00:1f.3: Digital mics found on Skylake+ platform, using SOF driver
[    3.201663] dyndbg param is supported only in CONFIG_DYNAMIC_DEBUG builds
[    3.216108] dyndbg param is supported only in CONFIG_DYNAMIC_DEBUG builds
[    3.226036] dyndbg param is supported only in CONFIG_DYNAMIC_DEBUG builds
[    3.286694] dyndbg param is supported only in CONFIG_DYNAMIC_DEBUG builds
[    3.306247] dyndbg param is supported only in CONFIG_DYNAMIC_DEBUG builds
[    3.316235] dyndbg param is supported only in CONFIG_DYNAMIC_DEBUG builds
[    3.319394] dyndbg param is supported only in CONFIG_DYNAMIC_DEBUG builds
[    3.329597] dyndbg param is supported only in CONFIG_DYNAMIC_DEBUG builds
[    3.340930] dyndbg param is supported only in CONFIG_DYNAMIC_DEBUG builds
[    3.347235] dyndbg param is supported only in CONFIG_DYNAMIC_DEBUG builds
[    3.349714] dyndbg param is supported only in CONFIG_DYNAMIC_DEBUG builds
[    3.353265] dyndbg param is supported only in CONFIG_DYNAMIC_DEBUG builds
[    3.358755] sof-audio-pci-intel-tgl 0000:00:1f.3: DSP detected with PCI class/subclass/prog-if 0x040380
[    4.190148] sof-audio-pci-intel-tgl 0000:00:1f.3: bound 0000:00:02.0 (ops 0xffffffff9df1fd70)
[    4.200396] sof-audio-pci-intel-tgl 0000:00:1f.3: use msi interrupt mode
[    4.218498] sof-audio-pci-intel-tgl 0000:00:1f.3: hda codecs found, mask 5
[    4.220992] sof-audio-pci-intel-tgl 0000:00:1f.3: using HDA machine driver skl_hda_dsp_generic now
[    4.223221] sof-audio-pci-intel-tgl 0000:00:1f.3: DMICs detected in NHLT tables: 2
[    4.228634] sof-audio-pci-intel-tgl 0000:00:1f.3: Firmware paths/files for ipc type 0:
[    4.230748] sof-audio-pci-intel-tgl 0000:00:1f.3:  Firmware file:     intel/sof/sof-tgl.ri
[    4.233721] sof-audio-pci-intel-tgl 0000:00:1f.3:  Topology file:     intel/sof-tplg/sof-hda-generic-2ch.tplg
[    4.237414] sof-audio-pci-intel-tgl 0000:00:1f.3: Firmware info: version 2:2:0-57864
[    4.240388] sof-audio-pci-intel-tgl 0000:00:1f.3: Firmware: ABI 3:22:1 Kernel ABI 3:23:0
[    4.240390] sof-audio-pci-intel-tgl 0000:00:1f.3: unknown sof_ext_man header type 3 size 0x30
[    4.356925] sof-audio-pci-intel-tgl 0000:00:1f.3: Firmware info: version 2:2:0-57864
[    4.358841] sof-audio-pci-intel-tgl 0000:00:1f.3: Firmware: ABI 3:22:1 Kernel ABI 3:23:0
[    4.368895] dyndbg param is supported only in CONFIG_DYNAMIC_DEBUG builds
[    4.376070] dyndbg param is supported only in CONFIG_DYNAMIC_DEBUG builds
[    4.381733] sof-audio-pci-intel-tgl 0000:00:1f.3: Topology: ABI 3:22:1 Kernel ABI 3:23:0
[    4.419845] snd_hda_codec_realtek ehdaudio0D0: autoconfig for ALC257: line_outs=1 (0x14/0x0/0x0/0x0/0x0) type:speaker
[    4.423355] snd_hda_codec_realtek ehdaudio0D0:    speaker_outs=0 (0x0/0x0/0x0/0x0/0x0)
[    4.425437] snd_hda_codec_realtek ehdaudio0D0:    hp_outs=1 (0x21/0x0/0x0/0x0/0x0)
[    4.427912] snd_hda_codec_realtek ehdaudio0D0:    mono: mono_out=0x0
[    4.429924] snd_hda_codec_realtek ehdaudio0D0:    inputs:
[    4.432369] snd_hda_codec_realtek ehdaudio0D0:      Mic=0x19
[    4.514894] input: sof-hda-dsp Mic as /devices/pci0000:00/0000:00:1f.3/skl_hda_dsp_generic/sound/card0/input15
[    4.525175] input: sof-hda-dsp Headphone as /devices/pci0000:00/0000:00:1f.3/skl_hda_dsp_generic/sound/card0/input16
[    4.538857] input: sof-hda-dsp HDMI/DP,pcm=3 as /devices/pci0000:00/0000:00:1f.3/skl_hda_dsp_generic/sound/card0/input17
[    4.588279] input: sof-hda-dsp HDMI/DP,pcm=4 as /devices/pci0000:00/0000:00:1f.3/skl_hda_dsp_generic/sound/card0/input18
[    4.640908] input: sof-hda-dsp HDMI/DP,pcm=5 as /devices/pci0000:00/0000:00:1f.3/skl_hda_dsp_generic/sound/card0/input19
[  211.833116] usbcore: registered new interface driver snd-usb-audio
tomg@haxboxen:~$ 

```

plbossart, 2024-07-30:
well that's good news @real-t0mg, at least you are in business with the latest firmware and kernel.

I am still surprised that this 6.6 kernel would prevent the latest firmware from booting. The TigerLake kernel stuff is a couple of years old, we haven't really touched it, same for the firmware which comes from a 2.2 stable branch. In general we don't expect users to update both kernel and firmware at the same time. Releases should be backwards-compatible.

@ujfalusi any idea on what could have caused this boot problem?

ujfalusi, 2024-07-30:
@plbossart, fwiw on my TGL laptop 2024.06 release works fine:
```
# pacman -Q --info sof-firmware
Name            : sof-firmware
Version         : 2024.06-1
Description     : Sound Open Firmware
Architecture    : x86_64
URL             : https://www.sofproject.org/
Licenses        : BSD-3-Clause AND ISC
...
```
```
# dmesg | grep -A2 "Firmware paths"
[   77.341809] sof-audio-pci-intel-tgl 0000:00:1f.3: Firmware paths/files for ipc type 0:
[   77.341812] sof-audio-pci-intel-tgl 0000:00:1f.3:  Firmware file:     intel/sof/sof-tgl.ri
[   77.341814] sof-audio-pci-intel-tgl 0000:00:1f.3:  Topology file:     intel/sof-tplg/sof-tgl-rt715-rt711-rt1308-mono.tplg
```
The error looks strange:
```
[    5.673437] sof-audio-pci-intel-tgl 0000:00:1f.3: 0x80000006: module: ROM, state: VERIFY_FEATURE_MASK, not running
[    5.673443] sof-audio-pci-intel-tgl 0000:00:1f.3: error code: 0x14 (unknown)
```
The ROM fails with Invalid feature mask (0x14 / 20) status code. The feature_mask is identical between 2024.03 and 2024.06 (0xffff, according to `tools/sof_ri_info/sof_ri_info.py`), furthermore the md5sum of the firmware files are the same:
```
$ md5sum ~/tmp/sof-tgl.ri-2024.* 
bc9d7a6ee8efff7318d608192155c061  /home/pujfalus/tmp/sof-tgl.ri-2024.03
bc9d7a6ee8efff7318d608192155c061  /home/pujfalus/tmp/sof-tgl.ri-2024.06
```
These are form the sof-bin release tarballs.

ujfalusi, 2024-07-30:
@plbossart, I think I know why the firmware was not booting:
```
[    2.613583] snd_sof:snd_sof_load_firmware_raw: sof-audio-pci-intel-tgl 0000:00:1f.3: request_firmware intel/sof/sof-tgl.ri successful
[    2.613585] snd_sof:ipc3_fw_ext_man_size: sof-audio-pci-intel-tgl 0000:00:1f.3: Unexpected extended manifest magic number: 0x31454124
[    2.613586] snd_sof:snd_sof_load_firmware_raw: sof-audio-pci-intel-tgl 0000:00:1f.3: firmware doesn't contain extended manifest
```
It is the IPC4 firmware file loaded as ipc3 image. In this case the header parsing is skipped and the downloaded firmware can be incorrect for booting.

@real-t0mg, can you re-check this with the 2024.06 release:
```
# cd /tmp
# wget https://github.com/thesofproject/sof-bin/releases/download/v2024.06/sof-bin-2024.06.tar.gz
# tar -xf sof-bin-2024.06.tar.gz
# ls -l sof-bin-2024.06/sof/intel-signed/sof-tgl.ri
525056 Jul 18 14:56 sof-bin-2024.06/sof/intel-signed/sof-tgl.ri

# md5sum sof-bin-2024.06/sof/intel-signed/sof-tgl.ri
bc9d7a6ee8efff7318d608192155c061  sof-bin-2024.06/sof/intel-signed/sof-tgl.ri

# ls -l sof-bin-2024.06/sof-ipc4/tgl/intel-signed/sof-tgl.ri 
694656 Jul 18 14:56 sof-bin-2024.06/sof-ipc4/tgl/intel-signed/sof-tgl.ri

# md5sum sof-bin-2024.06/sof-ipc4/tgl/intel-signed/sof-tgl.ri 
1d3eee19dc1cc3cd44900e2df53f4311  sof-bin-2024.06/sof-ipc4/tgl/intel-signed/sof-tgl.ri

and confirm that you have the `sof-bin-2024.06/sof/intel-signed/sof-tgl.ri` installed as `/lib/firmware/intel/sof/sof-tgl.ri` ?

Which distribution you are using and have you installed the sof-bin (2024.06) manually or via distro package?

Thanks

real-t0mg, 2024-07-30:
@ujfalusi thanks! Yes, I figured out that part yesterday re-reading https://thesofproject.github.io/latest/getting_started/intel_debug/introduction.html#user-space-and-filesystem-requirements

My chip/codec is a Realtek ALC257.

What @plbossart says here https://github.com/thesofproject/sof/issues/9339#issuecomment-2256680228 applies to my issue after having placed the correct IPC3 firmware and topology to my setup. My issue started after upgrading from  6.1.101 to  6.1.102. In the kernel. In the changelog I see these were changed:

```
sound/core/pcm_dmaengine.c             |    6 +++-
sound/pci/hda/patch_realtek.c          |    2 +
```
 
I the proceeded to  upgrade to 6.6.43, which you see in my comments above, which in turn logs these changes:

```
sound/core/pcm_dmaengine.c               |    6 +++-
 sound/core/seq/seq_ump_client.c          |   16 ++++++++++
 sound/pci/hda/patch_realtek.c            |    2 +
```

 If I reboot into 6.6.43, I get the error I logged in the issue with the topology file.

ujfalusi, 2024-08-06:
@real-t0mg, I think the subject of the issue has been rootcaused and fixed and now we are discussing unrelated issue(s) to iron out.
If you agree, can you close this and open new one to avoid confusion., thank you.

real-t0mg, 2024-08-06:
closing bug, the recommended fix is to run a current kernel. Thank you!

closed 2024-08-06 by real-t0mg