thesofproject/sof #9340: [BUG] rimage + openssl3 is broken in the stable-v2.2 branch, has always been

marc-hb, 2024-07-31:
**Describe the bug**

rimage + openssl3 is broken in the stable-v2.2 branch, always has been.

This was fixed 2 years ago by rimage commit https://github.com/thesofproject/rimage/commit/73a9d7c897d3658486dc5cb5d69c7a587dfc042a 

However that critical rimage fix never made it to https://github.com/thesofproject/sof/commits/stable-v2.2/rimage. stable-v2.2 fell 6 commits short.

That's because we tested https://github.com/thesofproject/sof/commits/stable-v2.3/rimage for a long time (which does have the openssl3 fix) and then decided to go backwards to stable-v2.2. I don't remember why.

In https://github.com/thesofproject/rimage/pull/97#pullrequestreview-1009485532 @lgirdwood wrote "We need to take this for v2.2, ..." but that never happened.

The investigation happened in https://github.com/thesofproject/sof/issues/5917 . I didn't re-open that bug and opened a new one instead because that investigation took an incredibly long time with many red herrings so this bug is very long and hard to read. One of the reasons it took so long: some people were using openssl1 (which worked) and other people were using openssl3 (which failed) but it took a long time to notice that.

**To Reproduce**

This affects only some SOF commits and not others and I have no idea why. Any OpenSSL expert around?

This was found in totally unrelated PR https://github.com/thesofproject/sof/pull/9336. When I fetch https://github.com/thesofproject/sof/commit/4ed65ed718109a346a7c573115c43918497729a4 which is the latest `pull/9336/merge` commit for that PR and run `git submodule update --recursive`, then the firmware never boots on my TGL Xtreme. After I cherry-pick openssl3 fix https://github.com/thesofproject/rimage/commit/73a9d7c897d3658486dc5cb5d69c7a587dfc042a, the firmware boots again every time.

(the git bisect wasn't straight-forward for unrelated reasons but it worked; you have been warned)

**Reproduction Rate**

100% for some SOF commits but 0% for other SOF commits like the current tag `v2.2.10`. No idea how some commits can boot anyway.

**Expected behavior**

The firmware boots.

**Impact**

Show stopper.

**Screenshots or console output**

```
juil. 31 02:39:51 tglup-mh kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: Attempting iteration 0 of Core En/ROM load...
juil. 31 02:39:51 tglup-mh kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: FW Poll Status: reg[0x4]=0xf010e0e successful
juil. 31 02:39:51 tglup-mh kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: FW Poll Status: reg[0x4]=0xf010e0e successful
juil. 31 02:39:51 tglup-mh kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: unstall/run core: core_mask = 1
juil. 31 02:39:51 tglup-mh kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: DSP core(s) enabled? 1 : core_mask 1
juil. 31 02:39:51 tglup-mh kernel: usbcore: registered new interface driver snd-usb-audio
juil. 31 02:39:52 tglup-mh kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: FW Poll Status: reg[0xd4]=0x0 timedout
juil. 31 02:39:52 tglup-mh kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: FW Poll Status: reg[0x4]=0xf010f0f successful
juil. 31 02:39:52 tglup-mh kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: FW Poll Status: reg[0x4]=0xf0f successful
juil. 31 02:39:52 tglup-mh kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: DSP core(s) enabled? 0 : core_mask 1
juil. 31 02:39:52 tglup-mh kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: Attempting iteration 1 of Core En/ROM load...
juil. 31 02:39:52 tglup-mh kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: FW Poll Status: reg[0x4]=0xf010f0f successful
juil. 31 02:39:52 tglup-mh kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: FW Poll Status: reg[0x4]=0xf010f0e successful
juil. 31 02:39:52 tglup-mh kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: unstall/run core: core_mask = 1
juil. 31 02:39:52 tglup-mh kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: DSP core(s) enabled? 1 : core_mask 1
juil. 31 02:39:52 tglup-mh kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: FW Poll Status: reg[0xd4]=0x80000000 successful
juil. 31 02:39:52 tglup-mh kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: FW Poll Status: reg[0x80000]=0x5000001 successful
juil. 31 02:39:52 tglup-mh kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: Code loader DMA starting
juil. 31 02:39:52 tglup-mh kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: Code loader DMA done, waiting for FW_ENTERED status
juil. 31 02:39:55 tglup-mh kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: FW Poll Status: reg[0x80000]=0x80000012 timedout
juil. 31 02:39:55 tglup-mh kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: hda_cl_copy_fw: timeout with rom_status_reg (0x80000) read
juil. 31 02:39:55 tglup-mh kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: FW Poll Status: reg[0x160]=0x140000 successful
juil. 31 02:39:55 tglup-mh kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: Code loader DMA stopped
juil. 31 02:39:55 tglup-mh kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ------------[ DSP dump start ]------------
juil. 31 02:39:55 tglup-mh kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: Firmware download failed
juil. 31 02:39:55 tglup-mh kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: fw_state: SOF_FW_BOOT_IN_PROGRESS (3)
juil. 31 02:39:55 tglup-mh kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: 0x80000012: module: ROM, state: CSE_VALIDATE_IMAGE_REQUEST, not running
juil. 31 02:39:55 tglup-mh kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: error code: 0x2c (error: signature verification failed)
juil. 31 02:39:55 tglup-mh kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: extended rom status:  0x80000012 0x2c 0x0 0x0 0x0 0x0 0x2510113 0x0
juil. 31 02:39:55 tglup-mh kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ------------[ DSP dump end ]------------
juil. 31 02:39:55 tglup-mh kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: Failed to start DSP
juil. 31 02:39:55 tglup-mh kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: error: failed to boot DSP firmware -110
juil. 31 02:39:55 tglup-mh kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: fw_state change: 3 -> 4
juil. 31 02:39:55 tglup-mh kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: FW Poll Status: reg[0x4]=0x1d003c timedout
juil. 31 02:39:55 tglup-mh kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: error: hda_dsp_core_reset_enter: timeout on HDA_DSP_REG_ADSPCS read
juil. 31 02:39:55 tglup-mh kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: error: dsp core reset failed: core_mask 1
juil. 31 02:39:55 tglup-mh kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: fw_state change: 4 -> 0
juil. 31 02:39:55 tglup-mh kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: error: sof_probe_work failed err: -110

```

marc-hb, 2024-07-31:
So now we have (at least) two obvious options:

- cherry-pick ONLY that rimage fix for stable-v2.2. Minimal risk of breaking something else BUT risk of missing other fixes.

- Fast forward sof/stable-v2.2 to the last "standalone" rimage commit before the rimage repo was merged into the sof repo. I smoke-tested that and it boots too.

cujomalainey, 2024-08-01:
rimage is pretty good at being backwards compatible right? How much risk is there in the fast forward in breaking something?

marc-hb, 2024-08-01:
Probably good but we have no evidence of that. There have been some big code changes later and they were just too big for me to audit them. On the other hand, I spotted no fix as important as this one. If anyone volunteers then please free to submit a bigger update after this one. The lack of branching and cherry-picks now will make a future upgrade easy _from a git perspective_.

marc-hb, 2024-08-01:
> rimage is pretty good at being backwards compatible right? How much risk is there in the fast forward in breaking something?

Sorry I assumed you meant "fast forward all the way" and answered accordingly. Now I see this may or may not be what you meant.

#9342 is the smallest (and small) fast forward with the fix.

cujomalainey, 2024-08-02:
@marc-hb we can close this now right?

marc-hb, 2024-09-03:
Re-opening for cavs2.5-001-drop-stable, see random firmware boot failures in https://sof-ci.01.org/sofpr/PR9427/build7557/devicetest/index.html
- #9427 

Exactly like what happened in #9336, a random, unrelated topology change is causing a "butterfly effect" in the firmware image and it stops booting.

I thought I could very quickly cherry-pick the same rimage upgrade than I did for stable-v2.2 (in #9342) and that has been working well. But that rimage submodule cherry-pick has a conflict because @abonislawski pushed a cavs25-only, "file error handling" rimage fix in https://github.com/thesofproject/rimage/commit/d287016caac744830ccfa9640bbd82ef56c635b5. I don't know why this file error handling fix was only on cavs25 and without any PR or code review. It looks like yet another static analysis "quick fix".

At this stage the best option is probably to create a new branch https://github.com/thesofproject/rimage/tree/cavs2.5-001-drop-stable-2 that points at the rimage commits used in stable-v2.2 + maybe a cherry-pick -x of ""file error handling" d287016caac ?

cc: @macchian , @mwasko

kv2019i, 2024-10-07:
I tested stable-v2.2 and cavs2.5-001-drop-stable both today and works after the backport. Closing.

PR #9342 (2024-08-02): [STABLE-v2.2] Update rimage to version 02abc5d342a3, including critical openssl3 fix
Fixes #9340

This adds the following commits from the main rimage branch:

```
commit 02abc5d342a3 ("rimage: ace signing functions need openssl 3.0 guards")
commit 73a9d7c897d3 ("rimage: fix openssl 3.0 support in ver25 signing")
commit 8ba3d1729656 ("adsp_config: fix name parsing error in parse_signed_pkg_ace_v1_5")
commit fe4dcaa62c13 ("mtl: Add ASRC module to the manifest")
commit af947cb63dd6 ("config: Add toml config for mtl")
commit c484d99abf89 ("rimage:  add ACE V1.5 handling")
commit 1b233f68bdee ("co [... cut]
commit 9e4dadc4 (2024-08-02): Update rimage to version 02abc5d342a3, including critical openssl3 fix

Fixes #9340

This adds the following commits from the main rimage branch:

```
commit 02abc5d342a3 ("rimage: ace signing functions need openssl 3.0 guards")
commit 73a9d7c897d3 ("rimage: fix openssl 3.0 support in ver25 signing")
commit 8ba3d1729656 ("adsp_config: fix name parsing error in parse_signed_pkg_ace_v1_5")
commit fe4dcaa62c13 ("mtl: Add ASRC module to the manifest")
commit af947cb63dd6 ("config: Add toml config for mtl")
commit c484d99abf89 ("rimage:  add ACE V1.5 handling")
commit 1b233f68bdee ("config: add rmb [... cut]
closed 2024-10-07 by kv2019i