thesofproject/sof #9341: [BUG]Intel Corporation Tiger Lake not start

coutouly, 2024-07-31:
**Describe the bug**
dmesg | grep -E 'sof-|snd|error'
[ 1876.586748] hp_wmi: query 0x4 returned error 0x5
[ 1876.595377] snd_hda_intel 0000:00:1f.3: DSP detected with PCI class/subclass/prog-if info 0x040100
[ 1876.599773] snd_hda_intel 0000:00:1f.3: Digital mics found on Skylake+ platform, using SOF driver
[ 1876.969520] sof-audio-pci-intel-tgl 0000:00:1f.3: DSP detected with PCI class/subclass/prog-if info 0x040100
[ 1876.969539] sof-audio-pci-intel-tgl 0000:00:1f.3: Digital mics found on Skylake+ platform, using SOF driver
[ 1876.969551] sof-audio-pci-intel-tgl 0000:00:1f.3: enabling device (0000 -> 0002)
[ 1876.969731] sof-audio-pci-intel-tgl 0000:00:1f.3: DSP detected with PCI class/subclass/prog-if 0x040100
[ 1878.120935] sof-audio-pci-intel-tgl 0000:00:1f.3: bound 0000:00:02.0 (ops i915_audio_component_bind_ops [i915])
[ 1879.300889] sof-audio-pci-intel-tgl 0000:00:1f.3: use msi interrupt mode
[ 1879.357736] sof-audio-pci-intel-tgl 0000:00:1f.3: hda codecs found, mask 5
[ 1879.357741] sof-audio-pci-intel-tgl 0000:00:1f.3: using HDA machine driver skl_hda_dsp_generic now
[ 1879.357744] sof-audio-pci-intel-tgl 0000:00:1f.3: DMICs detected in NHLT tables: 2
[ 1879.360077] sof-audio-pci-intel-tgl 0000:00:1f.3: Firmware info: version 2:2:0-57864
[ 1879.360079] sof-audio-pci-intel-tgl 0000:00:1f.3: Firmware: ABI 3:22:1 Kernel ABI 3:23:0
[ 1879.360084] sof-audio-pci-intel-tgl 0000:00:1f.3: unknown sof_ext_man header type 3 size 0x30
[ 1879.476406] sof-audio-pci-intel-tgl 0000:00:1f.3: Firmware info: version 2:2:0-57864
[ 1879.476409] sof-audio-pci-intel-tgl 0000:00:1f.3: Firmware: ABI 3:22:1 Kernel ABI 3:23:0
[ 1879.483843] sof-audio-pci-intel-tgl 0000:00:1f.3: Topology: ABI 3:22:1 Kernel ABI 3:23:0
[ 1879.483958] sof-audio-pci-intel-tgl 0000:00:1f.3: error: source BUF31.1@p\x1a\x06 not found
[ 1879.484042] sof-audio-pci-intel-tgl 0000:00:1f.3: error: tplg component load failed -22
[ 1879.484055] sof-audio-pci-intel-tgl 0000:00:1f.3: error: failed to load DSP topology -22
[ 1879.484066] sof-audio-pci-intel-tgl 0000:00:1f.3: ASoC: error at snd_soc_component_probe on 0000:00:1f.3: -22
[ 1879.484133] skl_hda_dsp_generic: probe of skl_hda_dsp_generic failed with error -22

marc-hb, 2024-07-31:
You deleted the https://github.com/thesofproject/sof/blob/main/.github/ISSUE_TEMPLATE/bug_report.md issue template entirely and provided zero system and version information. This means no one can help you.

plbossart, 2024-07-31:
Looks like some kernels have a bad patch that mangles the widget names, e.g.  "BUF31.1@p\x1a\x06", we had a similar issue in https://github.com/thesofproject/sof/issues/9339#issuecomment-2256385124

Yeah, at minimum we need to know what Linux kernel/distro is used. please attach the log of 'alsa-info'

coutouly, 2024-07-31:
`alsa-info
ALSA Information Script v 0.5.3
--------------------------------

This script visits the following commands/files to collect diagnostic
information about your ALSA installation and sound related hardware.

  dmesg
  lspci
  aplay
  amixer
  alsactl
  rpm, dpkg
  /proc/asound/
  /sys/class/sound/
  ~/.asoundrc (etc.)

See '/nix/store/fi2ka0rq8cfdqgif4n0q7mmx596wf040-alsa-utils-1.2.10/bin/.alsa-info-wrapped --help' for command line options.

/nix/store/fi2ka0rq8cfdqgif4n0q7mmx596wf040-alsa-utils-1.2.10/bin/.alsa-info-wrapped: line 659: tree: command not found

Your ALSA information is in /tmp/alsa-info.txt.z1DdMudq9Y
`
uname -a
Linux nanplc2 6.6.43 #1-NixOS SMP PREEMPT_DYNAMIC Sat Jul 27 09:34:11 UTC 2024 x86_64 GNU/Linux

Nixos distro

marc-hb, 2024-07-31:
> Linux nanplc2 6.6.43

This looks indeed like the same version as in #9339. Can you try downgrading or upgrading your kernel and test again?

coutouly, 2024-08-01:
you are right, I update the kernel and everything is right
thank you