thesofproject/sof #9360: SOF 2.11 Release Checklist

kv2019i, 2024-08-12:
Planned Timeline (updated on Aug 12th)
- Stable v2.11 Branch – ww36-37 
- RC1 – w37 / 13th Sep 
- Final tag
- Target Release – w38 / 20th Sep 
- Overflow week (in case multiple RCs are required) – ww39

Pre-RC1 Steps, Creation of Stable Branch
- [ ] Tag features and issue list with milestone v2.10: https://github.com/thesofproject/sof/milestones/v2.10
- [ ] Choose a SOF commit on main branch as candidate for stable branch
- [ ] Testing & sign off
- [ ] Branch stable v2.xx -- depending on main stability can be one week before RC1 or at the same time

RC1 Steps:
- [ ] Cherry pick critical features and bug fixes to stable-v2.xx for RC1
- [ ] Perform FW security scan and fix opens
- [ ] Testing & sign off
- [ ] Tag v2.xx-RC1

Final Release Steps:
- [ ] Cherry pick critical features and bug fixes to stable-v2.xx
- [ ] Testing & sign off
- [ ] Tag v2.xx release
- [ ] Publish release on https://github.com/thesofproject/sof/releases

Final Release, vendor specific steps / Intel:
- [ ] Production sign FW, upload FW and topology binaries to internal release folder
- [ ] Publish release on https://github.com/thesofproject/sof-bin/blob/main/HOWTO-new-release.md

Final Release, vendor specific steps / OtherVendor:
- add steps here

Final Steps:
- [ ] Declare the v2.xx release on https://www.sofproject.org/blog/
- [ ] Move any unfinished issues to next milestone

kv2019i, 2024-09-13:
rc1 and stable branch out delayed to next week. A new attempt on Monday 16th. We need to get https://github.com/thesofproject/sof/pull/9472 in. Everything else should be ready for rc1.

lgirdwood, 2024-09-16:
@amber fyi.

kv2019i, 2024-09-16:
rc1 tagged https://github.com/thesofproject/sof/tree/v2.11-rc1

kv2019i, 2024-09-27:
Release complete, planning started for v2.12 at https://github.com/thesofproject/sof/issues/9519