thesofproject/sof #9377: [BUG] Incdorrect LLP reported for SSP on LNL, causing broken delay reporting for user space

ujfalusi, 2024-08-20:
**Describe the bug**
The firmware reported llp value is incorrect for SSP link type.

**To Reproduce**
Run audio playback on SSP with application that uses the delay value, like `mpv`:
```
mpv --audio-device=alsa/hw:0,0 Echoplex-48KHz_S16_LE.wav
```

**Reproduction Rate**
10/10 (always)

**Expected behavior**
The LLP value reported by the firmware is correct.

**Impact**
video playback with audio using SSP link will fail as the audio delay is incorrect

**Environment**
LNL with nocodec topology

**Screenshots or console output**
Debug diff for the kernel:
```diff
diff --git a/sound/soc/sof/ipc4-pcm.c b/sound/soc/sof/ipc4-pcm.c
index 4df2be3d39eb..b816ffc454f2 100644
--- a/sound/soc/sof/ipc4-pcm.c
+++ b/sound/soc/sof/ipc4-pcm.c
@@ -966,6 +975,7 @@ static int sof_ipc4_pcm_pointer(struct snd_soc_component *component,
 	struct snd_sof_pcm_stream *sps;
 	u64 dai_cnt, host_cnt, host_ptr;
 	struct snd_sof_pcm *spcm;
+	u64 tmp_cnt;
 	int ret;
 
 	spcm = snd_sof_find_spcm_dai(component, rtd);
@@ -1008,6 +1018,7 @@ static int sof_ipc4_pcm_pointer(struct snd_soc_component *component,
 		sof_mailbox_read(sdev, time_info->llp_offset, &llp, sizeof(llp));
 		dai_cnt = ((u64)llp.reading.llp_u << 32) | llp.reading.llp_l;
 	}
+	tmp_cnt = dai_cnt;
 	dai_cnt += time_info->stream_end_offset;
 
 	/* In two cases dai dma counter is not accurate
@@ -1053,10 +1064,18 @@ static int sof_ipc4_pcm_pointer(struct snd_soc_component *component,
 	}
 
 	if (head_cnt < tail_cnt) {
+		pr_warn("[peter] | llp_offset: %#x, offset: %llu / %llu, tail_cnt: %lu (%llu), head_cnt: %lu, delay: %lu (OhNooo)\n",
+			time_info->llp_offset, time_info->stream_start_offset, time_info->stream_end_offset,
+			tail_cnt, tmp_cnt, head_cnt,
+			substream->runtime->boundary - tail_cnt + head_cnt);
 		time_info->delay = time_info->boundary - tail_cnt + head_cnt;
 		goto out;
 	}
 
+	pr_warn("[peter] | llp_offset: %#x, offset: %llu / %llu, tail_cnt: %lu (%llu), head_cnt: %lu, delay: %lu\n",
+		time_info->llp_offset, time_info->stream_start_offset, time_info->stream_end_offset,
+		tail_cnt, tmp_cnt, head_cnt, head_cnt - tail_cnt);
+
 	time_info->delay =  head_cnt - tail_cnt;
 
 out:
@@ -1067,6 +1086,7 @@ static int sof_ipc4_pcm_pointer(struct snd_soc_component *component,
 	div64_u64_rem(host_ptr, snd_pcm_lib_buffer_bytes(substream), &host_ptr);
 	*pointer = bytes_to_frames(substream->runtime, host_ptr);
 
+	pr_warn("[peter] | ldp pointer: %lu\n", *pointer);
 	return 0;
 }
```
Run:
```
mpv --audio-device=alsa/hw:0,0 Echoplex-48KHz_S16_LE.wav
```
On LNL-SSP:
```
[   36.261531] [peter] | llp_offset: 0x180740, offset: 208 / 0, tail_cnt: 2310751231805576626 (11534123268660352640), head_cnt: 192, delay: 3093568321039018766 (OhNooo)
[   36.261537] [peter] | ldp pointer: 192
[   36.261651] [peter] | llp_offset: 0x180740, offset: 208 / 0, tail_cnt: 2310751231805576626 (11534123268660352640), head_cnt: 192, delay: 3093568321039018766 (OhNooo)
[   36.261662] [peter] | ldp pointer: 192
[   36.261737] [peter] | llp_offset: 0x180740, offset: 208 / 0, tail_cnt: 2310751231805576626 (11534123268660352640), head_cnt: 192, delay: 3093568321039018766 (OhNooo)
[   36.261741] [peter] | ldp pointer: 192
[   36.262599] [peter] | llp_offset: 0x180740, offset: 208 / 0, tail_cnt: 2310749582538134962 (11534121619392910976), head_cnt: 240, delay: 3093569970306460478 (OhNooo)
[   36.262603] [peter] | ldp pointer: 240
[   36.281912] [peter] | llp_offset: 0x180740, offset: 208 / 0, tail_cnt: 2310749582538134962 (11534121619392910976), head_cnt: 1200, delay: 3093569970306461438 (OhNooo)
[   36.281917] [peter] | ldp pointer: 1200
[   36.286111] [peter] | llp_offset: 0x180740, offset: 208 / 0, tail_cnt: 2310749582538134962 (11534121619392910976), head_cnt: 1392, delay: 3093569970306461630 (OhNooo)
[   36.286117] [peter] | ldp pointer: 1392
[   36.286140] [peter] | llp_offset: 0x180740, offset: 208 / 0, tail_cnt: 2310749582538134962 (11534121619392910976), head_cnt: 1392, delay: 3093569970306461630 (OhNooo)
[   36.286142] [peter] | ldp pointer: 1392
```
On LNL-SDW and HDA the delay reporting seams to be disabled, so no information is taken from the firmware, we report 0 delay.

On MTL-SSP:
```
[  712.712353] [peter] | llp_offset: 0x180740, offset: 160 / 0, tail_cnt: 0 (99), head_cnt: 301, delay: 301
[  712.712358] [peter] | ldp pointer: 240
[  712.712403] [peter] | llp_offset: 0x180740, offset: 160 / 0, tail_cnt: 0 (99), head_cnt: 301, delay: 301
[  712.712412] [peter] | ldp pointer: 240
[  712.712560] [peter] | llp_offset: 0x180740, offset: 160 / 0, tail_cnt: 0 (99), head_cnt: 301, delay: 301
[  712.712563] [peter] | ldp pointer: 240
[  712.713336] [peter] | llp_offset: 0x180740, offset: 160 / 0, tail_cnt: 0 (147), head_cnt: 301, delay: 301
[  712.713339] [peter] | ldp pointer: 288
[  712.732365] [peter] | llp_offset: 0x180740, offset: 160 / 0, tail_cnt: 899 (1059), head_cnt: 1200, delay: 301
[  712.732368] [peter] | ldp pointer: 1200
[  712.737518] [peter] | llp_offset: 0x180740, offset: 160 / 0, tail_cnt: 1139 (1299), head_cnt: 1440, delay: 301
[  712.737525] [peter] | ldp pointer: 1440
[  712.737545] [peter] | llp_offset: 0x180740, offset: 160 / 0, tail_cnt: 1139 (1299), head_cnt: 1440, delay: 301
[  712.737546] [peter] | ldp pointer: 1440
```
On MTL-SDW:
```
[   25.807972] [peter] | llp_offset: 0x180740, offset: 120 / 0, tail_cnt: 0 (49), head_cnt: 263, delay: 263
[   25.807978] [peter] | ldp pointer: 192
[   25.808028] [peter] | llp_offset: 0x180740, offset: 120 / 0, tail_cnt: 0 (49), head_cnt: 263, delay: 263
[   25.808037] [peter] | ldp pointer: 192
[   25.808094] [peter] | llp_offset: 0x180740, offset: 120 / 0, tail_cnt: 0 (49), head_cnt: 263, delay: 263
[   25.808097] [peter] | ldp pointer: 192
[   25.808575] [peter] | llp_offset: 0x180740, offset: 120 / 0, tail_cnt: 0 (95), head_cnt: 265, delay: 265
[   25.808578] [peter] | ldp pointer: 240
[   25.828391] [peter] | llp_offset: 0x180740, offset: 120 / 0, tail_cnt: 935 (1055), head_cnt: 1200, delay: 265
[   25.828394] [peter] | ldp pointer: 1200
[   25.833133] [peter] | llp_offset: 0x180740, offset: 120 / 0, tail_cnt: 1127 (1247), head_cnt: 1392, delay: 265
[   25.833140] [peter] | ldp pointer: 1392
[   25.833155] [peter] | llp_offset: 0x180740, offset: 120 / 0, tail_cnt: 1127 (1247), head_cnt: 1392, delay: 265
[   25.833159] [peter] | ldp pointer: 1392
```

ujfalusi, 2024-08-20:
Description updated with correct LNL-SDW pointer log

lgirdwood, 2024-08-21:
@abonislawski @softwarecki could we have an incorrect DT entry that means we read the wrong LLP register ?

ujfalusi, 2024-08-29:
Zephyr PR with a fix: https://github.com/zephyrproject-rtos/zephyr/pull/77726