thesofproject/sof #9378: [BUG] Mismatch between Linux structure fields offsets and SOF

LaurentiuM1234, 2024-08-20:
**Describe the bug**
According to the GCC documentation (regarding enums):

```
Normally, the type is unsigned int if there are no negative values in the enumeration, otherwise int.
If -fshort-enums is specified, then if there are negative values it is the first of signed char,
short and int that can represent all the values, otherwise it is the first of unsigned char,
unsigned short and unsigned int that can represent all the values.

On some targets, -fshort-enums is the default; this is determined by the ABI.
```

enums are represented as ints (unsigned or not). If `-fshort-enums` is specified, the size of the enum is `char`, `short`, or `int` (or their unsigned versions if applicable).

Most importantly, some targets have this option enabled by default.

Although I wasn't able to get a confirmation on this, from testing it would seem like `arm-zephyr-eabi-gcc` has this behavior.

Now, the main problem with this is the fact that some structures used in the IPC protocol have enum fields, leading to Linux and SOF having the fields of these structure at different offsets. This results in SOF not being able to properly interpret IPC-related data from Linux.

The "problematic" structures we were able to find are:

`struct ipc4_pin_props`, which uses `enum ipc4_stream_type`. We don't actually use IPC4 but since this is also included somewhere it generates compilation warnings:

`warning: 'format' offset 1 in 'struct ipc4_pin_props' isn't aligned to 4`

`struct sof_ipc_comp`, which uses `enum sof_comp_type`. In this case, the firmware generates an error since it's not able to properly interpret the `struct sof_ipc_comp` data from Linux.

This all happens when SOF is built for a Cortex-M based platform with the following compiler: `arm-zephyr-eabi-gcc`.

My solution to this would be to replace the usage of enums with data types such as uint32_t, uint16_t, etc... (i.e: types for which we clearly know the size of) in structures used by the IPC protocol. For now, I think we should start with just `struct sof_ipc_comp` and (potentially) `struct ipc4_pin_props`, since these are the only ones causing problems.

LaurentiuM1234, 2024-08-20:
CC: @kv2019i @lgirdwood

lgirdwood, 2024-08-21:
> CC: @kv2019i @lgirdwood

Ack - fix looks good, lets get in for 2.11

closed 2024-08-22 by lgirdwood