thesofproject/sof #9418: [BUG] delay reporting starts goes out of sync and host position overtakes link position

kv2019i, 2024-08-28:
**Describe the bug**
Delay reporting stops working after some time of streaming.

**To Reproduce**
Play video to a HDA codec on Intel TGL platform with e.g. "mpv".

**Reproduction Rate**
Not 100% but high

**Expected behavior**
Delay reporting keeps working

**Impact**
Can cause lip sync to fail with certain video plays that rely on ALSA delay reporting interfaces

**Environment**
1) Branch name and commit hash of the 2 repositories: sof (firmware/topology) and linux (kernel driver).
    * Kernel: sof-dev as of 28th Aug 2024
    * SOF: SOF main as of 28th Aug 2024
2) Name of the topology file
    * Topology: sof-hda-generic
3) Name of the platform(s) on which the bug is observed.
    * Platform: TGL-H

**Screenshots or console output**

```
$ mpv --audio-device=alsa/hw:0,0 foobar.mov
 (+) Video --vid=1 (*) (hevc 1920x1080 23.976fps)
 (+) Audio --aid=1 (*) (pcm_s24le 2ch 48000Hz)
Using hardware decoding (vaapi).
AO: [alsa] 48000Hz stereo 2ch s32
VO: [gpu] 1920x1080 vaapi[p010]
AV: 00:00:38 / 00:02:00 (32%) A-V:  0.000

Audio/Video desynchronisation detected! Possible reasons include too slow
hardware, temporary CPU spikes, broken drivers, and broken files. Audio
position will not match to the video (see A-V status field).

AV: -13343998895:52:45 / 00:02:00 (0%) A-V:  0.000 Dropped: 1394
```

kv2019i, 2024-08-28:
FYI @ujfalusi  with whom I've already debugged this. Suspicion is that start_offset calculation has a bug and/or there is FW buffering that is not accounted. Logs show that delay reporting is correct at first, but in some test runs, the reported decreases in 1ms steps multiple times, and finally the values kernel sees wrap around leading to invalid values reported to application.

ujfalusi, 2024-08-29:
What I have observed is that the LDP (host pointer) and LLP (Link pointer) starts to converge and over time LLP overtakes LDP.
Afaik this is somehow happening in steps, confirmed by your plot graph.
What I have seen is that when the distance between LDP and LLP changes I see this print in firmware log:
```
[ 2317.902373] <inf> ll_schedule: zephyr_domain_thread_fn: ll core 0 timer avg 2350, max 2421, overruns 0
[ 2318.926373] <inf> ll_schedule: zephyr_domain_thread_fn: ll core 0 timer avg 2351, max 4103, overruns 0
[ 2319.761355] <inf> host_comp: host_get_copy_bytes_normal: comp:0 0x4 no bytes to copy, available samples: 0, free_samples: 192
[ 2319.950373] <inf> ll_schedule: zephyr_domain_thread_fn: ll core 0 timer avg 2350, max 3937, overruns 0
[ 2320.974373] <inf> ll_schedule: zephyr_domain_thread_fn: ll core 0 timer avg 2347, max 2430, overruns 0
```
It is _always_  there when the distance changes between the two pointer - which translates to delay values.

Furthermore, I can observe a small audible tick/tock when this happens when playing a 1KHz sine wav file. I cannot hear it with real music, but that is expected.

So far I have been able to reproduce this on: TGL HDA, TGL SSP, LNL SSP.
But not on MTL SDW, MTL SSP.

This is not related to the link type and the DMA servicing it.

ujfalusi, 2024-08-29:
> Tested FW fast_mode, did not help.

The fast mode would not copy if `available samples: 0`, so it cannot help.

ujfalusi, 2024-08-30:
I think I have found something interesting in logs. Added print for the host side before every `dma_reload()` call:
```
[ 2973.993401] <inf> host_comp: host_copy_normal: comp:0 0x4 host DMA reload 2: size: 192
[ 2973.994401] <inf> host_comp: host_copy_normal: comp:0 0x4 host DMA reload 2: size: 192
[ 2973.995401] <inf> host_comp: host_copy_normal: comp:0 0x4 host DMA reload 2: size: 192
[ 2973.996401] <inf> host_comp: host_copy_normal: comp:0 0x4 host DMA reload 2: size: 192
[ 2973.997398] <inf> host_comp: host_get_copy_bytes_normal: comp:0 0x4 no bytes to copy, available samples: 0, free_samples: 192, pending_length: 0, partial_size: 0
[ 2973.998401] <inf> host_comp: host_copy_normal: comp:0 0x4 host DMA reload 2: size: 192
[ 2973.999401] <inf> host_comp: host_copy_normal: comp:0 0x4 host DMA reload 2: size: 192
[ 2974.000401] <inf> host_comp: host_copy_normal: comp:0 0x4 host DMA reload 2: size: 192

[ 3005.581403] <inf> host_comp: host_copy_normal: comp:0 0x4 host DMA reload 2: size: 192
[ 3005.582401] <inf> host_comp: host_copy_normal: comp:0 0x4 host DMA reload 2: size: 192
[ 3005.583401] <inf> host_comp: host_copy_normal: comp:0 0x4 host DMA reload 2: size: 192
[ 3005.584398] <inf> host_comp: host_get_copy_bytes_normal: comp:0 0x4 no bytes to copy, available samples: 0, free_samples: 192, pending_length: 0, partial_size: 0
[ 3005.585401] <inf> host_comp: host_copy_normal: comp:0 0x4 host DMA reload 2: size: 192
[ 3005.586403] <inf> host_comp: host_copy_normal: comp:0 0x4 host DMA reload 2: size: 192
[ 3005.587403] <inf> host_comp: host_copy_normal: comp:0 0x4 host DMA reload 2: size: 192
[ 3005.588401] <inf> host_comp: host_copy_normal: comp:0 0x4 host DMA reload 2: size: 192
[ 3005.589401] <inf> host_comp: host_copy_normal: comp:0 0x4 host DMA reload 2: size: 192
[ 3005.590401] <inf> host_comp: host_copy_normal: comp:0 0x4 host DMA reload 2: size: 192

[ 3078.152401] <inf> host_comp: host_copy_normal: comp:0 0x4 host DMA reload 2: size: 192
[ 3078.153401] <inf> host_comp: host_copy_normal: comp:0 0x4 host DMA reload 2: size: 192
[ 3078.154398] <inf> host_comp: host_get_copy_bytes_normal: comp:0 0x4 no bytes to copy, available samples: 0, free_samples: 192, pending_length: 0, partial_size: 0
[ 3078.155401] <inf> host_comp: host_copy_normal: comp:0 0x4 host DMA reload 2: size: 192
[ 3078.156401] <inf> host_comp: host_copy_normal: comp:0 0x4 host DMA reload 2: size: 192

[ 3122.208401] <inf> host_comp: host_copy_normal: comp:0 0x4 host DMA reload 2: size: 192
[ 3122.209401] <inf> host_comp: host_copy_normal: comp:0 0x4 host DMA reload 2: size: 192
[ 3122.210398] <inf> host_comp: host_get_copy_bytes_normal: comp:0 0x4 no bytes to copy, available samples: 0, free_samples: 192, pending_length: 0, partial_size: 0
[ 3122.211401] <inf> host_comp: host_copy_normal: comp:0 0x4 host DMA reload 2: size: 192
[ 3122.212421] <inf> host_comp: host_copy_normal: comp:0 0x4 host DMA reload 2: size: 192

[ 3155.990401] <inf> host_comp: host_copy_normal: comp:0 0x4 host DMA reload 2: size: 192
[ 3155.991401] <inf> host_comp: host_copy_normal: comp:0 0x4 host DMA reload 2: size: 192
[ 3155.992398] <inf> host_comp: host_get_copy_bytes_normal: comp:0 0x4 no bytes to copy, available samples: 0, free_samples: 192, pending_length: 0, partial_size: 0
[ 3155.993400] <inf> host_comp: host_copy_normal: comp:0 0x4 host DMA reload 2: size: 192
[ 3155.994403] <inf> host_comp: host_copy_normal: comp:0 0x4 host DMA reload 2: size: 192

[ 4050.473290] <inf> host_comp: host_copy_normal: comp:0 0x4 host DMA reload 2: size: 192
[ 4050.474291] <inf> host_comp: host_copy_normal: comp:0 0x4 host DMA reload 2: size: 192
[ 4050.475286] <inf> host_comp: host_get_copy_bytes_normal: comp:0 0x4 no bytes to copy, available samples: 0, free_samples: 192, pending_length: 0, partial_size: 0
[ 4050.476290] <inf> host_comp: host_copy_normal: comp:0 0x4 host DMA reload 2: size: 192
[ 4050.477290] <inf> host_comp: host_copy_normal: comp:0 0x4 host DMA reload 2: size: 192
```
Most of the time the `dma_reload()` is called with timestamp ending with 400+ but when there is a report of nothing to copy, the timestamp end is less than 400, it looks like that the timer comes just a bit early and the DMA has not yet recorded the transfer, or the transfer yet to complete?

In any case, with this patch:
```diff
diff --git a/src/audio/host-zephyr.c b/src/audio/host-zephyr.c
index ac1751059eb5..bc7e52dfb9bb 100644
--- a/src/audio/host-zephyr.c
+++ b/src/audio/host-zephyr.c
@@ -372,8 +372,10 @@ static uint32_t host_get_copy_bytes_normal(struct host_data *hd, struct comp_dev
 	uint32_t free_samples;
 	uint32_t dma_sample_bytes;
 	uint32_t dma_copy_bytes;
+	bool retry = false;
 	int ret;
 
+again:
 	/* get data sizes from DMA */
 	ret = dma_get_status(hd->chan->dma->z_dev, hd->chan->index, &dma_stat);
 	if (ret < 0) {
@@ -403,9 +405,17 @@ static uint32_t host_get_copy_bytes_normal(struct host_data *hd, struct comp_dev
 	if (!(hd->ipc_host.feature_mask & BIT(IPC4_COPIER_FAST_MODE)))
 		dma_copy_bytes = MIN(hd->period_bytes, dma_copy_bytes);
 
-	if (!dma_copy_bytes)
-		comp_info(dev, "no bytes to copy, available samples: %d, free_samples: %d",
-			  avail_samples, free_samples);
+	if (!dma_copy_bytes) {
+		if (!retry && dma_stat.pending_length == 0 && hd->partial_size == 0) {
+			comp_info(dev, "no bytes to copy, retry");
+			k_usleep(5);
+			retry = true;
+			goto again;
+		}
+
+		comp_info(dev, "no bytes to copy, available samples: %d, free_samples: %d, pending_length: %d, partial_size: %d",
+			  avail_samples, free_samples, dma_stat.pending_length, hd->partial_size);
+	}
 
 	/* dma_copy_bytes should be aligned to minimum possible chunk of
 	 * data to be copied by dma.
```
The issue goes away. I can see occasional prints about the retry, but on second try with a short delay all is good, we have data available....

marc-hb, 2024-09-02:
Could this be related to?
- #8761

kv2019i, 2024-09-05:
Fixed via https://github.com/thesofproject/sof/pull/9437

closed 2024-09-05 by kv2019i