thesofproject/sof #9460: [BUG] Trace breaks build in 64 bit Posix build

cujomalainey, 2024-09-10:
**Describe the bug**
Traces fail to build in native sim 64 bit environment due to the fact that pointers exceed the 32bit container maximum

**To Reproduce**
Checkout https://github.com/cujomalainey/sof/tree/top
Run `scripts/fuzz.sh -b -a x86_64`

**Reproduction Rate**
100%

**Expected behavior**
Code builds

**Impact**
Blocks the ability to use logging with 64 bit builds

**Environment**
1) Branch name and commit hash of the 2 repositories: sof (firmware/topology) and linux (kernel driver).
    * Kernel: N/A
    * SOF: [{SHA}](https://github.com/cujomalainey/sof/tree/top)
2) Name of the topology file
    * Topology: N/A
3) Name of the platform(s) on which the bug is observed.
    * Platform: POSIX

**Screenshots or console output**
```
FAILED: modules/sof/CMakeFiles/modules_sof.dir/src/sof_workspace/sof/src/ipc/ipc3/host-page-table.c.obj                                                                                                                                                                                                                                                     
/usr/local/bin/clang -DCC_OPTIMIZE_FLAGS=\"-O2\" -DKERNEL -DK_HEAP_MEM_POOL_SIZE=2048 -DXCC_TOOLS_VERSION=\"llvm\" -D__ZEPHYR__=1 -DRELATIVE_FILE=\"../sof/src/ipc/ipc3/host-page-table.c\" -I/src/sof_workspace/build-fuzz/zephyr/include/generated/zephyr -I/src/sof_workspace/zephyr/include -I/src/sof_workspace/build-fuzz/zephyr/include/generated -I/
src/sof_workspace/zephyr/soc/native/inf_clock -I/src/sof_workspace/zephyr/boards/native/native_sim -I/src/sof_workspace/zephyr/scripts/native_simulator/common/src/include -I/src/sof_workspace/zephyr/scripts/native_simulator/native/src/include -I/src/sof_workspace/zephyr/drivers -I/src/sof_workspace/sof/zephyr/include -I/src/sof_workspace/sof/src/
platform/posix/include -I/src/sof_workspace/sof/src/audio/smart_amp/include/dsm_api/inc -I/src/sof_workspace/sof/tools/rimage/src/include -I/src/sof_workspace/sof/src/include -I/src/sof_workspace/sof/src/arch/host/include -I/src/sof_workspace/sof/third_party/include -I/src/sof_workspace/sof/posix/include -isystem /src/sof_workspace/zephyr/lib/lib
c/minimal/include -isystem /src/sof_workspace/zephyr/lib/libc/common/include -fno-strict-aliasing -O2 -imacros /src/sof_workspace/build-fuzz/zephyr/include/generated/zephyr/autoconf.h -ffreestanding -fno-common -g -gdwarf-4 -fcolor-diagnostics --config /src/sof_workspace/zephyr/cmake/toolchain/llvm/clang_libgcc.cfg -Wall -Wformat -Wformat-securit
y -Wno-format-zero-length -Wno-unused-but-set-variable -Wno-typedef-redefinition -Wno-deprecated-non-prototype -Wdouble-promotion -Wno-pointer-sign -Wpointer-arith -Wno-sometimes-uninitialized -Wno-shift-overflow -Wno-missing-braces -Wno-self-assign -Wno-address-of-packed-member -Wno-unused-function -Wno-initializer-overrides -Wno-section -Wno-un
known-warning-option -Wno-unused-variable -Wno-format-invalid-specifier -Wno-gnu -Wno-tautological-compare -Werror=implicit-int -fno-pic -fno-pie -fno-sanitize-recover=all -fno-asynchronous-unwind-tables -fstrict-overflow -Wno-vla -fmacro-prefix-map=/src/sof_workspace/sof/app=CMAKE_SOURCE_DIR -fmacro-prefix-map=/src/sof_workspace/zephyr=ZEPHYR_BA
SE -fmacro-prefix-map=/src/sof_workspace=WEST_TOPDIR -ffunction-sections -fdata-sections -m64 -fPIC -fvisibility=hidden -nostdinc -isystem /usr/local/lib/clang/18/include -include /src/sof_workspace/zephyr/arch/posix/include/undef_system_defines.h -fno-builtin -fsanitize-recover=all -fsanitize=address,fuzzer -std=c11 -fno-inline-functions -std=gn
u99 -MD -MT modules/sof/CMakeFiles/modules_sof.dir/src/sof_workspace/sof/src/ipc/ipc3/host-page-table.c.obj -MF modules/sof/CMakeFiles/modules_sof.dir/src/sof_workspace/sof/src/ipc/ipc3/host-page-table.c.obj.d -o modules/sof/CMakeFiles/modules_sof.dir/src/sof_workspace/sof/src/ipc/ipc3/host-page-table.c.obj -c /src/sof_workspace/sof/src/ipc/ipc3/
host-page-table.c                                                                                                                                                              
/src/sof_workspace/sof/src/ipc/ipc3/host-page-table.c:48:3: error: array size is negative                                                                                      
   48 |                 tr_err(&ipc_tr, "ipc_parse_page_descriptors(): There is no heap free with this block size: %zu",                                                       
      |                 ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~                                                       
   49 |                        sizeof(struct dma_sg_elem) * ring->pages);                                                                                                      
      |                        ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~                                                                                                       
/src/sof_workspace/sof/zephyr/include/sof/trace/../../../../src/include/sof/trace/trace.h:491:2: note: expanded from macro 'tr_err'                                            
  491 |         trace_error_with_ids(_TRACE_INV_CLASS, ctx, \                                                                                                                  
      |         ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~                                                                                                                  
  492 |                              _TRACE_INV_ID, _TRACE_INV_ID, fmt, ##__VA_ARGS__)                                                                                         
      |                              ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~                                                                                         
/src/sof_workspace/sof/zephyr/include/sof/trace/../../../../src/include/sof/trace/trace.h:363:2: note: expanded from macro 'trace_error_with_ids'                              
  363 |         _trace_error_with_ids(class, ctx, id_1, id_2, format, ##__VA_ARGS__)   
      |         ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/src/sof_workspace/sof/zephyr/include/sof/trace/../../../../src/include/sof/trace/trace.h:360:2: note: expanded from macro '_trace_error_with_ids'                                                                                                                                                                                                          
  360 |         _log_message(trace_log_filtered, true, LOG_LEVEL_CRITICAL, class, ctx, id_1,    \                                                                                                                                                                                                                                                           
      |         ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~                                                                                                                                                                                                                                                           
  361 |                      id_2, format, ##__VA_ARGS__)                                                                                                                                                                                                                                                                                                   
      |                      ~~~~~~~~~~~~~~~~~~~~~~~~~~~~                          
note: (skipping 21 expansions in backtrace; use -fmacro-backtrace-limit=0 to see all)
/src/sof_workspace/sof/posix/include/sof/trace/preproc-private.h:120:63: note: expanded from macro '_META_REQRS_2'                                                                                                                                                                                                                                          
  120 | #define _META_REQRS_2(...)    _META_REQRS_1(  _META_REQRS_1  (__VA_ARGS__))
      |                               ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~    
/src/sof_workspace/sof/posix/include/sof/trace/preproc-private.h:121:28: note: expanded from macro '_META_REQRS_1'                                                                                                                               [... cut ...]
1 error generated.                                                                     
[87/296] Building C object modules/sof/CMakeFiles/modules_sof.dir/src/sof_workspace/sof/src/ipc/ipc3/dai.c.obj                                                                                                                                                                                                                                              
FAILED: modules/sof/CMakeFiles/modules_sof.dir/src/sof_workspace/sof/src/ipc/ipc3/dai.c.obj 
```

@andyross FYI

kv2019i, 2025-01-03:
No owner, v2.13

kv2019i, 2025-04-23:
This would be good to solve for v2.13 but still lacking an owner.

kv2019i, 2025-05-02:
This is now fixed with changes to trace headers for posix builds. 

Tested:

"scripts/fuzz.sh -b -a x86_64"

with today's SOF main and build now passes.

closed 2025-05-02 by kv2019i