thesofproject/sof #9502: [BUG] Distorted "metallic" sound after snd_pcm_drain snd_pcm_prepare cycle

madenvel, 2024-09-20:
**Describe the bug**
Intermittently sounds becomes distorted after snd_pcm_drain / snd_pcm_prepare ALSA call. Does not happen if the device is reopened after snd_pcm_drain with snd_pcm_close / snd_pcm_open.

**To Reproduce**
Please find a sample code to reproduce the bug: https://bugzilla.kernel.org/show_bug.cgi?id=219260
Reproducible on ThinkBook 13x Gen4 hardware.

**Reproduction Rate**
Roughly 2/10 times with the test code attached.

**Expected behavior**
Device should not need to re-open before the playback.

**Impact**
Affects sound playback quite significantly, making devices unusable without external audio device.

**Environment**
1) Branch name and commit hash of the 2 repositories: sof (firmware/topology) and linux (kernel driver).
    * Kernel: 6.12.0-0.rc0.20240918gt4a39ac5b.306.vanilla.fc40.x86_64, I assume 20240918gt4a39ac5b is the 
    * SOF: v2024.06
2) Name of the topology file
    * Topology: sof-hda-generic-4ch.tplg
3) Name of the platform(s) on which the bug is observed.
    * Platform: Intel64

 ftrace file with snd* filter of the sequence of kernel calls: [trace.txt](https://github.com/user-attachments/files/17080964/trace.txt)

kv2019i, 2024-09-30:
Thank you @madenvel for the repro app in https://bugzilla.kernel.org/attachment.cgi?id=306866 -- we can reproduce the problem.

It seems it's the snd_pcm_drain() that triggers a bug in SOF stack. Just commeting out the drain between iterations, and the bug is not triggerred (but obviously this changes the behaviour so not a solution). Replacing snd_pcm_drain() with snd_pcm_drop() does not solve. 

@ujfalusi has a potential fix, would this be a kernel side or do we keep on FW side?

ujfalusi, 2024-09-30:
I would keep this in firmware for now.
The 'fix' in kernel is too intrusive for my liking and it results with IPC error on pcm_free.

ujfalusi, 2024-09-30:
@kv2019i, dropping the snd_pcm_drain() do changes the behavior since there will be no stop and no prepare happening, just continuous audio playback.

kv2019i, 2024-09-30:
Note on scope - affects cases where HD-DMA used on the link side.

ranj063, 2024-10-01:
@ujfalusi @madenvel @kv2019i I have an alternate proposal to fix this in the kernel. Basically cleaning up the link DMA during stop always also seems to help. Could you please verify if this works?
```
diff --git a/sound/soc/sof/intel/hda-dai.c b/sound/soc/sof/intel/hda-dai.c
index f1a8491bba9e..df8bf497e224 100644
--- a/sound/soc/sof/intel/hda-dai.c
+++ b/sound/soc/sof/intel/hda-dai.c
@@ -303,6 +303,7 @@ static int __maybe_unused hda_dai_trigger(struct snd_pcm_substream *substream, i
 
        switch (cmd) {
        case SNDRV_PCM_TRIGGER_SUSPEND:
+       case SNDRV_PCM_TRIGGER_STOP:
                ret = hda_link_dma_cleanup(substream, hext_stream, dai);
                if (ret < 0) {
                        dev_err(sdev->dev, "%s: failed to clean up link DMA\n", __func__);
```

ranj063, 2024-10-01:
I have restricted this to IPC4 and playback only to address this specific issue in my PR https://github.com/thesofproject/linux/pull/5197

madenvel, 2024-10-01:
@ranj063 I confirm that your **patch fixes the issue** for the test code (`repro.cpp`). I ran 20 iterations and did not hear any distortion while without the patch distortion is present at least in 2 runs out of 10 (usually on 5th and 6th iteration). 
I applied it on top of 6.11 coming with Fedora 41 Beta.

Would you have time to contribute it to the kernel?

@ujfalusi to comment on this.

ranj063, 2024-10-02:
> @ranj063 I confirm that your **patch fixes the issue** for the test code (`repro.cpp`). I ran 20 iterations and did not hear any distortion while without the patch distortion is present at least in [...]

@madenvel my PR still needs to be reviewed and tested in our CI before we can upstream it. We will keep you updated.

kv2019i, 2024-10-02:
Let me move this to kernel. Looking at this from pure FW point of view, the problematic sequence seems to be this (running repro.cpp modified to have 3sec delay between calls to play()):

```
# 1. at end of each test iteration, DAI pipeline is put to PAUSED state
[  122.155748] <inf> ipc: ipc_cmd: rx   : 0x13010003|0x0
[  122.155765] <inf> pipe: pipeline_trigger: pipe:1 0x0 pipe trigger cmd 2
# 2. a modified repro.cpp adds a 3sec delay which shows here, when next play() iteration 
# occurs, DAI pipeline is put to RESET state
[  125.165575] <inf> ipc: ipc_cmd: rx   : 0x13000002|0x1
[  125.165593] <inf> pipe: pipeline_trigger: pipe:1 0x0 pipe trigger cmd 0
[  125.165648] <inf> pipe: pipeline_trigger: pipe:0 0x0 pipe trigger cmd 0
[  125.165668] <wrn> copier: copier_comp_trigger: comp:0 0x4 dai is not ready
# 3. MOD_UNBIND and DELETE_PIPELINE are sent next
[  125.166176] <inf> ipc: ipc_cmd: rx   : 0x46000002|0x3
[  125.166938] <inf> ipc: ipc_cmd: rx   : 0x12000000|0x0
```

So on Intel platforms with HD-link-DMA, the host must clear the stream-tag in step3 (or arguably already for step2), as the FW can no longer setup the link DMA again for streaming as there is stale data from previous use and the DSP cannot reset this state. https://github.com/zephyrproject-rtos/zephyr/pull/79275 brings the HW to a state that does not cause artifacts upon next iteration, but it's really not correct as some state is retained when there should be none.