thesofproject/sof #9508: [BUG] rimage-extra.args not passed to llext module signing

kv2019i, 2024-09-24:
**Describe the bug**¨
The riamage extra args are not passed to llext module signing in xtensa-build-zephyr.py. 

build with extra command:
west config rimage.extra-args '-p -i 4'
 
Main image build command:
/buildagent/workspace/24951/build-rimage/rimage -o /buildagent/workspace/24951/build-lnl/zephyr/zephyr.ri -k /root/keys/mtl_private_key.pem -f 2.11.0 -b 1 -p -i 4 -e -c /buildagent/workspace/24951/build-lnl/zephyr/misc/generated/rimage_config.toml  

Llext build command:
 
/buildagent/workspace/24951/build-rimage/rimage -o /buildagent/workspace/24951/build-lnl/zephyr/drc_llext/drc.llext.ri -e -c /buildagent/workspace/24951/build-lnl/zephyr/drc_llext/rimage_config.toml -k /root/keys/mtl_private_key.pem -l -r /buildagent/workspace/24951/build-lnl/zephyr/drc_llext/drc.llext
 
**To Reproduce**
Build with extra args.

**Reproduction Rate**
100%

**Expected behavior**
rimage extra args passed in the same way to both main and module builds.

**Impact**
Cannot release binaries as modules.

lyakh, 2024-09-26:
#9509

kv2019i, 2024-09-27:
closed via #9509

closed 2024-09-27 by kv2019i