thesofproject/sof #9513: [BUG] enabling telemetry breaks shell and winconsole

kv2019i, 2024-09-25:
**Describe the bug**
It seems following debug features are broken on multiple Intel targets:
- app/shell_overlay.conf 
- app/winconsole_overlay.conf

... it turns out this is related to enabling SOF telemetry with debug window backend in https://github.com/thesofproject/sof/pull/9186 . 

In theory these shouldn't conflict, shell backend is using debug window slot 0 (2.0), while winconsole is using window 3 (3.0), while telemetry is hardcoded to use 2.1. 

**To Reproduce**
Build with above overlays.

**Reproduction Rate**
100% on Intel LNL.

**Expected behavior**
telemetry should not affect uses of other memory window users.

**Impact**
debug feature unavailable

**Environment**
1) Branch name and commit hash of the 2 repositories: sof (firmware/topology) and linux (kernel driver).
    * Kernel: n/a
    * SOF: f2efd4a64413da727b89c6ecb145ab0f140450ee 
2) Name of the topology file
    * Topology: n/a
3) Name of the platform(s) on which the bug is observed.
    * Platform: Intel lnl

kv2019i, 2024-09-25:
@tobonex @jsarha any ideas? We hit this with @ujfalusi

tobonex, 2024-09-25:
@kv2019i Window 3 is used by performance measurements. Try disabling  CONFIG_SOF_TELEMETRY_PERFORMANCE_MEASUREMENTS to check.

kv2019i, 2024-09-25:
@tobonex thanks, aa, I missed this:

```
#define ADSP_PMW ((volatile uint32_t *) \
»       »        (sys_cache_uncached_ptr_get((__sparse_force void __sparse_cache *) \
»       »       »       »            (WIN3_MBASE + WIN3_OFFSET))))
```

So winconsole is clearly conflicting and you can use either/or.

This doesn't explain the conflict with shell, but the shell overlay also enables the winconsole, so maybe that causes the problem.

kv2019i, 2024-09-26:
Closed via #9514

closed 2024-09-26 by kv2019i