thesofproject/sof #9541: [FEATURE][Zephyr] implement clk.h with native Zephyr interface and drop platform clk.h

kv2019i, 2024-10-03:
**Is your feature request related to a problem? Please describe.**
Implement the SOF clk.h interface without depending on SOF platform interface and rather implement the functions using direct calls to Zephyr. This allows to move the platform specifics to Zephyr side and reduce the work needed to be done on SOF side to add new platforms.

Part of https://github.com/thesofproject/sof/issues/5794

**Additional context**
Primary use in code base is for clock_get_freq() (e.g. in basefw.c). Most of the other uses are in XTOS drivers, which can be ignored for Zephyr builds. All in all, it's a good question whether clk.h interface is needed at all for application use and whether the clk.h could be just dropped (and only used for XTOS driver and platform code).

IOW, is there any need to query the CPU frequency from application code? After all, clock gating and scaling can be used and any logic dependent on exact frequency, is potentially sensitive to platform differences.

kv2019i, 2024-10-04:
Ok this is somewhat more complex as src/lib/cpu-clk-manager.c has non-trivial use of clk.h and this is used on many native Zephyr SOF platforms.

lgirdwood, 2024-10-15:
> Ok this is somewhat more complex as src/lib/cpu-clk-manager.c has non-trivial use of clk.h and this is used on many native Zephyr SOF platforms.

Ok, pls do create a new generic application header for cpu-clk manager and we can transition to that and remove the old headers.

kv2019i, 2024-11-19:
Further observations of current design:
- src/lib/clk.h still used in Zephyr builds (initialized by platform layer)
- SOF clk.h provides interfaces to enumerate platform DSP clocks
      - Zephyr clock_control.h and DTS can describe invididual clocks but AFAIK there is not way to query and enumerate core clocks
- SOF clk.h provides interfaces to use SOF notifier to get async events on DSP core freq change
      - no matching fucnction in Zephyr, but also no actual users of this in SOF Zephyr builds, so maybe can be dropped
- Intel platforms use zephyr/dts/bindings/clock/intel,adsp-shim-clkctl.yaml schema to describe the DT data 
- platform clk.h provides platform specific defaults to KCPS control (like PRIMARY_CORE_BASE_CPS_USAGE)

lgirdwood, 2024-11-21:
> Ripping out the cpu-clk-manager usage to its own interface might still be the best course of action, but above is stll useful to input even if we go down on this path.

We need to rip out the SOF side code that deals with clock IP and clock change notifications, i.e. any RTOS service/API should be used and the SOF code should be a client that asks for clock changes.

Agree, I dont think we have any notifier usage for clock change so can be dropped.

kv2019i, 2024-11-22:
Some study into implementing clock manage on top of Zephyr:
- Zephyr does not yet have a clock management framework like Linux https://www.kernel.org/doc/Documentation/devicetree/bindings/opp/opp-v2.yaml
- A number of enhancements are in progress:
    - https://github.com/zephyrproject-rtos/zephyr/pull/72102
    - https://github.com/zephyrproject-rtos/zephyr/pull/70467
- In terms of vendor specific interfaces in Zephyr, for SOF platforms supported, the Intel clock driver covers some of the functionaly but e.g. does not provide introspection of available clocks.

Next step would be to clean up the Intel ACE platform code in SOF to use more Zephyr definitions.

kv2019i, 2024-12-13:
Feature cutoff for v2.12. The remaining work has complex dependencies to Zephyr, so I won't move this to 2.13 directly. Needs to be separately planned, when SOF can intercept the clock framework.