thesofproject/sof #9548: [BUG] Fuzzer build broken upstream

cujomalainey, 2024-10-07:
**Describe the bug**
Due to [zephyr's requirement](https://github.com/zephyrproject-rtos/zephyr/commit/9d1b36126535c611617c5e8edcf06d70f5879287) to use > python3.10, the upstream docker container build has been broken for a few weeks. Looks like oss-fuzz failed file a bug reporting the issue. 

**To Reproduce**
see oss-fuzz build logs

**Reproduction Rate**
100%

**Expected behavior**
A clear and concise description of what you expected to happen.

**Impact**
new code is not fuzzed and fixes are not being checked

**Pending solution**
https://github.com/google/oss-fuzz/pull/12027

lgirdwood, 2024-10-15:
@cujomalainey to be clear, do you mean the SOF docker container needs an update for Python ? or oss-fuzz container (as oss-fuzz has a fix link) ?

cujomalainey, 2024-10-15:
> @cujomalainey to be clear, do you mean the SOF docker container needs an update for Python ? or oss-fuzz container (as oss-fuzz has a fix link) ?

Oss fuzz's container

lgirdwood, 2024-10-16:
> > @cujomalainey to be clear, do you mean the SOF docker container needs an update for Python ? or oss-fuzz container (as oss-fuzz has a fix link) ?
> 
> Oss fuzz's container

ok, but a good point in that I think our Docker containers are probably not asking for the updated Python directly, possibly inherit the value from other dependencies.

cujomalainey, 2024-10-16:
I added a hack to a oss fuzz PR to get around the limitation in zephyr, hopefully the upgrade finished before this breaks https://github.com/google/oss-fuzz/pull/12607

cujomalainey, 2024-10-31:
Work around implemented.

closed 2024-10-31 by cujomalainey