thesofproject/sof #9561: [FEATURE] Clean up SOF DMA interface(s)

kv2019i, 2024-10-09:
**Is your feature request related to a problem? Please describe.**
DMA engines are key part of the SOF and the DMA interface are used in many place (to transfer audio data, to transfer payloads of IPC message, to implement trace, to implement debug features, and so forth).

As described in the commit message and documentation added in commit https://github.com/thesofproject/sof/pull/9560/commits/816b143452725456ea50fa7b3ff23fafd1d43ab1 , the transition to use native Zephyr DMA interface has been one of the complex transition tasks, and this work is not complete.

**Describe the solution you'd like**
This issue is used to track cleanup of the DMA usage in SOF application code with following goals:
 - remove namespace overlap between SOF and Zephyr DMA APIs, making it clear which interface is used in application code (IOW, do not mix dma_ and DMA_ prefixes in definitions)
 - retain support for transition interfaces (building SOF with Zephyr but with XTOS drivers), but move the related code to separate files and/or clearly separate namespace

**Describe alternatives you've considered**
One option is to wait for all targets to move to Zephyr (also to native drivers) and then DMA cleanup could be done by dropping all the legacy interfaces. 

**Additional context**
Related to enhancement work done in #9015 and #5794

lgirdwood, 2024-10-15:
May be easiest to do a xtos interface removal after we have all ported to Zephyr.

lgirdwood, 2024-10-15:
Tagged for v2.12, some parts may be doable by this point, we can break down work if needed.

kv2019i, 2024-12-13:
Work in progress, but won't be complete for 2.12. Feature cutoff for v2.12, moving this to v2.13.

kv2019i, 2025-05-02:
Cleanup done for Zephyr-visible code in #9753 . 

Remaining cleanup depends on removal of XTOS in main (tracked in https://github.com/thesofproject/sof/issues/9983).

Moving this to TBD milestone. This needs to be scheduled with #9983

kv2019i, 2025-07-03:
Resuming, last bit in review at https://github.com/thesofproject/sof/pull/10090

kv2019i, 2025-07-04:
Closed via #10090 merge