thesofproject/sof #9567: [BUG][GH-CI]  Zephyr / compare-linux-win build failing

kv2019i, 2024-10-10:
**Describe the bug**
The compare-linux-win is failing in some pull requests. Example here from today 10th Oct:
https://github.com/thesofproject/sof/actions/runs/11238321498/job/31242814332?pr=9554

The Windows build seems to have stale data from previous builds and the compairson fails (FW build artifacts for on eplatform for Linux, but artifacts for multiple targets for Windows builds -> compare fails).

**To Reproduce**
Submit SOF pull-request.

**Reproduction Rate**
-

**Expected behavior**
-

**Impact**

lgirdwood, 2024-10-15:
Will need to live with this atm unless someone with Windows experience knows how to fix @marcinszkudlinski @abonislawski ?

marc-hb, 2024-10-16:
It was a regression in a Github action, fixed:

- https://github.com/actions/upload-artifact/issues/628#issuecomment-2402902713

closed 2024-10-16 by marc-hb