thesofproject/sof #9571: [BUG] [IPC3] ALH Pin5 ipc tx error for 0x30010000.

macchian, 2024-10-11:
**Describe the bug**
When creating a ALH Pin5 on SoundWire link 0,  relevant pipeline, widget, buf and pcm ipc tx errors : 

[    5.060518] snd_sof:sof_widget_setup_unlocked: sof-audio-pci-intel-tgl 0000:00:1f.3: widget PIPELINE.5.ALH5.IN setup complete
[    5.060520] snd_sof:ipc3_log_header: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx: 0x30010000
[    5.060725] sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx error for 0x30010000 (msg/reply size: 96/0): -22
[    5.060750] sof-audio-pci-intel-tgl 0000:00:1f.3: Failed to setup widget ALH5.IN
[    5.060765] snd_sof:ipc3_log_header: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx: 0x30110000
[    5.061391] snd_sof:sof_widget_free_unlocked: sof-audio-pci-intel-tgl 0000:00:1f.3: widget PIPELINE.5.ALH5.IN freed
[    5.061525] sof-audio-pci-intel-tgl 0000:00:1f.3: error: tplg component load failed -22
[    5.061545] sof-audio-pci-intel-tgl 0000:00:1f.3: error: failed to load DSP topology -22
[    5.061560] sof-audio-pci-intel-tgl 0000:00:1f.3: ASoC: error at snd_soc_component_probe on 0000:00:1f.3: -22
[    5.061597] sof_sdw sof_sdw: ASoC: failed to instantiate card -22

**To Reproduce**
Steps to reproduce the behavior: (e.g. list commands or actions used to reproduce the bug)
When create a ALH Pin5 on SoundWire link 0

#ALH Pin5 (ID: 4) 
DAI_CONFIG(ALH, 5, 4, `Capture-SmartMic',
        ALH_CONFIG(ALH_CONFIG_DATA(ALH, 5, 48000, 2)))

**Reproduction Rate**
100%

**Expected behavior**
A clear and concise description of what you expected to happen.
ALH Pin5 is successfully to create the relevant pipeline, widget, buf and pcm.

**Impact**
What impact does this issue have on your progress (e.g., annoyance, showstopper)
The Soundwire device not available to support on board.

**Environment**
1) Branch name and commit hash of the 2 repositories: sof (firmware/topology) and linux (kernel driver).
    * Kernel:  upstream
    * SOF: all IPC3 firmware/topology
2) Name of the topology file
    * Topology: new topology which requires ALH5 support will impact.
3) Name of the platform(s) on which the bug is observed.
    * Platform: IPC3 with Soundwire support

macchian, 2024-10-11:
@bardliao , could you loop someone who can support this? Thank you.

bardliao, 2024-10-11:
It seems only Pin 2, 3, 4 are supported by IPC3. @kv2019i @abonislawski can you take a look?

kv2019i, 2024-10-11:
@macchian @bardliao Do we have FW trace when this happens? Do we get the "stream_id %d out of range" error?
And any print from kernel showing what alh_stream_id is set by kernel when it sends the IPC? This can be compared against sof-stable-v2.2/./src/platform/tigerlake/include/platform/drivers/alh.h

lgirdwood, 2024-10-14:
@bardliao does this need a topology or FW fix ?

macchian, 2024-10-15:
@

> @macchian @bardliao Do we have FW trace when this happens? Do we get the "stream_id %d out of range" error? And any print from kernel showing what alh_stream_id is set by kernel when it sends the [...]

@bardliao @kv2019i , I share the firmware traces  and kernel logs with printed the alh_stream_id and dai_index. 

trace.logs:

similar messages:    Number of Blocks: total % used %d free %d
[      188339.627933] (          13.906249) c0 ipc                  src/ipc/ipc3/handler.c:1605 INFO ipc: new cmd 0x30010000
[      188373.325848] (          33.697914) c0 component             src/ipc/ipc3/helper.c:306  INFO comp new dai <c2b00d27-ffbc-4150-a51a-245c79c5e54b> type 2 id 5.24
[      188549.055008] (         175.729156) c0 dai                           src/lib/dai.c:172  ERROR dai_get: type 4 index 5 not found
[      188570.617507] (          21.562500) c0 dai                         src/audio/dai.c:179  ERROR dai_new(): dai_get() failed to create DAI.
[      188607.805005] (          37.187500) c0 dai                   src/ipc/ipc3/helper.c:313  ERROR comp_new(): unable to create the new component
[      188629.367505] (          21.562500) c0 ipc                   src/ipc/ipc3/helper.c:654  ERROR ipc_comp_new(): component cd = NULL
[      188651.815420] (          22.447916) c0 ipc                  src/ipc/ipc3/handler.c:1273 ERROR ipc: pipe 5 comp 24 creation failed -22
[      189009.523739] (         357.708313) c0 ipc                  src/ipc/ipc3/handler.c:1605 INFO ipc: new cmd 0x30110000
[      189030.044572] (          20.520832) c0 ipc                  src/ipc/ipc3/handler.c:1386 INFO ipc: comp 25 -> free
[      189056.086238] (          26.041666) c0 pipe         5.25  ......./pipeline-graph.c:199  INFO 

dmesg:

[    4.397558] snd_sof:sof_widget_setup_unlocked: sof-audio-pci-intel-tgl 0000:00:1f.3: widget BUF6.0 setup complete
[    4.397560] snd_sof:ipc3_log_header: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx: 0x30010000: GLB_TPLG_MSG: COMP_NEW
[    4.397717] snd_sof:ipc3_log_header: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx: 0x50010000: GLB_COMP_MSG: SET_VALUE
[    4.398057] snd_sof:ipc3_log_header: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx: 0x50020000: GLB_COMP_MSG: GET_VALUE
[    4.398399] snd_sof:sof_widget_setup_unlocked: sof-audio-pci-intel-tgl 0000:00:1f.3: widget PGA6.0 setup complete
[    4.398401] snd_sof:ipc3_log_header: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx: 0x30010000: GLB_TPLG_MSG: COMP_NEW
[    4.398598] snd_sof:sof_widget_setup_unlocked: sof-audio-pci-intel-tgl 0000:00:1f.3: widget PCM5P setup complete
[    4.398600] snd_sof:ipc3_log_header: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx: 0x30100000: GLB_TPLG_MSG: PIPE_NEW
[    4.399315] snd_sof:sof_widget_setup_unlocked: sof-audio-pci-intel-tgl 0000:00:1f.3: widget PIPELINE.5.ALH5.IN setup complete
[    4.399317] snd_sof:ipc3_log_header: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx: 0x30010000: GLB_TPLG_MSG: COMP_NEW
[    4.399634] sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx error for 0x30010000 (msg/reply size: 96/0): -22
[    4.399654] ALH5.IN dai_index 0x5
[    4.399662] sof-audio-pci-intel-tgl 0000:00:1f.3: Failed to setup widget ALH5.IN
[    4.399680] snd_sof:ipc3_log_header: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx: 0x30110000: GLB_TPLG_MSG: PIPE_FREE
[    4.400424] snd_sof:sof_widget_free_unlocked: sof-audio-pci-intel-tgl 0000:00:1f.3: widget PIPELINE.5.ALH5.IN freed
[    4.400642] sof-audio-pci-intel-tgl 0000:00:1f.3: error: tplg component load failed -22
[    4.400662] sof-audio-pci-intel-tgl 0000:00:1f.3: error: failed to load DSP topology -22
[    4.400677] sof-audio-pci-intel-tgl 0000:00:1f.3: ASoC: error at snd_soc_component_probe on 0000:00:1f.3: -22
[    4.400717] sof_sdw sof_sdw: ASoC: failed to instantiate card -22
[    4.400745] snd_soc_cs42l43:cs42l43_set_jack: cs42l43-codec cs42l43-codec: Configure accessory detect
[    4.401049] snd_soc_cs42l43:cs42l43_set_jack: cs42l43-codec cs42l43-codec: Successfully configured accessory detect
[    4.401259] sof_sdw sof_sdw: error -EINVAL: snd_soc_register_card failed -22
[    4.401279] sof_sdw sof_sdw: probe with driver sof_sdw failed with error -22
[    4.406184] soundwire_cadence:cdns_check_attached_status_dwork: soundwire_intel soundwire_intel.link.0: Peripheral 6 status: 1
[    5.371840] Bluetooth: hci0: Waiting for firmware download to complete

macchian, 2024-10-16:
@abonislawski , I share the fw traces. Could you add someone to check it? Thank you.

kv2019i, 2024-10-17:
Thanks @macchian for trace. This seems to be expected behaviour with current FW:

src/platform/intel/cavs/lib/dai.c sets up the ALH dais and there's rule:

```
»       »       »       dai[i].index = (i / DAI_NUM_ALH_BI_DIR_LINKS_GROUP) << 8 |
»       »       »       »       (i % DAI_NUM_ALH_BI_DIR_LINKS_GROUP);
```

On cAVS2.5,  DAI_NUM_ALH_BI_DIR_LINKS_GROUP is 4, so index 5 is not supported. 

@abonislawski Do you know why Icelake and Cannonlake seem to have  DAI_NUM_ALH_BI_DIR_LINKS_GROUP of 6 ?

lgirdwood, 2024-10-18:
@kv2019i @abonislawski do we need to update the value to 5 in cavs2.5 ipc3 FW to fix this ?

macchian, 2024-10-21:
@abonislawski @kv2019i ,do you want me to test any changes? Because this is a signed key machine, I am unable to build the signed key firmware on my end. Might need your test build if you have any suggestions.

abonislawski, 2024-10-23:
> On cAVS2.5, DAI_NUM_ALH_BI_DIR_LINKS_GROUP is 4, so index 5 is not supported.

In that scenario "alh pin 5" = group index? We have 4 groups so it would be invalid.

And btw we have incomplete handshake table in this file https://github.com/thesofproject/sof/blob/stable-v2.2/src/platform/tigerlake/include/platform/drivers/alh.h#L20
Im not sure why it is limited only to bidir streams

kv2019i, 2024-10-23:
Looking at this from FW point of view, following values are valid to be set in IPCs:
```
cAVS2.5 IPC3 support following SDW DAI index values:
   0,    1,    2,    3,
 256,  257,  258,  259,
 512,  513,  514,  515,
1024, 1025, 1026, 1027

And following stream-ids (config->alh.stream_id):
22, 23, 24, 25,
32, 33, 34, 35,
42, 43, 44, 45,
52, 53, 54, 55
```

closed 2024-10-29 by ranj063