thesofproject/sof #9587: DMIC Capture failure on HDA with cavs2.5

ranj063, 2024-10-16:
All capture usecases are failing with the following error
[  273.932645] kernel: snd_sof:snd_sof_get_nhlt_endpoint_data: sof-audio-pci-intel-tgl 0000:00:1f.3: dai index 0 nhlt type 2 direction 1 dev type 0
[  273.932650] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: no matching blob for sample rate: 48000 sample width: 16 channels: 4
[  273.932704] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: failed to prepare widget dai-copier.DMIC.dmic01.capture
[  273.932740] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: Failed to prepare connected widgets
[  273.932770] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: error: failed widget list set up for pcm 6 dir 1
[  273.932803] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ASoC: error at snd_soc_pcm_component_hw_params on 0000:00:1f.3: -22

Daily test plan:  46770

ranj063, 2024-10-16:
I checked the device and it looks like we're using the tplg NHLT and it fails to find the 32-bit DMIC blob. If I remove the module param to force the tplg NHLT, the test passes.

marc-hb, 2024-10-16:
Looking at result/planresult?planname=%5Edaily_test_new,

- Last successful daily run was 46534 

Start Time: 2024-10-04 13:21:25 UT
Linux Commit: [62f484728981](https://github.com/thesofproject/linux/commit/62f48472898105381a8061415c167b0d48019fe4)
KConfig Commit: [8189104a4f38](https://github.com/thesofproject/kconfig/commit/8189104a4f38167aaaa4e70eece20ba4b6c19ef1)
SOF Commit: [9cdef9fc13bb](https://github.com/thesofproject/sof/commit/9cdef9fc13bb214afce77e09accf7ec6807453cb)
Zephyr Commit: [155f3f3ba688](https://github.com/zephyrproject-rtos/zephyr/commit/155f3f3ba688dac3e1113ada3f9cfa894612f951)

- First failing daily run 46595

Start Time: 2024-10-10 13:11:05 UTC
Linux Commit: [7eab5d86f218](https://github.com/thesofproject/linux/commit/7eab5d86f218975574a780fa9b3dce28cc76f6bf)
KConfig Commit: [8189104a4f38](https://github.com/thesofproject/kconfig/commit/8189104a4f38167aaaa4e70eece20ba4b6c19ef1)
SOF Commit: [d2cf7c3cc859](https://github.com/thesofproject/sof/commit/d2cf7c3cc859c7d89ddd6cdec880ed20e26dc613)
Zephyr Commit: [155f3f3ba688](https://github.com/zephyrproject-rtos/zephyr/commit/155f3f3ba688dac3e1113ada3f9cfa894612f951)

There's a very clear and dramatic drop in the PASS rate between these two consecutive runs. There's a big date gap between the two because of planned power outage in Oregon.

It's less clear but I suspect the PASS rate of result/planresult?planname=%5Edaily_lnl dropped at around the same time? Harder to tell because unlike the other it's always been somewhat red so as expected everyone stopped paying attention.

ranj063, 2024-10-17:
> > I checked the device and it looks like we're using the tplg NHLT and it fails to find the 32-bit DMIC blob. If I remove the module param to force the tplg NHLT, the test passes.
> 
> @ranj063 So does this mean kernel cant find blob or the 32bit blob is missing form the topology(i.e. topology has 16bit blob only) ?

@lgirdwood it looks like the 32-bit blob is missing from the topology. Im verifying it this morning

ranj063, 2024-10-17:
the reason for the fialure was that the incorrect topology was being used on the device. For cavs2.5 platforms, when forcing the use of topology NHLT, we should copy sof-hda-generic-cavs25-4ch.tplg into sof-hda-generic-4ch.tplg before running the CI tests. But if the NHLT is included in the BIOS, that should be preferred over the topology NHLT.

closed 2024-10-17 by ranj063

marc-hb, 2024-10-18:
> (It's less clear but I suspect the PASS rate of result/planresult?planname=%5Edaily_lnl dropped at around the same time...? Harder to tell because unlike the other it's always been somewhat red so, as expected everyone stopped paying attention.)

This recent LNL regression and crash seems to be focused on `multiple-pipeline` tests. Has anyone filed a bug yet?

Here's a non-daily, public example: https://sof-ci.01.org/softestpr/PR1236/build782/devicetest/index.html

marc-hb, 2024-10-22:
> This recent, probably unrelated LNL regression and crash seems to be focused on multiple-pipeline tests. Has anyone filed a bug yet?

- https://github.com/thesofproject/sof/issues/9593

Thanks @kv2019i