thesofproject/sof #9600: [BUG] NULL pointer dereference with kernel 6.6.56-1-lts and sof-firmware 2024.09-1

pirr, 2024-10-19:
**Describe the bug**
I am dealing with a NULL pointer dereference after updating from `sof-firmware 2024.06-1` to `sof-firmware 2024.09-1`  on EndeavourOS (ArchLinux). The sound doesn't work at all. I tried it on 6.11.3.arch1-1 and there were no problems there (I just can't adjust the volume, but I had this problem before too).

I tried:
As write there https://thesofproject.github.io/latest/getting_started/intel_debug/introduction.html#pci-devices-introduced-after-2016
I used `options snd-intel-dspcfg dsp_driver=1` and the sound worked (can't adjust the volume).

<details>

<summary>System</summary>

```sh
System:
  Kernel: 6.6.56-1-lts arch: x86_64 bits: 64 compiler: gcc v: 14.2.1
  Desktop: KDE Plasma v: 6.2.0 tk: Qt v: N/A wm: kwin_wayland dm: SDDM
    Distro: EndeavourOS base: Arch Linux
Machine:
  Type: Laptop System: LENOVO product: 83E2 v: Yoga Pro 7 14IMH9
    serial: <superuser required> Chassis: type: 10 v: Yoga Pro 7 14IMH9
    serial: <superuser required>
  Mobo: LENOVO model: LNVNB161216 v: No DPK serial: <superuser required>
    part-nu: LENOVO_MT_83E2_BU_idea_FM_Yoga Pro 7 14IMH9 UEFI: LENOVO
    v: N9CN25WW date: 02/26/2024
Battery:
  ID-1: BAT0 charge: 73.0 Wh (99.1%) condition: 73.7/73.0 Wh (100.9%)
    volts: 17.4 min: 15.5 model: Sunwoda L22D4PF4 serial: <filter>
    status: charging
CPU:
  Info: 16-core (6-mt/10-st) model: Intel Core Ultra 9 185H bits: 64
    type: MST AMCP arch: Meteor Lake rev: 4 cache: 24 MiB note: check
  Speed (MHz): avg: 4734 min/max: 400/4700:5000:3800:2500 cores: 1: 4734
    2: 4734 3: 4734 4: 4734 5: 4734 6: 4734 7: 4734 8: 4734 9: 4734 10: 4734
    11: 4734 12: 4734 13: 4734 14: 4734 15: 4734 16: 4734 17: 4734 18: 4734
    19: 4734 20: 4734 21: 4734 22: 4734 bogomips: 135212
  Flags: avx avx2 ht lm nx pae sse sse2 sse3 sse4_1 sse4_2 ssse3 vmx
Graphics:
  Device-1: Intel Meteor Lake-P [Intel Arc Graphics] vendor: Lenovo
    driver: i915 v: kernel arch: Gen-13 ports: active: eDP-1 empty: DP-1, DP-2,
    DP-3, DP-4, HDMI-A-1 bus-ID: 00:02.0 chip-ID: 8086:7d55
  Device-2: Luxvisions Innotech Integrated RGB Camera driver: uvcvideo
    type: USB rev: 2.0 speed: 480 Mb/s lanes: 1 bus-ID: 3-9:2 chip-ID: 30c9:00c2
  Display: wayland server: X.org v: 1.21.1.13 with: Xwayland v: 24.1.2
    compositor: kwin_wayland driver: X: loaded: modesetting
    alternate: fbdev,intel,vesa dri: iris gpu: i915 display-ID: 0
  Monitor-1: eDP-1 res: 1536x960 size: N/A
  API: EGL v: 1.5 platforms: device: 0 drv: iris device: 1 drv: swrast gbm:
    drv: iris surfaceless: drv: iris wayland: drv: iris x11: drv: iris
  API: OpenGL v: 4.6 compat-v: 4.5 vendor: intel mesa v: 24.2.4-arch1.1
    glx-v: 1.4 direct-render: yes renderer: Mesa Intel Arc Graphics (MTL)
    device-ID: 8086:7d55 display-ID: :1.0
  API: Vulkan v: 1.3.295 surfaces: xcb,xlib,wayland device: 0
    type: integrated-gpu driver: mesa intel device-ID: 8086:7d55
Audio:
  Device-1: Intel Meteor Lake-P HD Audio vendor: Lenovo
    driver: sof-audio-pci-intel-mtl bus-ID: 00:1f.3 chip-ID: 8086:7e28
  API: ALSA v: k6.6.56-1-lts status: kernel-api
  Server-1: PipeWire v: 1.2.5 status: active with: 1: pipewire-pulse
    status: active 2: wireplumber status: active 3: pipewire-alsa type: plugin
    4: pw-jack type: plugin
Network:
  Device-1: Intel Meteor Lake PCH CNVi WiFi driver: iwlwifi v: kernel
    bus-ID: 00:14.3 chip-ID: 8086:7e40
  IF: wlan0 state: up mac: <filter>
  IF-ID-1: docker0 state: down mac: <filter>
Bluetooth:
  Device-1: Intel AX211 Bluetooth driver: btusb v: 0.8 type: USB rev: 2.0
    speed: 12 Mb/s lanes: 1 bus-ID: 3-10:3 chip-ID: 8087:0033
  Report: btmgmt ID: hci0 rfk-id: 5 state: down bt-service: enabled,running
    rfk-block: hardware: no software: no address: <filter> bt-v: 5.3 lmp-v: 12
Drives:
[... 152 lines cut ...]
[    9.287015] overlayfs: upper fs does not support file handles, falling back to index=off.
```

</details>

**To Reproduce**
I suppose this can be reproduced on a laptop with a sound card like mine (Intel Meteor Lake-P HD Audio vendor: Lenovo) on kernel 6.6.56-1-lts arch: x86_64 bits: 64 compiler: gcc v: 14.2.1.

**Reproduction Rate**
100%

**Expected behavior**
The sound works and there are no errors related to sof.

**Impact**
The Sound device doesn't work.

**Environment**
1) Branch name and commit hash of the 2 repositories: sof (firmware/topology) and linux (kernel driver).
    * Kernel: 6.6.56-1-lts arch: x86_64 bits: 64 compiler: gcc v: 14.2.1
    * SOF: -
2) Name of the topology file
    * Topology: -
3) Name of the platform(s) on which the bug is observed.
    * Platform: -

lgirdwood, 2024-10-21:
@ujfalusi @bardliao @ranj063 - new issue ?

ujfalusi, 2024-10-21:
@pirr, Can you copy [sof-dyndbg.conf.txt](https://github.com/user-attachments/files/17400684/sof-dyndbg.conf.txt) as `/etc/modprobe.d/sof-dyndbg.conf` reboot and attach the full kernel log with the NULL pointer?

pirr, 2024-10-21:
@ujfalusi here is the log
[dmesg.log](https://github.com/user-attachments/files/17465926/dmesg.log)

bardliao, 2024-10-22:
@pirr Are you saying kernel 6.11.3.arch1-1 + sof-firmware 2024.09-1 works, and kernel 6.6.56-1-lts + sof-firmware 2024.06-1 also works, but kernel 6.6.56-1-lts + sof-firmware 2024.09-1 doesn't work?

ujfalusi, 2024-10-22:
_I_ think it is the introduction of tdfb with Enum controls in sof 2024.09-1. The 6.6.y stable does not have support for enum type.

@singalsu, @kv2019i, I can try to see if it cna be backported, but it is a feature, not a bug fix - but  since the mix of new and old topology is a possibility, it can be viewed as bug fix.

@pirr, I think for MTL you should not use 6.6.y LTS kernel - the force_probe for MTL i915 support got dropped with 6.7 kernel for example, so the odds that the 6.6 LTS is not working is high.
Is there any particular reason you need to stick with 6.6 kernel?

ujfalusi, 2024-10-22:
Update: with these patches picked to stable 6.6.y, there is no longer NULL dereference:
```
07a866a41982 ASoC: SOF: ipc4-control: Add support for ALSA enum control
4a2fd607b7ca ASoC: SOF: ipc4-control: Add support for ALSA switch control
060a07cd9bc6 ASoC: SOF: ipc4-topology: Add definition for generic switch/enum control
```

ujfalusi, 2024-10-22:
@pirr, you need to use force_probe to get display working with the 6.6 LTS kernel? I cannot see that patch to be backported, for MTL 6.6 kernel is just too old to be used.
```
[    4.877750] i915 0000:00:02.0: Force probing unsupported Device ID 7d55, tainting kernel
```

pirr, 2024-10-22:
@bardliao yes it is. Yesterday I updated to 6.6.57-1-lts, but the error remained.

@ujfalus
> Is there any particular reason you need to stick with 6.6 kernel?

I use 6.6 LTS because on 6.11 speakers volume cannot be changed (only mute or at full volume). And yesterday working on kernel 6.11 I encountered that the laptop freezes. I don't know the reason yet, but I've seen errors that are related to sof. Here is log
[journal_6_11.log](https://github.com/user-attachments/files/17472415/journal_6_11.log). 
I'll try with the previous version `sof-firmware 2024.06-1`

>  you need to use force_probe to get display working with the 6.6 LTS kernel
I didn't change anything in the kernel. But apparently that's how it is.

ujfalusi, 2024-10-22:
@pirr, I think your issue is identical to https://github.com/thesofproject/sof/issues/9572 with 6.11 kernel regarding to the kernel errors.

The speaker volume control worked with 6.10 kernel or with older sof-firmware package? Have you tried with GUI or did you checked also with `alsamixer`?

Can you file an issue for it (volume control not working)?

ujfalusi, 2024-10-22:
Moving the issue under kernel.