thesofproject/sof #9602: [BUG] ] audio crashes and stops working after sleep (Dell 9720 w/ Alder Lake PCH-P audio)

eighthave, 2024-10-21:
**Describe the bug**

The built-in audio is totally stuck, both the speaker and when I plug in a headset.  It seems to be related to when the laptop is put to sleep.  When I plug in a USB audio device, everything works fine.  I'm the same machine as #7799, but it seems this is a different bug, based on the log output, but with the same or similar results.

**To Reproduce**

1. Boot computer
2. confirm audio works
3. use the computer normally
4. put the computer to sleep
5. wake it up

Audio stuck.

**Reproduction Rate**

It happens most of the time after a fresh reboot.  After a while, it sometimes gets unstuck and stays unstuck for a bit.

**Expected behavior**

Audio to work fine after sleep, and all the time.

**Impact**

Built-in audio does not work at all, it makes it impossible to do online meetings, listen to music, watch videos, etc.

**Environment**
1) Branch name and commit hash of the 2 repositories: sof (firmware/topology) and linux (kernel driver).
    * Kernel: Debian's linux-image-amd64 (6.10.11-1~bpo12+1) from bookworm-backports
    * SOF: Debian's firmware-sof-signed (2.2.5-1)
2) Name of the topology file
    * Topology: {FILE}
3) Name of the platform(s) on which the bug is observed.
    * Platform: Debian/bookworm on amd64.

**Screenshots or console output**

### `dmesg`

[audio-stuck-dmesg.txt.zip](https://github.com/user-attachments/files/17459162/audio-stuck-dmesg.txt.zip)

```console
# grep -i -e 'ASoC: error' -e soundwire -e pipewire -e alsa /var/log/syslog|grep 2024-10-21
2024-10-21T06:07:25.014047+02:00 monolith kernel: [38714.518345] soundwire_intel soundwire_intel.link.0: intel_pm_prepare: pm_runtime_resume failed: -16
2024-10-21T06:07:25.014048+02:00 monolith kernel: [38714.518364] soundwire_intel soundwire_intel.link.1: intel_pm_prepare: pm_runtime_resume failed: -16
2024-10-21T06:07:25.014048+02:00 monolith kernel: [38714.518372] soundwire_intel soundwire_intel.link.2: intel_pm_prepare: pm_runtime_resume failed: -16
2024-10-21T06:07:25.014049+02:00 monolith kernel: [38714.518378] soundwire_intel soundwire_intel.link.3: intel_pm_prepare: pm_runtime_resume failed: -16
2024-10-21T06:07:25.014049+02:00 monolith kernel: [38714.519961] soundwire_intel soundwire_intel.link.3: intel_suspend: invalid config: parent is suspended
2024-10-21T06:07:25.014050+02:00 monolith kernel: [38714.519970] soundwire_intel soundwire_intel.link.2: intel_suspend: invalid config: parent is suspended
2024-10-21T06:07:25.014050+02:00 monolith kernel: [38714.519977] soundwire_intel soundwire_intel.link.1: intel_suspend: invalid config: parent is suspended
2024-10-21T06:07:25.014051+02:00 monolith kernel: [38714.519983] soundwire_intel soundwire_intel.link.0: intel_suspend: invalid config: parent is suspended
2024-10-21T06:07:25.014064+02:00 monolith kernel: [70600.180063] Modules linked in: snd_usb_audio snd_usbmidi_lib snd_rawmidi hid_jabra typec_displayport cdc_mbim cdc_wdm cdc_ncm cdc_ether usbnet r8152 mii libphy usbhid cdc_acm ccm rfcomm snd_seq_dummy snd_hrtimer snd_seq snd_seq_device cmac algif_hash algif_skcipher af_alg xt_LOG nf_log_syslog xt_tcpudp wireguard xt_conntrack libchacha20poly1305 nf_conntrack chacha_x86_64 poly1305_x86_64 nf_defrag_ipv6 nf_defrag_ipv4 nft_compat curve25519_x86_64 libcurve25519_generic libchacha ip6_udp_tunnel udp_tunnel nf_tables nfnetlink overlay qrtr snd_ctl_led snd_soc_sof_sdw snd_sof_probes snd_soc_intel_hda_dsp_common bnep snd_soc_rt711_sdca snd_soc_rt715_sdca snd_hda_codec_hdmi snd_soc_rt1316_sdw regmap_sdw_mbq regmap_sdw snd_soc_dmic binfmt_misc snd_sof_pci_intel_tgl snd_sof_pci_intel_cnl snd_sof_intel_hda_generic soundwire_intel nls_ascii soundwire_generic_allocation soundwire_cadence nls_cp437 vfat snd_sof_intel_hda_common fat snd_sof_intel_hda_mlink snd_sof_intel_hda snd_sof_pci iwlmvm snd_sof_xtensa_dsp snd_sof snd_sof_utils
2024-10-21T06:07:25.014065+02:00 monolith kernel: [70600.180096]  snd_soc_hdac_hda snd_soc_acpi_intel_match snd_soc_acpi soundwire_bus mac80211 snd_soc_avs intel_uncore_frequency intel_uncore_frequency_common x86_pkg_temp_thermal snd_soc_hda_codec intel_powerclamp snd_hda_ext_core coretemp snd_soc_core kvm_intel snd_compress snd_pcm_dmaengine libarc4 uvcvideo snd_hda_intel kvm snd_intel_dspcfg snd_intel_sdw_acpi videobuf2_vmalloc snd_hda_codec uvc videobuf2_memops videobuf2_v4l2 snd_hda_core iwlwifi snd_hwdep btusb snd_pcm videodev hid_sensor_als btrtl hid_sensor_trigger mei_pxp processor_thermal_device_pci btintel mei_hdcp dell_laptop iTCO_wdt hid_sensor_iio_common snd_timer processor_thermal_device industrialio_triggered_buffer intel_pmc_bxt btbcm dell_wmi rapl kfifo_buf processor_thermal_wt_hint intel_rapl_msr dell_smbios videobuf2_common dell_wmi_sysman btmtk dcdbas intel_cstate bluetooth intel_uncore mc dell_wmi_ddv dell_smm_hwmon pcspkr firmware_attributes_class dell_wmi_descriptor wmi_bmof industrialio iTCO_vendor_support cfg80211 processor_thermal_rfim ucsi_acpi
2024-10-21T06:07:25.014116+02:00 monolith kernel: [70600.430644] soundwire_intel soundwire_intel.link.0: Failed to power up link: -11
2024-10-21T06:07:25.014116+02:00 monolith kernel: [70600.430659] soundwire_intel soundwire_intel.link.0: intel_resume failed: -11
2024-10-21T06:07:25.014140+02:00 monolith kernel: [70600.430664] soundwire_intel soundwire_intel.link.0: PM: dpm_run_callback(): pm_generic_resume returns -11
2024-10-21T06:07:25.014142+02:00 monolith kernel: [70600.430682] soundwire_intel soundwire_intel.link.0: PM: failed to resume: error -11
2024-10-21T06:07:25.014142+02:00 monolith kernel: [70600.431792] soundwire_intel soundwire_intel.link.1: Failed to power up link: -11
2024-10-21T06:07:25.014142+02:00 monolith kernel: [70600.431796] soundwire_intel soundwire_intel.link.1: intel_resume failed: -11
2024-10-21T06:07:25.014143+02:00 monolith kernel: [70600.431800] soundwire_intel soundwire_intel.link.1: PM: dpm_run_callback(): pm_generic_resume returns -11
2024-10-21T06:07:25.014143+02:00 monolith kernel: [70600.431811] soundwire_intel soundwire_intel.link.1: PM: failed to resume: error -11
2024-10-21T06:07:25.014144+02:00 monolith kernel: [70600.432911] soundwire_intel soundwire_intel.link.2: Failed to power up link: -11
2024-10-21T06:07:25.014144+02:00 monolith kernel: [70600.432915] soundwire_intel soundwire_intel.link.2: intel_resume failed: -11
2024-10-21T06:07:25.014144+02:00 monolith kernel: [70600.432919] soundwire_intel soundwire_intel.link.2: PM: dpm_run_callback(): pm_generic_resume returns -11
2024-10-21T06:07:25.014145+02:00 monolith kernel: [70600.432928] soundwire_intel soundwire_intel.link.2: PM: failed to resume: error -11
2024-10-21T06:07:25.014145+02:00 monolith kernel: [70600.434038] soundwire_intel soundwire_intel.link.3: Failed to power up link: -11
2024-10-21T06:07:25.014145+02:00 monolith kernel: [70600.434044] soundwire_intel soundwire_intel.link.3: intel_resume failed: -11
2024-10-21T06:07:25.014146+02:00 monolith kernel: [70600.434050] soundwire_intel soundwire_intel.link.3: PM: dpm_run_callback(): pm_generic_resume returns -11
2024-10-21T06:07:25.014146+02:00 monolith kernel: [70600.434068] soundwire_intel soundwire_intel.link.3: PM: failed to resume: error -11
2024-10-21T06:07:25.016435+02:00 monolith kernel: [70600.441685] soundwire_intel soundwire_intel.link.3: sdw_cdns_clock_stop failed: IP_MCP_CONTROL_SW_RST is not cleared
[... 63 lines cut ...]
2024-10-21T06:12:01.260150+02:00 monolith pipewire[2752]: D spa.alsa [alsa-pcm.c:1863:spa_alsa_enum_format]: opened:0 format:0 started:0
2024-10-21T06:12:01.260172+02:00 monolith pipewire[2752]: I spa.alsa [alsa-pcm.c:1096:spa_alsa_open]: 0x5618e978c408: ALSA device open '_ucm0001.hw:sofsoundwire,2' playback
2024-10-21T06:12:01.260190+02:00 monolith pipewire[2752]: E spa.alsa [alsa-pcm.c:1098:spa_alsa_open]: '_ucm0001.hw:sofsoundwire,2': playback open failed: Device or resource busy
2024-10-21T06:12:01.260209+02:00 monolith pipewire[2752]: spa.audioadapter: params Spa:Enum:ParamId:EnumFormat: 0:0 (follower format) Device or resource busy
2024-10-21T06:12:01.260228+02:00 monolith pipewire[2752]: pw.node: (alsa_output.pci-0000_00_1f.3-platform-sof_sdw.HiFi__Speaker__sink-59) suspended -> error (Start error: Device or resource busy)
2024-10-21T06:12:01.260425+02:00 monolith kernel: [70876.683291] soundwire_intel soundwire_intel.link.1: ASoC: error at snd_soc_pcm_component_pm_runtime_get on soundwire_intel.link.1: -16
[... cut ...]
2024-10-21T06:12:38.504435+02:00 monolith kernel: [70913.927279] soundwire_intel soundwire_intel.link.1: ASoC: error at snd_soc_pcm_component_pm_runtime_get on soundwire_intel.link.1: -16
2024-10-21T06:12:38.504442+02:00 monolith kernel: [70913.927281]  SDW1-Playback: ASoC: error at __soc_pcm_open on SDW1-Playback: -16
2024-10-21T06:12:38.504443+02:00 monolith kernel: [70913.927282]  Speaker: ASoC: error at dpcm_be_dai_startup on Speaker: -16
2024-10-21T06:12:38.504443+02:00 monolith kernel: [70913.927282]  Speaker: ASoC: error at dpcm_fe_dai_startup on Speaker: -16
```

bardliao, 2024-10-22:
@eighthave We need more debug trace to get what happened. Can you copy [sof-dyndbg.conf.txt](https://github.com/user-attachments/files/17400684/sof-dyndbg.conf.txt) as /etc/modprobe.d/sof-dyndbg.conf reboot and attach the full kernel log with the issue? Does the issue happen only when a headset is plugged in? Or it happens without headset, too?

```
2024-10-20T12:27:43.393697+02:00 monolith kernel: [ 7016.089097] soundwire_intel soundwire_intel.link.0: intel_pm_prepare: pm_runtime_resume failed: -16
2024-10-20T12:27:43.393698+02:00 monolith kernel: [ 7016.089117] soundwire_intel soundwire_intel.link.1: intel_pm_prepare: pm_runtime_resume failed: -16
2024-10-20T12:27:43.393700+02:00 monolith kernel: [ 7016.089126] soundwire_intel soundwire_intel.link.2: intel_pm_prepare: pm_runtime_resume failed: -16
2024-10-20T12:27:43.393701+02:00 monolith kernel: [ 7016.089136] soundwire_intel soundwire_intel.link.3: intel_pm_prepare: pm_runtime_resume failed: -16
```
and
```
2024-10-20T12:27:43.393724+02:00 monolith kernel: [ 7017.259839] soundwire_intel soundwire_intel.link.0: Failed to power up link: -11
2024-10-20T12:27:43.393725+02:00 monolith kernel: [ 7017.259850] soundwire_intel soundwire_intel.link.0: intel_resume failed: -11
2024-10-20T12:27:43.393726+02:00 monolith kernel: [ 7017.259857] soundwire_intel soundwire_intel.link.0: PM: dpm_run_callback(): pm_generic_resume returns -11
2024-10-20T12:27:43.393727+02:00 monolith kernel: [ 7017.259873] soundwire_intel soundwire_intel.link.0: PM: failed to resume: error -11
```

Where the error code -16 is -EBUSY and -11 is -EAGAIN. I suspect something has been wrong before the error was reported.

eighthave, 2024-10-22:
It happened with and without a headset plugged into the 3.5mm headphone jack.  I don't think it is related to plugging it in either, like I could reboot and not plug headphones in, and it would happen.  I might be wrong, but I think that's the case.

I've added that modprobe config, and I'll reply once it happens again.  It seems to be working OK now after a reboot.

@lgirdwood I'm not sure what "SDW codec ID" is exactly, could it be "rt714"?

ujfalusi, 2024-10-22:
Looks like a kernel issue, moving.