thesofproject/sof #9663: [BUG] Lenovo Yoga Slim 7 15 - SOF topology file not found

cgbowman, 2024-11-19:
**Describe the bug**
The SOF topology file is not detected (topology doesn't exist) on a LNL Lenovo Yoga Slim 7 15 laptop.

**To Reproduce**
Install Ubuntu 24.10 on the laptop, build & install 6.12.0-rc7+ kernel, then boot the laptop.
Errors will be present in dmesg logs.

**Reproduction Rate**
All the time

**Expected behavior**
The topology file should exist and be detected.

**Impact**
Showstopper
At best, results in no audio (dummy output). At worst, sometimes results in a failed boot.

**Environment**
1) Branch name and commit hash of the 2 repositories: sof (firmware/topology) and linux (kernel driver).
    * Kernel: 6.12.0-rc7+ (drm-tip)
    * SOF: sof-bin-2024.09.1 (release)
2) Name of the topology file
    * Topology: intel/sof-ipc4-tplg/sof-lnl-cs42l43-l0-4ch.tplg
3) Name of the platform(s) on which the bug is observed.
    * Platform: Lenovo Yoga Slim 7 15

**Screenshots or console output**
```
[    5.459112] sof-audio-pci-intel-lnl 0000:00:1f.3: hda codecs found, mask 4
[    5.459115] sof-audio-pci-intel-lnl 0000:00:1f.3: NHLT device BT(0) detected, ssp_mask 0x4
[    5.459116] sof-audio-pci-intel-lnl 0000:00:1f.3: BT link detected in NHLT tables: 0x4
[    5.459117] sof-audio-pci-intel-lnl 0000:00:1f.3: DMICs detected in NHLT tables: 4
[    5.460831] cs42l43 sdw:0:0:01fa:4243:01: supply vdd-p not found, using dummy regulator
[    5.460864] cs42l43 sdw:0:0:01fa:4243:01: supply vdd-d not found, using dummy regulator
[    5.460872] cs42l43 sdw:0:0:01fa:4243:01: supply vdd-a not found, using dummy regulator
[    5.460880] cs42l43 sdw:0:0:01fa:4243:01: supply vdd-io not found, using dummy regulator
[    5.460889] cs42l43 sdw:0:0:01fa:4243:01: supply vdd-cp not found, using dummy regulator
[    5.462775] sof-audio-pci-intel-lnl 0000:00:1f.3: SOF firmware and/or topology file not found.
[    5.462778] sof-audio-pci-intel-lnl 0000:00:1f.3: Supported default profiles
[    5.462779] sof-audio-pci-intel-lnl 0000:00:1f.3: - ipc type 1 (Requested):
[    5.462780] sof-audio-pci-intel-lnl 0000:00:1f.3:  Firmware file: intel/sof-ipc4/lnl/sof-lnl.ri
[    5.462781] sof-audio-pci-intel-lnl 0000:00:1f.3:  Topology file: intel/sof-ipc4-tplg/sof-lnl-cs42l43-l0-4ch.tplg
[    5.462782] sof-audio-pci-intel-lnl 0000:00:1f.3: Check if you have 'sof-firmware' package installed.
[    5.462782] sof-audio-pci-intel-lnl 0000:00:1f.3: Optionally it can be manually downloaded from:
[    5.462783] sof-audio-pci-intel-lnl 0000:00:1f.3:    https://github.com/thesofproject/sof-bin/
```
[lenovo_yoga_dmesg.txt](https://github.com/user-attachments/files/17810645/lenovo_yoga_dmesg.txt)

simontrimmer, 2024-11-19:
Having a crack at a PR with that topology change in, not my usual jam...

https://github.com/thesofproject/sof/pull/9666

cgbowman, 2024-11-27:
I was able to build the topology file with the docker image given by the SOF documentation, but after copying the topology over to the correct location, other errors are still present. I didn't rebuild the firmware binaries, as other targets weren't accepted by the docker image suggested by the instructions (even the 'cnl' target doesn't seem to be recognized as a valid target). It's possible that not rebuilding the rest of the lnl target may produce these errors, but I'm not familiar with this codebase, so I'm not sure.

```
[    7.483987] sof-audio-pci-intel-lnl 0000:00:1f.3: bound 0000:00:02.0 (ops i915_audio_component_bind_ops [xe])
[    7.491234] sof-audio-pci-intel-lnl 0000:00:1f.3: use msi interrupt mode
[    7.522036] sof-audio-pci-intel-lnl 0000:00:1f.3: hda codecs found, mask 4
[    7.522053] sof-audio-pci-intel-lnl 0000:00:1f.3: NHLT device BT(0) detected, ssp_mask 0x4
[    7.522057] sof-audio-pci-intel-lnl 0000:00:1f.3: BT link detected in NHLT tables: 0x4
[    7.522063] sof-audio-pci-intel-lnl 0000:00:1f.3: DMICs detected in NHLT tables: 4
[    7.529163] sof-audio-pci-intel-lnl 0000:00:1f.3: Firmware paths/files for ipc type 1:
[    7.529176] sof-audio-pci-intel-lnl 0000:00:1f.3:  Firmware file:     intel/sof-ipc4/lnl/sof-lnl.ri
[    7.529179] sof-audio-pci-intel-lnl 0000:00:1f.3:  Firmware lib path: intel/sof-ipc4-lib/lnl
[    7.529181] sof-audio-pci-intel-lnl 0000:00:1f.3:  Topology file:     intel/sof-ipc4-tplg/sof-lnl-cs42l43-l0-4ch.tplg
[    7.530298] sof-audio-pci-intel-lnl 0000:00:1f.3: Loaded firmware library: ADSPFW, version: 2.11.1.1
[    7.537583] cs42l43 sdw:0:0:01fa:4243:01: supply vdd-p not found, using dummy regulator
[    7.537623] cs42l43 sdw:0:0:01fa:4243:01: supply vdd-d not found, using dummy regulator
[    7.537641] cs42l43 sdw:0:0:01fa:4243:01: supply vdd-a not found, using dummy regulator
[    7.537651] cs42l43 sdw:0:0:01fa:4243:01: supply vdd-io not found, using dummy regulator
[    7.537658] cs42l43 sdw:0:0:01fa:4243:01: supply vdd-cp not found, using dummy regulator
[    7.540914] cs42l43 sdw:0:0:01fa:4243:01: devid: 0x042a43, rev: 0xa1, otp: 0x03
[    7.559314] fbcon: xedrmfb (fb0) is primary device
[    7.559321] fbcon: Deferring console take-over
[    7.559326] xe 0000:00:02.0: [drm] fb0: xedrmfb frame buffer device
[    7.563078] cs42l43 sdw:0:0:01fa:4243:01: Slave 6 state check1: UNATTACHED, status was 1
[    7.566856] xe 0000:00:02.0: [drm] GT1: found GSC cv104.1.0
[    7.572320] Bluetooth: BNEP (Ethernet Emulation) ver 1.3
[    7.572324] Bluetooth: BNEP filters: protocol multicast
[    7.572328] Bluetooth: BNEP socket layer initialized
[    7.587982] Adding alias for supply vdd-amp,(null) -> vdd-amp,sdw:0:0:01fa:4243:01
[    7.676905] sof-audio-pci-intel-lnl 0000:00:1f.3: Booted firmware version: 2.11.1.1
[    7.692936] cs35l56 spi-cs35l56-left: supply VDD_P not found, using dummy regulator
[    7.692989] cs35l56 spi-cs35l56-left: supply VDD_IO not found, using dummy regulator
[    7.692999] cs35l56 spi-cs35l56-left: supply VDD_A not found, using dummy regulator
[    7.704826] cs42l43-codec cs42l43-codec: supply vdd-amp not found, using dummy regulator
[    7.706871] loop15: detected capacity change from 0 to 8
[    7.708502] sof-audio-pci-intel-lnl 0000:00:1f.3: Topology: ABI 3:29:1 Kernel ABI 3:23:1
[    7.708636] sof-audio-pci-intel-lnl 0000:00:1f.3: error: can't connect DAI alh-copier.Playback-SmartAmp.0 stream Playback-SmartAmp
[    7.708642] sof-audio-pci-intel-lnl 0000:00:1f.3: failed to add widget type 27 name : alh-copier.Playback-SmartAmp.0 stream Playback-SmartAmp
[    7.708650] sof_sdw sof_sdw: ASoC: failed to load widget alh-copier.Playback-SmartAmp.0
[    7.708652] sof_sdw sof_sdw: ASoC: topology: could not load header: -22
[    7.708661] sof-audio-pci-intel-lnl 0000:00:1f.3: error: tplg component load failed -22
[    7.708669] sof-audio-pci-intel-lnl 0000:00:1f.3: error: failed to load DSP topology -22
[    7.708671] sof-audio-pci-intel-lnl 0000:00:1f.3: ASoC: error at snd_soc_component_probe on 0000:00:1f.3: -22
[    7.708691] sof_sdw sof_sdw: ASoC: failed to instantiate card -22
[    7.709398] sof_sdw sof_sdw: error -EINVAL: snd_soc_register_card failed -22
[    7.709409] sof_sdw sof_sdw: probe with driver sof_sdw failed with error -22
```

Added full dmesg here:
[lenovo_yoga_tplg_added.txt](https://github.com/user-attachments/files/17931107/lenovo_yoga_tplg_added.txt)

bardliao, 2024-11-27:
@cgbowman I guess your device is not listed in the sof_sdw_quirk_table[].
Can you share the output of `sudo dmidecode | grep SKU`?

simontrimmer, 2024-11-27:
The DMI information matches the unit I was using (including the updated bios)

simontrimmer, 2024-11-27:
_Tangent - I wasn't getting "[    8.008471] i2c_designware i2c_designware.2: i2c_dw_handle_tx_abort: lost arbitration"_

cgbowman, 2024-11-27:
> @cgbowman I guess your device is not listed in the sof_sdw_quirk_table[]. Can you share the output of `sudo dmidecode | grep SKU`?

```
$ sudo dmidecode | grep SKU
        SKU Number: LENOVO_MT_83HM_BU_idea_FM_Yoga Slim 7 15ILL9
        SKU Number: SKU Number
                Consumer SKU
```

bardliao, 2024-11-28:
@cgbowman Did you upgrade your kernel with https://github.com/thesofproject/linux/pull/5248?

cgbowman, 2024-11-28:
> @cgbowman Did you upgrade your kernel with https://github.com/thesofproject/linux/pull/5248?

Yes, I cherry-picked the necessary patches onto a drm-tip base, but can retry with just the branch by itself tomorrow morning if you'd like confirmation.

bardliao, 2024-11-28:
@cgbowman Can you put [sof-dyndbg.conf](https://github.com/user-attachments/files/17943589/sof-dyndbg.conf.txt) to the /etc/modprobe.d/ folder and share the full dmesg log? Note that the file should be rename to `sof-dyndbg.conf`. I.e. remove the `.txt` filename extension.

cgbowman, 2024-11-28:
> @cgbowman Can you put [sof-dyndbg.conf](https://github.com/user-attachments/files/17943589/sof-dyndbg.conf.txt) to the /etc/modprobe.d/ folder and share the full dmesg log? Note that the file should be rename to `sof-dyndbg.conf`. I.e. remove the `.txt` filename extension.

Got it, here you are: 
[lenovo_yoga_sof-dyndbg_dmesg.txt](https://github.com/user-attachments/files/17944221/lenovo_yoga_sof-dyndbg_dmesg.txt)

bardliao, 2024-11-28:
Can you try `options snd_soc_sof_sdw quirk=0x30000`?
It looks like your device doesn't match the DMI match.
```c
                .matches = {
                        DMI_MATCH(DMI_SYS_VENDOR, "LENOVO"),
                        DMI_MATCH(DMI_PRODUCT_NAME, "83HM")
                },
```
What I expect is `quirk SOC_SDW_SIDECAR_AMPS enabled` and `quirk SOC_SDW_CODEC_MIC enabled`. However, your dmesg shows:
```
[    7.691906] snd_soc_sof_sdw:log_quirks: sof_sdw sof_sdw: quirk realtek,jack-detect-source 1
```
As a result, the `Playback-SmartAmp` link is not created and the `Capture-SmartMic` link is still created.
```
[    7.691927] snd_soc_sdw_utils:asoc_sdw_init_dai_link: sof_sdw sof_sdw: create dai link SDW0-Playback-SimpleJack, id 0
[    7.691929] snd_soc_sdw_utils:asoc_sdw_init_dai_link: sof_sdw sof_sdw: create dai link SDW0-Capture-SmartMic, id 4
[    7.691932] snd_soc_sdw_utils:asoc_sdw_init_dai_link: sof_sdw sof_sdw: create dai link SDW0-Capture-SimpleJack, id 1
[    7.691934] snd_soc_sdw_utils:asoc_sdw_init_dai_link: sof_sdw sof_sdw: create dai link dmic01, id 5
[    7.691936] snd_soc_sdw_utils:asoc_sdw_init_dai_link: sof_sdw sof_sdw: create dai link dmic16k, id 6
[    7.691938] snd_soc_sdw_utils:asoc_sdw_init_dai_link: sof_sdw sof_sdw: create dai link iDisp1, id 7
[    7.691941] snd_soc_sdw_utils:asoc_sdw_init_dai_link: sof_sdw sof_sdw: create dai link iDisp2, id 8
[    7.691942] snd_soc_sdw_utils:asoc_sdw_init_dai_link: sof_sdw sof_sdw: create dai link iDisp3, id 9
```

cgbowman, 2024-11-28:
Updated the config with the line, here is the resulting dmesg:
[lenovo_yoga_sof-dyndbg_quirk_dmesg.txt](https://github.com/user-attachments/files/17951389/lenovo_yoga_sof-dyndbg_quirk_dmesg.txt)

bardliao, 2024-11-29:
```
[    7.641998] sof_sdw sof_sdw: Overriding quirk 0x1 => 0x30000
[    7.642001] snd_soc_sof_sdw:log_quirks: sof_sdw sof_sdw: quirk SOC_SDW_SIDECAR_AMPS enabled
```
Still `quirk SOC_SDW_CODEC_MIC enabled` is not shown in your dmesg.
Please make sure below PR/commits are applied to your kernel.
https://github.com/torvalds/linux/commit/a6f7afb39362ef70d08d23e5bfc0a14d69fafea1
https://github.com/thesofproject/linux/pull/5248

Also,
``[    0.000000] Linux version 6.12.0-rc7+ (cgbowman@cgbowman-desk6) (gcc (Ubuntu 14.2.0-4ubuntu2) 14.2.0, GNU ld (GNU Binutils for Ubuntu) 2.43.1) #1 SMP PREEMPT_DYNAMIC Wed Nov 13 13:45:10 PST 2024``
It seems your kernel is built on Nov 13. Can you double check that you boot with your own kernel with the required commits?

cgbowman, 2024-11-29:
You are correct, I turned on CONFIG_LOCALVERSION_AUTO to confirm and it was loading another kernel by default.
After loading up the kernel with the proper SHA, I saw that the volume up/down controls are no longer giving off a dummy output (progress!), now showing 'Speakers' (although still no sound yet).

Here are the kernel commits I picked up:
```
013ad9da61dd (HEAD -> sof-testing) ASoC: Intel: sof_sdw: Add a dev_dbg message for the SOC_SDW_CODEC_MIC quirk
aa2d07bf6dc5 ASoC: Intel: sof_sdw: Correct quirk for Lenovo Yoga Slim 7
0be0c6f9ace0 ASoC: Intel: sof_sdw: Add quirks for some new Lenovo laptops
ebd7b7a4be3a ASoC: Intel: sof_sdw: Add quirk for cs42l43 system using host DMICs
db85b34f8a7a ASoC: sdw_utils: Add a quirk to allow the cs42l43 mic DAI to be ignored
4ee8684ae09b ASoC: sdw_utils: Add support for exclusion DAI quirks
739a2506c44a (origin/drm-tip, origin/HEAD) drm-tip: 2024y-11m-26d-13h-49m-46s UTC integration manifest
```

Here is the journalctl -k output (dmesg became too long):
[lenovo_yoga_sof-dyndbg_kernel_confirmed_journalctl.txt](https://github.com/user-attachments/files/17955643/lenovo_yoga_sof-dyndbg_kernel_confirmed_journalctl.txt)

bardliao, 2024-11-29:
No error from the log. I guess that is the mixer setting issue. @charleskeepax Can you help on it?

simontrimmer, 2024-11-29:
Hiya,
Looking at the log I suspect you are missing the amplifier firmware and tuning - have you update to a point beyond https://git.kernel.org/pub/scm/linux/kernel/git/firmware/linux-firmware.git/commit/?id=879bd9ae0b893ee00ca4069f09d018a9efd25ac3 ?
-Simon

cgbowman, 2024-11-29:
Yes, I've updated again with today's latest just to double check, along with ensuring initramfs gets updated with `sudo update-initramfs -u`

It seems the files from the given SHA are installed in /lib/firmware correctly:
```
$ find /lib/firmware -iname "cs35l56*" | grep -e 17aa380e -e Rev3.11.20
/lib/firmware/cirrus/cs35l56-b0-dsp1-misc-17aa380e-spkid0-ampl.bin
/lib/firmware/cirrus/cs35l56-b0-dsp1-misc-17aa380e-spkid0.wmfw
/lib/firmware/cirrus/cs35l56-b0-dsp1-misc-17aa380e-spkid1.wmfw
/lib/firmware/cirrus/cs35l56/CS35L56_Rev3.11.20.wmfw
/lib/firmware/cirrus/cs35l56-b0-dsp1-misc-17aa380e-spkid1-ampl.bin
/lib/firmware/cirrus/cs35l56-b0-dsp1-misc-17aa380e-spkid1-ampr.bin
/lib/firmware/cirrus/cs35l56-b0-dsp1-misc-17aa380e-spkid0-ampr.bin
```

simontrimmer, 2024-11-29:
Hi @cgbowman,
It does feel like it's not looking for the files, I guess the only other thing that stands out to me is that this is a product that has the spkid gpio derived from the cs42l43 driver - this was a fairly new patch that was delivered via Mark's SPI tree, I wonder if that's the missing piece?

https://git.kernel.org/pub/scm/linux/kernel/git/broonie/spi.git/commit/?id=f3c605147741e0ad8f1c51a7decef2040debfd16

I'll be able to double check the device I have here on Monday,
-Simon

cgbowman, 2024-11-29:
Sounds good, I'll try on my end as well! Always happy to test :)

simontrimmer, 2024-12-02:
@cgbowman 

Hmm, as a sanity check I just built at:

86183cdca9d6 (HEAD) spi: cs42l43: Add GPIO speaker id support to the bridge configuration
49e1628e8983 (upstream/topic/sof-dev, upstream/HEAD) ASoC: Intel: soc-acpi: arl: Add match entries for new cs42l43 laptops

Which is the sof-dev with the patch from mark's tree

git remote add spi git://git.kernel.org/pub/scm/linux/kernel/git/broonie/spi.git
git cherry-pick f3c605147741

and add

options snd-soc-wm_adsp dyndbg=+pmf
options spi-cs42l43 dyndbg=+pmf

To the modprobe.d dynamic debug config, I can see:

root@crus-83hm:~# dmesg | grep -i spi_cs42l43
[   10.982372] spi_cs42l43:cs42l43_get_speaker_id_gpios: cs42l43-spi cs42l43-spi: spk-id-gpios = 1
root@crus-83hm:~# dmesg | grep wm_adsp
[   11.887606] snd_soc_wm_adsp:wm_adsp_request_firmware_file: cs35l56 spi-cs35l56-right: DSP1: Failed to request 'cirrus/cs35l56-b0-dsp1-misc-17aa380e-spkid1-ampr.wmfw'
[   11.891569] snd_soc_wm_adsp:wm_adsp_request_firmware_file: cs35l56 spi-cs35l56-right: DSP1: Found 'cirrus/cs35l56-b0-dsp1-misc-17aa380e-spkid1.wmfw'
[   11.892501] snd_soc_wm_adsp:wm_adsp_request_firmware_file: cs35l56 spi-cs35l56-right: DSP1: Found 'cirrus/cs35l56-b0-dsp1-misc-17aa380e-spkid1-ampr.bin'
[   11.904986] snd_soc_wm_adsp:wm_adsp_request_firmware_file: cs35l56 spi-cs35l56-left: DSP1: Failed to request 'cirrus/cs35l56-b0-dsp1-misc-17aa380e-spkid1-ampl.wmfw'
[   11.905081] snd_soc_wm_adsp:wm_adsp_request_firmware_file: cs35l56 spi-cs35l56-left: DSP1: Found 'cirrus/cs35l56-b0-dsp1-misc-17aa380e-spkid1.wmfw'
[   11.905915] snd_soc_wm_adsp:wm_adsp_request_firmware_file: cs35l56 spi-cs35l56-left: DSP1: Found 'cirrus/cs35l56-b0-dsp1-misc-17aa380e-spkid1-ampl.bin'
root@crus-83hm:~#

The firmware version should then jump up to v3.11.20
root@crus-83hm:~# dmesg |grep v3.11.20
[   15.352566] cs35l56 spi-cs35l56-right: DSP1: Firmware: 1a00d6 vendor: 0x2 v3.11.20, 41 algorithms
[   15.361456] cs35l56 spi-cs35l56-left: DSP1: Firmware: 1a00d6 vendor: 0x2 v3.11.20, 41 algorithms
[   15.474488] cs35l56 spi-cs35l56-right: DSP1: cirrus/cs35l56-b0-dsp1-misc-17aa380e-spkid1-ampr.bin: v3.11.20
[   15.591591] cs35l56 spi-cs35l56-left: DSP1: cirrus/cs35l56-b0-dsp1-misc-17aa380e-spkid1-ampl.bin: v3.11.20
root@crus-83hm:~#

cgbowman, 2024-12-02:
gotcha, thanks for the sanity check, I'll try to replicate with that setup.

cgbowman, 2024-12-04:
Just an update, I think my issue may be related to non-updated sof-lnl.ri firmware, trying to go through sof build docs, but still have to figure out why the specified targets aren't recognized as valid options in the signed/unsigned lists.

Will keep working on it, maybe I should switch to the zephyr build, but unsure what is best practice for building this firmware.

bardliao, 2024-12-05:
@cgbowman I don't think your issue is related to sof-lnl.ri. However, you could upgrade your sof-lnl.ri from https://github.com/thesofproject/sof-bin/tree/main/v2.11.x/sof-ipc4-v2.11.1/lnl/intel-signed if you want.
BTW, have you tried @simontrimmer's suggestion https://github.com/thesofproject/sof/issues/9663#issuecomment-2511504203?

cgbowman, 2024-12-10:
Apologies for the very late reply, I was caught up in other tasks.

>BTW, have you tried @simontrimmer's suggestion https://github.com/thesofproject/sof/issues/9663#issuecomment-2511504203?

I have tried the suggestion and it didn't seem to yield results for my device. I keep suspecting there's something else wrong with it, but I'm unsure what yet.

Here is my kernel with the sof-dev branch (on the day of testing):
```
1b1da367dd0a (HEAD -> sof-sanity-check) x86/cpu: Add INTEL_LUNARLAKE_M to X86_BUG_MONITOR
54e3355085c3 spi: cs42l43: Add GPIO speaker id support to the bridge configuration
49e1628e8983 (sof/topic/sof-dev) ASoC: Intel: soc-acpi: arl: Add match entries for new cs42l43 laptops
```

Here is my journalctl output:
[sof-dev-test.txt](https://github.com/user-attachments/files/18073214/sof-dev-test.txt)

bardliao, 2024-12-10:
@cgbowman Just to confirm the issue you still have is that there is no sound from the speaker, right? If so, can we close the issue and create another one for the speaker no sound issue?

simontrimmer, 2024-12-10:
@cgbowman 

Yes weird - at least the trace does show that the amplifiers are hooking up with the expected firmware and features now.

It's a tangent, but I'd be extremely interested in the contents of the /sys/firmware/efi/efivars/CirrusSmart* files - when you get a chance could you drop them here or on the new ticket we use to further investigate this system?
Thanks!
-Simon

cgbowman, 2024-12-11:
> @cgbowman Just to confirm the issue you still have is that there is no sound from the speaker, right? If so, can we close the issue and create another one for the speaker no sound issue?

Yes, that's correct. I've filed a new issue here: https://github.com/thesofproject/sof/issues/9720

>It's a tangent, but I'd be extremely interested in the contents of the /sys/firmware/efi/efivars/CirrusSmart* files - when you get a chance could you drop them here or on the new ticket we use to further investigate this system?

I'll make sure to post this information in the new issue.

Thanks for all your work! 😃

closed 2024-12-11 by cgbowman