thesofproject/sof #9674: [BUG] annoying kernel message from SOF loader

Shawn-Huang724, 2024-11-22:
**Describe the bug**
SOF loader of Linux kernel always prompt an error message of "unknown sof_ext_man header type 3 size 0x30" during firmware loading on all platforms.

**To Reproduce**
Boot SOF firmware and check the dmesg log of kernel.

**Reproduction Rate**
10/10, 100%.

**Expected behavior**
Saw "unknown sof_ext_man header type 3 size 0x30" message in dmesg log.

**Impact**
Annoyance.

**Environment**
1) Branch name and commit hash of the 2 repositories: sof (firmware/topology) and linux (kernel driver).
    * Kernel: 5f650cf5eca3d74c7bcf4837128afa88cb037b07
    * SOF: ef51d72b1e6ef992925e259290c6f11dd81f364a
2) Name of the topology file
    * Topology: anyone.
3) Name of the platform(s) on which the bug is observed.
    * Platform: all platforms.

**Screenshots or console output**
``` unknown sof_ext_man header type 3 size 0x30 ``` in dmesg log.

Shawn-Huang724, 2024-11-22:
Found a recent commit which might be a solution from kernel in sof-dev branch:
https://github.com/thesofproject/linux/commit/1a4efb45a6beadb10565a0bafdd33f999d5d5286

lgirdwood, 2024-11-22:
@ujfalusi pls confirm if your PR will fix and if so we can close. Thanks !

wszypelt, 2025-04-25:
@ujfalusi  can you confirm? Thanks!

ujfalusi, 2025-04-25:
The kernel no longer warns for this since 6.13

closed 2025-04-25 by wszypelt