thesofproject/sof #9678: [BUG] Left and right audio channels swapped on speakers and Microphone not working on Razer Blade 15

callummarshall9, 2024-11-23:
**Describe the bug**
Hi,

I have a Razer Blade 15 RZ09-0421 (2022)

When I play audio, the left channel comes out the right speaker and the right channel comes out the left speaker. I've confirmed this by using the speaker test and when I click "left" it sounds like it's coming out the right speaker.

When I try to use the built in microphone no audio is picked up.

What have you tried to diagnose or workaround this issue?

I've tried the topology fix listed here:
https://abarry.org/ubuntu-on-razer-blade-15-2022-advanced/
however this seems to not work and I've had to revert the sound card.

**To Reproduce**
Boot up system and try to record or play any audio.

**Reproduction Rate**
Reliably fails every time I boot up system.

**Expected behavior**
A clear and concise description of what you expected to happen.

**Impact**
Annoyance
What impact does this issue have on your progress (e.g., annoyance, showstopper)

**Environment**

Output of uname -a
Linux fedora 6.11.7-300.fc41.x86_64 #1 SMP PREEMPT_DYNAMIC Fri Nov  8 19:23:10 UTC 2024 x86_64 GNU/Linux
Using KDE Fedora, Pipewire and X11.

Here's the output of alsa-info.sh to provide more information about the sound card.

http://alsa-project.org/db/?f=36a0ff155bab8d5bb8447713f55468562f8fe6db

Branch names unknown as is topology and name of platform

sof-logger yields command not found.

callummarshall9, 2024-11-23:
[image]

Screenshot of settings window.
"Digital microphone" picks up no output

lgirdwood, 2024-11-25:
@callummarshall9 the L/R swap is likely to be the codec configuration or PCB layout issue (someone else reported a similar L/R issue recently - pls search).  It is worth trying alsamixer in case the default mixer config is wrong.
DMICs can be attached to the codec or the SoC, codec DMICs may need some alsamixer tuning too. SoC DMICs should just work. Adding in @ujfalusi who may have some more hints.

callummarshall9, 2024-11-25:
@lgirdwood I apologise, I looked over this ticket https://github.com/thesofproject/sof/issues/5989#but I downloaded the wrong topology file mentioned further up and that fix didn't work... However using this exact topology file has fixed it https://github.com/thesofproject/sof/issues/5989#issuecomment-1176605277

closed 2024-11-25 by callummarshall9