thesofproject/sof #9681: [BUG] dmic recording volume is too low

pma1, 2024-11-26:
**Describe the bug**
The dmic volume is too low even the max volume of input/output is set.

**To Reproduce**
Set the max volume in gnome-setting.
$ arecord -f cd b.wav

**Reproduction Rate**
100%

**Expected behavior**
Clear audio recording.

**Impact**
It's barely to hear the dmic.

**Environment**
1)
    * Kernel: 6.12 9f16d5e6f2206 (linux/master) Merge tag 'for-linus' of git://git.kernel.org/pub/scm/virt/kvm/kvm
    * SOF: db771bf v2.11.x: add new 2.11.2 topology2 production binaries
    
2) Name of the topology file
    * Topology: intel/sof-ipc4-tplg/sof-lnl-rt713-l0-rt1318-l1-2ch.tplg
3) Name of the platform(s) on which the bug is observed.
    * Platform: ThinkPad X1G13 Intel CPU: Intel(R) Core(TM) Ultra 7 258V

**Screenshots or console output**
[x1g13-mic-low-dmesg.txt](https://github.com/user-attachments/files/17917668/x1g13-mic-low-dmesg.txt)

pma1, 2024-11-27:
[image]
screenshot of alsamixer, the caputure volume is max.

bardliao, 2024-11-27:
This is PCH DMIC. @pma1 Can you share the result of `amixer -c0 contents`?
And try `amixer -c0 cset name='Dmic0 Capture Volume' 45`

pma1, 2024-11-27:
$ amixer -c0 cset name='Dmic0 Capture Volume' 45
numid=40,iface=MIXER,name='Dmic0 Capture Volume'
  ; type=INTEGER,access=rw---R--,values=2,min=0,max=45,step=0
  : values=45,45
  | dBscale-min=-90.00dB,step=2.00dB,mute=1

After set command above, the volume is still the same low as before.

output of amixer -c0 contents:
[amixer-contents.txt](https://github.com/user-attachments/files/17930658/amixer-contents.txt)

bardliao, 2024-11-27:
@pma1 The mixer setting looks good to me. Can you share the captured wav file? Can you try different formats? Like `-fS16_LE -c2 -r48000`, `-fS24_LE -c2 -r48000`, `-fS32_LE -c2 -r48000`. @singalsu Could it be an `DMIC0 Capture IIR Eq` issue?

pma1, 2024-11-27:
[arecord-16-24-32.tar.gz](https://github.com/user-attachments/files/17932581/arecord-16-24-32.tar.gz)

arecord file attached, thanks.

bardliao, 2024-11-27:
@pma1 Can you try `arecord -Dhw:0,10 -fS32_LE -r48000 -c2 file.wav`?

pma1, 2024-11-27:
The volume is as low as before.

singalsu, 2024-11-28:
We are in progress of adding DRC and other capture pre-processing. I'll do a pull request to enable it on this topology. It will boost the capture level.

singalsu, 2024-11-28:
@pma1 Are you able to replace the topology (keep a backup if this fails) with this from my test build? Please unzip it since GH does not allow .tplg attachment. If the installed topology is with different file name with compression, move it so there's no lottery for which topology gets loaded.

[sof-lnl-rt713-l0-rt1318-l1-2ch.zip](https://github.com/user-attachments/files/17945406/sof-lnl-rt713-l0-rt1318-l1-2ch.zip)

Then you need to find the control for DRC from "alsamixer -c0" and enable it with key m. You can also enable the TDFB beam for beamformer and set the beam angle to 0 degrees with cursor up/down keys. There's only single angle in the blob so there won't be a steerable beam but future topologies may have it, so it doesn't hurt to set it too.

singalsu, 2024-11-28:
Once the controls are set, they should persists if you keep using same topology. You can also set the controls from command line, e.g.

```
amixer -c0 controls | grep Dmic0
amixer -c0 cset name='Dmic0 Capture DRC switch' on
amixer -c0 cset nname='Dmic0 Capture TDFB beam switch' on
amixer -c0 cset name='Dmic0 Capture TDFB angle set enum' 0

```

singalsu, 2024-11-28:
@pma1 @bardliao It is possible to get even more gain with custom DRC and IIR blobs but I'd not do it with generic topologies but instead with UCM2 runtime rules with matched DMI ID. We have already a framework to control SOF for playback with sof-hda-generic (see https://github.com/alsa-project/alsa-ucm-conf/tree/master/ucm2/blobs/sof) and such should be added capture also and other PC platforms with SDW codec.

pma1, 2024-11-28:
@singalsu 
Thanks for your kindly help.

The tplg you uploaded is failed to be loaded, please refer to the dmesg:
[6.12-x1cg13-sof-load-fail-dmesg.txt](https://github.com/user-attachments/files/17949342/6.12-x1cg13-sof-load-fail-dmesg.txt)

lgirdwood, 2024-11-28:
@singalsu @ujfalusi see
```
[    3.937788] sof_sdw sof_sdw: ASoC: Parent card not yet available, widget card binding deferred
[    3.937812] sof-audio-pci-intel-lnl 0000:00:1f.3: Direct firmware load for intel/sof-ipc4-lib/lnl/B36EE4DA-006F-47F9-A06D-FECBE2D8B6CE.bin failed with error -2
[    3.937814] sof-audio-pci-intel-lnl 0000:00:1f.3: Library file 'intel/sof-ipc4-lib/lnl/B36EE4DA-006F-47F9-A06D-FECBE2D8B6CE.bin' is missing
[    3.937815] sof-audio-pci-intel-lnl 0000:00:1f.3: failed to find module info for widget drc.100.1 with UUID B36EE4DA-006F-47F9-A06D-FECBE2D8B6CE
[    3.937816] sof-audio-pci-intel-lnl 0000:00:1f.3: failed updating IPC struct for drc.100.1
[    3.937915] sof-audio-pci-intel-lnl 0000:00:1f.3: error: tplg component load failed -22
[    3.937917] sof-audio-pci-intel-lnl 0000:00:1f.3: error: failed to load DSP topology -22
[    3.937918] sof-audio-pci-intel-lnl 0000:00:1f.3: ASoC: error at snd_soc_component_probe on 0000:00:1f.3: -22
```
@pma1 can you ls -l the contents of ```'intel/sof-ipc4-lib/lnl/``` in the firmware directory. Thanks.

pma1, 2024-11-29:
@lgirdwood 

After copied the intel/sof-ipc4-lib/lnl/ file,

The sound card works, the mic volume it better, but not loud enough.
[x1cg13-mic-arecod-better-f-cd.tar.gz](https://github.com/user-attachments/files/17953318/x1cg13-mic-arecod-better-f-cd.tar.gz)

but it keep the same volume even set "amixer -c0 cset name='Dmic0 Capture TDFB angle set enum' 90"
[image]

Set 'Dmic0 Capture TDFB angle set enum' to -90, the vomlume is the same.

dmesg:
[6.12-x1cg13-sof-mic-volume-better.txt](https://github.com/user-attachments/files/17953324/6.12-x1cg13-sof-mic-volume-better.txt)

singalsu, 2024-11-29:
Thanks for testing it! I noticed there's in similar hda-generic topology gain applied in IIR, so it is still louder than this. I'll do another change to change IIR to amplifying type.

The beam angle should be 0 towards user, that's how to coordinates were defined.  But there is no 90 degree beam in the blob, so there is no impact except for beam on/off. Blobs with better control are possible with DMI ID based load by UCM.

singalsu, 2024-11-29:
Can you please try this topology:  
[sof-lnl-rt713-l0-rt1318-l1-2ch-plus20db.zip](https://github.com/user-attachments/files/17957380/sof-lnl-rt713-l0-rt1318-l1-2ch-plus20db.zip)

It's very good that you are able to test since I don't have access to a similar PC. Can you try to talk very loud to the microphone when capturing, like people might do in a loud environment like cafe with babble. Does the microphone signal clip or sound distorted?

singalsu, 2024-11-29:
Thanks @pma1 ! I'll proceed with this latest change in this pull request: https://github.com/thesofproject/sof/pull/9690

The next SOF release should be OK.

pma1, 2024-11-29:
Thanks to all of you for the great help.

singalsu, 2024-12-02:
@pma1 Can you please check this smaller fix that would be quick to release as stable-v2.11 update. This change only adds gain to IIR, there's no added DRC and TDFB components. What do you think, does this sufficiently address the issue?

[sof-lnl-rt713-l0-rt1318-l1-2ch-iir-only-plus20db.zip](https://github.com/user-attachments/files/17975567/sof-lnl-rt713-l0-rt1318-l1-2ch-iir-only-plus20db.zip)

pma1, 2024-12-02:
@singalsu there is no obviously difference with previous tplg, maybe minor lower volume but hardly noticed.

I think this can fix the issue.

singalsu, 2024-12-02:
Thanks a lot for testing @pma1 !  @kv2019i @lgirdwood I think then we can proceed with https://github.com/thesofproject/sof/pull/9698 fix for 2.11.

lgirdwood, 2024-12-02:
> Thanks a lot for testing @pma1 ! @kv2019i @lgirdwood I think then we can proceed with #9698 fix for 2.11.

Will merge this as soon as CI is back later today.

kv2019i, 2024-12-05:
Fix merged, closing. Will be released in sof-bin-2024.09.2 binary release.

closed 2024-12-05 by kv2019i