thesofproject/sof #9682: [BUG] no sof-lnl-cs42l43-l0-2ch.tplg found

pma1, 2024-11-26:
**Describe the bug**
No sound cards are initialized.

**Reproduction Rate**
1/1

**Expected behavior**
The right tplg is loaded.

**Impact**
no audio

**Environment**
1) Branch name and commit hash of the 2 repositories: sof (firmware/topology) and linux (kernel driver).
    * Kernel: 6.12+ 9f16d5e6f2206 Merge tag 'for-linus' of git://git.kernel.org/pub/scm/virt/kvm/kvm
    * SOF: db771bf v2.11.x: add new 2.11.2 topology2 production binaries
2) Name of the topology file
    * Topology: intel/sof-ipc4-tplg/sof-lnl-cs42l43-l0-2ch.tplg
3) Name of the platform(s) on which the bug is observed.
    * Platform: CPU0: Intel(R) Core(TM) Ultra 7 258V

**Screenshots or console output**
[aladdin-1-6.12-dmesg-nosoundcards.txt](https://github.com/user-attachments/files/17917786/aladdin-1-6.12-dmesg-nosoundcards.txt)

lgirdwood, 2024-11-26:
@charleskeepax I guess this topology is now upstream but not in the current release ?

charleskeepax, 2024-11-27:
Alas a bunch of these SKUs ended up looking slightly different to what we expected. This needs some updates, we have a 4-channel topology, but not a 2-channel one. This will also need some quirk updates for the kernel side.

rfvirgil, 2024-11-27:
Can you try it with these two patches?

Kernel: https://github.com/CirrusLogic/sof-linux/commit/5025ded7f001bc587f294a489cac5c4876ad4268
Topology: https://github.com/CirrusLogic/sof/commit/985e34c9ebe730de28152c3716fc897fa9986599

pma1, 2024-11-28:
@rfvirgil ,
Thanks for the quick change.

Would you mind share a sof-bin? Sorry I am not familiar with sof building.

rfvirgil, 2024-11-28:
tplg file attached
[sof-lnl-cs42l43-l0-2ch.tplg.gz](https://github.com/user-attachments/files/17946464/sof-lnl-cs42l43-l0-2ch.tplg.gz)

lgirdwood, 2024-11-28:
@ujfalusi @bardliao pls see above copier error and dmesg. Missing kernel patch or topology ?

rfvirgil, 2024-11-28:
@pma1 Did you apply the kernel patch? https://github.com/thesofproject/linux/commit/f6f710b39cc155cd0bd67d8ce0470d741aef4217

pma1, 2024-11-29:
@rfvirgil 

Sorry for my mistake.

After applied patch, with new tplg, still show:
sof-audio-pci-intel-lnl 0000:00:1f.3: no hda codecs found!

[Aladdin-sof-tplg-kernel-6.12-patch-load-fail-dmesg.txt](https://github.com/user-attachments/files/17953197/Aladdin-sof-tplg-kernel-6.12-patch-load-fail-dmesg.txt)

bardliao, 2024-11-29:
The main issue is:
```
[    4.014846] sof_sdw sof_sdw: ASoC: physical link dmic01 (id 4) not exist
[    4.015984] sof_sdw sof_sdw: ASoC: topology: could not load header: -22
[    4.017044] sof-audio-pci-intel-lnl 0000:00:1f.3: error: tplg component load failed -22
[    4.017758] sof-audio-pci-intel-lnl 0000:00:1f.3: error: failed to load DSP topology -22
[    4.018707] sof-audio-pci-intel-lnl 0000:00:1f.3: ASoC: error at snd_soc_component_probe on 0000:00:1f.3: -22
[    4.019675] sof_sdw sof_sdw: ASoC: failed to instantiate card -22
```
Please rename [sof-dyndbg.conf.txt](https://github.com/user-attachments/files/17953612/sof-dyndbg.conf.txt) to sof-dyndbg.conf and put it in the /etc/modprobe.d/ folder and share the dmesg again. Thanks.

pma1, 2024-11-29:
@bardliao ,

dmesg attached:
[Aladdin-sof-tplg-kernel-6.12-patch-load-fail-dmesg-dyndbg.txt](https://github.com/user-attachments/files/17954941/Aladdin-sof-tplg-kernel-6.12-patch-load-fail-dmesg-dyndbg.txt)

bardliao, 2024-11-29:
```
	Line 3681: [    3.675201] snd_soc_sdw_utils:asoc_sdw_init_dai_link: sof_sdw sof_sdw: create dai link SDW0-Playback-SimpleJack, id 0
	Line 3682: [    3.675862] snd_soc_sdw_utils:asoc_sdw_init_dai_link: sof_sdw sof_sdw: create dai link SDW0-Capture-SimpleJack, id 1
	Line 3683: [    3.676936] snd_soc_sdw_utils:asoc_sdw_init_dai_link: sof_sdw sof_sdw: create dai link SDW0-Playback-SmartAmp, id 2
	Line 3684: [    3.677793] snd_soc_sdw_utils:asoc_sdw_init_dai_link: sof_sdw sof_sdw: create dai link dmic01, id 3
	Line 3685: [    3.678640] snd_soc_sdw_utils:asoc_sdw_init_dai_link: sof_sdw sof_sdw: create dai link dmic16k, id 4
	Line 3686: [    3.679337] snd_soc_sdw_utils:asoc_sdw_init_dai_link: sof_sdw sof_sdw: create dai link iDisp1, id 5
	Line 3687: [    3.679976] snd_soc_sdw_utils:asoc_sdw_init_dai_link: sof_sdw sof_sdw: create dai link iDisp2, id 6
	Line 3688: [    3.680805] snd_soc_sdw_utils:asoc_sdw_init_dai_link: sof_sdw sof_sdw: create dai link iDisp3, id 7
```
The dmic01 id is 3.
@pma1 Can you try [sof-lnl-cs42l43-l0-2ch.zip](https://github.com/user-attachments/files/17955064/sof-lnl-cs42l43-l0-2ch.zip)?

pma1, 2024-11-29:
[Aladdin-sof-tplg-kernel-6.12-patch-audio-show-no-input-nomic-dmesg.txt](https://github.com/user-attachments/files/17956082/Aladdin-sof-tplg-kernel-6.12-patch-audio-show-no-input-nomic-dmesg.txt)

With the new tplg, the audio is shown:
$ cat /proc/asound/cards 
 0 [sofsoundwire   ]: sof-soundwire - sof-soundwire
                      LENOVO-21Q6SIT019-ThinkPadAladdin

But still no sound output, no mic.

bardliao, 2024-11-29:
```
[    3.065555] calling  xe_init+0x0/0x100 [xe] @ 275
[    3.065616] initcall xe_init+0x0/0x100 [xe] returned -19 after 0 usecs
```
Also some
```
sof-audio-pci-intel-lnl 0000:00:1f.3: ASoC: error at snd_soc_dai_hw_params on iDisp1 Pin: -22
```
are shown.
@kv2019i Do you know if display audio work on 6.12.0+?

@pma1 Can you test with aplay/arecord first?

pma1, 2024-11-29:
There is no sound from aplay or arecord.

xe is disabled because of some issues.

After enabled xe, the sound can not output or input too.
[Aladdin-sof-tplg-kernel-6.12-patch-xe-audio-show-no-input-nomic-dmesg.txt](https://github.com/user-attachments/files/17956842/Aladdin-sof-tplg-kernel-6.12-patch-xe-audio-show-no-input-nomic-dmesg.txt)

kv2019i, 2024-11-29:
@pma1 wrote:
> After enabled xe, the sound can not output or input too. [Aladdin-sof-tplg-kernel-6.12-patch-xe-audio-show-no-input-nomic-dmesg.txt](https://github.com/user-attachments/files/17956842/Aladdin-sof-tplg-kernel-6.12-patch-xe-audio-show-no-input-nomic-dmesg.txt)

ACk to @bardliao , display audio should work and this log at least looks good for driver probes. Xe driver for display is up, SOF and codec drivers are probed without errors. I wonder if some alsamixer settings are wrong now. Output from "alsa-info.sh" would be useful (when playing back to one device with aplay). It will show the state of all codec settings.

rfvirgil, 2024-11-29:
Speakers will not work because you don't have firmware for the CS35L56. We would need permission from Lenovo to publish the firmware.

Did you buy this laptop from a shop, or is this a pre-production device?

pma1, 2024-11-29:
alsa-info:
[aladdin-alsa-info.txt](https://github.com/user-attachments/files/17958738/aladdin-alsa-info.txt)

This is a pre-production device.

Does it need a kernel patch to match the CS35L56?

rfvirgil, 2024-12-02:
> Does it need a kernel patch to match the CS35L56?

The patch I linked that changes the DMI matches should be the only kernel patch you need.

bardliao, 2024-12-06:
@rfvirgil 
```c
"cavs-sdw\;sof-lnl-cs42l43-l0-2ch\;PLATFORM=lnl,NUM_SDW_AMP_LINKS=1,NUM_DMICS=2,\
PDM1_MIC_A_ENABLE=0,PDM1_MIC_B_ENABLE=0,DMIC0_ID=3,DMIC1_ID=4,\
PREPROCESS_PLUGINS=nhlt,NHLT_BIN=nhlt-sof-lnl-cs42l43-l0-2ch.bin,\
SDW_AMP_FEEDBACK=false,SDW_SPK_STREAM=Playback-SmartAmp,\
SDW_JACK_OUT_STREAM=Playback-SimpleJack,SDW_JACK_IN_STREAM=Capture-SimpleJack,\
HDMI1_ID=5,HDMI2_ID=6,HDMI3_ID=7,DMIC0_ENHANCED_CAPTURE=true,\
EFX_DMIC0_TDFB_PARAMS=line4_pass,EFX_DMIC0_DRC_PARAMS=dmic_default"
```
The point is `DMIC0_ID=3,DMIC1_ID=4`. They are 4 and 5 by default. But from
```
	Line 3684: [    3.677793] snd_soc_sdw_utils:asoc_sdw_init_dai_link: sof_sdw sof_sdw: create dai link dmic01, id 3
	Line 3685: [    3.678640] snd_soc_sdw_utils:asoc_sdw_init_dai_link: sof_sdw sof_sdw: create dai link dmic16k, id 4
```
They should be 3 and 4.

RyanJiao, 2024-12-08:
[image]
Hi, I'm using kernel 6.12 on my Intel Ultra 258V and the sof driver was not initialized properly resulting in no audio. 

But I can't even find a sof-lnl.**  file in this repo's release zip. How can I load my audio card driver properly? It would be so grateful if anyone can give any instruction.

bardliao, 2024-12-09:
@RyanJiao You could find the sof-lnl.ri from https://github.com/thesofproject/sof-bin/tree/main/v2.11.x/sof-ipc4-v2.11.1/lnl/intel-signed

rfvirgil, 2024-12-16:
The tplg file (with Bard's change) is merged into the sof repository here:
https://github.com/thesofproject/sof/commit/f4cd55c2e879e7f822cb793bd65449fa1da36b04

The firmware files for the CS35L56 for Aladdin are now available here:
https://git.kernel.org/pub/scm/linux/kernel/git/firmware/linux-firmware.git/commit/?id=9c46d10f8ea7a72d910ba5f5ebed7fcd2de31c71

**Do not copy the firmware files manually. It must be installed correctly**
First **make a backup of your existing /lib/firmware content so you can restore it if this all goes wrong!**

```
sudo apt-get install rdfind
git clone git://git.kernel.org/pub/scm/linux/kernel/git/firmware/linux-firmware.git
cd linux-firmware
sudo make install
sudo make dedup

```

pma1, 2025-02-11:
@rfvirgil and @bardliao 
After I copied the sof-lnl-cs42l43-l0-2ch.tplg in the latest sof-bin, and with 6.14-rc2.

There is no sound.

Errors are shown:
kernel: soundwire_intel soundwire_intel.link.0: IO transfer timed out, cmd 2 device 0 addr 50 len 6
kernel: soundwire sdw-master-0-0: trf on Slave 0 failed:-110 read addr 50 count 0
kernel: soundwire sdw-master-0-0: DEVID read fail:-110

[Aladdin-1-6.14-rc2-alsa-info.txt](https://github.com/user-attachments/files/18746240/Aladdin-1-6.14-rc2-alsa-info.txt)

bardliao, 2025-02-11:
```
Simple mixer control 'cs42l43 Speaker L Input 1',0
  Capabilities: volume volume-joined enum
  Items: 'None' 'Tone Generator 1' 'Tone Generator 2' 'Decimator 1' 'Decimator 2' 'Decimator 3' 'Decimator 4' 'ASPRX1' 'ASPRX2' 'ASPRX3' 'ASPRX4' 'ASPRX5' 'ASPRX6' 'DP5RX1' 'DP5RX2' 'DP6RX1' 'DP6RX2' 'DP7RX1' 'DP7RX2' 'ASRC INT1' 'ASRC INT2' 'ASRC INT3' 'ASRC INT4' 'ASRC DEC1' 'ASRC DEC2' 'ASRC DEC3' 'ASRC DEC4' 'ISRC1 INT1' 'ISRC1 INT2' 'ISRC1 DEC1' 'ISRC1 DEC2' 'ISRC2 INT1' 'ISRC2 INT2' 'ISRC2 DEC1' 'ISRC2 DEC2' 'EQ1' 'EQ2'
  Item0: 'None'
```
```
Simple mixer control 'cs42l43 Speaker R Input 1',0
  Capabilities: volume volume-joined enum
  Items: 'None' 'Tone Generator 1' 'Tone Generator 2' 'Decimator 1' 'Decimator 2' 'Decimator 3' 'Decimator 4' 'ASPRX1' 'ASPRX2' 'ASPRX3' 'ASPRX4' 'ASPRX5' 'ASPRX6' 'DP5RX1' 'DP5RX2' 'DP6RX1' 'DP6RX2' 'DP7RX1' 'DP7RX2' 'ASRC INT1' 'ASRC INT2' 'ASRC INT3' 'ASRC INT4' 'ASRC DEC1' 'ASRC DEC2' 'ASRC DEC3' 'ASRC DEC4' 'ISRC1 INT1' 'ISRC1 INT2' 'ISRC1 DEC1' 'ISRC1 DEC2' 'ISRC2 INT1' 'ISRC2 INT2' 'ISRC2 DEC1' 'ISRC2 DEC2' 'EQ1' 'EQ2'
  Item0: 'None'
```
```
Simple mixer control 'cs42l43 Headphone L Input 1',0
  Capabilities: volume volume-joined enum
  Items: 'None' 'Tone Generator 1' 'Tone Generator 2' 'Decimator 1' 'Decimator 2' 'Decimator 3' 'Decimator 4' 'ASPRX1' 'ASPRX2' 'ASPRX3' 'ASPRX4' 'ASPRX5' 'ASPRX6' 'DP5RX1' 'DP5RX2' 'DP6RX1' 'DP6RX2' 'DP7RX1' 'DP7RX2' 'ASRC INT1' 'ASRC INT2' 'ASRC INT3' 'ASRC INT4' 'ASRC DEC1' 'ASRC DEC2' 'ASRC DEC3' 'ASRC DEC4' 'ISRC1 INT1' 'ISRC1 INT2' 'ISRC1 DEC1' 'ISRC1 DEC2' 'ISRC2 INT1' 'ISRC2 INT2' 'ISRC2 DEC1' 'ISRC2 DEC2' 'EQ1' 'EQ2'
  Item0: 'None'
```
```
Simple mixer control 'cs42l43 Headphone R Input 1',0
  Capabilities: volume volume-joined enum
  Items: 'None' 'Tone Generator 1' 'Tone Generator 2' 'Decimator 1' 'Decimator 2' 'Decimator 3' 'Decimator 4' 'ASPRX1' 'ASPRX2' 'ASPRX3' 'ASPRX4' 'ASPRX5' 'ASPRX6' 'DP5RX1' 'DP5RX2' 'DP6RX1' 'DP6RX2' 'DP7RX1' 'DP7RX2' 'ASRC INT1' 'ASRC INT2' 'ASRC INT3' 'ASRC INT4' 'ASRC DEC1' 'ASRC DEC2' 'ASRC DEC3' 'ASRC DEC4' 'ISRC1 INT1' 'ISRC1 INT2' 'ISRC1 DEC1' 'ISRC1 DEC2' 'ISRC2 INT1' 'ISRC2 INT2' 'ISRC2 DEC1' 'ISRC2 DEC2' 'EQ1' 'EQ2'
  Item0: 'None'
```

@pma1 Looks like the mixer settings are incorrect. Can you update your ucm files and test again?
You can get the latest ucm from https://github.com/alsa-project/alsa-ucm-conf/

pma1, 2025-02-12:
@bardliao copied the latest alsa-ucm-conf, the speaker sof-firmware is in the sound ouput list, but no sound output, no mic input at all.

alsa-info 

[aladdin-1-ucm250212-6.14-rc2-alsa-info.txt](https://github.com/user-attachments/files/18761619/aladdin-1-ucm250212-6.14-rc2-alsa-info.txt)

bardliao, 2025-02-12:
@pma1 Can you run "sudo alsaucm reload" and check is there any error?

pma1, 2025-02-13:
@bardliao Thanks for the reminder, after update alsa-utils to 1.2.13.

with the lastest alsa-ucm-conf, the audio input and output works fine.

Could you please share the specific commits of ucm those works with this tplg?
I have to backport these commits to the old release of Ubuntu.

Thank you very much.

bardliao, 2025-02-13:
@pma1 I think @rfvirgil and @charleskeepax may be better people to answer the question. :)

pma1, 2025-02-14:
@bardliao 
Neither the Speaker mute LED or the mic mute LED can work.
There is no codec list in card sofsoundwire:
$ cat /proc/asound/cards 
 0 [sofsoundwire   ]: sof-soundwire - sof-soundwire
                      LENOVO-21Q6SIT019-ThinkPadAladdin
29 [ThinkPadEC     ]: ThinkPad EC - ThinkPad Console Audio Control
                      ThinkPad Console Audio Control at EC reg 0x30, fw N4CHT24W

Only 1 codec: /proc/asound/card0/codec#2
Codec: Intel Lunar Lake HDMI
Address: 2
AFG Function Id: 0x1 (unsol 0)
Vendor Id: 0x80862820
Subsystem Id: 0x80860101
Revision Id: 0x100000