thesofproject/sof #9687: [BUG] DSP crash after Zephyr 26th Nov update

kv2019i, 2024-11-27:
**Describe the bug**
DSP panic seen with Zephyr update in https://github.com/thesofproject/sof/pull/9671 . This pull-request without the Zephyr update didn't trigger issue, so must be related to the Zephyr version update.

**To Reproduce**
Tests triggered by CI

**Reproduction Rate**
100% 

**Expected behavior**
No DSP crash

**Impact**
Blocking Zephyr updates as a test in PR testing is failing.

**Environment**
1) Branch name and commit hash of the 2 repositories: sof (firmware/topology) and linux (kernel driver).
    * see https://github.com/thesofproject/sof/pull/9671 
2) Name of the topology file
    * Topology: Intel test 01_21_TestCopierIpc
3) Name of the platform(s) on which the bug is observed.
    * Platform: Intel ptl
 
**Screenshots or console output**

```
*** Booting Zephyr OS build v4.0.0-831-ga14ae39e7447 ***
[00:00:01.946,813] <inf> main: sof_app_main: SOF on intel_adsp
[00:00:01.946,936] <inf> main: sof_app_main: SOF initialized
[00:00:02.037,930] <err> os: print_fatal_exception:  ** FATAL EXCEPTION
[00:00:02.038,041] <err> os: print_fatal_exception:  ** CPU 0 EXCCAUSE 29 (store prohibited)
[00:00:02.038,071] <err> os: print_fatal_exception:  **  PC 0xa007f894 VADDR 0xa00303bc
[00:00:02.038,100] <err> os: print_fatal_exception:  **  PS 0x60823
[00:00:02.038,126] <err> os: print_fatal_exception:  **    (INTLEVEL:3 EXCM: 0 UM:1 RING:0 WOE:1 OWB:8 CALLINC:2)
[00:00:02.038,216] <err> os: xtensa_dump_stack:  **  A0 0xa0063640  SP 0xa00fbd70  A2 0xa0149100  A3 0x40148ba8
[00:00:02.038,248] <err> os: xtensa_dump_stack:  **  A4 (nil)  A5 0xa01491b0  A6 0xb8  A7 0x19fff
[00:00:02.038,275] <err> os: xtensa_dump_stack:  **  A8 0x19fff  A9 0xa00fbd40 A10 0x40147000 A11 0x448
[00:00:02.038,301] <err> os: xtensa_dump_stack:  ** A12 0x40149240 A13 0x10006 A14 0xa01491b0 A15 (nil)
[00:00:02.038,328] <err> os: xtensa_dump_stack:  ** LBEG 0xa004f591 LEND 0xa004f59d LCOUNT 0xf
[00:00:02.038,355] <err> os: xtensa_dump_stack:  ** SAR 0x20
[00:00:02.038,381] <err> os: xtensa_dump_stack:  **  THREADPTR (nil)

Backtrace:0xa007f891:0xa00fbd70 0xa006363d:0xa00fbd90 0xa0061dde:0xa00fbdb0 0xa006b2a7:0xa00fbdd0 0xa0070ac1:0xa00fbe00 0xa006d021:0xa00fbe40 0xa006c7f6:0xa00fbe70 0xa003e25c:0xa00fbea0 0xa0060b05:0xa00fbf00 0xa003d555:0xa00fbf30 0xa0061f4a:0xa00fbf60 0xa0061a68:0xa00fbf80 0xa00647f2:0xa00fbfa0 0xa00581c1:0xa00fbfc0 0xa005bcf3:0xa00fbff0 

[00:00:02.039,546] <err> os: z_fatal_error: >>> ZEPHYR FATAL ERROR 0: CPU exception on CPU 0
[00:00:02.039,598] <err> os: z_fatal_error: Current thread: 0x40146998 (unknown)
[00:00:02.058,460] <err> zephyr: k_sys_fatal_error_handler: Halting system
```

kv2019i, 2024-11-27:
Backtrace parsed shows following:

comp_new_ipc4
- copier_init
- copier_ipcgtw_create 
- comp_buffer_connect
- buffer_attach
-> next->prev = item;  /* next is NULL */

Code analysis did not reveal any obvious problems and how the attached item could be NULL. 

It does seem this is not a new bug however, but changes in Zephyr upstream for PTL to make the zero address invalid, may just allow to trap stores with a NULL base address (on other platforms these do not trigger a fault).

tmleman, 2024-11-28:
@kv2019i  fix: https://github.com/thesofproject/sof/pull/9689

tmleman, 2024-11-28:
Next fix: https://github.com/thesofproject/sof/pull/9691

wszypelt, 2024-12-13:
@kv2019i please verify

kv2019i, 2024-12-13:
Bug fixes merged, so this can be closed.

closed 2024-12-13 by kv2019i