thesofproject/sof #9695: [BUG] DSP Panic on Intel MTL

as400l, 2024-11-29:
**Describe the bug**
DSP Panic seen and full freeze of the OS.

**To Reproduce**
Open pavucontrol mute/unmute microphone few times. Close pavucontrol. Wait for freeze.

**Reproduction Rate**
100%

**Expected behavior**
No DSP Panic.

**Impact**
Cannot use builtin microphone.

**Environment**
1) Branch name and commit hash of the 2 repositories: sof (firmware/topology) and linux (kernel driver).
    * Kernel: 6.12.1
    * SOF: sof-bin 2024.09.1
2) Name of the topology file
    * Topology: sof-hda-generic-2ch.tplg
3) Name of the platform(s) on which the bug is observed.
    * Platform: Intel Meteor Lake Ultra 9 185H, Asus Zenbook 14 OLED UX3405M, Alpine Linux

**Screenshots or console output**
```
[  186.448058] sof-audio-pci-intel-mtl 0000:00:1f.3: ------------[ DSP dump start ]------------
[  186.448069] sof-audio-pci-intel-mtl 0000:00:1f.3: DSP panic!
[  186.448071] sof-audio-pci-intel-mtl 0000:00:1f.3: fw_state: SOF_FW_BOOT_COMPLETE (7)
[  186.448078] sof-audio-pci-intel-mtl 0000:00:1f.3: 0x50000005: module: ROM_EXT, state: FW_ENTERED, running
[  186.448083] sof-audio-pci-intel-mtl 0000:00:1f.3: Firmware state: 0x5, status/error code: 0x0
[  186.448116] sof-audio-pci-intel-mtl 0000:00:1f.3: Unknown toolchain is used
[  186.448120] sof-audio-pci-intel-mtl 0000:00:1f.3: error: DSP Firmware Oops
[  186.448121] sof-audio-pci-intel-mtl 0000:00:1f.3: error: Exception Cause: AllocaCause, MOVSP instruction, if caller’s registers are not in the register file
[  186.448123] sof-audio-pci-intel-mtl 0000:00:1f.3: EXCCAUSE 0x00000005 EXCVADDR 0x00000000 PS       0x00060d20 SAR     0x0000000c
[  186.448126] sof-audio-pci-intel-mtl 0000:00:1f.3: EPC1     0xa007626d EPC2     0x00000000 EPC3     0x00000000 EPC4    0x00000000
[  186.448128] sof-audio-pci-intel-mtl 0000:00:1f.3: EPC5     0x00000000 EPC6     0x00000000 EPC7     0x00000000 DEPC    0x00000000
[  186.448129] sof-audio-pci-intel-mtl 0000:00:1f.3: EPS2     0x00000000 EPS3     0x00000000 EPS4     0x00000000 EPS5    0x00000000
[  186.448131] sof-audio-pci-intel-mtl 0000:00:1f.3: EPS6     0x00000000 EPS7     0x00000000 INTENABL 0x00000000 INTERRU 0x00000000
[  186.448132] sof-audio-pci-intel-mtl 0000:00:1f.3: stack dump from 0x00000000
[  186.448134] sof-audio-pci-intel-mtl 0000:00:1f.3: AR registers:
[  186.448136] sof-audio-pci-intel-mtl 0000:00:1f.3: 0x0: a004ed15 a0111680 00000000 4015a7c0
[  186.448138] sof-audio-pci-intel-mtl 0000:00:1f.3: 0x10: a0166b00 00000018 401492b0 a0111680
[  186.448140] sof-audio-pci-intel-mtl 0000:00:1f.3: 0x20: a005fb41 a0111640 401492b0 a006506c
[  186.448142] sof-audio-pci-intel-mtl 0000:00:1f.3: 0x30: a005fb41 a0111640 401492b0 a006506c
[  186.448144] sof-audio-pci-intel-mtl 0000:00:1f.3: ------------[ DSP dump end ]------------
[  186.946817] sof-audio-pci-intel-mtl 0000:00:1f.3: ipc timed out for 0xe030001|0x300
[  186.946837] sof-audio-pci-intel-mtl 0000:00:1f.3: ------------[ IPC dump start ]------------
[  186.946851] sof-audio-pci-intel-mtl 0000:00:1f.3: Host IPC initiator: 0x8e030001|0x300|0x0, target: 0x1b0a0000|0x0|0x0, ctl: 0x3
[  186.946856] sof-audio-pci-intel-mtl 0000:00:1f.3: ------------[ IPC dump end ]------------
[  186.946859] sof-audio-pci-intel-mtl 0000:00:1f.3: IPC timeout
[  186.946866] sof-audio-pci-intel-mtl 0000:00:1f.3: ASoC: error at soc_component_trigger on 0000:00:1f.3: -110
[  186.946878]  HDMI2: ASoC: trigger FE cmd: 1 failed: -110
[  186.946897] sof-audio-pci-intel-mtl 0000:00:1f.3: ipc4_tx_msg_unlocked: ipc message send for 0xe010001|0x0 failed: -19
[  186.946902] sof-audio-pci-intel-mtl 0000:00:1f.3: ASoC: error at soc_component_trigger on 0000:00:1f.3: -19
[  186.946904]  HDMI2: ASoC: trigger FE cmd: 0 failed: -19
[  186.947086] sof-audio-pci-intel-mtl 0000:00:1f.3: ipc4_tx_msg_unlocked: ipc message send for 0x13000003|0x1 failed: -19
[  186.947091] sof-audio-pci-intel-mtl 0000:00:1f.3: failed to pause all pipelines
[  186.947093] sof-audio-pci-intel-mtl 0000:00:1f.3: ASoC: error at soc_component_trigger on 0000:00:1f.3: -19
[  186.947096]  DMIC Raw: ASoC: trigger FE cmd: 0 failed: -19
[  186.947198] sof-audio-pci-intel-mtl 0000:00:1f.3: ipc4_tx_msg_unlocked: ipc message send for 0x46060004|0x19 failed: -19
[  186.947203] sof-audio-pci-intel-mtl 0000:00:1f.3: failed to unbind modules module-copier.12.2:0 -> tdfb.11.1:0
[  186.947208] sof-audio-pci-intel-mtl 0000:00:1f.3: ipc4_tx_msg_unlocked: ipc message send for 0x12040000|0x0 failed: -19
[  186.947212] sof-audio-pci-intel-mtl 0000:00:1f.3: failed to free pipeline widget pipeline.12
[  186.947219] sof-audio-pci-intel-mtl 0000:00:1f.3: ipc4_tx_msg_unlocked: ipc message send for 0x12050000|0x0 failed: -19
[  186.947222] sof-audio-pci-intel-mtl 0000:00:1f.3: failed to free pipeline widget pipeline.11
[  186.947225] sof-audio-pci-intel-mtl 0000:00:1f.3: Failed to free connected widgets
[  186.947233] sof-audio-pci-intel-mtl 0000:00:1f.3: sof_pcm_stream_free: sof_widget_list_free failed -19
[  186.947236] sof-audio-pci-intel-mtl 0000:00:1f.3: ASoC: error at snd_soc_pcm_component_prepare on 0000:00:1f.3: -19
[  186.947240]  DMIC Raw: ASoC: error at __soc_pcm_prepare on DMIC Raw: -19
[  186.947243]  DMIC Raw: ASoC: error at dpcm_fe_dai_prepare on DMIC Raw: -19
[  186.947344] sof-audio-pci-intel-mtl 0000:00:1f.3: ipc4_tx_msg_unlocked: ipc message send for 0x13020003|0x0 failed: -19
[  186.947348] sof-audio-pci-intel-mtl 0000:00:1f.3: ASoC: error at soc_dai_trigger on Analog CPU DAI: -19
[  186.947354]  HDA Analog: ASoC: error at dpcm_be_dai_trigger on HDA Analog: -19
[  186.947357]  HDA Analog: ASoC: trigger FE cmd: 0 failed: -19
```

Full dmesg in attachment.
[dmesg.txt](https://github.com/user-attachments/files/17959577/dmesg.txt)

Please also include the relevant sections from the firmware log and kernel log in the report (and attach the full logs for complete reference). Kernel log is taken from *dmesg* and firmware log from *sof-logger*. See https://thesofproject.github.io/latest/developer_guides/debugability/logger/index.html

lgirdwood, 2024-11-29:
@as400l are you able to see this with alsamixer ? and if so which DMIC Kcontrol ?
@ujfalusi any additional kernel debug options to enable ?

ujfalusi, 2024-11-29:
As usual, @as400l:
Can you add this file [sof-dyndbg.conf.txt](https://github.com/user-attachments/files/16927209/sof-dyndbg.conf.txt)
as /etc/modprobe.d/sof-dyndbg.conf, reboot and re-attach the dmesg log which contains the boot and the error itself?

In case the log is truncated because of a small log buffer, please add `log_buf_len=4M` to the kernel command line parameter (passed by the bootloader to the kernel)

as400l, 2024-11-29:
Here is dmesg with the error and sof-dyndbg.conf enabled.

BTW - isn't it strange that it uses sof-hda-generic-2ch.tplg file ?

[dmesg.log.gz](https://github.com/user-attachments/files/17961014/dmesg.log.gz)

ujfalusi, 2024-11-29:
@as400l, for some reason the dyndbg did not enabled the debug prints, we don't see what was the last message that was sent to the firmware, we know that the next would have been 0xe010002|0x0, which is not sent as the firmware has crashed.
Can you check again if the dyndbg is in place? The probing should be much more verbose with lots of prints about modules and stuff.

sof-hda-generic-2ch.tplg is chosen, because you have DMIC in your system
```
[   15.097047] sof-audio-pci-intel-mtl 0000:00:1f.3: DMICs detected in NHLT tables: 2
```
you also have BT offload advertised:
```
[   15.097044] sof-audio-pci-intel-mtl 0000:00:1f.3: NHLT device BT(0) detected, ssp_mask 0x4
[   15.097046] sof-audio-pci-intel-mtl 0000:00:1f.3: BT link detected in NHLT tables: 0x4
```
I'm not sure if that can cause any issues.

You can disable the dmic for testing the analog path (you will loose the laptop microphones) :
```
options snd_sof_intel_hda_generic dyndbg=+pmf dmic_num=0
```
in for example /etc/modprobe.d/no-dmic.conf

as400l, 2024-11-29:
I tried multiple times with "wpctl set-mute @DEFAULT_AUDIO_SOURCE@ toggle". But could not reproduce this behaviour.

So maybe the real cause of this is actually XE drm module crash or hang related to pavucontrol ? Which may be seen at the end of dmesg I've sent ? Is this even possible ?

As to the debug prints. My kernel may is really slimmed down. So that may be the reason. May have to try with default distro kernel.

lgirdwood, 2024-12-02:
> Here is dmesg with the error and sof-dyndbg.conf enabled.
> 
> BTW - isn't it strange that it uses sof-hda-generic-2ch.tplg file ?
> 
> @lgirdwood - I tried with alsamixer but can't reproduce it. But, on the other hand, with alsamixer I can't unmute the mic. I have this LED on keyboard and no matter what I tried with alsamixer it's just constantly on. Which means that the mic was not unmuted.
> 
> [dmesg.log.gz](https://github.com/user-attachments/files/17961014/dmesg.log.gz)

Ok, its strange that alsamixer wont unmute the mic, I assume you tried ```alsamixer -c N``` (where N is card number) to make sure all kcontrols have been tried.

Btw, is the keyboard LED on a key ? i.e. can it be pressed with Fn/Alt/Ctrl/shift combinations to switch LED on/off ? This should be mapped to the kcontrol that will mute/unmute the mic.

Please do try the stock kernel. We need to figure out what has happened here with stock kernel logs.

as400l, 2024-12-02:
@lgirdwood - as I mentioned above - I tried with "wpctl" and it correctly mutes/unmutes microphone. LED goes off/on as it should. But I could not reproduce this error.

I'm leaning towards something else causing this panic.

Stock Alpine kernel was also not helpful since it's probably also stripped. Let me try with loglevel=7.

ujfalusi, 2024-12-03:
Based on the log I think it is the ChainDMA (HDMI audio) which is causing the firmware panic:
```
[  313.391960] snd_sof:sof_pcm_trigger: sof-audio-pci-intel-mtl 0000:00:1f.3: pcm: trigger stream 4 dir 0 cmd 1
[  313.391963] snd_sof:sof_ipc4_trigger_pipelines: sof-audio-pci-intel-mtl 0000:00:1f.3: trigger cmd: 1 state: 4
[  313.391966] snd_sof:sof_ipc4_log_header: sof-audio-pci-intel-mtl 0000:00:1f.3: ipc tx      : 0xe030001|0x300
[  313.393682] snd_sof:sof_ipc4_log_header: sof-audio-pci-intel-mtl 0000:00:1f.3: ipc rx      : 0x1b0a0000|0x0
[  313.393687] sof-audio-pci-intel-mtl 0000:00:1f.3: ------------[ DSP dump start ]------------
[  313.393689] sof-audio-pci-intel-mtl 0000:00:1f.3: DSP panic!
[  313.393691] sof-audio-pci-intel-mtl 0000:00:1f.3: fw_state: SOF_FW_BOOT_COMPLETE (7)
[  313.393695] sof-audio-pci-intel-mtl 0000:00:1f.3: 0x50000005: module: ROM_EXT, state: FW_ENTERED, running
[  313.393700] sof-audio-pci-intel-mtl 0000:00:1f.3: Firmware state: 0x5, status/error code: 0x0
[  313.393733] sof-audio-pci-intel-mtl 0000:00:1f.3: Unknown toolchain is used
[  313.393735] sof-audio-pci-intel-mtl 0000:00:1f.3: error: DSP Firmware Oops
[  313.393737] sof-audio-pci-intel-mtl 0000:00:1f.3: error: Exception Cause: AllocaCause, MOVSP instruction, if caller’s registers are not in the register file
[  313.393741] sof-audio-pci-intel-mtl 0000:00:1f.3: EXCCAUSE 0x00000005 EXCVADDR 0x00000000 PS       0x00060d20 SAR     0x0000000c
[  313.393745] sof-audio-pci-intel-mtl 0000:00:1f.3: EPC1     0xa007626d EPC2     0x00000000 EPC3     0x00000000 EPC4    0x00000000
[  313.393748] sof-audio-pci-intel-mtl 0000:00:1f.3: EPC5     0x00000000 EPC6     0x00000000 EPC7     0x00000000 DEPC    0x00000000
[  313.393750] sof-audio-pci-intel-mtl 0000:00:1f.3: EPS2     0x00000000 EPS3     0x00000000 EPS4     0x00000000 EPS5    0x00000000
[  313.393752] sof-audio-pci-intel-mtl 0000:00:1f.3: EPS6     0x00000000 EPS7     0x00000000 INTENABL 0x00000000 INTERRU 0x00000000
[  313.393754] sof-audio-pci-intel-mtl 0000:00:1f.3: stack dump from 0x00000000
[  313.393756] sof-audio-pci-intel-mtl 0000:00:1f.3: AR registers:
[  313.393759] sof-audio-pci-intel-mtl 0000:00:1f.3: 0x0: a004ed15 a0111680 00000000 40152c80
[  313.393762] sof-audio-pci-intel-mtl 0000:00:1f.3: 0x10: a0166740 00000018 40149740 a0111680
[  313.393764] sof-audio-pci-intel-mtl 0000:00:1f.3: 0x20: a005fb41 a0111640 40149740 a006506c
[  313.393766] sof-audio-pci-intel-mtl 0000:00:1f.3: 0x30: a005fb41 a0111640 40149740 a006506c
[  313.393768] sof-audio-pci-intel-mtl 0000:00:1f.3: ------------[ DSP dump end ]------------
[  313.393770] snd_sof:sof_set_fw_state: sof-audio-pci-intel-mtl 0000:00:1f.3: fw_state change: 7 -> 8
[  313.393774] snd_sof:sof_ipc4_log_header: sof-audio-pci-intel-mtl 0000:00:1f.3: ipc rx done : 0x1b0a0000|0x0
[  313.898625] sof-audio-pci-intel-mtl 0000:00:1f.3: ipc timed out for 0xe030001|0x300
[  313.898645] sof-audio-pci-intel-mtl 0000:00:1f.3: ------------[ IPC dump start ]------------
[  313.898657] sof-audio-pci-intel-mtl 0000:00:1f.3: Host IPC initiator: 0x8e030001|0x300|0x0, target: 0x1b0a0000|0x0|0x0, ctl: 0x3
[  313.898661] sof-audio-pci-intel-mtl 0000:00:1f.3: ------------[ IPC dump end ]------------
[  313.898663] sof-audio-pci-intel-mtl 0000:00:1f.3: IPC timeout
```
0xe030001 is ChainDMA with ALLOCATE and ENABLE bit set, but what is not right is that the Host DMA ID is 1 while the Link DMA ID is 0.
We had similar issue in past (https://github.com/thesofproject/linux/issues/5116) which supposed to be fixed by https://github.com/thesofproject/linux/pull/5119.

There are lots of things happening in the log, but looks like something (PW?) is trying PCMs at random keeping them open and stopping, starting, reconfiguring them.

as400l, 2024-12-03:
@ujfalusi - just to remind - this happens only while using pavucontrol which is actually PulseAudio tool.
I could not reproduce this while using native WirePlumber tool - wpctl.

ujfalusi, 2024-12-03:
OK, so to reproduce the issue:
aplay -Dhw:0,3 -c8 -r48000 -fS32_LE /dev/zero -d 120
```
[ 2810.282081] snd_sof:sof_pcm_trigger: sof-audio-pci-intel-tgl 0000:00:1f.3: pcm3 (HDMI1), dir 0: Entry: trigger (cmd: 1)
[ 2810.282087] snd_sof:sof_ipc4_trigger_pipelines: sof-audio-pci-intel-tgl 0000:00:1f.3: pcm3 (HDMI1), dir 0: cmd: 1, state: 4
[ 2810.282093] snd_sof:sof_ipc4_log_header: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx      : 0xe030000|0xc00: GLB_CHAIN_DMA
[ 2810.282656] snd_sof:sof_ipc4_log_header: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx reply: 0x2e000000|0xc00: GLB_CHAIN_DMA
[ 2810.282692] snd_sof:sof_ipc4_log_header: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx done : 0xe030000|0xc00: GLB_CHAIN_DMA
[ 2810.283232] snd_sof_intel_hda_common:hda_dsp_stream_trigger: sof-audio-pci-intel-tgl 0000:00:1f.3: FW Poll Status: reg[0x160]=0x2014001e successful
```
Press <`CTRL+z`> to freeze aplay
```
[ 2814.029625] snd_sof:sof_pcm_trigger: sof-audio-pci-intel-tgl 0000:00:1f.3: pcm3 (HDMI1), dir 0: Entry: trigger (cmd: 0)
[ 2814.029633] snd_sof:sof_ipc4_trigger_pipelines: sof-audio-pci-intel-tgl 0000:00:1f.3: pcm3 (HDMI1), dir 0: cmd: 0, state: 3
[ 2814.029645] snd_sof:sof_ipc4_log_header: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx      : 0xe010000|0x0: GLB_CHAIN_DMA
[ 2814.030855] snd_sof:sof_ipc4_log_header: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx reply: 0x2e000000|0x0: GLB_CHAIN_DMA
[ 2814.031022] snd_sof:sof_ipc4_log_header: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx done : 0xe010000|0x0: GLB_CHAIN_DMA
[ 2814.031034] snd_soc_core:dpcm_be_dai_trigger:  iDisp1: ASoC: trigger BE iDisp1 cmd 0
[ 2814.031045] snd_sof_intel_hda_common:hda_dai_trigger: sof-audio-pci-intel-tgl 0000:00:1f.3: cmd=0 dai iDisp1 Pin direction 0
```
wait a sec or two then start a new HDMI playback (while the :0,3 is frozen):
aplay -Dhw:0,4 -c8 -r48000 -fS32_LE /dev/zero -d 120

```
[ 2823.354025] snd_sof:sof_ipc4_trigger_pipelines: sof-audio-pci-intel-tgl 0000:00:1f.3: pcm4 (HDMI2), dir 0: cmd: 1, state: 4
[ 2823.354033] snd_sof:sof_ipc4_log_header: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx      : 0xe030001|0xc00: GLB_CHAIN_DMA
[ 2823.357361] snd_sof:sof_ipc4_log_header: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc rx      : 0x1b0a0000|0x0: GLB_NOTIFICATION|EXCEPTION_CAUGHT
[ 2823.357367] sof-audio-pci-intel-tgl 0000:00:1f.3: ------------[ DSP dump start ]------------
[ 2823.357370] sof-audio-pci-intel-tgl 0000:00:1f.3: DSP panic!
[ 2823.357373] sof-audio-pci-intel-tgl 0000:00:1f.3: fw_state: SOF_FW_BOOT_COMPLETE (7)
[ 2823.357381] sof-audio-pci-intel-tgl 0000:00:1f.3: 0x00000005: module: ROM, state: FW_ENTERED, running
[ 2823.357490] sof-audio-pci-intel-tgl 0000:00:1f.3: FW is built with Zephyr toolchain
[ 2823.357493] sof-audio-pci-intel-tgl 0000:00:1f.3: error: DSP Firmware Oops
[ 2823.357496] sof-audio-pci-intel-tgl 0000:00:1f.3: error: Exception Cause: AllocaCause, MOVSP instruction, if caller’s registers are not in the register file
[ 2823.357499] sof-audio-pci-intel-tgl 0000:00:1f.3: EXCCAUSE 0x00000005 EXCVADDR 0x00000000 PS       0x00060f20 SAR     0x0000001d
[ 2823.357503] sof-audio-pci-intel-tgl 0000:00:1f.3: EPC1     0xbe04126c EPC2     0x00000000 EPC3     0x00000000 EPC4    0x00000000
[ 2823.357507] sof-audio-pci-intel-tgl 0000:00:1f.3: EPC5     0x00000000 EPC6     0x00000000 EPC7     0x00000000 DEPC    0x00000000
[ 2823.357510] sof-audio-pci-intel-tgl 0000:00:1f.3: EPS2     0x00000000 EPS3     0x00000000 EPS4     0x00000000 EPS5    0x00000000
[ 2823.357513] sof-audio-pci-intel-tgl 0000:00:1f.3: EPS6     0x00000000 EPS7     0x00000000 INTENABL 0x00000000 INTERRU 0x00000000
[ 2823.357515] sof-audio-pci-intel-tgl 0000:00:1f.3: stack dump from 0x00000000
[ 2823.357518] sof-audio-pci-intel-tgl 0000:00:1f.3: AR registers:
[ 2823.357521] sof-audio-pci-intel-tgl 0000:00:1f.3: 0x0: be04156b be0a2eb0 9e0b1700 be0b17c0
[ 2823.357525] sof-audio-pci-intel-tgl 0000:00:1f.3: 0x10: fff001ff 00000000 00003000 be0a2eb0
[ 2823.357540] sof-audio-pci-intel-tgl 0000:00:1f.3: 0x20: 00000000 be0a2e90 9e0a8630 00060f25
[ 2823.357546] sof-audio-pci-intel-tgl 0000:00:1f.3: 0x30: 00000000 be0a2e90 9e0a8630 00060f25
[ 2823.357551] sof-audio-pci-intel-tgl 0000:00:1f.3: ------------[ DSP dump end ]------------
```

ujfalusi, 2024-12-03:
Reverting `7eab5d86f218 ("ASoC: SOF: Intel: hda: Always clean up link DMA during stop")` is fixing this particular issue.

That patch was part of https://github.com/thesofproject/linux/pull/5197, which was fixing various metallic noise issues around similar sequences.

The issue is not limited to ChainDMA
TGL HDA machine will fail:
```
aplay -Dhw:0,3 -c2 -r48000 -fS32_LE /dev/zero -d 120
CTRL+z
aplay -Dhw:0,0 -c2 -r48000 -fS32_LE /dev/zero -d 120
```
or will cause fw panic:
```
aplay -Dhw:0,0 -c2 -r48000 -fS32_LE /dev/zero -d 120
CTRL+z
aplay -Dhw:0,3 -c2 -r48000 -fS32_LE /dev/zero -d 120
```
On LNL sdw it is the same with all endpoints:
```
aplay -Dhw:0,0 -c2 -r48000 -fS32_LE /dev/zero -d 120
CTRL+z
aplay -Dhw:0,2 -c2 -r48000 -fS32_LE /dev/zero -d 120
```
Only the ChainDMA PCMs will cause panic, others will fail.

@ranj063, I think it might be because we release the LinkDMA channel in sof/intel/ but we don't inform the firmware about this (we don't do a full stop) and this is causing a race if a new PCM comes in between the stop and the prepare/hw_params/start of the other PCM.

lgirdwood, 2024-12-09:
@abonislawski fyi - for FW panic, it probably worth checking if this panic is due to HW state transition (fixed above in SW) and if it needs a FW fix too. Thanks !

closed 2024-12-13 by ranj063

lvanderree, 2024-12-30:
I also had this bug on my Lenovo X1 Carbon gen 12 with Build in Audio (Meteor Lake-P HD)
Running Fedora 41 up till kernel 6.12.7-200
When using WebEx (44.10.2) 

I could reproduce this by start-stopping the Audio-Test in WebEx-settings multiple times after each other

I Installed the experimental kernel 6.13.0-0.rc4.36.fc42.x86_64 (hoping this work was already merged), but unfortunately this didn't fix the problem.
Then I compiled the kernel with this source (517a41a6df6ac78e7a8da35856531ad432 (HEAD -> topic/sof-dev), 7aae4a3729e4627ca25bddba2023eb7f9afc95a4 (HEAD -> rawhide)) and I can confirm this FIXED the DSP panic

Thanks!

ujfalusi, 2025-01-02:
@lvanderree, 6.13-rc4 should include the patch and should be working (6.12.8 will also have the backport).

I wonder what are we missing... Let me test 6.13-rc5 here.

ujfalusi, 2025-01-02:
> @lvanderree, 6.13-rc4 should include the patch and should be working (6.12.8 will also have the backport).
> 
> I wonder what are we missing... Let me test 6.13-rc5 here.

6.13-rc5 is working fine and indeed the patch is first available in -rc5, -rc4 should still fail, @lvanderree, can you re-test with -rc5 if you have time to confirm that you no longer need to do git merges to get things working?
```
e8d0ba147d90 ASoC: SOF: Intel: hda-dai: Do not release the link DMA on STOP
```

lvanderree, 2025-01-02:
Unfortunately I immediately was able to reproduce an audio problem with WebEx under kernel **6.13.0-0.rc5.42.fc42.x86_64**
When play-pausing the test-tune under WebEx Settings it stopped all audio-output (originally playing over the internal-speakers)
However it didn't kernel panic this time, only showing broken pipes in journalctl

> jan 02 16:09:22 X1-Leon pipewire[4892]: spa.alsa: hw:sofhdadspp: (0 suppressed) snd_pcm_avail after recover: Gebroken pijp
> jan 02 16:09:28 X1-Leon pipewire[4892]: spa.alsa: hw:sofhdadspp: (375 suppressed) snd_pcm_avail after recover: Gebroken pijp
> jan 02 16:09:30 X1-Leon gnome-shell[5105]: JS ERROR: Error: Expected an object of type GvcMixerStream for argument 'stream' but got type undefined
>                                            readline_callback@file:///home/leon/.local/share/gnome-shell/extensions/quick-settings-audio-panel@rayzeq.github.io/libs/widgets.js:462:51
>                                            @resource:///org/gnome/shell/ui/init.js:21:20
> jan 02 16:09:35 X1-Leon pipewire[4892]: spa.alsa: hw:sofhdadspp: (375 suppressed) snd_pcm_avail after recover: Gebroken pijp
> jan 02 16:09:42 X1-Leon pipewire[4892]: spa.alsa: hw:sofhdadspp: (375 suppressed) snd_pcm_avail after recover: Gebroken pijp
> jan 02 16:09:49 X1-Leon pipewire[4892]: spa.alsa: hw:sofhdadspp: (375 suppressed) snd_pcm_avail after recover: Gebroken pijp
> jan 02 16:09:56 X1-Leon pipewire[4892]: spa.alsa: hw:sofhdadspp: (375 suppressed) snd_pcm_avail after recover: Gebroken pijp
> jan 02 16:10:03 X1-Leon pipewire[4892]: spa.alsa: hw:sofhdadspp: (375 suppressed) snd_pcm_avail after recover: Gebroken pijp
> jan 02 16:10:09 X1-Leon pipewire[4892]: spa.alsa: hw:sofhdadspp: (375 suppressed) snd_pcm_avail after recover: Gebroken pijp
> jan 02 16:10:16 X1-Leon pipewire[4892]: spa.alsa: hw:sofhdadspp: (375 suppressed) snd_pcm_avail after recover: Gebroken pijp
> jan 02 16:10:23 X1-Leon pipewire[4892]: spa.alsa: hw:sofhdadspp: (375 suppressed) snd_pcm_avail after recover: Gebroken pijp
> jan 02 16:10:30 X1-Leon pipewire[4892]: spa.alsa: hw:sofhdadspp: (375 suppressed) snd_pcm_avail after recover: Gebroken pijp
> jan 02 16:10:37 X1-Leon pipewire[4892]: spa.alsa: hw:sofhdadspp: (375 suppressed) snd_pcm_avail after recover: Gebroken pijp
> jan 02 16:10:44 X1-Leon pipewire[4892]: spa.alsa: hw:sofhdadspp: (375 suppressed) snd_pcm_avail after recover: Gebroken pijp
> jan 02 16:10:46 X1-Leon pipewire[4892]: spa.alsa: hw:sofhdadspp: (127 suppressed) snd_pcm_avail after recover: Gebroken pijp
> jan 02 16:10:57 X1-Leon systemd[4874]: app-gnome-webex-17064.scope: Couldn't move process 17064 to requested cgroup '/user.slice/user-1000.slice/user@1000.service/app.slice/app-gnome-webex-17064.scope': No such process
> jan 02 16:10:57 X1-Leon systemd[4874]: app-gnome-webex-17064.scope: Failed to add PIDs to scope's control group: No such process
> jan 02 16:10:57 X1-Leon systemd[4874]: app-gnome-webex-17064.scope: Failed with result 'resources'.
> jan 02 16:10:57 X1-Leon systemd[4874]: Failed to start app-gnome-webex-17064.scope - Application launched by gnome-shell.
> jan 02 16:11:01 X1-Leon systemd[4874]: app-gnome-webex-17111.scope: Couldn't move process 17111 to requested cgroup '/user.slice/user-1000.slice/user@1000.service/app.slice/app-gnome-webex-17111.scope': No such process
> jan 02 16:11:01 X1-Leon systemd[4874]: app-gnome-webex-17111.scope: Failed to add PIDs to scope's control group: No such process
> jan 02 16:11:01 X1-Leon systemd[4874]: app-gnome-webex-17111.scope: Failed with result 'resources'.
> jan 02 16:11:01 X1-Leon systemd[4874]: Failed to start app-gnome-webex-17111.scope - Application launched by gnome-shell.
> jan 02 16:11:02 X1-Leon /usr/libexec/gdm-x-session[4945]: (II) modeset(0): EDID vendor "BOE", prod id 3072
> jan 02 16:11:02 X1-Leon /usr/libexec/gdm-x-session[4945]: (II) modeset(0): Using hsync ranges from config file
> jan 02 16:11:02 X1-Leon /usr/libexec/gdm-x-session[4945]: (II) modeset(0): Using vrefresh ranges from config file
> jan 02 16:11:02 X1-Leon /usr/libexec/gdm-x-session[4945]: (II) modeset(0): Printing DDC gathered Modelines:
> jan 02 16:11:02 X1-Leon /usr/libexec/gdm-x-session[4945]: (II) modeset(0): Modeline "1920x1200"x0.0  154.76  1920 1968 2000 2080  1200 1203 1209 1240 +hsync -vsync (74.4 kHz eP)
> jan 02 16:11:09 X1-Leon pipewire[4892]: pw.node: (alsa_input.pci-0000_00_1f.3-platform-skl_hda_dsp_generic.HiFi__Mic1__source-63) graph xrun not-triggered (49 suppressed)
> jan 02 16:11:09 X1-Leon pipewire[4892]: pw.node: (alsa_input.pci-0000_00_1f.3-platform-skl_hda_dsp_generic.HiFi__Mic1__source-63) xrun state:0x7fc776434008 pending:1/2 s:1164561764670 a:1164561796505 f:1164561798193 waiting:31835 process:1688 status:triggered
> 

Besides that I didn't have any audio when setting output over HDMI (1).  Spotify would play over the Internal speakers, but when switching to HDMI(1) I couldn't hear anything anymore. Switching back to Internal and music was back.
I could unplug and replug my screen (USB-C). But this didn't help.
The only thing that resolved this was sleeping my system, and awake. Than audio over HDMI (1) worked again.
I don' t think however that this (no audio over HDMI until sleep-awake) is related to the original issue (DSP Panic).

However after rebooting again with my custom kernel **6.13.0-rc3-sof** I could reproduce both problems (crashing audio when testing with WebEx, and no audio over HDMI1 until sleep/wake-up), I don' t know why I didn't  [... cut ...]
> jan 02 16:42:32 X1-Leon pipewire[4892]: spa.alsa: hw:sofhdadsp,3p: (375 suppressed) snd_pcm_avail after recover: Gebroken pijp
> jan 02 16:42:34 X1-Leon pipewire[4892]: spa.alsa: hw:sofhdadsp,3p: (375 suppressed) snd_pcm_avail after recover: Gebroken pijp
> 

If there is anything else I can do to provide more insights please say so (and maybe point me in the right direction how to get this info). Or if this is not directly related to the original  DSP Panic, and release DMA solution, but I should open a new ticket I can do that as well.

carlinigraphy, 2025-01-02:
I am also experiencing this issue. Also on a 12th Gen X1 Carbon.

If you need someone else to test--please let me know.

ujfalusi, 2025-01-03:
@carlinigraphy, also triggered by WebEx?

ujfalusi, 2025-01-03:
@lvanderree, can you clarify these:
- with 6.13-rc5 there is no DSP panic
- with topic/sof-dev kernel there is no DSP panic
- With both kernel when using WebEx you experience loss of audio which can be fixed by suspend/resume

When the audio broke, can it be resurrected by stopping all audio activities and letting PA/PW to close the PCM devices?
This can be checked with `cat /proc/asound/card0/pcm*/sub0/status` (or better `for pcm in /proc/asound/card*/pcm*; do echo ${pcm}; cat ${pcm}/sub0/status; done`) if it returns only `closed` then there is no audio activity going on and the card should power of and in theory audio would work?

ujfalusi, 2025-01-03:
@lvanderree, @carlinigraphy, please follow https://github.com/thesofproject/sof/issues/9695#issuecomment-2507983475 and provide raw dmesg log without filtering which contains the events around the time the issue happens, but as long as possible to see the prior events.

If you don't see the DSP panic with 6.13-rc5 / 6.12.8 / sof-dev kernel then please open a new issue to track this, thank you.

lvanderree, 2025-01-03:
> @lvanderree, can you clarify these:
 
* with 6.13-rc5 there is no DSP panic
* with topic/sof-dev kernel there is no DSP panic
* with both kernel when using WebEx you experience loss of audio (also of alternative sources playing at the same time), which can be fixed by closing the WebEx stream
*  Audio out via HDMI (1) does not work, until after a suspend/resume

> When the audio broke, can it be resurrected by stopping all audio activities and letting PA/PW to close the PCM devices? This can be checked with `cat /proc/asound/card0/pcm*/sub0/status` (or better [...]

When I play music (Spotify) and then start/stop/start/stop/start... the audio-test in WebEx I can break alle audio. I let Spotify continue playing, but close the (already stopped test) WebEx settings-window, letting WebEx itself continu to run, and the music starts playing again.

**So in short** , I don't even have to close all audio activity to restore audio. Only closing the Web-Ex streams seem to be enough.

_I haven' t had time to reboot with the [sof-dyndbg.conf.txt](https://github.com/user-attachments/files/16927209/sof-dyndbg.conf.txt) config, but I can do that in a new ticket, which probably is more suitable than continue in this thread._

### Some debug results:

When I've closed both Spotify and WebEx no PCM streams are opened:
```bash
for pcm in /proc/asound/card*/pcm*; do echo ${pcm}; cat ${pcm}/sub0/status; done
/proc/asound/card0/pcm0c
closed
/proc/asound/card0/pcm0p
closed
/proc/asound/card0/pcm31p
closed
/proc/asound/card0/pcm3p
closed
/proc/asound/card0/pcm4p
closed
/proc/asound/card0/pcm5p
closed
/proc/asound/card0/pcm6c
closed
```

and after starting both (Spotify and WebEx), but not starting sound yet, the PCM streams remain closed.

Then when playing only in Spotify over the internal speaker I see:

```bash
for pcm in /proc/asound/card*/pcm*; do echo ${pcm}; cat ${pcm}/sub0/status; done
/proc/asound/card0/pcm0c
closed
/proc/asound/card0/pcm0p
state: RUNNING
owner_pid   : 3530
trigger_time: 10541.693975450
tstamp      : 10576.476283753
delay       : 3702
avail       : 29280
avail_max   : 32304
-----
hw_ptr      : 1669728
appl_ptr    : 1673216
/proc/asound/card0/pcm31p
closed
/proc/asound/card0/pcm3p
closed
/proc/asound/card0/pcm4p
closed
/proc/asound/card0/pcm5p
closed
/proc/asound/card0/pcm6c
closed
```

Then opening the Audio-settings in WebEx, while Spofity remains playing, but without starting tests I see:
```bash
for pcm in /proc/asound/card*/pcm*; do echo ${pcm}; cat ${pcm}/sub0/status; done
[... 271 lines cut ...]
for pcm in /proc/asound/card*/pcm*; do echo ${pcm}; cat ${pcm}/sub0/status; done
/proc/asound/card0/pcm0c
closed
/proc/asound/card0/pcm0p
closed
/proc/asound/card0/pcm31p
closed
/proc/asound/card0/pcm3p
state: RUNNING
owner_pid   : 3530
trigger_time: 11661.147549638
tstamp      : 11665.175596114
delay       : 2988
avail       : 29780
avail_max   : 32732
-----
hw_ptr      : 193620
appl_ptr    : 196608
/proc/asound/card0/pcm4p
closed
/proc/asound/card0/pcm5p
closed
/proc/asound/card0/pcm6c
closed
```

ujfalusi, 2025-01-03:
@lvanderree, thank you for the details!
In all cases WebEx is involved if I read it right, have you checked if it happens also if you run WebEx in browser instead of the application?

I don't have WebEx account and it looks like that the configuration is account-walled.

That is really surprising that WebEx using analog out+DMIC can break HDMI playback...
Can you check the PCM states
1) when WebEx test is strated
2) when WebEx test is stopped

What is even more strange is that when you play audio to speaker and start WebEx (which will use the same PCM device) it will break.
All audio is going through PA/PW audio servers and it is the only entity which allowed to use the hardware, adding, removing new streams must not have any effect on the opened stream, PA/PW is doing the mixing and keeps the audio device running.

Unless WebEx takes over the audio somehow...

ujfalusi, 2025-01-03:
OK, I can reproduce this with WebEx now, wow.

lvanderree, 2025-01-03:
For now WebEx was the only app I've experienced issues with. Good to hear you are able to install it as well, and can enjoy this wonderful product as well :crossed_fingers: 
I hadn't thought about using their web-app yet, but tried it out right now, and the web app didn't seem to have any troubles, even though they seem to have a extensive web implementation as well, capable of iterating over all my devices. 

The web app doesn't have a settings window, so I cannot test the settings via the web app. However I made a call via web to the native (linux) app and that did made my audio system crash (with broken pipes) again. But as soon as I ended the app-side my spotify sound restored again.

So it definitely is some magic the app does. I can partly follow what you say and that indeed sounds surprising that they somehow interfere with PA/PW. Please let me know what else I can do, now you can test and see for yourself as well.

vanushwashere, 2025-01-08:
I've got the same freezes on my laptop:
OS: KDE neon 6.2 x86_64 
Host: 21MA002XRT ThinkPad E16 Gen 2 
Kernel: 6.8.0-51-generic 

```
2025-01-08T00:08:01.169603+04:00 mangata kernel: sof-audio-pci-intel-mtl 0000:00:1f.3: ASoC: error at soc_dai_trigger on Analog CPU DAI: -22                                                                                              │
│2025-01-08T00:08:01.169613+04:00 mangata kernel:  HDA Analog: ASoC: trigger FE cmd: 1 failed: -22                                                                                                                                         │
│2025-01-08T00:08:01.169614+04:00 mangata kernel: sof-audio-pci-intel-mtl 0000:00:1f.3: ASoC: error at soc_dai_trigger on Analog CPU DAI: -22                                                                                              │
│2025-01-08T00:08:01.169615+04:00 mangata kernel:  HDA Analog: ASoC: trigger FE cmd: 1 failed: -22                                                                                                                                         │
│2025-01-08T00:08:01.169616+04:00 mangata kernel: sof-audio-pci-intel-mtl 0000:00:1f.3: ASoC: error at soc_dai_trigger on Analog CPU DAI: -22                                                                                              │
│2025-01-08T00:08:01.169617+04:00 mangata kernel:  HDA Analog: ASoC: trigger FE cmd: 1 failed: -22                                                                                                                                         │
│2025-01-08T00:08:01.169628+04:00 mangata kernel: sof-audio-pci-intel-mtl 0000:00:1f.3: ASoC: error at soc_dai_trigger on Analog CPU DAI: -22                                                                                              │
│2025-01-08T00:08:01.169629+04:00 mangata kernel:  HDA Analog: ASoC: trigger FE cmd: 1 failed: -22                                                                                                                                         │
│2025-01-08T00:08:01.169630+04:00 mangata kernel: sof-audio-pci-intel-mtl 0000:00:1f.3: ASoC: error at soc_dai_trigger on Analog CPU DAI: -22                                                                                              │
│2025-01-08T00:08:01.169631+04:00 mangata kernel:  HDA Analog: ASoC: trigger FE cmd: 1 failed: -22                                                                                                                                         │
│2025-01-08T00:08:01.169632+04:00 mangata kernel: sof-audio-pci-intel-mtl 0000:00:1f.3: ASoC: error at soc_dai_trigger on Analog CPU DAI: -22                                                                                              │
│2025-01-08T00:08:01.169633+04:00 mangata kernel:  HDA Analog: ASoC: trigger FE cmd: 1 failed: -22                                                                                                                                         │
│2025-01-08T00:08:01.169634+04:00 mangata kernel: sof-audio-pci-intel-mtl 0000:00:1f.3: ASoC: error at soc_dai_trigger on Analog CPU DAI: -22                                                                                              │
│2025-01-08T00:08:01.169634+04:00 mangata kernel:  HDA Analog: ASoC: trigger FE cmd: 1 failed: -22                                                                                                                                         │
│2025-01-08T00:08:01.169635+04:00 mangata kernel: sof-audio-pci-intel-mtl 0000:00:1f.3: ASoC: error at soc_dai_trigger on Analog CPU DAI: -22                                                                                              │
│2025-01-08T00:08:01.169636+04:00 mangata kernel:  HDA Analog: ASoC: trigger FE cmd: 1 failed: -22                                                                                                                                         │
│2025-01-08T00:08:01.169636+04:00 mangata kernel: sof-audio-pci-intel-mtl 0000:00:1f.3: ASoC: error at soc_dai_trigger on Analog CPU DAI: -22                                                                                              │
│2025-01-08T00:08:01.169637+04:00 mangata kernel:  HDA Analog: ASoC: trigger FE cmd: 1 failed: -22                                                                                                                                         │
│2025-01-08T00:08:01.169638+04:00 mangata kernel: sof-audio-pci-intel-mtl 0000:00:1f.3: ASoC: error at soc_dai_trigger on Analog CPU DAI: -22                                                                                              │
│2025-01-08T00:08:01.169639+04:00 mangata kernel:  HDA Analog: ASoC: trigger FE cmd: 1 failed: -22                                                                                                                                         │
│2025-01-08T00:08:01.169640+04:00 mangata kernel: sof-audio-pci-intel-mtl 0000:00:1f.3: ASoC: error at soc_dai_trigger on Analog CPU DAI: -22                                                                                              │
│2025-01-08T00:08:01.169640+04:00 mangata kernel:  HDA Analog: ASoC: trigger FE cmd: 1 failed: -22                                                                                                                                         │
[... cut ...]
│2025-01-08T00:08:01.169663+04:00 mangata kernel:  HDA Analog: ASoC: trigger FE cmd: 1 failed: -22                                                                                                                                         │
│2025-01-08T00:08:01.169664+04:00 mangata kernel: sof-audio-pci-intel-mtl 0000:00:1f.3: ASoC: error at soc_dai_trigger on Analog CPU DAI: -22                                                                                              │
│2025-01-08T00:08:01.169665+04:00 mangata kernel:  HDA Analog: ASoC: trigger FE cmd: 1 failed: -22                                                  
```

vanushwashere, 2025-01-08:
Changing to pulseaudio didn't help with the problem for me, I am still getting that errors and experience freezes

ujfalusi, 2025-01-08:
@vanushwashere, I think your issue is different and unrelated but I cannot see what goes wrong.
Can you follow https://github.com/thesofproject/sof/issues/9695#issuecomment-2507983475 and open a new issue with the full kernel log attached?

Thank you.

vanushwashere, 2025-01-08:
It just spams that messages to log every millisecond :/ 
I will open a new issue later