thesofproject/sof #9699: [BUG] Problem with external DMIC on Lenovo ThinkBook 16 G7 IML

SebO-L, 2024-12-01:
**Describe the bug**
Hi all, 

I have a issue with my new laptop, a lenovo Thinkbook 16 G7 IML, Meteor lake proc.

The sound is working either the mic when i plug an headphone, but the external mic is not working at all.
I tried to follow a lot of threads like with one https://github.com/thesofproject/linux/issues/2460 without success.

dmesg sof-part
```
[   10.338963] sof-audio-pci-intel-mtl 0000:00:1f.3: enabling device (0000 -> 0002)
[   10.339132] sof-audio-pci-intel-mtl 0000:00:1f.3: DSP detected with PCI class/subclass/prog-if 0x040100
[   10.339236] sof-audio-pci-intel-mtl 0000:00:1f.3: bound 0000:00:02.0 (ops i915_audio_component_bind_ops [i915])
[   10.346564] sof-audio-pci-intel-mtl 0000:00:1f.3: use msi interrupt mode
[   10.379242] sof-audio-pci-intel-mtl 0000:00:1f.3: hda codecs found, mask 5
[   10.379249] sof-audio-pci-intel-mtl 0000:00:1f.3: using HDA machine driver skl_hda_dsp_generic now
[   10.379253] sof-audio-pci-intel-mtl 0000:00:1f.3: DMICs detected in NHLT tables: 2
[   10.383121] sof-audio-pci-intel-mtl 0000:00:1f.3: Firmware paths/files for ipc type 1:
[   10.383125] sof-audio-pci-intel-mtl 0000:00:1f.3:  Firmware file:     intel/sof-ipc4/mtl/sof-mtl.ri
[   10.383127] sof-audio-pci-intel-mtl 0000:00:1f.3:  Firmware lib path: intel/sof-ipc4-lib/mtl
[   10.383128] sof-audio-pci-intel-mtl 0000:00:1f.3:  Topology file:     intel/sof-ace-tplg/sof-hda-generic-2ch.tplg
[   10.383856] sof-audio-pci-intel-mtl 0000:00:1f.3: Loaded firmware library: ADSPFW, version: 2.10.0.1
[   10.493906] sof-audio-pci-intel-mtl 0000:00:1f.3: Booted firmware version: 2.10.0.1
[   10.503028] sof-audio-pci-intel-mtl 0000:00:1f.3: Topology: ABI 3:29:0 Kernel ABI 3:23:1
[   10.587964] input: sof-hda-dsp Mic as /devices/pci0000:00/0000:00:1f.3/skl_hda_dsp_generic/sound/card0/input15
[   10.588013] input: sof-hda-dsp Headphone as /devices/pci0000:00/0000:00:1f.3/skl_hda_dsp_generic/sound/card0/input16
[   10.588050] input: sof-hda-dsp HDMI/DP,pcm=3 as /devices/pci0000:00/0000:00:1f.3/skl_hda_dsp_generic/sound/card0/input17
[   10.588105] input: sof-hda-dsp HDMI/DP,pcm=4 as /devices/pci0000:00/0000:00:1f.3/skl_hda_dsp_generic/sound/card0/input18
[   10.588167] input: sof-hda-dsp HDMI/DP,pcm=5 as /devices/pci0000:00/0000:00:1f.3/skl_hda_dsp_generic/sound/card0/input19
```

I tried to replace the topology file from sof-hda-generic-2ch.tplg to sof-hda-generic-2ch-pdm1.tplg from this link https://github.com/thesofproject/linux/issues/2460#issuecomment-898423772 without success.

I also join the alsa-info.txt file for further information.

I would really appreciate your help 

bab

[alsa-info.txt](https://github.com/user-attachments/files/17969410/alsa-info.txt)

SebO-L, 2024-12-02:
Hi, 

As it seems that my laptop have 2 capture hardware devices, I tried to capture on each commands : 

```
arecord -f S16_LE -r 48000 -c 2 -d 5 -t wav output.wav
arecord -D hw:0,0 -f S16_LE -r 48000 -c 2 -d 5 -t wav output2.wav
arecord -D hw:0,6 -f S32_LE -r 48000 -c 2 -d 5 -t wav output3.wav     //doesn't work with S16_LE option
```
No errors have been printed ( shell output or dmesg)
plz, have a look at the uploaded results, but it's look like  i obtain a flat signal.

[output-wav.zip](https://github.com/user-attachments/files/17979197/output-wav.zip)

@singalsu "Pls also check the WAV file, you should see a WAV header"
U mean an hexdump ?

```
hexdump -C output2.wav | head -n 10
00000000  52 49 46 46 24 a6 0e 00  57 41 56 45 66 6d 74 20  |RIFF$...WAVEfmt |
00000010  10 00 00 00 01 00 02 00  80 bb 00 00 00 ee 02 00  |................|
00000020  04 00 10 00 64 61 74 61  00 a6 0e 00 d2 ff cf ff  |....data........|
00000030  d1 ff cd ff d7 ff cc ff  d7 ff c9 ff da ff d0 ff  |................|
00000040  d8 ff d0 ff d7 ff cd ff  d6 ff d2 ff dd ff cf ff  |................|
00000050  d5 ff cd ff d6 ff cf ff  d6 ff d4 ff d7 ff d6 ff  |................|
00000060  d9 ff d5 ff d6 ff d3 ff  dc ff d3 ff d7 ff d3 ff  |................|
00000070  d8 ff d0 ff dd ff d5 ff  d7 ff d5 ff db ff d2 ff  |................|
00000080  d6 ff d4 ff d8 ff d3 ff  d9 ff d9 ff db ff d4 ff  |................|
00000090  dd ff d5 ff de ff db ff  df ff d9 ff e0 ff d5 ff  |................|
```

Isn't my issue can be related to the proc generation like this one https://github.com/thesofproject/linux/issues/5226

```
 cat /proc/cpuinfo 
processor	: 0
vendor_id	: GenuineIntel
cpu family	: 6
model		: 170
model name	: Intel(R) Core(TM) Ultra 5 125U
```

Thanks

singalsu, 2024-12-02:
It looks like output.wav is all zeros (default capture device whatever that was).

Output2.wav from HDA codec hw:0,0 is silent noise. Did you happen to have a headset with microphone connected and checked the codec analog gains?

Output3.wav from DMIC hw:0,6 is all zeros. Did you check that capture volume and capture switches were enabled?

```
amixer -c0 cset name='Dmic0 Capture Volume' 45
amixer -c0 cset name='Dmic0 Capture Switch' on

```
Also check the original if you have the pdm1 version of sof-hda-generic-2ch.tplg in use.

SebO-L, 2024-12-02:
ok, for the first time I have a record, but very very low

```
$ amixer -c0 cset name='Dmic0 Capture Volume' 45
numid=42,iface=MIXER,name='Dmic0 Capture Volume'
  ; type=INTEGER,access=rw---R--,values=2,min=0,max=45,step=0
  : values=45,45
  | dBscale-min=-90.00dB,step=2.00dB,mute=1
$ amixer -c0 cset name='Dmic0 Capture Switch' on
numid=43,iface=MIXER,name='Dmic0 Capture Switch'
  ; type=BOOLEAN,access=rw------,values=4
  : values=on,on,on,on
$ arecord -D hw:0,6 -f S32_LE -r 48000 -c 2 -d 5 -t wav output4.wav
Capture WAVE 'output4.wav' : Signed 32 bit Little Endian, Fréquence 48000 Hz, Stéréo
```

[output4.zip](https://github.com/user-attachments/files/17983406/output4.zip)

What is weird, is that I don't have the capture peripheral on gnome panel on the same time
[image]

> Did you happen to have a headset with microphone connected and checked the codec analog gains?
With an headset, microphone is working (micro jack and usbc headset

>Did you check that capture volume and capture switches were enabled?
I am not exactly sure of what you mean. If I pressed the capture keyboard key, nothing happen. (volume sound keys are working)

>Also check the original if you have the pdm1 version of sof-hda-generic-2ch.tplg in use.
Again not sure. I tried to replace sof-hda-generic-2ch.tplg by  sof-hda-generic-2ch-pdm1.tplg without success. (firmware load failed with pdm1

dmesg | grep sof
```
[   21.911056] sof-audio-pci-intel-mtl 0000:00:1f.3: enabling device (0000 -> 0002)
[   21.911191] sof-audio-pci-intel-mtl 0000:00:1f.3: DSP detected with PCI class/subclass/prog-if 0x040100
[   21.911257] sof-audio-pci-intel-mtl 0000:00:1f.3: bound 0000:00:02.0 (ops i915_audio_component_bind_ops [i915])
[   21.918187] sof-audio-pci-intel-mtl 0000:00:1f.3: use msi interrupt mode
[   21.962665] sof-audio-pci-intel-mtl 0000:00:1f.3: hda codecs found, mask 5
[   21.962670] sof-audio-pci-intel-mtl 0000:00:1f.3: using HDA machine driver skl_hda_dsp_generic now
[   21.962674] sof-audio-pci-intel-mtl 0000:00:1f.3: DMICs detected in NHLT tables: 2
[   21.966555] sof-audio-pci-intel-mtl 0000:00:1f.3: Firmware paths/files for ipc type 1:
[   21.966558] sof-audio-pci-intel-mtl 0000:00:1f.3:  Firmware file:     intel/sof-ipc4/mtl/sof-mtl.ri
[   21.966558] sof-audio-pci-intel-mtl 0000:00:1f.3:  Firmware lib path: intel/sof-ipc4-lib/mtl
[   21.966559] sof-audio-pci-intel-mtl 0000:00:1f.3:  Topology file:     intel/sof-ace-tplg/sof-hda-generic-2ch.tplg
[   21.967099] sof-audio-pci-intel-mtl 0000:00:1f.3: Loaded firmware library: ADSPFW, version: 2.10.0.1
[   22.077607] sof-audio-pci-intel-mtl 0000:00:1f.3: Booted firmware version: 2.10.0.1
[   22.089699] sof-audio-pci-intel-mtl 0000:00:1f.3: Topology: ABI 3:29:0 Kernel ABI 3:23:1
```

Thank you, it is not resolved completely, but I have a sound from the microphone ;O

singalsu, 2024-12-03:
Yep, that is very silent if it was you talking next to notebook. What do you see if you do command?

`amixer -c0 controls | grep Dmic0`

If you have new enough FW and topology you can enable DRC and TDFB processing components to boost the level and improve SNR. Those should be available if you can upgrade to SOF v2.11. The above log shows you are using v2.10. 

See https://github.com/thesofproject/sof-bin/tree/main/v2.11.x

```
amixer -c0 cset name='Dmic0 Capture DRC switch' on
amixer -c0 cset name='Dmic0 Capture TDFB beam switch' on
amixer -c0 cset name='Dmic0 Capture TDFB angle set enum' 0
```

singalsu, 2024-12-03:
Concerning your Gnome mixer screenshot, looks like there's something wrong. In my previous generation sof-hda-generic-2ch.tplg laptop the mixer looks like this (running Fedora 41). If I move the microphone gain slider, also the alsamixer ascii gain control follows it. Somehow your system had muted the microphone.

[image]

closed 2025-05-08 by wszypelt