thesofproject/sof #9715: [BUG] Having trouble loading sof driver for Intel Lunar Lake system with kernel 6.12.

RyanJiao, 2024-12-08:
[image]

Hi, I'm using kernel 6.12 on my Intel Ultra 258V and the sof driver was not initialized properly resulting in no audio. 

But I can't even find a sof-lnl.**  file in this repo's release zip. How can I load my audio card driver properly? It would be so grateful if anyone can give any instruction.

kv2019i, 2024-12-09:
@RyanJiao The firmware for this platform  is in sof-ipc4 folder, so it is there, but in another folder.

Do you have kernel logs of what happens? Output of alsa-info.sh and kernel log would be needed to investigate what happens. See https://thesofproject.github.io/latest/getting_started/intel_debug/suggestions.html for some help on how to get alsa-info.sh.

RyanJiao, 2024-12-09:
Thank you for your kind instruction! This could help me a lot. I will try loading the file and if everything ok i will close this issue. Sorry for silly question.

RyanJiao, 2024-12-09:
I run the install.sh and reboot , the lunar lake driver is installed but still can't find analog audio output device. 

ryan@zenbook:~$ aplay -l
**** List of PLAYBACK Hardware Devices ****
card 0: PCH [HDA Intel PCH], device 3: Generic Digital [Generic Digital]
  Subdevices: 0/1
  Subdevice #0: subdevice #0

Here is the log file.

ryan@zenbook:~$ sudo dmesg | grep audio
[    2.903669] sof-audio-pci-intel-lnl 0000:00:1f.3: SoundWire enabled on CannonLake+ platform, using SOF driver
[    2.903793] sof-audio-pci-intel-lnl 0000:00:1f.3: DSP detected with PCI class/subclass/prog-if 0x040100
[    3.253588] snd_hda_intel 0000:00:1f.3: bound 0000:00:02.0 (ops i915_audio_component_bind_ops [xe])
[    3.272302] snd_hda_codec_generic hdaudioC0D2: autoconfig for Generic: line_outs=0 (0x0/0x0/0x0/0x0/0x0) type:line
[    3.272311] snd_hda_codec_generic hdaudioC0D2:    speaker_outs=0 (0x0/0x0/0x0/0x0/0x0)
[    3.272313] snd_hda_codec_generic hdaudioC0D2:    hp_outs=0 (0x0/0x0/0x0/0x0/0x0)
[    3.272314] snd_hda_codec_generic hdaudioC0D2:    mono: mono_out=0x0
[    3.272315] snd_hda_codec_generic hdaudioC0D2:    dig-out=0x4/0x0
[    3.272315] snd_hda_codec_generic hdaudioC0D2:    inputs:

ryan@zenbook:~$ lsmod | grep snd_hda_intel
snd_hda_intel          61440  3
snd_intel_dspcfg       45056  4 snd_hda_intel,snd_sof,snd_sof_intel_hda_common,snd_sof_intel_hda_generic
snd_hda_codec         208896  5 snd_hda_codec_generic,snd_hda_codec_hdmi,snd_hda_intel,snd_soc_hdac_hda,snd_sof_intel_hda
snd_hda_core          147456  8 snd_hda_codec_generic,snd_hda_codec_hdmi,snd_hda_intel,snd_hda_ext_core,snd_hda_codec,snd_sof_intel_hda_common,snd_soc_hdac_hda,snd_sof_intel_hda
snd_pcm               196608  12 snd_hda_codec_hdmi,snd_hda_intel,snd_hda_codec,soundwire_intel,snd_sof,snd_sof_intel_hda_common,snd_compress,snd_soc_core,snd_sof_utils,snd_hda_core,snd_pcm_dmaengine
snd                   143360  19 snd_ctl_led,snd_hda_codec_generic,snd_seq,snd_seq_device,snd_hda_codec_hdmi,snd_hwdep,snd_hda_intel,snd_hda_codec,snd_sof,snd_timer,snd_compress,snd_soc_core,snd_pcm,snd_rawmidi

ryan@zenbook:~$ lsmod | grep snd_sof
snd_sof_pci_intel_lnl    16384  0
snd_sof_pci_intel_mtl    40960  1 snd_sof_pci_intel_lnl
snd_sof_intel_hda_generic    36864  2 snd_sof_pci_intel_mtl,snd_sof_pci_intel_lnl
soundwire_intel        77824  1 snd_sof_intel_hda_generic
snd_sof_intel_hda_common   184320  3 snd_sof_pci_intel_mtl,snd_sof_pci_intel_lnl,snd_sof_intel_hda_generic
snd_soc_hdac_hda       24576  1 snd_sof_intel_hda_common
snd_sof_intel_hda_mlink    45056  4 snd_sof_pci_intel_lnl,soundwire_intel,snd_sof_intel_hda_common,snd_sof_intel_hda_generic
snd_sof_intel_hda      24576  2 snd_sof_intel_hda_common,snd_sof_intel_hda_generic
snd_sof_pci            24576  3 snd_sof_pci_intel_mtl,snd_sof_pci_intel_lnl,snd_sof_intel_hda_generic
snd_sof_xtensa_dsp     12288  1 snd_sof_intel_hda_generic
snd_sof               397312  6 snd_sof_pci_intel_mtl,snd_sof_pci_intel_lnl,snd_sof_pci,snd_sof_intel_hda_common,snd_sof_intel_hda_generic,snd_sof_intel_hda
snd_sof_utils          16384  1 snd_sof
snd_hda_ext_core       32768  4 snd_sof_intel_hda_common,snd_soc_hdac_hda,snd_sof_intel_hda_mlink,snd_sof_intel_hda
snd_soc_acpi_intel_match   118784  3 snd_sof_pci_intel_mtl,snd_sof_pci_intel_lnl,snd_sof_intel_hda_generic
snd_soc_acpi           16384  2 snd_soc_acpi_intel_match,snd_sof_intel_hda_generic
snd_soc_core          434176  4 soundwire_intel,snd_sof,snd_sof_intel_hda_common,snd_soc_hdac_hda
snd_intel_dspcfg       45056  4 snd_hda_intel,snd_sof,snd_sof_intel_hda_common,snd_sof_intel_hda_generic
snd_intel_sdw_acpi     16384  2 snd_intel_dspcfg,snd_sof_intel_hda_generic
snd_hda_codec         208896  5 snd_hda_codec_generic,snd_hda_codec_hdmi,snd_hda_intel,snd_soc_hdac_hda,snd_sof_intel_hda
snd_hda_core          147456  8 snd_hda_codec_generic,snd_hda_codec_hdmi,snd_hda_intel,snd_hda_ext_core,snd_hda_codec,snd_sof_intel_hda_common,snd_soc_hdac_hda,snd_sof_intel_hda
snd_pcm               196608  12 snd_hda_codec_hdmi,snd_hda_intel,snd_hda_codec,soundwire_intel,snd_sof,snd_sof_intel_hda_common,snd_compress,snd_soc_core,snd_sof_utils,snd_hda_core,snd_pcm_dmaengine
snd                   143360  19 snd_ctl_led,snd_hda_codec_generic,snd_seq,snd_seq_device,snd_hda_codec_hdmi,snd_hwdep,snd_hda_intel,snd_hda_codec,snd_sof,snd_timer,snd_compress,snd_soc_core,snd_pcm,snd_rawmidi

Would you kindly take a look on this.

kv2019i, 2024-12-09:
@RyanJiao It would seem your kernel is missing SoundWire support and the display codec is not handled by the right driver. 

- for Soundwire, distribution kernels will have this enabled by default, but if you are building on your own, the list of configs to enable is here:
- https://github.com/thesofproject/kconfig/blob/master/soundwire-codecs-defconfig
- https://github.com/thesofproject/kconfig/blob/master/soundwire-defconfig

For the display codec, is this still with v6.12 kernel? I see you have the correct "snd_hda_codec_hdmi" module in the build and loaded. This codec should be bound to the display codec with patch "ALSA: hda: add HDMI codec ID for Intel LNL" and v6.12 kernel does have this.

What happens in your log that SOF driver does not compelte the probe, but instead the non-DSP snd_hda_intel driver is used. This driver doesn't know how to handle SoundWire codecs, so it only finds the HDMI/DP audio device. But as the snd_hda_codec_generic is used as the codec driver, the display audio is not correctly setup.

RyanJiao, 2024-12-17:
Thank you! 

But I don't want to reinstall my kernel in recent time. So maybe I would just have to bear with this problems.

closed 2024-12-17 by RyanJiao