thesofproject/sof #9720: [BUG] Lenovo Yoga Slim 7 15ILL9 - No sound coming from speakers

cgbowman, 2024-12-11:
**Describe the bug**
Despite a topology in place, the Lenovo machine doesn't produce sound from the speakers.

**To Reproduce**
Install Ubuntu 24.10 on the laptop, build & install 6.12.0-rc7+ kernel, then boot the laptop. With sof topology updated, sound is not being made from the speakers.

**Reproduction Rate**
Always

**Expected behavior**
Sound should be heard from the speakers.

**Impact**
No audio, results in an annoyance, but doesn't stop the system from functioning.

**Environment**
1) Branch name and commit hash of the 2 repositories: sof (firmware/topology) and linux (kernel driver).
    * Kernel: 49e1628e8983 (sof/topic/sof-dev)
    * SOF: sof-bin-2024.09.1 (release)
2) Name of the topology file
    * Topology: intel/sof-ipc4-tplg/sof-lnl-cs42l43-l0-4ch.tplg
3) Name of the platform(s) on which the bug is observed.
    * Platform: Lenovo Yoga Slim 7 15

**Screenshots or console output**
[image]

journalctl -k log:
[sof-dev-test.txt](https://github.com/user-attachments/files/18089500/sof-dev-test.txt)

cgbowman, 2024-12-11:
>It's a tangent, but I'd be extremely interested in the contents of the /sys/firmware/efi/efivars/CirrusSmart* files - when you get a chance could you drop them here or on the new ticket we use to further investigate this system?

@simontrimmer , here are the contents of the directory, as requested from https://github.com/thesofproject/sof/issues/9663:
```
/sys/firmware/efi/efivars$ ls | grep "CirrusSmart*"
CirrusSmartAmpCalibrationData-02f9af02-7734-4233-b43d-93fe5aa35db3
CirrusSmartCalInfo-ea064247-46bd-433b-9249-e31b63a64172
```

simontrimmer, 2024-12-11:
@cgbowman Thanks! I'm out of the workshop now but will compare with the one I have tomorrow, but I'm 99% certain that I have more Sinks than you see in your wpctl output - I wonder if this is one of those ucm problems?

(The unit I have will almost certainly have had all the files updated to the latest from https://github.com/alsa-project/alsa-ucm-conf  as I'd recently had some fun with another product and this Lenovo would have been installed off a live USB image I'd made for the one that needed the updated ucm configuration)

simontrimmer, 2024-12-11:
Actually perhaps 24.10 is missing this?

https://github.com/alsa-project/alsa-ucm-conf/commit/e4277a204cd61f8f68f616df0d8c42e5bdeb1475

cgbowman, 2024-12-11:
> @cgbowman - Actually perhaps 24.10 is missing this?
> 
> [alsa-project/alsa-ucm-conf@e4277a2](https://github.com/alsa-project/alsa-ucm-conf/commit/e4277a204cd61f8f68f616df0d8c42e5bdeb1475)

@simontrimmer, I updated the files, rebooted, and the Settings menu's 'output device' shows as "Dummy Output". Specifically, adding the change to the sof-soundwire.conf file causes the change.

I've also tried changing the entire ucm2 folder and replacing it with the latest dir available in the alsa-ucm-conf repo and I observe the same issue.

simontrimmer, 2024-12-12:
@cgbowman Interesting - I suspect we're in the right area .... Is there an error message from the UCM?

Looking at the one I have here:
...
 │
 ├─ Sinks:
 │      56. Lunar Lake-M HD Audio Controller HDMI / DisplayPort 3 Output [vol: 1.00]
 │      57. Lunar Lake-M HD Audio Controller HDMI / DisplayPort 2 Output [vol: 1.00]
 │      58. Lunar Lake-M HD Audio Controller HDMI / DisplayPort 1 Output [vol: 1.00]
 │  *   59. Lunar Lake-M HD Audio Controller Headphones + Speaker [vol: 0.37]
 │
...

which mirrors the outputs to HDMI and the cs42l43/cs35l56 pair, and it looks like when I upgraded the alsa-ucm-conf files they were matched with similarly dated alsa-lib and alsa-utils, I think a new file syntax (7) was introduced that might be confusing it, this was important to make a different system I was using work, but it could also apply here too.

-Simon

lgirdwood, 2024-12-12:
@cgbowman just checking, was your alsa-info data above produced with active playback (if not, pls run again with music as speaker test is quite fast/short) ? Just want to rule out userspace isn't incorrectly modifying mixer values when a stream is started/stopped.

cgbowman, 2024-12-12:
> @cgbowman just checking, was your alsa-info data above produced with active playback (if not, pls run again with music as speaker test is quite fast/short) ? Just want to rule out userspace isn't incorrectly modifying mixer values when a stream is started/stopped. 

Apologies, it was not produced with active playback. I will upload more info when I have the chance.

cgbowman, 2024-12-13:
>@cgbowman Interesting - I suspect we're in the right area .... Is there an error message from the UCM?

Attached full journalctl output after updating ucm2 directory: 
[lenovo-yoga-slim-updated-ucm-journalctl.txt](https://github.com/user-attachments/files/18119285/lenovo-yoga-slim-updated-ucm-journalctl.txt)

Some notable entries from alsactl:
```
$ cat ~/lenovo-yoga-slim-updated-ucm-journalctl.txt | grep -i alsa
Dec 12 16:44:58 cgbowman-desk6 (udev-worker)[2246]: controlC0: Process '/usr/sbin/alsactl -E HOME=/run/alsa -E XDG_RUNTIME_DIR=/run/alsa/runtime restore /dev/snd/controlC0' failed with exit code 99.
Dec 12 16:44:58 cgbowman-desk6 systemd[1]: alsa-state.service - Manage Sound Card State (restore and store) was skipped because of an unmet condition check (ConditionPathExists=/etc/alsa/state-daemon.conf).
Dec 12 16:44:58 cgbowman-desk6 systemd[1]: Starting alsa-restore.service - Save/Restore Sound Card State...
Dec 12 16:44:58 cgbowman-desk6 alsactl[2345]: Found hardware: "sof-soundwire" "Intel Lunar Lake HDMI" "HDA:80862820,80860101,00100000  cfg-amp:3 mic:dmic cfg-mics:4 iec61937-pcm:7,6,5 spk:cs35l56-bridge hs:cs42l43" "" ""
Dec 12 16:44:58 cgbowman-desk6 alsactl[2345]: Hardware is initialized using a generic method
Dec 12 16:44:58 cgbowman-desk6 alsactl[2345]: /usr/sbin/alsactl: set_control:1475: Cannot write control '2:0:0:cs42l43 Jack Override:0' : Device or resource busy
Dec 12 16:44:58 cgbowman-desk6 systemd[1]: Finished alsa-restore.service - Save/Restore Sound Card State.
Dec 12 16:46:46 cgbowman-desk6 sudo[4062]: cgbowman : TTY=pts/1 ; PWD=/home/cgbowman ; USER=root ; COMMAND=/usr/sbin/alsa-info
Dec 12 16:51:16 cgbowman-desk6 sudo[4637]: cgbowman : TTY=pts/1 ; PWD=/home/cgbowman ; USER=root ; COMMAND=/usr/sbin/alsa-info
```

Updated alsa-info output with music playing in Rhythmbox: 
[lenovo-yoga-slim-alsa-info-music-playing.txt](https://github.com/user-attachments/files/18119237/lenovo-yoga-slim-alsa-info-music-playing.txt)

simontrimmer, 2024-12-16:
Er,

@charleskeepax would not having a headset plugged in be the most likely reason for the Jack Override returning -EBUSY?

I was saying earlier today I should come back to this - I do think that something in the alsa-ucm-conf / alsa-utils / alsa-lib isn't happy and that is just snowballing, but the only way I could think of reproducing the problem and then working out what was happening was to sacrifice my working system running Debian12+updates and basically starting again with a clean Ubuntu noble install and working through the problems. I did that recently on a different product and whilst I got to do what I needed in the sound domain, the UI experience was less than enjoyable and put me off rushing into it 😁 Might be a good week for it.

@cgbowman - just checking, did you update the alsa-utils & alsa-lib when you updated the UCM files? (thinking about syntax 7)

-Simon

charleskeepax, 2024-12-16:
Yes Jack Override will return EBUSY if there is no jack present, its purpose is to set the jack to various types of device that can't be automatically detected. However, an error from alsactl restore should not have any impact on the UCM.

cgbowman, 2024-12-16:
> @cgbowman - just checking, did you update the alsa-utils & alsa-lib when you updated the UCM files? (thinking about syntax 7)

Good call, I'll double check

cgbowman, 2024-12-16:
I updated alsa-lib & alsa-utils (using the tarball releases) from 1.2.11 to 1.2.13 and it didn't seem to make an impact yet:
- [lenovo-yoga-slim-alsa-info-lib-update.txt](https://github.com/user-attachments/files/18156940/lenovo-yoga-slim-alsa-info-lib-update.txt)
- [lenovo-yoga-slim-journalctl-lib-update.txt](https://github.com/user-attachments/files/18156950/lenovo-yoga-slim-journalctl-lib-update.txt)

Do you think I should try directly from the repo sources? There didn't seem to be much more than a few commits on top of the 1.2.13 tags for both upstream sources.

lgirdwood, 2024-12-17:
> Yes Jack Override will return EBUSY if there is no jack present, its purpose is to set the jack to various types of device that can't be automatically detected. However, an error from alsactl restore should not have any impact on the UCM.

@charleskeepax do you mean no jack connector/pin present or connected to the codec on the PCB ? Would it be better to return ENODEV here ? i.e. EBUSY makes me think codec is busy with some other work so cant do the jack detect work atm

Agree, UCM should just carry on.

Scharkenberg, 2025-02-18:
There was a UEFI update about a week ago, but sound isn't working yet. The system recognizes seven different output devices which are all non-functional, plus the dummy output. I would like to help diagnose and hopefully solve this problem, but I don't know where to begin - SOF and sound in general are new to me.

bfog, 2025-02-18:
Can confirm I am seeing the same behaviour as @Scharkenberg . Using EndeavourOS, with the default kernel 6.13.2  shows Dummy Output. Upgrading to 6.14-rc3 now at least detects Lunar Lake-M HD Audio Controller, but still no audio and 7 different audio output options. Happy to provide any logs to get this addressed.

Unrelated, but for anyone reading this, I found my laptop fans weren't working on kernel 6.13.2. 6.14 has resolved this.

simontrimmer, 2025-02-19:
Random tagging of potentially interested parties - @Scharkenberg @bfog @cgbowman 

I just updated my test unit to the public bios release "DMI: LENOVO 83HM/LNVNB161216, BIOS NYCN66WW 01/15/2025" and early indications are extremely positive - I used a kernel built the head of Linus' branch and even the annoying "it hangs at some point, probably power management related" has gone.

I have an early ubuntu 25.04 build on here at the moment, but the key feature for people is to make sure that they have the latest alsa components (some of the alsa-ucm-conf files now have file format dependencies on the newer alsa-lib/alsa-utils).

The distro has:
> ii  alsa-ucm-conf                                 1.2.12-1ubuntu1                            all          ALSA Use Case Manager configuration files
> ii  alsa-utils                                    1.2.13-1ubuntu1                            amd64        Utilities for configuring and using ALSA

and wpctl status was only showing the Dummy Output, updating my alsa-ucm-conf files to the head solved that problem.

Depending on your distribution, you might need to get update the linux-firmware repository - I published the initial tuning files a long time ago now but that's not very long when considering Linux distribution cadences. The files I booted up with **_were the ones I published yesterday_** but the originals will produce sound.

> root@:~# dmesg | grep DSP1
> [    4.338570] cs35l56 spi-cs35l56-right: DSP1: cirrus/cs35l56-b0-dsp1-misc-17aa380e-spkid1.wmfw: format 3 timestamp 0x6662200a
> [    4.338587] cs35l56 spi-cs35l56-right: DSP1: cirrus/cs35l56-b0-dsp1-misc-17aa380e-spkid1.wmfw: Thu 06 Jun 2024 21:46:02 GMT Daylight Time
> [    4.344364] cs35l56 spi-cs35l56-left: DSP1: cirrus/cs35l56-b0-dsp1-misc-17aa380e-spkid1.wmfw: format 3 timestamp 0x6662200a
> [    4.344373] cs35l56 spi-cs35l56-left: DSP1: cirrus/cs35l56-b0-dsp1-misc-17aa380e-spkid1.wmfw: Thu 06 Jun 2024 21:46:02 GMT Daylight Time
> [    5.798523] cs35l56 spi-cs35l56-left: DSP1: Firmware: 1a00d6 vendor: 0x2 v3.11.20, 41 algorithms
> [    5.981874] cs35l56 spi-cs35l56-left: DSP1: cirrus/cs35l56-b0-dsp1-misc-17aa380e-spkid1-ampl.bin: v3.11.20
> [    5.981883] cs35l56 spi-cs35l56-left: DSP1: Misc: C:\Cirrus_Projects\Lenovo_PC\S790\SIT\Bin\20241226_LS_DeltaDBE_14Max\init\CS35L56 [I2C - 0x60] @ Vir
> [    7.719539] cs35l56 spi-cs35l56-right: DSP1: Firmware: 1a00d6 vendor: 0x2 v3.11.20, 41 algorithms
> [    7.888613] cs35l56 spi-cs35l56-right: DSP1: cirrus/cs35l56-b0-dsp1-misc-17aa380e-spkid1-ampr.bin: v3.11.20
> [    7.888621] cs35l56 spi-cs35l56-right: DSP1: Misc: C:\Cirrus_Projects\Lenovo_PC\S790\SIT\Bin\20241226_LS_DeltaDBE_14Max\init\CS35L56 [I2C - 0x62] @ Vir

alsa-ucm-conf
> commit c3314b9ca29861d19164d2b3987745b7170dab06 (HEAD, origin/master, origin/HEAD)
> Author: WeirdTreeThing <bradyn127@protonmail.com>
> Date:   Sat Jan 25 15:21:21 2025 -0500
> 
>     ucm2: add mt8183_mt6358_ts3a227_max98357
> 
>     Closes: https://github.com/alsa-project/alsa-ucm-conf/pull/502
>     Signed-off-by: WeirdTreeThing <bradyn127@protonmail.com>
>     Signed-off-by: Jaroslav Kysela <perex@perex.cz>

Kernel
> commit 6537cfb395f352782918d8ee7b7f10ba2cc3cbf2 (HEAD, torvalds/master)
> 
> Merge: 2408a807bfc3 e77aa4b2eaa7
> Author: Linus Torvalds <torvalds@linux-foundation.org>
> Date:   Tue Feb 18 09:00:31 2025 -0800
> 
>     Merge tag 'sound-6.14-rc4' of git://git.kernel.org/pub/scm/linux/kernel/git/tiwai/sound
> 
>     Pull sound fixes from Takashi Iwai:
>      "A slightly large collection of fixes, spread over various drivers.
> 
>       Almost all are small and device-specific fixes and quirks in ASoC SOF
>       Intel and AMD, Renesas, Cirrus, HD-audio, in addition to a small fix
>       for MIDI 2.0"
> 
>     * tag 'sound-6.14-rc4' of git://git.kernel.org/pub/scm/linux/kernel/git/tiwai/sound: (41 commits)
>       ALSA: seq: Drop UMP events when no UMP-conversion is set
>       ALSA: hda/conexant: Add quirk for HP ProBook 450 G4 mute LED
>       ALSA: hda/cirrus: Reduce codec resume time
>       ALSA: hda/cirrus: Correct the full scale volume set logic
>       virtio_snd.h: clarify that `controls` depends on VIRTIO_SND_F_CTLS
>       ALSA: hda: Add error check for snd_ctl_rename_id() in snd_hda_create_dig_out_ctls()
>       ALSA: hda/tas2781: Fix index issue in tas2781 hda SPI driver
>       ASoC: imx-audmix: remove cpu_mclk which is from cpu dai device
>       ALSA: hda/realtek: Fixup ALC225 depop procedure
>       ALSA: hda/tas2781: Update tas2781 hda SPI driver
>       ASoC: cs35l41: Fix acpi_device_hid() not found
>       ASoC: SOF: amd: Add branch prediction hint in ACP IRQ handler
>       ASoC: SOF: amd: Handle IPC replies before FW_BOOT_COMPLETE
>       ASoC: SOF: amd: Drop unused includes from Vangogh driver
>       ASoC: SOF: amd: Add post_fw_run_delay ACP quirk
>       ASoC: Intel: soc-acpi-intel-ptl-match: revise typo of rt713_vb_l2_rt1320_l13
>       ASoC: Intel: soc-acpi-intel-ptl-match: revise typo of rt712_vb + rt1320 support
>       ALSA: Switch to use hrtimer_setup()
>       ALSA: hda: hda-intel: add Panther Lake-H support
>       ASoC: SOF: Intel: pci-ptl: Add support for PTL-H
>       ...

cgbowman, 2025-02-19:
Awesome! I've updated my device as of yesterday for firmware, kernel, & other bits. I'll update the alsa components and report my observations.

cgbowman, 2025-02-19:
Can confirm, after updating linux-firmware & alsa-ucm-conf (whilst using an updated drm-tip kernel) and rebooting, I'm greeted with working audio!

Thanks @simontrimmer and others for all your work! If you think there's no other work to do here, I think we can close this issue as fixed.

simontrimmer, 2025-02-19:
@cgbowman - Great!

closed 2025-02-19 by cgbowman

bfog, 2025-02-20:
Just thought I'd pipe in to also confirm that I now have audio working!

Steps I took:
- Update alsa-libs and alsa-utils to latest from git
- Update alsa-ucm-conf to latest from git
- Install latest linux-firmware from git (suggest using yay to install linux-firmware-git, arch packages aren't latest)

I also updated my linux kernel to latest commit. However, after I got sound working I reverted back to the default for arch which is 6.13.3 and found sound was still working. So seems as though 6.13 kernel should be sufficient, the main culprits are grabbing latest linux-firmware and alsa components.

Scharkenberg, 2025-02-21:
I can confirm that on my Manjaro installation I only needed the newest linux-firmware (specifically the cs42l* and cs35l56 goodies) and the newest alsa-ucm-conf from the master branch.

There is a very small delay at the beginning of a playback session (and perhaps dropped buffers?) but I can live with it.

pavelToman, 2025-03-03:
Can confirm too that update linux-firmware and alsa-ucm-conf from master works for my Yoga 7 Aura. 
I tried it with Fedora 41, but it did not work because of the kernel version (6.12). After update to 6.14.rc4 it works!

high3eam, 2025-10-25:
Experienced missing ucm2 profiles when I upgraded my 15ILL9 from Ubuntu 25.04 plucky to 25.10 questing (no sound coming from built-in speakers, only dummy output displayed in settings). Internal speakers were working fine before in 25.04.

Fixed it by overwriting the current ucm2 files with the files from the newest version.

```
$ wget https://www.alsa-project.org/files/pub/lib/alsa-ucm-conf-1.2.14.tar.bz2
$ tar -xjf alsa-ucm-conf-1.2.14.tar.bz2
$ sudo cp -r alsa-ucm-conf-1.2.14/ucm2/* /usr/share/alsa/ucm2/
$ sudo reboot
```

After that, sound devices were recognized properly again and sound output from internal speakers was working again.

t7tran, 2025-10-31:
> Experienced missing ucm2 profiles when I upgraded my 15ILL9 from Ubuntu 25.04 plucky to 25.10 questing (no sound coming from built-in speakers, only dummy output displayed in settings). Internal [...]

For the next looker, I'm using Lenovo x9-15 without sound in 24.04. Had a bit of joy as the sound worked in 25.04 before losing it again after upgrading to 25.10. Patching with that latest version v1.2.14 works.