thesofproject/sof #9723: No sound on ASUS Vivobook S 14 lunar lake

nivekuil, 2024-12-12:
Kernel 6.12.4-200.fc41.x86_64, Fedora Kinoite testing
```
journalctl -k | grep sof-audio

Dec 11 22:22:43 fedora kernel: sof-audio-pci-intel-lnl 0000:00:1f.3: enabling device (0000 -> 0002)
Dec 11 22:22:43 fedora kernel: sof-audio-pci-intel-lnl 0000:00:1f.3: DSP detected with PCI class/subclass/prog-if 0x040100
Dec 11 22:22:43 fedora kernel: sof-audio-pci-intel-lnl 0000:00:1f.3: bound 0000:00:02.0 (ops i915_audio_component_bind_ops [xe])
Dec 11 22:22:43 fedora kernel: sof-audio-pci-intel-lnl 0000:00:1f.3: use msi interrupt mode
Dec 11 22:22:43 fedora kernel: sof-audio-pci-intel-lnl 0000:00:1f.3: hda codecs found, mask 5
Dec 11 22:22:43 fedora kernel: sof-audio-pci-intel-lnl 0000:00:1f.3: using HDA machine driver skl_hda_dsp_generic now
Dec 11 22:22:43 fedora kernel: sof-audio-pci-intel-lnl 0000:00:1f.3: NHLT device BT(0) detected, ssp_mask 0x4
Dec 11 22:22:43 fedora kernel: sof-audio-pci-intel-lnl 0000:00:1f.3: BT link detected in NHLT tables: 0x4
Dec 11 22:22:43 fedora kernel: sof-audio-pci-intel-lnl 0000:00:1f.3: DMICs detected in NHLT tables: 2
Dec 11 22:22:43 fedora kernel: sof-audio-pci-intel-lnl 0000:00:1f.3: Firmware paths/files for ipc type 1:
Dec 11 22:22:43 fedora kernel: sof-audio-pci-intel-lnl 0000:00:1f.3:  Firmware file:     intel/sof-ipc4/lnl/sof-lnl.ri
Dec 11 22:22:43 fedora kernel: sof-audio-pci-intel-lnl 0000:00:1f.3:  Firmware lib path: intel/sof-ipc4-lib/lnl
Dec 11 22:22:43 fedora kernel: sof-audio-pci-intel-lnl 0000:00:1f.3:  Topology file:     intel/sof-ipc4-tplg/sof-hda-generic-2ch.tplg
Dec 11 22:22:43 fedora kernel: sof-audio-pci-intel-lnl 0000:00:1f.3: Loaded firmware library: ADSPFW, version: 2.11.1.1
Dec 11 22:22:43 fedora kernel: sof-audio-pci-intel-lnl 0000:00:1f.3: Booted firmware version: 2.11.1.1
Dec 11 22:22:43 fedora kernel: sof-audio-pci-intel-lnl 0000:00:1f.3: Topology: ABI 3:29:1 Kernel ABI 3:23:1
Dec 11 22:22:43 fedora kernel: sof-audio-pci-intel-lnl 0000:00:1f.3: Direct firmware load for intel/sof-ipc4-lib/lnl/B36EE4DA-006F-47F9-A06D-FECBE2D8B6CE.bin failed with error -2
Dec 11 22:22:43 fedora kernel: sof-audio-pci-intel-lnl 0000:00:1f.3: Library file 'intel/sof-ipc4-lib/lnl/B36EE4DA-006F-47F9-A06D-FECBE2D8B6CE.bin' is missing
Dec 11 22:22:43 fedora kernel: sof-audio-pci-intel-lnl 0000:00:1f.3: failed to find module info for widget drc.11.1 with UUID B36EE4DA-006F-47F9-A06D-FECBE2D8B6CE
Dec 11 22:22:43 fedora kernel: sof-audio-pci-intel-lnl 0000:00:1f.3: failed updating IPC struct for drc.11.1
Dec 11 22:22:43 fedora kernel: sof-audio-pci-intel-lnl 0000:00:1f.3: error: tplg component load failed -22
Dec 11 22:22:43 fedora kernel: sof-audio-pci-intel-lnl 0000:00:1f.3: error: failed to load DSP topology -22
Dec 11 22:22:43 fedora kernel: sof-audio-pci-intel-lnl 0000:00:1f.3: ASoC: error at snd_soc_component_probe on 0000:00:1f.3: -22
```

```
rpm -q alsa-sof-firmware -i

Name        : alsa-sof-firmware
Version     : 2024.09.2
Release     : 1.fc41
Architecture: noarch
Install Date: Wed 11 Dec 2024 05:31:46 PM PST
Group       : Unspecified
Size        : 8828485
License     : BSD-3-Clause Apache-2.0
Signature   : RSA/SHA256, Mon 09 Dec 2024 06:53:49 AM PST, Key ID d0622462e99d6ad1
Source RPM  : alsa-sof-firmware-2024.09.2-1.fc41.src.rpm
Build Date  : Fri 06 Dec 2024 01:02:48 AM PST
```

lgirdwood, 2024-12-12:
@nivekuil can you try installing the latest sof-bin release, its looks like you are missing a file.
https://github.com/thesofproject/sof-bin

lgirdwood, 2024-12-12:
@ujfalusi @kv2019i if we are missing a UUID module on the FS we should still enumerate the DSP/driver, yes the pipelines that need that module will fail on creation (FW will report error that can come back up the stack), but the other audio endpoints should at least be functional.

kv2019i, 2024-12-12:
@perexg The Fedora packages for SOF 2024.09 seem to be missing the library directories:
https://bodhi.fedoraproject.org/updates/FEDORA-2024-5f8ce29da9

```
$ rpm -qlp alsa-sof-firmware-2024.09.2-1.fc41.noarch.rpm  |grep ipc4-lib
$ 
$ find sof-bin-2024.09.2 |grep ipc4-lib
sof-bin-2024.09.2/sof-ipc4-lib
sof-bin-2024.09.2/sof-ipc4-lib/lnl
sof-bin-2024.09.2/sof-ipc4-lib/lnl/B36EE4DA-006F-47F9-A06D-FECBE2D8B6CE.bin
sof-bin-2024.09.2/sof-ipc4-lib/lnl/community
sof-bin-2024.09.2/sof-ipc4-lib/lnl/community/B36EE4DA-006F-47F9-A06D-FECBE2D8B6CE.bin
sof-bin-2024.09.2/sof-ipc4-lib/lnl/community/drc.llext
sof-bin-2024.09.2/sof-ipc4-lib/lnl/intel-signed
sof-bin-2024.09.2/sof-ipc4-lib/lnl/intel-signed/B36EE4DA-006F-47F9-A06D-FECBE2D8B6CE.bin
sof-bin-2024.09.2/sof-ipc4-lib/lnl/intel-signed/drc.llext
$
```

We have more modular binaries upcoming in 2025 SOF releases, so the /lib/firmware/intel/sof-ipc4-lib is essential for SOF in the future so sof-bin contens would have to be all included. The full layout is documented at https://thesofproject.github.io/latest/getting_started/intel_debug/introduction.html#user-space-and-filesystem-requirements 

And apologies, I now realize we didn't have any visible warnings to distro maintainers about this in the 2024.09 release notes https://github.com/thesofproject/sof-bin/releases/tag/v2024.09

@nivekuil As a short term fix, you can copy the above files from https://github.com/thesofproject/sof-bin/releases/download/v2024.09/sof-bin-2024.09.tar.gz to /lib/firmware/intel/sof-ipc4-lib . Later the updated RPM will add these files in.

@lgirdwood I'll let @ujfalusi comment, probably a good safety option. But, but, this is really quite a big error case if the files are not updated together. We have no binary interface tracking within the FW now, so it really is expected that the full set of sof-bin is installed to a machine, so it is good this is picked up. A missing file would end to an error, but mixing files could create much harder to debug/detect errors, so I think it's key we make sof-bin such that it's deployed as a whole.

ranj063, 2024-12-12:
> @ujfalusi @kv2019i if we are missing a UUID module on the FS we should still enumerate the DSP/driver, yes the pipelines that need that module will fail on creation (FW will report error that can come back up the stack), but the other audio endpoints should at least be functional.

@lgirdwood We made the decision to fail big and fail early if we know that one or more of the endpoints will be non-functional at the time of topology loading and I think thats the right direction to go instead of keeping partial functionality

ujfalusi, 2024-12-13:
> @lgirdwood We made the decision to fail big and fail early if we know that one or more of the endpoints will be non-functional at the time of topology loading and I think thats the right direction [...]

@lgirdwood, @ranj063, I agree that the fail early, fail big is the right thing to do. This case is unfortunate but it caught - in a dramatic way - a distro packaging issue with a clean error for the reason.
Sidenote: in this particular case (sof-hda-generic-2ch.tplg w/o DRC module) the user would be left without working analog playback and DMIC capture, HDMI playback and analog capture does not have DRC in path.
If we not fail early, but only fail with the PCMs that are missing a module then it is just a "small step" to let the kernel fix the topology on the fly by rewriting it to exclude the missing module(s) and that is not something we can really do or plan to do.

The SOF release for LNL consists of sof-lnl.ri and B36EE4DA-006F-47F9-A06D-FECBE2D8B6CE.bin.

On the other hand on SDW machines with no DMIC (atm) does not have DRC used in topology, so they will probe fine.

kv2019i, 2024-12-18:
I filed an upstream bug to Fedora:
https://bugzilla.redhat.com/show_bug.cgi?id=2332975

We can close this here if @nivekuil you can confirm audio does work after adding the missing files.

perexg, 2024-12-18:
I have not noticed this new directory layout extension. Better release comments are always useful.

@nivekuil : There are linked fedora updates in https://bugzilla.redhat.com/show_bug.cgi?id=2332975 now with updated rpm package contents. Any feedback would be really nice.

kv2019i, 2024-12-18:
Thanks @perexg ! Fedora was not only one impacted. I also filed a bug to Debian https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=1090726 .

mrhpearson, 2024-12-18:
Debian packaging updated: https://salsa.debian.org/mpearson/firmware-sof/-/commit/50dc57acceadd01cdfcf0076b6126e3e7b3992c7
I'll ask for this to be uploaded for release.

I'll drop a note to the Ubuntu folk as well - they handle the packaging differently sometimes and might not pick up the Debian package for a while.

nivekuil, 2024-12-21:
It's working now, thank you.

closed 2024-12-21 by nivekuil