thesofproject/sof #9735: [BUG] Topology sof-lnl-cs42l43-l0-2ch contains incorrect 4ch beamformer blob

singalsu, 2024-12-18:
**Describe the bug**
The 2ch DMIC topology is built with embedded beamformer blob for 4ch system.

The target build is defined as:

"cavs-sdw\;sof-lnl-cs42l43-l0-**2ch**\;PLATFORM=lnl,NUM_SDW_AMP_LINKS=1,NUM_DMICS=2,\
PDM1_MIC_A_ENABLE=0,PDM1_MIC_B_ENABLE=0,DMIC0_ID=3,DMIC1_ID=4,\
PREPROCESS_PLUGINS=nhlt,NHLT_BIN=nhlt-sof-lnl-cs42l43-l0-**2ch**.bin,\
SDW_AMP_FEEDBACK=false,SDW_SPK_STREAM=Playback-SmartAmp,\
SDW_JACK_OUT_STREAM=Playback-SimpleJack,SDW_JACK_IN_STREAM=Capture-SimpleJack,\
HDMI1_ID=5,HDMI2_ID=6,HDMI3_ID=7,DMIC0_ENHANCED_CAPTURE=true,\
**EFX_DMIC0_TDFB_PARAMS=line4_pass**,EFX_DMIC0_DRC_PARAMS=dmic_default"

The valid options for 2ch beamformer are "line2_generic_pm10deg" and "line2_pass".

**To Reproduce**
Build the topologies from git main or stable-v2.12 with scripts/build-tools.sh. Install the topology to PC
(I can't do this, I don't have such device)

**Reproduction Rate**
Always.

**Expected behavior**
EFX_DMIC0_TDFB_PARAMS=line2_generic_pm10deg or EFX_DMIC0_TDFB_PARAMS=line2_pass

**Impact**
Likely fail of capture, not sure what happens.

**Environment**
1) Branch name and commit hash of the 2 repositories: sof (firmware/topology) and linux (kernel driver).
    * Kernel: {SHA}
    * SOF: {SHA}
2) Name of the topology file
    * Topology: {FILE}
3) Name of the platform(s) on which the bug is observed.
    * Platform: {PLATFORM}

**Screenshots or console output**
If applicable, add a screenshot (drag-and-drop an image), or console logs
(cut-and-paste text and put a code fence (\`\`\`) before and after, to help
explain the issue.

Please also include the relevant sections from the firmware log and kernel log in the report (and attach the full logs for complete reference). Kernel log is taken from *dmesg* and firmware log from *sof-logger*. See https://thesofproject.github.io/latest/developer_guides/debugability/logger/index.html

singalsu, 2024-12-18:
Issue confirmed, when I simulate the issue with sof-hda-generic-2ch.tplg with this change:

```
diff --git a/tools/topology/topology2/production/tplg-targets-hda-generic.cmake b/tools/topology/topology2/production/tplg-targets-hda-generic.cmake
index e2f64cbe2..18b365c8c 100644
--- a/tools/topology/topology2/production/tplg-targets-hda-generic.cmake
+++ b/tools/topology/topology2/production/tplg-targets-hda-generic.cmake
@@ -17,7 +17,7 @@ EFX_HDA_MIC_TDFB_PARAMS=line2_pass,EFX_HDA_MIC_DRC_PARAMS=passthrough"
 # passthrough pipelines for HDMI and
 # 2 or 4 DMIC, no NHLT blob included in topology
 "sof-hda-generic\;sof-hda-generic-2ch\;HDA_CONFIG=mix,NUM_DMICS=2,\
-DMIC0_ENHANCED_CAPTURE=true,EFX_DMIC0_TDFB_PARAMS=line2_generic_pm10deg,\
+DMIC0_ENHANCED_CAPTURE=true,EFX_DMIC0_TDFB_PARAMS=line4_pass,\
 EFX_DMIC0_DRC_PARAMS=dmic_default"
 
 "sof-hda-generic\;sof-hda-generic-4ch\;HDA_CONFIG=mix,NUM_DMICS=4,\

The arecord fails and mtrace shows this error:

```
[  163.769421] <inf> tdfb: tdfb_prepare: comp:1 0x14 tdfb_prepare()
[  163.769441] <err> tdfb: tdfb_init_coef: comp:1 0x14 tdfb_init_coef(), stream output channels count 2 does not match configuration 4
[  163.769451] <err> tdfb: tdfb_prepare: comp:1 0x14 tdfb_prepare() error: tdfb_setup failed.
[  163.769461] <inf> component: comp_set_state: comp:1 0x14 comp_set_state(), state already set to 2
[  163.769475] <err> module_adapter: module_prepare: comp:1 0x14 module_prepare() error -22: module specific prepare failed, comp_id 20
[  163.769485] <err> module_adapter: module_adapter_prepare: comp:1 0x14 module_adapter_prepare() error ffffffea: module prepare failed
[  163.769496] <err> pipe: pipeline_prepare: pipe:1 0x0 pipeline_prepare(): ret = -22, dev->comp.id = 0x20004
[  163.769505] <err> ipc: ipc4_pcm_params: ipc: pipe 1 comp 0 prepare failed -22

```
```

lgirdwood, 2024-12-19:
@singalsu fix merged, can you testest and close ?

singalsu, 2024-12-20:
The fixes are in git main commit 7650d0ac21f2a51e7dfb63b785142e7bb5a0921a  and in stable-v2.12 in commit b03f0e40b0cbb9a318f8d03048d0ea1f970d02eb ("Tools: Topology2: Fix DMIC TDFB blob in sof-lnl-cs42l43-l0-2ch").

closed 2024-12-20 by singalsu