thesofproject/sof #9748: [BUG] sof-hda-generic-1ch.tplg not found

stefdb, 2024-12-19:
**Describe the bug**
No Soundcards are initialised.

**To Reproduce**
Boot the System (ASUS NUC14LNS)

**Reproduction Rate**
100%

**Expected behavior**
Topology is loaded, and soundcard is initialised

**Impact**
No audio

**Environment**
1) Branch name and commit hash of the 2 repositories: sof (firmware/topology) and linux (kernel driver).
    * Kernel: Linux version 6.12.0-next-20241127-test-kernel (dd@dd-ASUS-EXPERTBOOK) (gcc (Ubuntu 14.2.0-4ubuntu2) 14.2.0, GNU ld (GNU Binutils for Ubuntu) 2.43.1) #4 SMP PREEMPT_DYNAMIC Wed Nov 27 17:53:05 CST 2024
    * SOF: v2.11.1
2) Name of the topology file
    * Topology: sof-hda-generic-1ch.tplg
3) Name of the platform(s) on which the bug is observed.
    * Platform: Intel(R) Core(TM) Ultra 9 288V 3.30GHz
    * ASUS NUC14LNS

**Screenshots or console output**
Initial log with missing file:
[NUC14LNS_dmesg.txt](https://github.com/user-attachments/files/18200319/NUC14LNS_dmesg.txt)

I also attempted to generate sof-hda-generic-1ch.tplg by modifying tools/topology/topology2/production/tplg-targets-hda-generic.cmake, to add the configuration where NUM_DMICS=1:

///"sof-hda-generic\;sof-hda-generic-1ch\;HDA_CONFIG=mix,NUM_DMICS=1,\
DMIC0_ENHANCED_CAPTURE=true,EFX_DMIC0_TDFB_PARAMS=line2_generic_pm10deg,\
EFX_DMIC0_DRC_PARAMS=dmic_default"///

But this didn't work due, producing these error messages:

[    7.026064]  DMIC Raw: substream DMIC Raw has no playback, no capture
[    7.026071] skl_hda_dsp_generic skl_hda_dsp_generic: ASoC: can't create pcm DMIC Raw :-22
[    7.026722] skl_hda_dsp_generic skl_hda_dsp_generic: probe with driver skl_hda_dsp_generic failed with error -22

Full log attached here:
[1219_dmesg.txt](https://github.com/user-attachments/files/18200361/1219_dmesg.txt)

kv2019i, 2024-12-19:
This is a curious case, I don't believe we've seen a single dmic configuration before. @singalsu  @bardliao @ujfalusi do you recall?

@stefdb Can quick workaround to try is to force the dmic count to 2 with:

/etc/modprobe.d/sof.conf (or some file under modprobe.d):
```
options snd_sof_intel_hda_generic dmic_num=2
```

This should instruct the drive to load sof-hda-generic-2ch.tplg which is part of standard distribution. If this works, you'll probably have audio only on the left channel of the recorded stream, but at least allows audio device to probe successfully.

bardliao, 2024-12-20:
> This is a curious case, I don't believe we've seen a single dmic configuration before. @singalsu @bardliao @ujfalusi do you recall?
> 

No, it is the first time I am aware of that.

singalsu, 2024-12-30:
First time for me too. Please use as @kv2019i suggested the 2ch topology as replacement. Due to PDM bus characteristic, it will give a dual mono capture stream as e.g. HDA generic headset (mono) mic also appears to system.

I will try to replicate the issue with my devices.

@stefdb Can you please also dump the NHLT content of your PC and attach here (`sudo cat /sys/firmware/acpi/tables/NHLT >NHLT.bin`).

singalsu, 2025-01-02:
Confirming that I'm able to replicate the issue by forcing `options snd_sof_intel_hda_generic dmic_num=1` plus and `options snd_sof_intel_hda_common sof_use_tplg_nhlt=1` and similarly creating a topology sof-hda-generic-cavs25-1ch.tplg renamed to sof-hda-generic-1ch.tplg. I get the same error as reported.

singalsu, 2025-01-03:
Update - I'm testing a fix in #9763.

stefdb, 2025-01-07:
[NHLT.zip](https://github.com/user-attachments/files/18333043/NHLT.zip)

Hi,

I've attached NHLT.bin (inside NHLT.zip) from the system, as requested.

Thanks,
Stefan

singalsu, 2025-01-07:
@stefdb Please wait testing attached sof-hda-generic-ace1-1ch.tplg. I don't have suitable MTL/LNL hardware to test this so I ran it here with the cAVS version that worked. There may be a bug in alsa-utils topology NHLT generator for ACE.  I will do more checks and try to fix it.

singalsu, 2025-01-08:
Here's new topologies bundle (fix in https://github.com/alsa-project/alsa-utils/pull/286). Please try sof-hda-generic-ace1-1ch.tplg from here:

[dmic_1ch_tplgs_20250108.tar.gz](https://github.com/user-attachments/files/18350718/dmic_1ch_tplgs_20250108.tar.gz)

stefdb, 2025-01-16:
Hi,

We've tested these topologies and the dmic seems to work now, and the soundcard probes.
However, the speaker is not currently working, and I've been trying to work out why.
Previously we tested with SOF disabled, and the speaker worked ok.
The strange thing is that it seems like the play command isn't being propagated to the kernel, so it seems something in userspace is not happy, but I'm still investigating.

I've attached 2 dmesg logs of using the new topologies.

Thanks,
Stefan

[20240113_generic-ace1-1ch.txt](https://github.com/user-attachments/files/18443053/20240113_generic-ace1-1ch.txt)
[20250113_generic-1ch.txt](https://github.com/user-attachments/files/18443054/20250113_generic-1ch.txt)

singalsu, 2025-01-17:
@stefdb Great that the microphone works now! Please create another issue for the speaker and attach the above logs there. Especially if it needs help from Cirrus (some fw upload fails seen in dmesg), it's more clear to handle it there.

bardliao, 2025-01-17:
@stefdb  Yes, please create another issue for the speaker issue for better tracking. Please also attach the output of `alsa-info`.
And what is your playback command? Did you try aplay?

lgirdwood, 2025-02-10:
> We can close this issue after these are merged and verified:
> 
> * [x]  [Tools: Topology2: Add build of 1ch DMIC topologies #9763](https://github.com/thesofproject/sof/pull/9763)[ ]  [Topology: NHLT: Intel: Fix mono DMIC configure for MTL platform alsa-project/alsa-utils#286](https://github.com/alsa-project/alsa-utils/pull/286)

@singalsu can we close now ?

singalsu, 2025-02-11:
Yep, everything related to DMIC is now completed. Please continue in other issue with speaker problem.

closed 2025-02-11 by singalsu