thesofproject/sof #9756: [BUG] no sof-arl-rt711-l0.tplg found

pma1, 2024-12-24:
**Describe the bug**
On ARL-U, it shows:
[    4.303907] sof-audio-pci-intel-mtl 0000:00:1f.3: enabling device (0000 -> 0002)
[    4.304072] sof-audio-pci-intel-mtl 0000:00:1f.3: DSP detected with PCI class/subclass/prog-if 0x040100
[    4.365668] sof-audio-pci-intel-mtl 0000:00:1f.3: bound 0000:00:02.0 (ops i915_audio_component_bind_ops [i915])
[    4.372400] sof-audio-pci-intel-mtl 0000:00:1f.3: use msi interrupt mode
[    4.386091] sof-audio-pci-intel-mtl 0000:00:1f.3: hda codecs found, mask 4
[    4.386097] sof-audio-pci-intel-mtl 0000:00:1f.3: NHLT device BT(0) detected, ssp_mask 0x4
[    4.386099] sof-audio-pci-intel-mtl 0000:00:1f.3: BT link detected in NHLT tables: 0x4
[    4.386101] sof-audio-pci-intel-mtl 0000:00:1f.3: DMICs detected in NHLT tables: 0
[    4.390331] sof-audio-pci-intel-mtl 0000:00:1f.3: SOF firmware and/or topology file not found.
[    4.390338] sof-audio-pci-intel-mtl 0000:00:1f.3: Supported default profiles
[    4.390338] sof-audio-pci-intel-mtl 0000:00:1f.3: - ipc type 1 (Requested):
[    4.390339] sof-audio-pci-intel-mtl 0000:00:1f.3:  Firmware file: intel/sof-ipc4/arl/sof-arl.ri
[    4.390340] sof-audio-pci-intel-mtl 0000:00:1f.3:  Topology file: intel/sof-ace-tplg/sof-arl-rt711-l0.tplg
[    4.390341] sof-audio-pci-intel-mtl 0000:00:1f.3: Check if you have 'sof-firmware' package installed.
[    4.390342] sof-audio-pci-intel-mtl 0000:00:1f.3: Optionally it can be manually downloaded from:
[    4.390343] sof-audio-pci-intel-mtl 0000:00:1f.3:    https://github.com/thesofproject/sof-bin/
[    4.392031] sof-audio-pci-intel-mtl 0000:00:1f.3: error: sof_probe_work failed err: -2

**To Reproduce**
bootup and check dmesg.

**Reproduction Rate**
100%

**Expected behavior**
Audio input/ouput works fine.

**Impact**
No audio.

**Environment**
1) Branch name and commit hash of the 2 repositories: sof (firmware/topology) and linux (kernel driver).
    * Kernel: 6.13-rc1
    * SOF: 05f6b0f  v2.11.x: add new 2.11.3 topology2 production binaries
2) Name of the topology file
    * Topology:  intel/sof-ace-tplg/sof-arl-rt711-l0.tplg
3) Name of the platform(s) on which the bug is observed.
    * Platform: ARL-U
[arlu-6.13-rc1-dmesg.txt](https://github.com/user-attachments/files/18237118/arlu-6.13-rc1-dmesg.txt)

kv2019i, 2025-01-03:
This seems to be a config with no PCH-DMICs and single RT711:
[    4.386101] sof-audio-pci-intel-mtl 0000:00:1f.3: DMICs detected in NHLT tables: 0

@CBala21 You added the kernel mach entries in https://github.com/thesofproject/linux/pull/4841 , but I see no matching PR for the topology to SOF repo. This is sometimes the case where the configuration is not in any shipping product, but given this bug, I'm now not sure if this is the case anymore.

kv2019i, 2025-01-07:
@pma1 Can you try copying sof-mtl-rt711-4ch.rplg to sof-arl-rt711-4ch (file part of v2.11 release:
https://github.com/thesofproject/sof-bin/blob/main/v2.11.x/sof-ipc4-tplg-v2.11.3/sof-mtl-rt711-4ch.tplg ).

In addition, either a) enable the PCH-DMIC in BIOS (so that SOF driver will look up topology file with "-4ch" suffix), or b) override the BIOS setting with:
```
/etc/modprobe.d/sof.conf:
options snd_sof_intel_hda_generic dmic_num=4
```

It's not a problem if there are no DMICs connected on the board, above recipe will still work.

pma1, 2025-01-08:
After loaded the mtl sof-bin and sof module parameter,

the dmesg show error:

[    4.881662] sof-audio-pci-intel-mtl 0000:00:1f.3: no matching blob for sample rate: 48000 sample width: 16 channels: 4
[    4.881674] sof-audio-pci-intel-mtl 0000:00:1f.3: failed to prepare widget dai-copier.DMIC.dmic01.capture
[    4.881676] sof-audio-pci-intel-mtl 0000:00:1f.3: Failed to prepare connected widgets
[    4.881677] sof-audio-pci-intel-mtl 0000:00:1f.3: error: failed widget list set up for pcm 10 dir 1
[    4.881679] sof-audio-pci-intel-mtl 0000:00:1f.3: ASoC: error at snd_soc_pcm_component_hw_params on 0000:00:1f.3: -22

[arlu-6.13-rc3-dmesg-sof-mtl-4ch.txt](https://github.com/user-attachments/files/18341355/arlu-6.13-rc3-dmesg-sof-mtl-4ch.txt)

CBala21, 2025-01-08:
Looks blob configuration not picked from topology and its still refer NHLT for dmic.
Please set this flag in sof.conf
options snd_sof_intel_hda_common sof_use_tplg_nhlt=1

pma1, 2025-01-08:
@CBala21 
Thanks for your help.
With "sof_use_tplg_nhlt=1", there is no error in dmesg.

But the audio output is still dummy.

[arlu-6.13-rc3-alsa-info.txt](https://github.com/user-attachments/files/18345029/arlu-6.13-rc3-alsa-info.txt)

kv2019i, 2025-01-08:
@pma1 The drivers are now fine, from the alsa-info you shared:
```
**** List of PLAYBACK Hardware Devices ****
card 0: sofsoundwire [sof-soundwire], device 0: Jack Out (*) []
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 0: sofsoundwire [sof-soundwire], device 5: HDMI1 (*) []
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 0: sofsoundwire [sof-soundwire], device 6: HDMI2 (*) []
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 0: sofsoundwire [sof-soundwire], device 7: HDMI3 (*) []
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 0: sofsoundwire [sof-soundwire], device 31: Deepbuffer Jack Out (*) []
  Subdevices: 1/1
  Subdevice #0: subdevice #0

ARECORD

**** List of CAPTURE Hardware Devices ****
card 0: sofsoundwire [sof-soundwire], device 1: Jack In (*) []
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 0: sofsoundwire [sof-soundwire], device 10: DMIC Raw (*) []
  Subdevices: 1/1
  Subdevice #0: subdevice #0
```

You should be able to play/capture audio with apps using ALSA directly. 

To get audio working with Pipewire, there's similar problem with UCM files as you had with the tplg file. As this is a R&D configuration, the upstream UCM doesn't necessarily support the particular configuration you have, so UCM lookup will fail and e.g. GNOME sound settings will just show a dummy card. PW logs should have errors about UCM file parsing.

pma1, 2025-01-08:
@kv2019i 

Thanks for your detail explanation.

"speaker-test -c 2 " can not output sound.

Since this is a R&D configuration, I would like to close this issue.
I don't want to bother you too much.
If you have anything to make it work, please let me know.

closed 2025-01-08 by pma1

kv2019i, 2025-01-09:
@pma1 To get audio audible in speaker-test, you are probably just setting the initial mixer settings normally done via UCM. You can set these manually (commands from BootSequence in /usr/share/alsa/ucm2/codecs/rt711/init.conf):

E.g.
```
amixer -Dhw:sofsoundwire  cset name='rt711 DAC Surr Playback Volume' 87
amixer -Dhw:sofsoundwire  cset name='rt711 ADC 08 Capture Volume' 63
amixer -Dhw:sofsoundwire  cset name='rt711 ADC 23 Mux' 'MIC2'
amixer -Dhw:sofsoundwire  cset name='rt711 ADC 08 Capture Switch' 1
amixer -Dhw:sofsoundwire  cset name='rt711 AMIC Volume' 1
```

This should get speaker-test working.