thesofproject/sof #9759: [BUG] ASUS Zenbook S 14: No sound from DMIC

dantmnf, 2024-12-26:
**Describe the bug**
Recording from DMIC generates a few pulses then a flat signal.

**To Reproduce**
```console
% arecord -l
**** List of CAPTURE Hardware Devices ****
card 0: sofsoundwire [sof-soundwire], device 1: Jack In (*) []
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 0: sofsoundwire [sof-soundwire], device 4: Microphone (*) []
  Subdevices: 1/1
  Subdevice #0: subdevice #0
% arecord -D hw:0,4 -f s16_le -r 48000 -c 2 -t wav -d 5 record.wav
Recording WAVE 'record.wav' : Signed 16 bit Little Endian, Rate 48000 Hz, Stereo
```

**Reproduction Rate**
All the time

**Expected behavior**
Should be able to record some sound

**Impact**
Unusable built-in microphone

**Environment**
1) Branch name and commit hash of the 2 repositories: sof (firmware/topology) and linux (kernel driver).
    * Kernel: stable 6.12.6
    * SOF: release v2024.09.2
2) Name of the topology file
    * Topology: intel/sof-ipc4-tplg/sof-lnl-cs42l43-l0-cs35l56-l23.tplg
3) Name of the platform(s) on which the bug is observed.
    * Platform: Lunar Lake

**Screenshots or console output**
[image]

alsa-info output for reference: 
[alsa-info.txt](https://github.com/user-attachments/files/18252368/alsa-info.txt)

P.S. recording from audio jack works.

iradraconis, 2024-12-31:
Same here.. Using Linux Kernel 13.0 RC5 on Fedora and still the microphone is not functioning. 

Inxi output for Audio:

```
Audio:
  Device-1: Intel Lunar Lake-M HD Audio driver: sof-audio-pci-intel-lnl
  API: ALSA v: k6.13.0-0.rc5.42.fc42.x86_64 status: kernel-api
  Server-1: PipeWire v: 1.2.7 status: active
```

bardliao, 2025-01-08:
That might be UCM issue. @dantmnf Can you try the latest UCM from https://github.com/alsa-project/alsa-ucm-conf? @charleskeepax @stuhenderson Could you take a look?

pcgrosen, 2025-01-08:
Hi @bardliao, I have the same system and am experiencing the same issue. I tried the latest from that repo (installed via instructions in the README, including state purge), but no change -- still just a click at the beginning, then silence.

Is there something else I can try, or additional debug info I can gather? Thanks!

bardliao, 2025-01-08:
@pcgrosen Can you upgrade [cs42l43.bin](https://git.kernel.org/pub/scm/linux/kernel/git/firmware/linux-firmware.git/tree/cs42l43.bin) ? I thought it is a muted issue but it should not be. Because you could record a pop noise at the beginning. Now I guess you may need an upgraded cs42l43 firmware.

dantmnf, 2025-01-08:
> @pcgrosen Can you upgrade [cs42l43.bin](https://git.kernel.org/pub/scm/linux/kernel/git/firmware/linux-firmware.git/tree/cs42l43.bin) ? I thought it is a muted issue but it should not be. Because you could record a pop noise at the beginning. Now I guess you may need an upgraded cs42l43 firmware.

I think we are already on the latest version since it has been in linux-firmware for (exactly) one year.

bardliao, 2025-01-08:
Does anyone know exactly which DMIC is used by this device?
```
[    3.831584] sof-audio-pci-intel-lnl 0000:00:1f.3: DMICs detected in NHLT tables: 2
```
It seems PCH DMIC is detected in the NHLT table, but I am not sure if it matches the real HW design or not.
If PCH DMIC is used, we probably need to create sof-lnl-cs42l43-l0-cs35l56-l23-2ch.tplg and some kernel change is required.

dantmnf, 2025-01-08:
It do have 2-channel microphone and can be verified in Windows by moving a speaker around two microphones while recording:
[image]

charleskeepax, 2025-01-08:
This laptop has host DMICs but is missing a quirk for such, we are looking at putting together a patch.

bardliao, 2025-01-08:
> This laptop has host DMICs but is missing a quirk for such, we are looking at putting together a patch.

@charleskeepax  We may need to do some change other than adding a quirk.
```c
			/*
			 * DMICs use up to 4 pins and are typically pin-muxed with SoundWire
			 * link 2 and 3, or link 1 and 2, thus we only try to enable dmics
			 * if all conditions are true:
			 * a) 2 or fewer links are used by SoundWire
			 * b) the NHLT table reports the presence of microphones
			 */
			if (hweight_long(mach->link_mask) <= 2)
				dmic_fixup = true;
			else
				mach->mach_params.dmic_num = 0;
```
https://github.com/thesofproject/linux/blob/topic/sof-dev/sound/soc/sof/intel/hda.c#L1332
I think that's why the kernel didn't look for sof-lnl-cs42l43-l0-cs35l56-l23-2ch.tplg

rfvirgil, 2025-01-08:
@dantmnf can you try the top two patches from here: https://github.com/CirrusLogic/sof-linux/commits/topic/asus-UX5406SA/

dantmnf, 2025-01-08:
> @dantmnf can you try the top two patches from here: https://github.com/CirrusLogic/sof-linux/commits/topic/asus-UX5406SA/

@rfvirgil I think manually applying the quirk via module parameter is equivalent to the two patches?

I tried `snd_soc_sof_sdw.quirk=0x20001` and seems we then need a new topology:

```
[    3.999128] sof-audio-pci-intel-lnl 0000:00:1f.3: hda codecs found, mask 4
[    3.999136] sof-audio-pci-intel-lnl 0000:00:1f.3: NHLT device BT(0) detected, ssp_mask 0x4
[    3.999139] sof-audio-pci-intel-lnl 0000:00:1f.3: BT link detected in NHLT tables: 0x4
[    3.999142] sof-audio-pci-intel-lnl 0000:00:1f.3: DMICs detected in NHLT tables: 2
[    4.003051] sof-audio-pci-intel-lnl 0000:00:1f.3: Firmware paths/files for ipc type 1:
[    4.003056] sof-audio-pci-intel-lnl 0000:00:1f.3:  Firmware file:     intel/sof-ipc4/lnl/sof-lnl.ri
[    4.003059] sof-audio-pci-intel-lnl 0000:00:1f.3:  Firmware lib path: intel/sof-ipc4-lib/lnl
[    4.003060] sof-audio-pci-intel-lnl 0000:00:1f.3:  Topology file:     intel/sof-ipc4-tplg/sof-lnl-cs42l43-l0-cs35l56-l23.tplg
[    4.003874] sof-audio-pci-intel-lnl 0000:00:1f.3: Loaded firmware library: ADSPFW, version: 2.11.1.1
[    4.007945] cs42l43 sdw:0:0:01fa:4243:01: supply vdd-p not found, using dummy regulator
[    4.007981] cs42l43 sdw:0:0:01fa:4243:01: supply vdd-d not found, using dummy regulator
[    4.007988] cs42l43 sdw:0:0:01fa:4243:01: supply vdd-a not found, using dummy regulator
[    4.007993] cs42l43 sdw:0:0:01fa:4243:01: supply vdd-io not found, using dummy regulator
[    4.007999] cs42l43 sdw:0:0:01fa:4243:01: supply vdd-cp not found, using dummy regulator
[    4.016428] cs42l43 sdw:0:0:01fa:4243:01: devid: 0x042a43, rev: 0xa1, otp: 0x03
[    4.030094] xe 0000:00:02.0: [drm] GT1: found GSC cv104.1.0
[    4.034112] cs35l56 sdw:0:2:01fa:3556:01:0: supply VDD_P not found, using dummy regulator
[    4.034152] cs35l56 sdw:0:2:01fa:3556:01:0: supply VDD_IO not found, using dummy regulator
[    4.034162] cs35l56 sdw:0:2:01fa:3556:01:0: supply VDD_A not found, using dummy regulator
[    4.044964] cs42l43 sdw:0:0:01fa:4243:01: Slave 6 state check1: UNATTACHED, status was 1
[    4.045921] cs35l56 sdw:0:2:01fa:3556:01:0: Cirrus Logic CS35L56 Rev B0 OTP3 fw:3.4.4 (patched=0)
[    4.052244] cs35l56 sdw:0:2:01fa:3556:01:0: Slave 2 state check1: UNATTACHED, status was 1
[    4.052287] cs35l56 sdw:0:2:01fa:3556:01:1: supply VDD_P not found, using dummy regulator
[    4.052312] cs35l56 sdw:0:2:01fa:3556:01:1: supply VDD_IO not found, using dummy regulator
[    4.052319] cs35l56 sdw:0:2:01fa:3556:01:1: supply VDD_A not found, using dummy regulator
[    4.058855] cs35l56 sdw:0:2:01fa:3556:01:1: Cirrus Logic CS35L56 Rev B0 OTP3 fw:3.4.4 (patched=0)
[    4.064508] cs35l56 sdw:0:3:01fa:3556:01:2: supply VDD_P not found, using dummy regulator
[    4.064545] cs35l56 sdw:0:3:01fa:3556:01:2: supply VDD_IO not found, using dummy regulator
[    4.064553] cs35l56 sdw:0:3:01fa:3556:01:2: supply VDD_A not found, using dummy regulator
[    4.070052] cs35l56 sdw:0:2:01fa:3556:01:1: Slave 1 state check1: UNATTACHED, status was 1
[    4.072270] cs35l56 sdw:0:3:01fa:3556:01:2: Cirrus Logic CS35L56 Rev B0 OTP3 fw:3.4.4 (patched=0)
[    4.073957] Adding alias for supply vdd-amp,(null) -> vdd-amp,sdw:0:0:01fa:4243:01
[    4.077751] cs35l56 sdw:0:3:01fa:3556:01:3: supply VDD_P not found, using dummy regulator
[    4.077758] cs35l56 sdw:0:3:01fa:3556:01:2: Slave 2 state check1: UNATTACHED, status was 1
[    4.077772] cs35l56 sdw:0:3:01fa:3556:01:3: supply VDD_IO not found, using dummy regulator
[    4.077777] cs35l56 sdw:0:3:01fa:3556:01:3: supply VDD_A not found, using dummy regulator
[    4.089061] cs35l56 sdw:0:3:01fa:3556:01:3: Cirrus Logic CS35L56 Rev B0 OTP3 fw:3.4.4 (patched=0)
[    4.100557] cs35l56 sdw:0:3:01fa:3556:01:3: Slave 1 state check1: UNATTACHED, status was 1
[    4.280920] sof-audio-pci-intel-lnl 0000:00:1f.3: Booted firmware version: 2.11.1.1
[    4.303828] sof_sdw sof_sdw: Overriding quirk 0x1 => 0x20001
[    4.304043] cs42l43-codec cs42l43-codec: supply vdd-amp not found, using dummy regulator
[    4.306534] sof-audio-pci-intel-lnl 0000:00:1f.3: Topology: ABI 3:29:1 Kernel ABI 3:23:1
[    4.306703] sof-audio-pci-intel-lnl 0000:00:1f.3: error: can't connect DAI alh-copier.Capture-SmartMic.0 stream Capture-SmartMic
[    4.306706] sof-audio-pci-intel-lnl 0000:00:1f.3: failed to add widget type 28 name : alh-copier.Capture-SmartMic.0 stream Capture-SmartMic
[    4.306713] sof_sdw sof_sdw: ASoC: failed to load widget alh-copier.Capture-SmartMic.0
[    4.306714] sof_sdw sof_sdw: ASoC: topology: could not load header: -22
[    4.306729] sof-audio-pci-intel-lnl 0000:00:1f.3: error: tplg component load failed -22
[    4.306734] sof-audio-pci-intel-lnl 0000:00:1f.3: error: failed to load DSP topology -22
[    4.306736] sof-audio-pci-intel-lnl 0000:00:1f.3: ASoC: error at snd_soc_component_probe on 0000:00:1f.3: -22
[    4.306753] sof_sdw sof_sdw: ASoC: failed to instantiate card -22
[    4.308033] sof_sdw sof_sdw: error -EINVAL: snd_soc_register_card failed -22
[    4.308040] sof_sdw sof_sdw: probe with driver sof_sdw failed with error -22
```

rfvirgil, 2025-01-08:
@dantmnf Ah, that will be the thing @bardliao was referring to. It should be loading sof-lnl-cs42l43-l0-cs35l56-l23-2ch.tplg, but the logic here:
https://github.com/thesofproject/linux/blob/topic/sof-dev/sound/soc/sof/intel/hda.c#L1332

makes it decide that it doesn't have host mics, so it loads the wrong topology.
If I've understood that code correctly, because there are 3 SoundWire links in use it decides that there cannot be host dmics.
I'm not sure what the correct fix is there, presumably that limitation is there for a good reason so we can't just delete it.

bardliao, 2025-01-09:
@dantmnf @pcgrosen Can you try https://github.com/thesofproject/linux/pull/5287 ?
And the topology file: [sof-lnl-cs42l43-l0-cs35l56-l23-2ch.zip](https://github.com/user-attachments/files/18356730/sof-lnl-cs42l43-l0-cs35l56-l23-2ch.zip)
The topology PR is https://github.com/thesofproject/sof/pull/9770.
snd_soc_sof_sdw.quirk=0x20001 is still needed.
Please also add "options snd_soc_sdw_utils dyndbg=+pmf" to get more debug trace.
Thanks.

dantmnf, 2025-01-09:
@bardliao Just before your comment, I've made some success:

Found another quirk that would "fix" `dmic_num`:

https://github.com/gregkh/linux/blob/v6.12.8/sound/soc/intel/boards/sof_sdw.c#L1065-L1071

Built `sof-lnl-cs42l43-l0-cs35l56-l23-2ch` topology modeled after `sof-lnl-cs42l43-l0-cs35l56-l3` vs `sof-lnl-cs42l43-l0-cs35l56-l3-2ch`

```patch
diff --git a/tools/topology/topology2/production/tplg-targets-ace2.cmake b/tools/topology/topology2/production/tplg-targets-ace2.cmake
index 96237b413..74e4fc8ea 100644
--- a/tools/topology/topology2/production/tplg-targets-ace2.cmake
+++ b/tools/topology/topology2/production/tplg-targets-ace2.cmake
@@ -74,4 +74,12 @@ EFX_DMIC0_TDFB_PARAMS=line2_generic_pm10deg,EFX_DMIC0_DRC_PARAMS=dmic_default"
 "cavs-sdw\;sof-lnl-cs42l43-l0-cs35l56-l23\;PLATFORM=lnl,NUM_SDW_AMP_LINKS=2,SDW_DMIC=1,\
 SDW_AMP_FEEDBACK=false,SDW_SPK_STREAM=Playback-SmartAmp,SDW_DMIC_STREAM=Capture-SmartMic,\
 SDW_JACK_OUT_STREAM=Playback-SimpleJack,SDW_JACK_IN_STREAM=Capture-SimpleJack"
+
+"cavs-sdw\;sof-lnl-cs42l43-l0-cs35l56-l23-2ch\;PLATFORM=lnl,\
+NUM_DMICS=2,PDM1_MIC_A_ENABLE=0,PDM1_MIC_B_ENABLE=0,\
+PREPROCESS_PLUGINS=nhlt,NHLT_BIN=sof-lnl-cs42l43-l0-cs35l56-l23-2ch.bin,\
+NUM_SDW_AMP_LINKS=2,SDW_AMP_FEEDBACK=false,SDW_SPK_STREAM=Playback-SmartAmp,\
+SDW_JACK_OUT_STREAM=Playback-SimpleJack,SDW_JACK_IN_STREAM=Capture-SimpleJack,\
+HDMI1_ID=6,HDMI2_ID=7,HDMI3_ID=8,DMIC0_ENHANCED_CAPTURE=true,\
+EFX_DMIC0_TDFB_PARAMS=line2_generic_pm10deg,EFX_DMIC0_DRC_PARAMS=dmic_default"
 )
```

And finally with quirks and topology override

```
# quirk=RT711_JD1|SOC_SDW_PCH_DMIC|SOC_SDW_CODEC_MIC
options snd_soc_sof_sdw quirk=0x20041
options snd_sof_pci tplg_filename=sof-lnl-cs42l43-l0-cs35l56-l23-2ch.tplg
```

After reboot, The microphone mute LED on keyboard is on. 

No UCM configuration available. 

After `amixer cset name='Dmic0 Capture Switch' on`, the mute LED is off and I can record some channel-swapped sound (beeping the right mic goes to channel 0)

---

However, replacing the topology file with yours doesn't work (with unpatched kernel and `options snd_soc_sdw_utils dyndbg=+pmf` applied):
```
[    4.131649] sof_sdw sof_sdw: ASoC: physical link dmic01 (id 3) not exist
[    4.131650] sof_sdw sof_sdw: ASoC: topology: could not load header: -22
[    4.131682] sof-audio-pci-intel-lnl 0000:00:1f.3: error: tplg component load failed -22
[    4.131684] sof-audio-pci-intel-lnl 0000:00:1f.3: error: failed to load DSP topology -22
[    4.131685] sof-audio-pci-intel-lnl 0000:00:1f.3: ASoC: error at snd_soc_component_probe on 0000:00:1f.3: -22
[    4.131695] sof_sdw sof_sdw: ASoC: failed to instantiate card -22
[    4.132766] sof_sdw sof_sdw: error -EINVAL: snd_soc_register_card failed -22
[    4.132768] sof_sdw sof_sdw: probe with driver sof_sdw failed with error -22
```

pcgrosen, 2025-01-09:
@bardliao I applied that patch onto my otherwise-clean 6.12.6 kernel, but I'm getting the same topology load error that @dantmnf just posted. Is there something else in the sof kernel fork (that isn't upstreamed) that I need? If so, I can try a clean build from that repo, but that will probably have to be tomorrow.

bardliao, 2025-01-09:
@dantmnf @pcgrosen Can you try [sof-lnl-cs42l43-l0-cs35l56-l23-2ch.zip](https://github.com/user-attachments/files/18359155/sof-lnl-cs42l43-l0-cs35l56-l23-2ch.zip)? I updated the link IDs.
Also, can you share the dmesg log with "options snd_soc_sdw_utils dyndbg=+pmf"?
I would like to get the "create dai link %s, id %d\n" trace.

@singalsu Any idea about the channel-swapped issue?

dantmnf, 2025-01-09:
@bardliao This file works but still have two channels swapped
```
[    3.288346] sof-audio-pci-intel-lnl 0000:00:1f.3: enabling device (0000 -> 0002)
[    3.288472] sof-audio-pci-intel-lnl 0000:00:1f.3: DSP detected with PCI class/subclass/prog-if 0x040100
[    3.714134] sof-audio-pci-intel-lnl 0000:00:1f.3: bound 0000:00:02.0 (ops i915_audio_component_bind_ops [xe])
[    3.721002] sof-audio-pci-intel-lnl 0000:00:1f.3: use msi interrupt mode
[    3.753954] sof-audio-pci-intel-lnl 0000:00:1f.3: hda codecs found, mask 4
[    3.753963] sof-audio-pci-intel-lnl 0000:00:1f.3: NHLT device BT(0) detected, ssp_mask 0x4
[    3.753965] sof-audio-pci-intel-lnl 0000:00:1f.3: BT link detected in NHLT tables: 0x4
[    3.753967] sof-audio-pci-intel-lnl 0000:00:1f.3: DMICs detected in NHLT tables: 2
[    3.759921] sof-audio-pci-intel-lnl 0000:00:1f.3: Firmware paths/files for ipc type 1:
[    3.759928] sof-audio-pci-intel-lnl 0000:00:1f.3:  Firmware file:     intel/sof-ipc4/lnl/sof-lnl.ri
[    3.759930] sof-audio-pci-intel-lnl 0000:00:1f.3:  Firmware lib path: intel/sof-ipc4-lib/lnl
[    3.759931] sof-audio-pci-intel-lnl 0000:00:1f.3:  Topology file:     intel/sof-ipc4-tplg/sof-lnl-cs42l43-l0-cs35l56-l23-2ch.tplg
[    3.760659] sof-audio-pci-intel-lnl 0000:00:1f.3: Loaded firmware library: ADSPFW, version: 2.11.1.1
[    4.035244] sof-audio-pci-intel-lnl 0000:00:1f.3: Booted firmware version: 2.11.1.1
[    4.057241] sof_sdw sof_sdw: Overriding quirk 0x1 => 0x20041
[    4.057254] snd_soc_sdw_utils:asoc_sdw_count_sdw_endpoints: sof_sdw sof_sdw: Found 5 devices with 8 endpoints
[    4.057257] snd_soc_sdw_utils:asoc_sdw_parse_sdw_endpoints: sof_sdw sof_sdw: Adding prefix cs42l43 for cs42l43-codec
[    4.057259] snd_soc_sdw_utils:asoc_sdw_parse_sdw_endpoints: sof_sdw sof_sdw: Add dev: 0, 0x3001fa424301 end: 0, dai: 0, P/- to solo: 0
[    4.057261] snd_soc_sdw_utils:asoc_sdw_parse_sdw_endpoints: sof_sdw sof_sdw: Add dev: 0, 0x3001fa424301 end: 2, dai: 0, -/C to solo: 0
[    4.057264] snd_soc_sdw_utils:asoc_sdw_parse_sdw_endpoints: sof_sdw sof_sdw: Adding prefix AMP1 for sdw:0:2:01fa:3556:01:0
[    4.057265] snd_soc_sdw_utils:asoc_sdw_parse_sdw_endpoints: sof_sdw sof_sdw: Add dev: 2, 0x23001fa355601 end: 0, dai: 1, P/C to group: 1
[    4.057267] snd_soc_sdw_utils:asoc_sdw_parse_sdw_endpoints: sof_sdw sof_sdw: Adding prefix AMP2 for sdw:0:2:01fa:3556:01:1
[    4.057268] snd_soc_sdw_utils:asoc_sdw_parse_sdw_endpoints: sof_sdw sof_sdw: Add dev: 2, 0x23101fa355601 end: 0, dai: 1, P/C to group: 1
[    4.057270] snd_soc_sdw_utils:asoc_sdw_parse_sdw_endpoints: sof_sdw sof_sdw: Adding prefix AMP3 for sdw:0:3:01fa:3556:01:2
[    4.057270] snd_soc_sdw_utils:asoc_sdw_parse_sdw_endpoints: sof_sdw sof_sdw: Add dev: 3, 0x33201fa355601 end: 0, dai: 1, P/C to group: 1
[    4.057272] snd_soc_sdw_utils:asoc_sdw_parse_sdw_endpoints: sof_sdw sof_sdw: Adding prefix AMP4 for sdw:0:3:01fa:3556:01:3
[    4.057272] snd_soc_sdw_utils:asoc_sdw_parse_sdw_endpoints: sof_sdw sof_sdw: Add dev: 3, 0x33301fa355601 end: 0, dai: 1, P/C to group: 1
[    4.057275] snd_soc_sdw_utils:asoc_sdw_init_dai_link: sof_sdw sof_sdw: create dai link SDW0-Playback-SimpleJack, id 0
[    4.057276] snd_soc_sdw_utils:asoc_sdw_init_dai_link: sof_sdw sof_sdw: create dai link SDW0-Capture-SimpleJack, id 1
[    4.057278] snd_soc_sdw_utils:asoc_sdw_init_dai_link: sof_sdw sof_sdw: create dai link SDW2-Playback-SmartAmp, id 2
[    4.057280] snd_soc_sdw_utils:asoc_sdw_init_dai_link: sof_sdw sof_sdw: create dai link SDW2-Capture-SmartAmp, id 3
[    4.057281] snd_soc_sdw_utils:asoc_sdw_init_dai_link: sof_sdw sof_sdw: create dai link dmic01, id 4
[    4.057282] snd_soc_sdw_utils:asoc_sdw_init_dai_link: sof_sdw sof_sdw: create dai link dmic16k, id 5
[    4.057285] snd_soc_sdw_utils:asoc_sdw_init_dai_link: sof_sdw sof_sdw: create dai link iDisp1, id 6
[    4.057286] snd_soc_sdw_utils:asoc_sdw_init_dai_link: sof_sdw sof_sdw: create dai link iDisp2, id 7
[    4.057287] snd_soc_sdw_utils:asoc_sdw_init_dai_link: sof_sdw sof_sdw: create dai link iDisp3, id 8
[    4.059077] sof-audio-pci-intel-lnl 0000:00:1f.3: Topology: ABI 3:29:1 Kernel ABI 3:23:1
[    4.059469] sof_sdw sof_sdw: ASoC: Parent card not yet available, widget card binding deferred
[    4.060126] sof-audio-pci-intel-lnl 0000:00:1f.3: Loaded firmware library: ADSPFW, version: 2.11.1.1
[    4.103514] sof_sdw sof_sdw: hda_dsp_hdmi_build_controls: no PCM in topology for HDMI converter 3
```

singalsu, 2025-01-09:
@dantmnf @bardliao Do you mean that PCH DMIC channels are swapped?

If with PCH DMIC and not with a microphone at SDW codec, that might be a circuit schematic issue in the PC. There's a number of ways to swap it. Easiest would be likely with custom beamformer blob set up by UCMv2 SOF customization based on DMI ID (in SOF v2.11, but patches for UCM not yet merged). The NHLT DMIC blob could also swap it at HW level but it would need BIOS NHLT update to set up channels swap or load with topology a custom NHLT.

dantmnf, 2025-01-09:
Just noticed that recording on Windows with exclusive mode is also channel-swapped. Looks like the channels are swapped back by a software audio processing component.

bardliao, 2025-01-09:
@singalsu Yes, the laptop uses PCH DMIC and not a microphone at SDW codec.

singalsu, 2025-01-09:
> Just noticed that recording on Windows with exclusive mode is also channel-swapped. Looks like the channels are swapped back by a software audio processing component.

Then maybe it would be best to follow Windows and prepare a beamformer (TDFB) blob that swaps the channels (please create an enhancement type issue for SOF for a channels swap solution). An example of DMI ID Based processing customize is in:

https://github.com/alsa-project/alsa-ucm-conf/blob/master/ucm2/blobs/sof/product_configs/AAEON/UPX-TGL01.conf

We have also talked about developing a SOF example template component for developers that would swap channels as a minimal processing example. It could be used with a custom topology.

dantmnf, 2025-01-10:
> The NHLT DMIC blob could also swap it at HW level but it would need BIOS NHLT update to set up channels swap or load with topology a custom NHLT.

The NHLT table also supposes that channel 0 is on right (well, assuming positive horizontal-offset means right)
```console
root@archlinux ~ # nhlt-dmic-info                   
{
        "mics-data-version":1,
        "mics-data-source":"acpi-nhlt",
        "mics":[
                {
                        "channel":0,
                        "type":"subcardoid",
                        "location":"laptop-front-panel",
                        "horizontal-offset":45,
                        "freq-low-band":100,
                        "freq-high-band":20000,
                        "vertical-angle-begin":-60,
                        "vertical-angle-end":60,
                        "horizontal-angle-begin":-50,
                        "horizontal-angle-end":50
                },
                {
                        "channel":1,
                        "type":"subcardoid",
                        "location":"laptop-front-panel",
                        "horizontal-offset":-45,
                        "freq-low-band":100,
                        "freq-high-band":20000,
                        "vertical-angle-begin":-60,
                        "vertical-angle-end":60,
                        "horizontal-angle-begin":-50,
                        "horizontal-angle-end":50
                }
        ]
}
```

Seems it is not designed to simply take channel 0 as left…

rfvirgil, 2025-01-10:
@bardliao Are you going to submit a patch to add the new tplg to SOF?

bardliao, 2025-01-20:
@rfvirgil Could you submit a PR for https://github.com/CirrusLogic/sof-linux/commits/topic/asus-UX5406SA/?

rfvirgil, 2025-01-20:
Did anyone test that https://github.com/CirrusLogic/sof-linux/commits/topic/asus-UX5406SA/ correctly detects this laptop and sets the quirk? @dantmnf ?

dantmnf, 2025-01-20:
Confirmed that with 3 patches in this thread I can get it working without extra kernel/module parameters

rfvirgil, 2025-01-20:
@dantmnf Thanks

rpolyano, 2025-02-03:
Sorry for the silly question, but what "3 patches" are you referring to? Is it the [patch](https://github.com/thesofproject/linux/pull/5301.patch) of https://github.com/thesofproject/linux/pull/5301? 

I'm dealing with the same issue on the same laptop, but for me the internal mic only doesn't work if there is a bluetooth headphone connected (not sure if related to this), even if the headphone is not selected as input, only as output.

Wachamuli, 2025-02-05:
It works for me too!
Recommended guide: https://github.com/dantmnf/zenbook-s14-linux?tab=readme-ov-file#audio

simontrimmer, 2025-02-10:
Just following up from an aside with (at least) @bardliao and @rfvirgil  this morning, I've been having less success...

If I have a kernel at 
commit 926287855e7fc3a9c74ddeeeff2a7b0f41ca85db (HEAD, upstream/topic/sof-dev, upstream/HEAD)
Author: Pierre-Louis Bossart <pierre-louis.bossart@linux.dev>
Date:   Mon Nov 20 13:15:33 2023 -0600

    ALSA: hda: hda-intel: add Panther Lake-H support

    One more PCI ID.

    Signed-off-by: Pierre-Louis Bossart <pierre-louis.bossart@linux.intel.com>
    Signed-off-by: Peter Ujfalusi <peter.ujfalusi@linux.intel.com>

And build some topologies at 
commit b5e8e455c8ec3d786d5f36c438b1838d1f2c816a (HEAD, upstream/main)
Author: Adrian Bonislawski <adrian.bonislawski@intel.com>
Date:   Thu Feb 6 09:12:28 2025 +0100

    rimage: config: add ptl to generic platform toml

    This is required to build single PTL loadable module

    Signed-off-by: Adrian Bonislawski <adrian.bonislawski@intel.com>

The on the ux5406
[    0.000000] DMI: ASUSTeK COMPUTER INC. ASUS Zenbook S 14 UX5406SA_UX5406SA/UX5406SA, BIOS UX5406SA.303 11/08/2024

Then I can see

[    3.660120] snd_soc_sof_sdw:sof_card_dai_links_create: sof_sdw sof_sdw: DAI link numbers: sdw 3, ssp 0, dmic 2, hdmi 3, bt: 0
[    3.660128] snd_soc_sdw_utils:asoc_sdw_init_dai_link: sof_sdw sof_sdw: create dai link SDW0-Playback-SimpleJack, id 0
[    3.660130] snd_soc_sdw_utils:asoc_sdw_init_dai_link: sof_sdw sof_sdw: create dai link SDW0-Capture-SimpleJack, id 1
[    3.660132] snd_soc_sdw_utils:asoc_sdw_init_dai_link: sof_sdw sof_sdw: create dai link SDW2-Playback-SmartAmp, id 2
**[    3.660133] snd_soc_sdw_utils:asoc_sdw_init_dai_link: sof_sdw sof_sdw: create dai link dmic01, id 3**
[    3.660134] snd_soc_sdw_utils:asoc_sdw_init_dai_link: sof_sdw sof_sdw: create dai link dmic16k, id 4
[    3.660136] snd_soc_sdw_utils:asoc_sdw_init_dai_link: sof_sdw sof_sdw: create dai link iDisp1, id 5
[    3.660137] snd_soc_sdw_utils:asoc_sdw_init_dai_link: sof_sdw sof_sdw: create dai link iDisp2, id 6
[    3.660138] snd_soc_sdw_utils:asoc_sdw_init_dai_link: sof_sdw sof_sdw: create dai link iDisp3, id 7
**[    3.662066] sof_sdw sof_sdw: ASoC: physical link dmic01 (id 4) not exist**
[    3.662070] sof_sdw sof_sdw: ASoC: topology: could not load header: -22
[    3.662144] sof_sdw sof_sdw: ASoC: failed to instantiate card -22
[    3.663267] sof_sdw sof_sdw: error -EINVAL: snd_soc_register_card failed -22
[    3.663272] sof_sdw sof_sdw: probe with driver sof_sdw failed with error -22

I think we should have all the necessary bits in the trees now, but as an insanity test adding the quirk in the guide didn't help either
[    3.204182] sof_sdw sof_sdw: Overriding quirk 0x20000 => 0x20041
[    3.204184] snd_soc_sof_sdw:log_quirks: sof_sdw sof_sdw: quirk realtek,jack-detect-source 1
[    3.204185] snd_soc_sof_sdw:log_quirks: sof_sdw sof_sdw: quirk SOC_SDW_PCH_DMIC enabled
[    3.204186] snd_soc_sof_sdw:log_quirks: sof_sdw sof_sdw: quirk SOC_SDW_CODEC_MIC enabled

For reference, the normal quirks on this platform is only
[    3.411416] snd_soc_sof_sdw:log_quirks: sof_sdw sof_sdw: quirk SOC_SDW_CODEC_MIC enabled

-Simon