thesofproject/sof #9790: [BUG] update SOF CI docker with new alsa-utils

kv2019i, 2025-01-22:
Update SOF docker image used in public CI github flows once alsa-utils https://github.com/alsa-project/alsa-utils/pull/286 is merged upstream.

This is needed to complete work in https://github.com/thesofproject/sof/pull/9763

FYI @singalsu

kv2019i, 2025-02-04:
@cgturner1 Any updates with this one?

cgturner1, 2025-02-05:
Building a new docker today...once that is complete will need to verify it pulled in the right components before I push upstream hopefully it works first shot for me (since the build takes a few hours) but I think a new docker version should be pushed by end of week

kv2019i, 2025-02-18:
The SOF docker update pushed on Feb 6 has the alsa-utils fix for SSP  encoding:
"Topology: NHLT: Intel: Fix DMA slots config in SSP blob"

But the fixes for dmic 1ch (and https://github.com/thesofproject/sof/pull/9763 ) is not yet included, so let's keep this open.

cgturner1, 2025-04-03:
@kv2019i forgot to circle back on this ... i think we are ok to close since the fixes were included

kv2019i, 2025-04-03:
@cgturner1 ? Did you do an update yet that includes #9763 fix (https://github.com/alsa-project/alsa-utils/pull/286 )? 

So patch with ```commit fd67c11b6d3d84308db14150997ae773283caec2 (sof/master, origin/master)
Topology: NHLT: Intel: Improve error message for DMIC enable conflict
```

This was merged close to the time we did the Feb 5 update. I've not pinged about this as I've been waiting whether https://github.com/thesofproject/sof/pull/9870 matures on time for 2.13 release in May. If #9870 is merged, then we can stop these docker updates for alsa-utils/lib altogether.

lgirdwood, 2025-04-03:
Should we not stop here and merge the local ALSA build PR and have the Docker use local ALSA to build topology.
https://github.com/thesofproject/sof/pull/9870

cgturner1, 2025-04-03:
> Should we not stop here and merge the local ALSA build PR and have the Docker use local ALSA to build topology. [#9870](https://github.com/thesofproject/sof/pull/9870)

Yea that makes sense i had thought this had been dealt with on the previous update to the docker but can focus on changing to use the local ALSA instead as the future path

cgturner1, 2025-04-16:
still working on debugging the work from #9870 in a alsa-less docker...I think im running into path issues i expect this to be working soon (within a week) if we ned to update the alsa in our current docker ASAP i can do that in parrallel so to unblock any issues while i debug and implement the changes to use the local alsa

kv2019i, 2025-05-13:
@cgturner1 Should we keep this open and track the remaining Jenkins/docker issues separate? Or do we keep this open to track?

cgturner1, 2025-05-13:
> [@cgturner1](https://github.com/cgturner1) Should we keep this open and track the remaining Jenkins/docker issues separate? Or do we keep this open to track?

I think we should close this as the new docker does have the latest alsa utils and we can open a separate task for fixing the compatability issues on the docker with the new local alsa tools install

closed 2025-05-13 by kv2019i