thesofproject/sof #9797: [BUG] compare-linux-win CI build check fails to binary difference in imx8/imx8 zephyr.elf

kv2019i, 2025-01-27:
**Describe the bug**
Starting today 27th Jan 2025, multiple pull-requests are showing a failure in build reproducibility between Linux and Windows builds with
error:

https://github.com/thesofproject/sof/actions/runs/12947318306/job/36114439673?pr=9794
```
 Files linux-build  imx8 imx8x imx8m imx8ulp/build-sof-staging/sof-info/imx8/stripped-zephyr.elf and windows-build  imx8 imx8x imx8m imx8ulp/build-sof-staging/sof-info/imx8/stripped-zephyr.elf differ
Files linux-build  imx8 imx8x imx8m imx8ulp/build-sof-staging/sof-info/imx8/zephyr.lst and windows-build  imx8 imx8x imx8m imx8ulp/build-sof-staging/sof-info/imx8/zephyr.lst differ
Files linux-build  imx8 imx8x imx8m imx8ulp/build-sof-staging/sof-info/imx8/zephyr_version.h and windows-build  imx8 imx8x imx8m imx8ulp/build-sof-staging/sof-info/imx8/zephyr_version.h differ
Files linux-build  imx8 imx8x imx8m imx8ulp/build-sof-staging/sof-info/imx8m/stripped-zephyr.elf and windows-build  imx8 imx8x imx8m imx8ulp/build-sof-staging/sof-info/imx8m/stripped-zephyr.elf differ
Files linux-build  imx8 imx8x imx8m imx8ulp/build-sof-staging/sof-info/imx8m/zephyr.lst and windows-build  imx8 imx8x imx8m imx8ulp/build-sof-staging/sof-info/imx8m/zephyr.lst differ
Files linux-build  imx8 imx8x imx8m imx8ulp/build-sof-staging/sof-info/imx8m/zephyr_version.h and windows-build  imx8 imx8x imx8m imx8ulp/build-sof-staging/sof-info/imx8m/zephyr_version.h differ
Files linux-build  imx8 imx8x imx8m imx8ulp/build-sof-staging/sof-info/imx8ulp/stripped-zephyr.elf and windows-build  imx8 imx8x imx8m imx8ulp/build-sof-staging/sof-info/imx8ulp/stripped-zephyr.elf differ
```

This is shown in multiple open PRs:
 - https://github.com/thesofproject/sof/pull/9794 , https://github.com/thesofproject/sof/pull/9795 , https://github.com/thesofproject/sof/pull/9793
 - ... so does not seem to be related to the pull-requests.

To add to the mystery, I can't see the failure in any of the pull requests failed last week.
 
**To Reproduce**
Submit a pull request to SOF repository.

**Reproduction Rate**
100%

**Expected behavior**
Linux and Windows host builds should create identical binaries.

**Impact**
Failure seen on every CI request, can mask other errors.

kv2019i, 2025-01-27:
@thesofproject/nxp  Rings any bells to you? I checked multiple PRs merged last week, but I couldn't find a PR where this failure would have been seen (already pre merge). I could have missed some.

dbaluta, 2025-01-27:
@kv2019i looks like this is not only for imx builds.

```
 Files linux-build  tgl tgl-h/build-sof-staging/sof-info/tgl-h/stripped-zephyr.elf and windows-build  tgl tgl-h/build-sof-staging/sof-info/tgl-h/stripped-zephyr.elf differ
Files linux-build  tgl tgl-h/build-sof-staging/sof-info/tgl-h/zephyr.lst and windows-build  tgl tgl-h/build-sof-staging/sof-info/tgl-h/zephyr.lst differ
Files linux-build  tgl tgl-h/build-sof-staging/sof-info/tgl-h/zephyr_version.h and windows-build  tgl tgl-h/build-sof-staging/sof-info/tgl-h/zephyr_version.h differ
Files linux-build -d mtl/build-sof-staging/sof-info/mtl/boot.mod and windows-build -d mtl/build-sof-staging/sof-info/mtl/boot.mod differ
Files linux-build -d mtl/build-sof-staging/sof-info/mtl/stripped-main.elf and windows-build -d mtl/build-sof-staging/sof-info/mtl/stripped-main.elf differ
Files linux-build -d mtl/build-sof-staging/sof-info/mtl/stripped-zephyr.elf and windows-build -d mtl/build-sof-staging/sof-info/mtl/stripped-zephyr.elf differ
Files linux-build -d mtl/build-sof-staging/sof-info/mtl/zephyr.lst and windows-build -d mtl/build-sof-staging/sof-info/mtl/zephyr.lst differ
Files linux-build -d mtl/build-sof-staging/sof-info/mtl/zephyr_version.h and windows-build -d mtl/build-sof-staging/sof-info/mtl/zephyr_version.h differ
```

dbaluta, 2025-01-27:
@kv2019i  Relevant issue in the past https://github.com/thesofproject/sof/issues/9034 was solved by opting out on adding the source code on the assembly listing file.

marc-hb, 2025-01-27:
> You can downloads artefacts from every PR page,

Actually, daily tests are more convenient for this: because zero PR interference.

Started failing 4 days ago: https://github.com/thesofproject/sof/actions/runs/12940934767

> please compare these.

Don't wait because artefacts expire after some time.

marc-hb, 2025-01-27:
There's been a progressive rollout of a new Windows image ( Version: 20250113.1.0 -> 20250120.2.0) around the same time but it does not correlate with the failure. 

For instance, the IMX build got the newer Windows image in 794 but 794 was still all green:
https://github.com/thesofproject/sof/actions/runs/12898787138/job/35966473504

As you can tell from the version numbers, minor upgrades like these happen all the time.

The first step is really to download and compare `zephyr_version.h`.

kv2019i, 2025-01-28:
It would seem the builds really are different now. I wonder if this is somehow related to reent cold/llext change -- the .cold section at least looks different. E.g. here for github build 796 (https://github.com/thesofproject/sof/actions/runs/12940934767#artifacts)

```
--- linux/build-sof-staging/sof-info/mtl/stripped-zephyr.elf.lst        2025-01-28 14:20:56.645624296 +0200
+++ windows/build-sof-staging/sof-info/mtl/stripped-zephyr.elf.lst      2025-01-28 14:21:12.225625128 +0200
@@ -83,13 +83,13 @@
  a10484e0 0c38a088 20c02000 89094603 00a83b98  .8.. . ...F...;.
  a10484f0 0b0c18a0 8820c020 0089090c 021df0    ..... . ....... 
 Contents of section .cold:
- a1049000 006002a0 609708a0 949808a0 40200000  .`..`.......@ ..
- a1049010 f8870aa0 789708a0 dc4104a0 2c0e08a0  ....x....A..,...
+ a1049000 006002a0 6c9708a0 a09808a0 40200000  .`..l.......@ ..
+ a1049010 10880aa0 849708a0 dc4104a0 2c0e08a0  .........A..,...
```

The debug sections have been different already before, so I wonder if this just pushes out the sections to different addresses (and use of cold sections just triggers this).

marc-hb, 2025-01-28:
Thanks @kv2019i . Indeed, the difference can be observed in the build log:

```
[7/445] Generating include/generated/zephyr/version.h
-- Zephyr version: 4.0.99 (D:/a/sof/sof/workspace/zephyr), build: v4.0.0-2813-g42701fdb2729
```
versus

```
[9/447] Generating include/generated/zephyr/version.h
-- Zephyr version: 4.0.99 (/zep_workspace/zephyr), build: 42701fdb2729
```

A long time ago, I added magit west and git tricks to speed things up (a lot). They worked fine for a very long time but maybe some subtle git version difference or other environmental difference just broke them. I'll take a quick look.

marc-hb, 2025-01-28:
Around the time of daily build 795, there was a progressive rollout of Ubuntu 22 VM version 20250120.2, up from  20250105.1

This upgrade seems to correlate 100% with the loss of the git tag in git describe.

The newer Ubuntu VM upgraded git from version 2.47.1 to version Git 2.48.1

Meanwhile, git on Windows did not change and stuck to `Git 2.47.1.windows.1`

https://github.com/actions/runner-images/blob/ubuntu22/20250105.1/images/ubuntu/Ubuntu2204-Readme.md
https://github.com/actions/runner-images/blob/ubuntu22/20250120.2/images/ubuntu/Ubuntu2204-Readme.md

kv2019i, 2025-01-30:
Closed via https://github.com/thesofproject/sof/pull/9801 . Thanks @marc-hb !

closed 2025-01-30 by kv2019i

marc-hb, 2025-01-31:
After dropping sof-docs from them, daily tests are back to green!

https://github.com/thesofproject/sof/actions/workflows/daily-tests.yml

[image]