thesofproject/sof #9821: [BUG]No input from microphones on Fujitsu Intel Corporation Alder Lake PCH-P High Definition Audio Controller

u20380, 2025-02-09:
**Describe the bug**
I am not able to get any input from the microphone(s) on my system.
Neither the internal microphone nor the microphone on the plugged in earphones with microphone produce any sound.

See also https://bugzilla.suse.com/show_bug.cgi?id=1236863

**To Reproduce**
1.Use either internal microphones or microphone from earplugs
2.Use arecord to test microphone input
3.Playback recording from arecord
Actual Results:  
No sound is captured from either microphone.
Same result with other audio recording applications, like Audacity, Zoom or Teams

**Reproduction Rate**
All the time

**Expected behavior**
Sound from microphones to be available

**Impact**
Not able to attend zoom or teams meetings from this laptop

**Environment**
Not sure how to get the following information:

1) Branch name and commit hash of the 2 repositories: sof (firmware/topology) and linux (kernel driver).
    * Kernel: {SHA}
    * SOF: {SHA}
2) Name of the topology file
    * Topology: {FILE}
3) Name of the platform(s) on which the bug is observed.
    * Platform: {PLATFORM}

**Screenshots or console output**
\`\`\`
inxi -Aa
Audio:
  Device-1: Intel Alder Lake PCH-P High Definition Audio
    vendor: Fujitsu Client driver: sof-audio-pci-intel-tgl
    alternate: snd_hda_intel, snd_soc_avs, snd_sof_pci_intel_tgl
    bus-ID: 00:1f.3 chip-ID: 8086:51c8 class-ID: 0401
  API: ALSA v: k6.13.1-lp156.6.g93072cc-default status: kernel-api
    with: aoss type: oss-emulator tools: alsactl,alsamixer,amixer
  Server-1: PipeWire v: 1.0.5 status: active with: 1: pipewire-pulse
    status: active 2: wireplumber status: active
    tools: pactl,pw-cat,pw-cli,wpctl
\`\`\`
I have attached the output of alsa-info.sh in the attachment.

[alsa-info.txt](https://github.com/user-attachments/files/18724903/alsa-info.txt)

If applicable, add a screenshot (drag-and-drop an image), or console logs
(cut-and-paste text and put a code fence (\`\`\`) before and after, to help
explain the issue.

Please also include the relevant sections from the firmware log and kernel log in the report (and attach the full logs for complete reference). Kernel log is taken from *dmesg* and firmware log from *sof-logger*. See https://thesofproject.github.io/latest/developer_guides/debugability/logger/index.html

lgirdwood, 2025-02-10:
Important logs parts are below. NHLT reports 2 DMICs, also Mic could be connected to ALC257.
@bardliao @ujfalusi any ideas ?
```
[    4.874713] sof-audio-pci-intel-tgl 0000:00:1f.3: enabling device (0000 -> 0002)
[    4.876839] sof-audio-pci-intel-tgl 0000:00:1f.3: DSP detected with PCI class/subclass/prog-if 0x040100
[    4.876976] sof-audio-pci-intel-tgl 0000:00:1f.3: bound 0000:00:02.0 (ops i915_audio_component_bind_ops [i915])
[    4.885361] sof-audio-pci-intel-tgl 0000:00:1f.3: use msi interrupt mode
[    4.923333] sof-audio-pci-intel-tgl 0000:00:1f.3: hda codecs found, mask 5
[    4.923344] sof-audio-pci-intel-tgl 0000:00:1f.3: using HDA machine driver skl_hda_dsp_generic now
[    4.923349] sof-audio-pci-intel-tgl 0000:00:1f.3: NHLT device BT(0) detected, ssp_mask 0x4
[    4.923353] sof-audio-pci-intel-tgl 0000:00:1f.3: BT link detected in NHLT tables: 0x4
[    4.923357] sof-audio-pci-intel-tgl 0000:00:1f.3: DMICs detected in NHLT tables: 2
[    4.927245] sof-audio-pci-intel-tgl 0000:00:1f.3: Firmware paths/files for ipc type 0:
[    4.927254] sof-audio-pci-intel-tgl 0000:00:1f.3:  Firmware file:     intel/sof/sof-adl.ri
[    4.927257] sof-audio-pci-intel-tgl 0000:00:1f.3:  Topology file:     intel/sof-tplg/sof-hda-generic-2ch.tplg
[    4.928200] sof-audio-pci-intel-tgl 0000:00:1f.3: Firmware info: version 2:2:0-57864
[    4.928208] sof-audio-pci-intel-tgl 0000:00:1f.3: Firmware: ABI 3:22:1 Kernel ABI 3:23:1
[    5.027831] sof-audio-pci-intel-tgl 0000:00:1f.3: Firmware info: version 2:2:0-57864
[    5.027842] sof-audio-pci-intel-tgl 0000:00:1f.3: Firmware: ABI 3:22:1 Kernel ABI 3:23:1
[    5.039056] sof-audio-pci-intel-tgl 0000:00:1f.3: Topology: ABI 3:22:1 Kernel ABI 3:23:1
[    5.039692] skl_hda_dsp_generic skl_hda_dsp_generic: ASoC: Parent card not yet available, widget card binding deferred
--
[    5.071253] iwlwifi 0000:00:14.3: Detected RF GF, rfid=0x2010d000
[    5.075307] snd_hda_codec_realtek ehdaudio0D0: ALC257: SKU not ready 0x909701f0
[    5.075543] snd_hda_codec_realtek ehdaudio0D0: autoconfig for ALC257: line_outs=1 (0x14/0x0/0x0/0x0/0x0) type:speaker
[    5.075548] snd_hda_codec_realtek ehdaudio0D0:    speaker_outs=0 (0x0/0x0/0x0/0x0/0x0)
[    5.075552] snd_hda_codec_realtek ehdaudio0D0:    hp_outs=1 (0x21/0x0/0x0/0x0/0x0)
[    5.075555] snd_hda_codec_realtek ehdaudio0D0:    mono: mono_out=0x0
[    5.075557] snd_hda_codec_realtek ehdaudio0D0:    inputs:
[    5.117628] skl_hda_dsp_generic skl_hda_dsp_generic: hda_dsp_hdmi_build_controls: no PCM in topology for HDMI converter 3
```

singalsu, 2025-02-11:
Please check that Dmic0 capture switch is set on, and also make sure Dmic0 volume is set to max.

amixer -c0 cset name='Dmic0 Capture Volume' 70
amixer -c0 cset name='Dmic0 Capture Switch' on

If the problem still persists, attach the the wav file (screenshot is not enough) from e.g. "arecord -Dhw:0,6 -f S32_LE -r 48000 -c 2 -d 5 rec.wav" here. I'll try to determine from start glitch shape if there's a connected microphone or not. If no microphone in PCH-DMIC then it would be in ALC257 codec and a different kind of issue.

bardliao, 2025-02-11:
We could focus on the PCH-DMIC issue as Takashi is helping the ALC257 issue on https://thesofproject.github.io/latest/developer_guides/debugability/logger/index.html

BTW, I saw 
	control.38 {
		iface MIXER
		name 'Dmic0 Capture Switch'
		value.0 true
		value.1 true
		comment {
			access 'read write'
			type BOOLEAN
			count 2
		}
	}
from the alsa-info.txt. It seems 'Dmic0 Capture Switch' is already on.

u20380, 2025-02-15:
I have tried to record from the internal microphone as per your recommendations.
Again the result was a short glitch.

~> amixer -c0 cset name='Dmic0 Capture Volume' 70
numid=37,iface=MIXER,name='Dmic0 Capture Volume'
  ; type=INTEGER,access=rw---R--,values=2,min=0,max=70,step=0
  : values=70,70
  | dBscale-min=-50.00dB,step=1.00dB,mute=1

:~> amixer -c0 cset name='Dmic0 Capture Switch' on
numid=38,iface=MIXER,name='Dmic0 Capture Switch'
  ; type=BOOLEAN,access=rw------,values=2
  : values=on,on

~> arecord -Dhw:0,6 -f S32_LE -r 48000 -c 2 -d 5 rec.wav
Recording WAVE 'rec.wav' : Signed 32 bit Little Endian, Rate 48000 Hz, Stereo

Uploading a wav file is not accepted by the system. 

I am attaching a screenshot from Audacity, 

[image]

u20380, 2025-02-15:
[20250211 rec.zip](https://github.com/user-attachments/files/18810739/20250211.rec.zip)

I found out that zip files can be uploaded. The audio wav file is zipped up in the attached archive.

u20380, 2025-02-17:
I hope this is what you wanted me to do.

In /lib/firmware/intel/sof-tplg/ I renamed intel/sof-tplg/sof-hda-generic-**2**ch.tplg to save it temporarily.

Then copied intel/sof-tplg/sof-hda-generic-**4**ch.tplg to intel/sof-tplg/sof-hda-generic-**2**ch.tplg 
And then I started recording with  arecord -Dhw:0,6 -f S32_LE -r 48000 -c **4** -d 5 rec.wav

Helas, same result, a glitch as mentioned earlier.

singalsu, 2025-02-18:
> Helas, same result, a glitch as mentioned earlier.

So, all four channels look the same? Did you check that capture switches and volumes are enabled for every channel.

u20380, 2025-02-18:
Checking the settings of the soundcard with the four channel driver:

**~> amixer contents**
numid=4,iface=MIXER,name='Master Playback Switch'
  ; type=BOOLEAN,access=rw------,values=1
  : **values=off**
numid=3,iface=MIXER,name='Master Playback Volume'
  ; type=INTEGER,access=rw------,values=2,min=0,max=65536,step=1
  : values=65536,65536
numid=2,iface=MIXER,name='Capture Switch'
  ; type=BOOLEAN,access=rw------,values=1
  : **values=on**
numid=1,iface=MIXER,name='Capture Volume'
  ; type=INTEGER,access=rw------,values=2,min=0,max=65536,step=1
  : values=65536,65536

The Master Playback Switch is off, the Capture Switch is on.

Trying to toggle the Master Playback Switch on:

**~> amixer -c 0 -- sset Master Playback Switch 100%**
Simple mixer control 'Master',0
  Capabilities: pvolume pvolume-joined pswitch pswitch-joined
  Playback channels: Mono
  Limits: Playback 0 - 87
  Mono: Playback 87 [100%] [0.00dB] [on]

But still the Master Playback Switch stays off:

**~> amixer contents**
numid=4,iface=MIXER,name='Master Playback Switch'
  ; type=BOOLEAN,access=rw------,values=1
  : values=off
numid=3,iface=MIXER,name='Master Playback Volume'
  ; type=INTEGER,access=rw------,values=2,min=0,max=65536,step=1
  : values=65536,65536
numid=2,iface=MIXER,name='Capture Switch'
  ; type=BOOLEAN,access=rw------,values=1
  : values=on
numid=1,iface=MIXER,name='Capture Volume'
  ; type=INTEGER,access=rw------,values=2,min=0,max=65536,step=1
  : values=65536,65536

And no sound is recorded.

Should I be seeing more channels, with sof-hda-generic-4ch.tplg?

u20380, 2025-02-18:
arecord -Dhw:0,6 -f S32_LE -r 48000 -c 4 -d 5 rec.wav
Recording WAVE 'rec.wav' : Signed 32 bit Little Endian, Rate 48000 Hz, Channels 4

Does "Channels 4" indicate that I do try to record from any of four channels?

singalsu, 2025-02-18:
> arecord -Dhw:0,6 -f S32_LE -r 48000 -c 4 -d 5 rec.wav Recording WAVE 'rec.wav' : Signed 32 bit Little Endian, Rate 48000 Hz, Channels 4
> 
> Does "Channels 4" indicate that I do try to record from any of four channels?

Yes, you should get a four channels wav file.

singalsu, 2025-02-18:
> numid=2,iface=MIXER,name='Capture Switch'
> ; type=BOOLEAN,access=rw------,values=1
> : values=on

The control name should be "Dmic0 Capture Switch". When channels count increased 2 to 4, the controls for the new channels 3-4 may be now with attenuation or mute applied.

u20380, 2025-02-18:
Here it shows only two capture devices:

cat /proc/asound/devices
  1:        : sequencer
  2: [ 0- 6]: digital audio capture
  3: [ 0- 7]: digital audio capture
  4: [ 0- 0]: digital audio playback
  5: [ 0- 0]: digital audio capture
  6: [ 0-31]: digital audio playback
  7: [ 0- 3]: digital audio playback
  8: [ 0- 4]: digital audio playback
  9: [ 0- 5]: digital audio playback
 10: [ 0- 2]: hardware dependent
 11: [ 0- 0]: hardware dependent
 12: [ 0]   : control
 33:        : timer

singalsu, 2025-02-18:
> Here it shows only two capture devices:
>
> cat /proc/asound/devices 1: : sequencer 2: [ 0- 6]: digital audio capture 3: [ 0- 7]: digital audio capture 4: [ 0- 0]: digital audio playback 5: [ 0- 0]: [...]

Yes, you can capture from headset mic hw:0,0, DMIC 48 kHz hw:0,6 or DMIC 16 kHz hw:0,7. The DMIC capture devices access the same physical microphones but with different sample rate. The 16 kHz DMIC has a separate volume control.

u20380, 2025-02-18:
Alsamixer shows two digital microphones, each with L and R, I can only toggle Dmic0 on and off.

[image]

singalsu, 2025-02-18:
> Alsamixer shows two digital microphones, each with L and R, I can only toggle Dmic0 on and off.

Looks like you have capture switch set for all four Dmic0 channels. If you still get same zero PCM code capture wav for all channels except start transient, then I wonder if the BIOS that indicates 2 DMIC can be correct. 

What if the microphones are analog? You could try to disable SOF with "options snd_intel_dspcfg dsp_driver=1" in a /etc/modprobe.d file for sof, e.g. "sof.conf". In that case the driver would attempt to use analog microhone switched with headphone mic. This is again just a guess but if no other ideas here you could try. The capture device would be then hw:0,0.

bardliao, 2025-02-18:
BTW @u20380 Have you tried the same device on Windows?

u20380, 2025-02-18:
I did - with a positive outcome.

bardliao, 2025-02-19:
@u20380 Do you have the product name of the device? Or the audio design is even better. We need to know where does the host dmic connect exactly.

u20380, 2025-02-19:
Its a Fujitsu/Siemens Lifebook E5512.

I have submitted a request for Fujitsu customer support to provide information about the audio setup.

u20380, 2025-02-21:
Here is the response from Fujitsu customer support. 
I have asked for more detailed information, like an electrical wiring scheme.

**Quote**
Unfortunately, this was the feedback from our team:
“Linux distros are not under the scope of the standard support.”

Nevertheless, here is some information that we were able to gather.

This is the sub board used, the sun board is connect to the MB with an FPC cable.
(see attached image)

 These are the speakers:
(see attached image)
 

The audio chip is a Realtek ALC 257

For further information about the audio chip, we would advise to contact Realtek support.

**

[image]
[image]

**

bardliao, 2025-02-24:
@u20380 I think the most important question is that does the host microphone connect to the Realtek ALC 257 codec or to Intel PCH?
In the meantime, you could add "options snd_sof_intel_hda_common sof_use_tplg_nhlt=1" to `/etc/modprobe.d/alsa-base.conf` and see if it helps..

With that option you will use the topology nhlt instead of the BIOS nhlt.

singalsu, 2025-02-24:
Also, did you try my SOF disable suggestion above?

u20380, 2025-02-24:
I have added your suggestion above.
Then I tried to record from the microphones:

arecord -Dhw:0,6 -f S32_LE -r 48000 -c 4 -d 5 20250224_recording.wav

Resulting in the following error:

arecord: main:850: audio open error: No such file or directory `

singalsu, 2025-02-24:
> arecord: main:850: audio open error: No such file or directory `

If the microphone would be connected to ALC257, the capture device would be Hw:0,0.

u20380, 2025-02-24:
This is what I get with Hw:0,0

> arecord -Dhw:0,0 -f S32_LE -r 48000 -c 4 -d 5 rec 20240224_recording.wav
arecord: set_params:1398: Channels count non available

bardliao, 2025-02-25:
> I have added your suggestion above. Then I tried to record from the microphones:
> 
> arecord -Dhw:0,6 -f S32_LE -r 48000 -c 4 -d 5 20250224_recording.wav
> 
> Resulting in the following error:
> 
> arecord: main:850: audio open error: No such file or directory `

@u20380 Can you share the alsa-info when you disable SOF? Probably DMIC will not be enabled with the legacy HDA driver.
BTW, I think we need Realtek's help on the ALC257 part. I don't know how we can work around if the BIOS doesn't provide verb table properly. @KailangYang Can you help?

u20380, 2025-02-25:
As requested. 
[20250225 - alsa-info.txt](https://github.com/user-attachments/files/18971760/20250225.-.alsa-info.txt)

BTW, I will not be getting additional information from Fujitsu. They refer to their standard download page.

bardliao, 2025-02-26:
```
**** List of CAPTURE Hardware Devices ****
card 0: PCH [HDA Intel PCH], device 0: ALC257 Analog [ALC257 Analog]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 0: PCH [HDA Intel PCH], device 2: ALC257 Alt Analog [ALC257 Alt Analog]
  Subdevices: 2/2
  Subdevice #0: subdevice #0
  Subdevice #1: subdevice #1496 
```

There is no dmic device if SOF is disabled. That's why you can't use arecord -Dhw:0,6

singalsu, 2025-02-27:
> This is what I get with Hw:0,0
> 
> > arecord -Dhw:0,0 -f S32_LE -r 48000 -c 4 -d 5 rec 20240224_recording.wav
> > arecord: set_params:1398: Channels count non available

hw:0,0 is 2ch device, please try with `-c 2`.

u20380, 2025-02-28:
I thought I would try with a bluetooth headset:

~> arecord -Dhw:0,0 -f S32_LE -r 48000 -c 2 -d 5 rec 2024028_bluetooth recording.wav
Recording WAVE 'rec' : Signed 32 bit Little Endian, Rate 48000 Hz, Stereo
arecord: set_params:1456: Unable to install hw params:
ACCESS:  RW_INTERLEAVED
FORMAT:  S32_LE
SUBFORMAT:  STD
SAMPLE_BITS: 32
FRAME_BITS: 64
CHANNELS: 2
RATE: 48000
PERIOD_TIME: 125000
PERIOD_SIZE: 6000
PERIOD_BYTES: 48000
PERIODS: 4
BUFFER_TIME: 500000
BUFFER_SIZE: 24000
BUFFER_BYTES: 192000
TICK_TIME: 0

And the alsa-info 

[20250228_alsa-info.txt](https://github.com/user-attachments/files/19032554/20250228_alsa-info.txt)