thesofproject/sof #9834: [BUG] Internal microphone never worked on new ASUS Zenbook S 14 UX5406SA Lunar Lake cs42l43

sdwerwed, 2025-02-15:
**Describe the bug**
Internal microphone never works on new laptop  ASUS Zenbook S 14 UX5406SA Lunar Lake, similar issues are reported on various community forums as well, it is related to `mic:cs42l43-dmic`. Here is the bug link https://bugzilla.suse.com/show_bug.cgi?id=1237161 for opensuse, [Takashi Iwai](mailto:tiwai@suse.com) commented to report here to the upstream first.

**To Reproduce**
Just use laptop with similar hardware, and install latest kernel, I use OpenSuse Tumbleweed

**Reproduction Rate**
It is permanent issue, never works, not supported probably

**Expected behavior**
I expect mic to work out of the box ideally without any custom configurations. If not, better to have some manual steps to make it work.

**Impact**
For me it is serious issue because I can not do online meetings, I join from my mobile to participate on meetings

**Environment**
Kernel: 6.13.1-1-default
SOF firmware version: [2.11.1.1](https://2.11.1.1/) (Loaded firmware library: ADSPFW, version: [2.11.1.1](https://2.11.1.1/))
SOF topology ABI: 3:29:1
Kernel ABI: 3:23:1
Firmware path: intel/sof-ipc4/lnl/sof-lnl.ri
Firmware library path: intel/sof-ipc4-lib/lnl
Name of the topology file

Topology: intel/sof-ipc4-tplg/sof-lnl-cs42l43-l0-cs35l56-l23.tplg
Name of the platform(s) on which the bug is observed

Platform: Intel Lunar Lake-M HD Audio Controller (8086:a828)
Device: ASUS Zenbook S 14 UX5406SA
Board Name: UX5406SA

**Screenshots or console output**
Alsa logs: https://bugzilla.suse.com/attachment.cgi?id=880329

sdwerwed, 2025-02-16:
I made mic work by doing those changes manually https://github.com/dantmnf/zenbook-s14-linux?tab=readme-ov-file#microphone-array-dmic

lgirdwood, 2025-02-17:
> I made mic work by doing those changes manually https://github.com/dantmnf/zenbook-s14-linux?tab=readme-ov-file#microphone-array-dmic
> 
> There is issue tracking on https://github.com/thesofproject/sof/issues/9759 if it is exactly the same issue maintainers can close this as duplicate

@sdwerwed thanks for the update - are all patches needed now upstream for next release ?

closed 2025-02-25 by lgirdwood