thesofproject/sof #9845: [BUG] topology2: ASRC pipelines not working in sof-hda-benchmark-asrc32.tplg

lyakh, 2025-02-19:
**Describe the bug**
Trying to play or record audio with sof-hda-benchmark-asrc32.tplg on "HDA Analog" PCMs on MTL exits with an error:

```
$ aplay -d 4 -D hw:1,0 -c 2 -r 48000 -f S32_LE /dev/zero
Playing raw data '/dev/zero' : Signed 32 bit Little Endian, Rate 48000 Hz, Stereo
aplay: set_params:1416: Unable to install hw params:
ACCESS:  RW_INTERLEAVED
FORMAT:  S32_LE
SUBFORMAT:  STD
SAMPLE_BITS: 32
FRAME_BITS: 64
CHANNELS: 2
RATE: 48000
PERIOD_TIME: 125000
PERIOD_SIZE: 6000
PERIOD_BYTES: 48000
PERIODS: 4
BUFFER_TIME: 500000
BUFFER_SIZE: 24000
BUFFER_BYTES: 192000
TICK_TIME: 0
$ arecord -D hw:1,0 -r 48000 -c 2 -f S32_LE -d 3 /dev/null
Recording WAVE '/dev/null' : Signed 32 bit Little Endian, Rate 48000 Hz, Stereo
arecord: set_params:1416: Unable to install hw params:
ACCESS:  RW_INTERLEAVED
FORMAT:  S32_LE
SUBFORMAT:  STD
SAMPLE_BITS: 32
FRAME_BITS: 64
CHANNELS: 2
RATE: 48000
PERIOD_TIME: 125000
PERIOD_SIZE: 6000
PERIOD_BYTES: 48000
PERIODS: 4
BUFFER_TIME: 500000
BUFFER_SIZE: 24000
BUFFER_BYTES: 192000
TICK_TIME: 0
```

singalsu, 2025-02-19:
I'm able to replicate the issue with UPX board. The dmesg shows

```
Feb 19 18:08:34 ekstremisti kernel: snd_sof:sof_ipc4_widget_setup: sof-audio-pci-intel-tgl 0000:00:1f.3: Create widget dai-copier.HDA.Analog.playback (pipe 2) - ID 4, instance 1, core 0
Feb 19 18:08:34 ekstremisti kernel: snd_sof:sof_ipc4_log_header: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx      : 0x40010004|0x10015: MOD_INIT_INSTANCE [data size: 84]
Feb 19 18:08:34 ekstremisti kernel: snd_sof:sof_ipc4_log_header: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx reply: 0x60000000|0x10015: MOD_INIT_INSTANCE
Feb 19 18:08:34 ekstremisti kernel: snd_sof:sof_ipc4_log_header: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx done : 0x40010004|0x10015: MOD_INIT_INSTANCE [data size: 84]
Feb 19 18:08:34 ekstremisti kernel: snd_sof:sof_widget_setup_unlocked: sof-audio-pci-intel-tgl 0000:00:1f.3: widget dai-copier.HDA.Analog.playback setup complete
Feb 19 18:08:34 ekstremisti kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: cannot bind host-copier.0.playback -> asrc.1.1, no firmware module for:  sink
Feb 19 18:08:34 ekstremisti kernel: snd_sof:sof_ipc4_log_header: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx      : 0x12000000|0x0: GLB_DELETE_PIPELINE
Feb 19 18:08:34 ekstremisti kernel: snd_sof:sof_ipc4_log_header: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx reply: 0x32000000|0x0: GLB_DELETE_PIPELINE
Feb 19 18:08:34 ekstremisti kernel: snd_sof:sof_ipc4_log_header: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx done : 0x12000000|0x0: GLB_DELETE_PIPELINE
Feb 19 18:08:34 ekstremisti kernel: snd_sof:sof_widget_free_unlocked: sof-audio-pci-intel-tgl 0000:00:1f.3: widget pipeline.1 freed
Feb 19 18:08:34 ekstremisti kernel: snd_sof:sof_widget_free_unlocked: sof-audio-pci-intel-tgl 0000:00:1f.3: widget host-copier.0.playback freed
Feb 19 18:08:34 ekstremisti kernel: snd_sof:sof_ipc4_log_header: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx      : 0x12010000|0x0: GLB_DELETE_PIPELINE
Feb 19 18:08:34 ekstremisti kernel: snd_sof:sof_ipc4_log_header: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx reply: 0x32000000|0x0: GLB_DELETE_PIPELINE
Feb 19 18:08:34 ekstremisti kernel: snd_sof:sof_ipc4_log_header: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx done : 0x12010000|0x0: GLB_DELETE_PIPELINE
Feb 19 18:08:34 ekstremisti kernel: snd_sof:sof_widget_free_unlocked: sof-audio-pci-intel-tgl 0000:00:1f.3: widget pipeline.2 freed
Feb 19 18:08:34 ekstremisti kernel: snd_sof:sof_widget_free_unlocked: sof-audio-pci-intel-tgl 0000:00:1f.3: widget dai-copier.HDA.Analog.playback freed
Feb 19 18:08:34 ekstremisti kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: sof_pcm_setup_connected_widgets: pcm0 (HDA Analog), dir 0: Widget list set up failed
Feb 19 18:08:34 ekstremisti kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ASoC: error at snd_soc_pcm_component_hw_params on 0000:00:1f.3: -19
Feb 19 18:08:34 ekstremisti kernel:  HDA Analog: ASoC: error at __soc_pcm_hw_params on HDA Analog: -19
Feb 19 18:08:34 ekstremisti kernel:  HDA Analog: ASoC: error at dpcm_fe_dai_hw_params on HDA Analog: -19

```

singalsu, 2025-02-21:
The above fix PRs for Linux and SOF work now but need cleanup. Also the IPC4 ASRC version in FW isn't exactly like described for init IPC, needs a fix as well.

singalsu, 2025-02-21:
> The above fix PRs for Linux and SOF work now but need cleanup. Also the IPC4 ASRC version in FW isn't exactly like described for init IPC, needs a fix as well.

Checked it, the current IPC implementation is right so no need for further patches.

singalsu, 2025-02-25:
@lyakh The PRs are now merged. OK to close this issue?

closed 2025-02-26 by lyakh