thesofproject/sof #9847: [BUG] Distorted audio output ASUS Chromebook Plus CX34 (board name: MARASOV)

lamekino, 2025-02-20:
**Describe the bug**
Audio from the speakers on the ASUS Chromebook Plus CX34 does not work; however, using wired or bluetooth headphones works without issue. The audio emanating from the speakers is quiet yet distorted. Adjusting the volume seems to have little effect on output quality or loudness. The microphone also does not work at all. Some dmesg logs seem suspect:
```
[  +0.850830] sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx error for 0x60010000 (msg/reply size: 108/20): -22
[  +0.000015] sof-audio-pci-intel-tgl 0000:00:1f.3: HW params ipc failed for stream 1
[  +0.000004] sof-audio-pci-intel-tgl 0000:00:1f.3: ASoC: error at snd_soc_pcm_component_hw_params on 0000:00:1f.3: -22
[  +0.001059] sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx error for 0x60010000 (msg/reply size: 108/20): -22
[  +0.000011] sof-audio-pci-intel-tgl 0000:00:1f.3: HW params ipc failed for stream 1
[  +0.000004] sof-audio-pci-intel-tgl 0000:00:1f.3: ASoC: error at snd_soc_pcm_component_hw_params on 0000:00:1f.3: -22
[  +0.001001] sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx error for 0x60010000 (msg/reply size: 108/20): -22
[  +0.000005] sof-audio-pci-intel-tgl 0000:00:1f.3: HW params ipc failed for stream 1
[  +0.000003] sof-audio-pci-intel-tgl 0000:00:1f.3: ASoC: error at snd_soc_pcm_component_hw_params on 0000:00:1f.3: -22
[  +0.001024] sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx error for 0x60010000 (msg/reply size: 108/20): -22
[  +0.000008] sof-audio-pci-intel-tgl 0000:00:1f.3: HW params ipc failed for stream 1
[  +0.000009] sof-audio-pci-intel-tgl 0000:00:1f.3: ASoC: error at snd_soc_pcm_component_hw_params on 0000:00:1f.3: -22
```
**Reproduction rate**
100%.

**Expected behavior**
- Speaker audio shouldn't have distortion.
- Changing the volume should have a greater effect on the sound output.
- Microphone should capture audio.

**Environment**
Arch Linux, using MrChromeBox's coreboot firmware. I use the Topology from WeirdTreeThing's [repo](https://github.com/WeirdTreeThing/chromebook-linux-audio/tree/main/conf/sof/tplg). Their `setup-audio` script is in order for the speaker to be detected at all (otherwise it's called 'Dummy Output'). There is also an original report at the script's repo [here](https://github.com/WeirdTreeThing/chromebook-linux-audio/issues/175).

I'm going to attach my alsa-info.txt file and record the output of speaker-test for an example, recorded with my phone. I hope this is the correct place to report this issue, as I don't know much about low-level audio. Thanks 😁

**Attachments**
[alsa-info.txt](https://github.com/user-attachments/files/18879817/alsa-info.txt)
[pink-noise-speaker-test.mp3.zip](https://github.com/user-attachments/files/18879842/pink-noise-speaker-test.mp3.zip)

lgirdwood, 2025-02-20:
@lamekino I would check the audio configuration/topology used on Chrome where audio does work and copy it. If that doesnt help then coreboot may need checked for correct pinmux alignment with Chrome coreboot.

lgirdwood, 2025-02-21:
> > I would check the audio configuration/topology used on Chrome where audio does work and copy it.
> 
> I've tried copying the /lib/firmware/ files and the /usr/share/alsa/ucm files to no avail. It results in the speaker going back to dummy input. Is that what you mean? I don't really know if I'm doing it right. I'm using the chrome recovery images as my source.

Please also make sure topology is reused from Chrome too (its next to the FW directory in Chrome). I would also confirm your dmesg versus the Chrome dmesg, it should state the audio config that can be compared.

> 
> > coreboot may need checked for correct pinmux alignment with Chrome coreboot.
> 
> I believe that I could try and look at that but I don't know where to start. I'm experienced in C, but wouldn't know where to start. If it's as simple as changing the value, recompiling and reflashing the board I could get that working if I know what files to look for; however, I haven't done embedded or firmware C, but I'm willing to learn 😄
> 
> I would want to make 100% sure that it's a problem with coreboot before going that direction and do a reflash... I have a feeling that I didn't copy the configuration over correctly. What files specifically should I be copying over? I did specifically ones for the `sofrt5682` driver for the alsa UCM files, I don't remember what /lib/firmware files I did since that was a while ago.

I would probably reflash Chrome here and take a copy of the top level UCM, FW, topology folders for reuse on Linux. Sorry, I cant help on coreboot.

lamekino, 2025-02-21:
Okay, I'll reflash the stock fw and reinstall ChromeOS today. I'll capture the dmesg and post it later tonight when I have it finished.

lamekino, 2025-02-21:
Alright, I'm on ChromeOS again and I've run the alsa-info script again. Something that stood out to me was that the device "attaches" (ie "attach allowed to drvr) to `intel_ish_ipc`, `snd_soc_avs` and `sof-audio-pci-intel-tgl` before loading the firmware and topology files. I am attaching the alsa-info.txt that I got on ChromeOS and the dmesg log I captured after booting. From the looks of things I would guess that there is either a missing driver or something is misconfigured, but I'm not sure.

I'm going to hold off on returning to Linux for now. I'm going to archive the `/lib/firmware/intel` and the UCM directory. Is there anything else that I should save before heading to doing a reinstall of Linux?

lgirdwood, 2025-02-24:
@lamekino The Chrome logs look correct and its using following FW/topology. Pls make sure these are loaded on Linux and the dmesg looks similar. I dont think you need to copy any other files other than the topologies, FWs and UCMS as you have mentioned above.
```
[    5.014815] sof-audio-pci-intel-tgl 0000:00:1f.3:  Firmware file:     intel/sof/community/sof-rpl.ri
[    5.014817] sof-audio-pci-intel-tgl 0000:00:1f.3:  Topology file:     intel/sof-tplg/sof-rpl-rt1019-rt5682.tplg
```
Once Linux shows the same FW/topology and versions - then I would look at UCM is its still not working.

akuehntopf, 2025-03-10:
I've also encountered this issue.
Asus CX34, "Marasov" on Fedora 41.

I'm back on stock now, but if there's anything i can do to help resolve this issue, let me know.

closed 2025-05-08 by wszypelt