thesofproject/sof #9852: [BUG] Internal Mic not working - Dell Precision

gravit8n, 2025-02-22:
**Describe the bug**
After upgrading to the Linux image 6.11.0-17 from 6.8.0.52 on Ubuntu 24.04.2 LTS, the microphone on my DELL Precision 5690 stopped working. The microphone appears to receive a wrong signal in the audio settings (or pavucontrol), and the signal fluctuates continuously. When I attempt to record, a crackling sound is heard.

Capture : https://github.com/user-attachments/assets/6a08441d-7fdc-42d2-b2df-3c9f438da9ea

This issue is not new; it has been happening with some kernel since at least 6.8.0.3x. Multiple users have reported similar issues for Dell Precision in various forums: [link to forum post](https://askubuntu.com/questions/1519369/internal-mic-not-working-on-dell-precision-5490/1541865#1541865) or [here](https://github.com/intel/ipu6-drivers/issues/228#issuecomment-2673246547).

**To Reproduce**
Can not reproduce, it depends of the kernel and packages.

**Reproduction Rate**
All the time.
It worked sometimes with the kernel 6.8.0.48, 6.8.0.52, but never with 6.11.0-17 .

**Expected behavior**
The microphone should work correctly, and audio sinks should be active and capable of outputting sound. No problem on Windows.

**Impact**
It's not possible to hold a conference or a remote meeting.

**Environment**
- Kernel: 6.11.0-17
- SOF firmware version: 2.11.1.1
- SOF topology : ABI 3:29:1 
- Kernel ABI 3:23:1
- Firmware file:     intel/sof-ipc4/mtl/sof-mtl.ri
- Firmware lib path: intel/sof-ipc4-lib/mtl
- Topology file:     intel/sof-ace-tplg/sof-mtl-rt713-l0-rt1316-l12-rt1713-l3.tplg
- Distribution: Ubuntu 24.04.2 LTS
- PulseAudio : 16.1
- PipeWire version: 1.0.5
- pipewire-pulse : 1.0.5-1ubuntu2
- Hardware: DELL Precision 5690 with Meteor Lake-P HD Audio Controller

**What I’ve tried so far:**
- Purging and reinstalling pulseaudio, pipeWire, pipewire-pulse, linux-firmware, ubuntu-drivers-common, firmware-sof-signed, alsa-utils, linux-image-6.11.0-17-generic, linux-image-generic-hwe-24.04, ...
- Resetting ~/.config/pulse and ~/.config/pipewire directories.
- Installing BIOS and firmware updates and experimenting with older kernel versions ( linux-image-6.8.0-52-generic - but not working)

**Screenshots or console output**
\`\`\`
$ lspci | grep -i audio
00:1f.3 Multimedia audio controller: Intel Corporation Meteor Lake-P HD Audio Controller (rev 20)
01:00.1 Audio device: NVIDIA Corporation Device 22be (rev a1)
\`\`\`

\`\`\`
$ cat /proc/asound/cards
 0 [NVidia         ]: HDA-Intel - HDA NVidia
                      HDA NVidia at 0x7d000000 irq 17
 1 [sofsoundwire   ]: sof-soundwire - sof-soundwire
                      DellInc.-Precision5690--02RYDD
\`\`\`

\`\`\`
$ arecord -l
**** Liste des périphériques matériels CAPTURE ****
carte 1 : sofsoundwire [sof-soundwire], périphérique 1 : Jack In (*) []
  Sous-périphériques : 1/1
  Sous-périphérique #0 : subdevice #0
carte 1 : sofsoundwire [sof-soundwire], périphérique 3 : Amp feedback (*) []
  Sous-périphériques : 1/1
  Sous-périphérique #0 : subdevice #0
carte 1 : sofsoundwire [sof-soundwire], périphérique 4 : Microphone (*) []
  Sous-périphériques : 1/1
  Sous-périphérique #0 : subdevice #0
\`\`\`

\`\`\`
$ pactl list short sources
48	alsa_output.pci-0000_00_1f.3-platform-sof_sdw.HiFi__hw_sofsoundwire_7__sink.monitor	PipeWire	s32le 2ch 48000Hz	SUSPENDED
49	alsa_output.pci-0000_00_1f.3-platform-sof_sdw.HiFi__hw_sofsoundwire_6__sink.monitor	PipeWire	s32le 2ch 48000Hz	SUSPENDED
50	alsa_output.pci-0000_00_1f.3-platform-sof_sdw.HiFi__hw_sofsoundwire_5__sink.monitor	PipeWire	s32le 2ch 48000Hz	SUSPENDED
51	alsa_output.pci-0000_00_1f.3-platform-sof_sdw.HiFi__hw_sofsoundwire__sink.monitor	PipeWire	s32le 2ch 48000Hz	SUSPENDED
52	alsa_output.pci-0000_00_1f.3-platform-sof_sdw.HiFi__hw_sofsoundwire_2__sink.monitor	PipeWire	s32le 2ch 48000Hz	SUSPENDED
53	alsa_input.pci-0000_00_1f.3-platform-sof_sdw.HiFi__hw_sofsoundwire_1__source	PipeWire	s32le 2ch 48000Hz	SUSPENDED
54	alsa_input.pci-0000_00_1f.3-platform-sof_sdw.HiFi__hw_sofsoundwire_4__source	PipeWire	s32le 2ch 48000Hz	SUSPENDED
\`\`\`

\`\`\`
$ systemctl --user unmask pipewire.service
systemctl --user unmask pipewire-pulse.service
systemctl --user enable --now pipewire pipewire-pulse
(base) glion@chronop5690:~$ systemctl --user status pipewire pipewire-pulse
● pipewire.service - PipeWire Multimedia Service
     Loaded: loaded (/usr/lib/systemd/user/pipewire.service; enabled; preset: enabled)
     Active: active (running) since Fri 2025-02-21 21:55:16 CET; 13h ago
TriggeredBy: ● pipewire.socket
   Main PID: 19441 (pipewire)
      Tasks: 3 (limit: 76244)
     Memory: 9.8M (peak: 10.8M)
        CPU: 1.209s
     CGroup: /user.slice/user-1000.slice/user@1000.service/session.slice/pipewire.service
             └─19441 /usr/bin/pipewire

févr. 21 21:55:16 chronop5690 systemd[3113]: Started pipewire.service - PipeWire Multimedia Service.
févr. 21 21:55:16 chronop5690 pipewire[19441]: mod.jackdbus-detect: Failed to receive jackdbus reply: org.freedesktop.>

● pipewire-pulse.service - PipeWire PulseAudio
     Loaded: loaded (/usr/lib/systemd/user/pipewire-pulse.service; enabled; preset: enabled)
     Active: active (running) since Fri 2025-02-21 21:50:39 CET; 13h ago
TriggeredBy: ● pipewire-pulse.socket
   Main PID: 19047 (pipewire-pulse)
      Tasks: 3 (limit: 76244)
     Memory: 14.1M (peak: 15.5M)
        CPU: 870ms
     CGroup: /user.slice/user-1000.slice/user@1000.service/session.slice/pipewire-pulse.service
             └─19047 /usr/bin/pipewire-pulse

févr. 21 21:50:39 chronop5690 systemd[3113]: Started pipewire-pulse.service - PipeWire PulseAudio.
\`\`\`

[log.mic.debug.txt](https://github.com/user-attachments/files/18921811/log.mic.debug.txt)

lgirdwood, 2025-02-24:
> Update 2025/02/25: The issue seems to come from the microphone with the F4 key on the keyboard (mute/unmute) When I switch from secondary to primary F keys, the microphone behavior changes and the microphone works, even if the microphone is not working before...

This could be a HW configured Microphone mute, it may be routed to the soundiwre codec directly (where the SDW coded will mute/unmute in HW) .

abonislawski, 2025-02-25:
@gravit8n looks like we can close it?

gravit8n, 2025-03-02:
> [@gravit8n](https://github.com/gravit8n) looks like we can close it?

Not really.

Update 2022/03/02: The problem reappeared after a week of normal usage.
It happened after switching to Windows. However, the usual method of toggling the Fn keys between primary and secondary functions did not work this time.
I tried reinstalling all relevant packages (PulseAudio, PipeWire, ALSA, etc.) without success.
I then went into the BIOS to disable the microphone and observe Ubuntu's behavior with the microphone turned off. After rebooting, the microphone worked normally using the primary Fn key… yet in the BIOS, the microphone was still disabled.

lgirdwood, 2025-03-03:
> I then went into the BIOS to disable the microphone and observe Ubuntu's behavior with the microphone turned off. After rebooting, the microphone worked normally using the primary Fn key… yet in the BIOS, the microphone was still disabled.

This looks like a board integration issue here and is probably out of scope for the SOF project since we have no access to how the mic mute HW is wired up on this device.

wszypelt, 2025-04-25:
@gravit8n was the issue fixed?

gravit8n, 2025-04-25:
Not really fixed. I haven't had any issues since I had to disable the microphone in the UEFI in order for it to work on Ubuntu. But by doing that, I can no longer use it on Windows because it sees it as disabled. I contacted DELL, but I haven't received a response so far.

closed 2025-05-16 by abonislawski

NinjaTuna007, 2026-07-10:
Following up on my comment above: since this issue is closed and the failure mode I'm seeing is materially different from the OP's (mute-key has no effect, no BIOS changes, failed state survives warm reboot / full driver reload and clears only on cold power-off), I've filed it separately with full details: #10988