thesofproject/sof #9884: [BUG] alsabat failing on PTL HDA config

kv2019i, 2025-03-10:
**Describe the bug**
Seen in CI at least twice:
- https://sof-ci.01.org/linuxpr/PR5355/build6549/devicetest/index.html (Linux kernel PR)
- https://sof-ci.01.org/sofpr/PR9881/build11343/devicetest/index.html (SOF FW PR)

lgirdwood, 2025-03-11:
Seen again today, all sightings on ba-ptlh-rvp-hda-01 so far. Validation team have now added another anti static bag under speaker PCB to see if this improves the issue.

kv2019i, 2025-03-18:
Still seen once on 17th March in Intel internal run 51031 . Rest of the runs are clean this week, so this is appearing less often now.

lgirdwood, 2025-03-19:
> Still seen once on 17th March in Intel internal run 51031 . Rest of the runs are clean this week, so this is appearing less often now.

Agree - seeing it far less often now.

kv2019i, 2025-03-20:
One case today on 20th March: https://sof-ci.01.org/sofpr/PR9906/build11647/devicetest/index.html

lgirdwood, 2025-03-24:
Ok, I'm putting this down to the DUT/HW electrical issue in this case especially since reproduction has changed by adding another anti-static mat.

kv2019i, 2025-06-11:
No longer visible in daily test plans, closing.

closed 2025-06-11 by kv2019i